jblance/mpp-solar #254: Running as daemon on Raspberry Pi

RoySalisbury, 2022-08-27:
On Raspberry Pi if I use the "--daemon" option I get the following error:

2022-08-26 19:05:40,701:DEBUG:__init__:main@237: Using Mongo None with mppsolar
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 248, in main
    import systemd.daemon
ModuleNotFoundError: No module named 'systemd'

Trying to install 'systemd' using 'pip install systemd' fails..

How do I run mpp-solar as a daemon (so it runs as a service all the time)?

jblance, 2022-08-27:
> 2022-08-26 ***@***.***: Using Mongo None with

You need to install systemd for python

Try:
sudo apt-get install python3-systemd

RoySalisbury, 2022-08-27:
Excellent.. That works ... Thanks

closed 2022-08-27 by RoySalisbury

michas79de, 2023-01-07:
Thanks @jblance , that helped me too!
But when I close PuTTY it seems to stop working.
At least Home Assistant doesn't get any data via MQTT any longer, after closing PuTTY.

Isn't it possible to start mpp-solar in a way, allowing to close the SSH connection?
(How?)

jblance, 2023-01-07:
How did you install?
What is your config and setup?
What is the service/daemon status?

Saentist, 2023-01-07:
> Thanks @jblance , that helped me too! But when I close PuTTY it seems to stop working. At least Home Assistant doesn't get any data via MQTT any longer, after closing PuTTY.
> 
> Isn't it possible to start mpp-solar in a way, allowing to close the SSH connection? (How?)

Use command SCREEN as separate shell
` SCREEN mpp-solar -p /dev/hidraw1 -o hassd_mqtt,screen -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG -q HA-IP --mqttuser USER --mqttpass PASS -n DEV-NAME --daemon`
then press CTRL+A CTRL+D
leave remote session
To reattach later `screen -R `

Or setup it as service.

michas79de, 2023-01-07:
> How did you install? What is your config and setup? What is the service/daemon status?

I used `sudo pip install mppsolar` from your [README.md](https://github.com/jblance/mpp-solar#readme) to install it to my RPi3.

My config contains:
```
[SETUP]
pause=10
mqtt_broker=192.168.178.44
mqtt_port=1883
mqtt_user=mqtt-user
mqtt_pass=*********

[EASUN]
protocol=PI30
type=mppsolar
port=/dev/ttyUSB0
command=QDI#QFLAG#QID#QMOD#QOPM#QPIGS#QPIWS#QVFW
tag=Inverter
outputs=hass_mqtt
```

And I'm trying to start with `mpp-solar -C /docker/mpp-solar/mpp-solar.conf --daemon`* just from SSH console via PuTTy.
(* I made this path though it isn't a docker container. I grew up with DOS and WIN and I'm new to Linux and docker. But Mosquitto and HomeAssistant are running now!)

michas79de, 2023-01-07:
> Use command SCREEN as separate shell ` SCREEN mpp-solar -p /dev/hidraw1 -o hassd_mqtt,screen -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG -q HA-IP --mqttuser USER --mqttpass PASS -n DEV-NAME --daemon` then press CTRL+A CTRL+D leave remote session To reattach later `screen -R `

results to `-bash: SCREEN: command not found`

> Or setup it as service.

How? (for dummies)

jblance, 2023-01-07:
> > Or setup it as service.
> 
> How? (for dummies)

try https://github.com/jblance/mpp-solar/blob/master/docs/service-README.md for a start

screen will needs to be installed and is not part of this software (it is a useful linux tool that does achieve what you are after though)

michas79de, 2023-01-07:
Thanks, will try and report later