jblance/mpp-solar #257: JK BMS and TTL FTDI

preussal, 2022-09-05:
BMS queries work with Bluetooth.
But unfortunately not over TTL 232

**BMS 1**

jkbms -p C8:47:8C:E2:90:FA -n Battery-1 -P JK02 -c getInfo
```

Command: getInfo - BLE Device Information inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
header                        	55aaeb90       	    
record_type                   	03             	    
record_counter                	118            	    
device_model                  	JK-B2A24S15P   	    
hardware_version              	10.XW          	    
software_version              	10.08          	    
up_time                       	39D6H30M0S     	    
power-on_times                	16             	    
device_name                   	Battery-1      	    
device_passcode               	1234           	    
manufacturing_date            	220726         	    
serial_number                 	2042102267     	    
passcode                      	0000           	    
user_data                     	Battery #1a    	 
```

**BMS 2**

jkbms -p C8:47:8C:E1:E7:30 -n Battery-2  -P JK02 -c getInfo

```
Command: getInfo - BLE Device Information inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
header                        	55aaeb90       	    
record_type                   	03             	    
record_counter                	122            	    
device_model                  	JK-B2A24S20P   	    
hardware_version              	10.XW          	    
software_version              	10.07          	    
up_time                       	36D7H15M0S     	    
power-on_times                	23             	    
device_name                   	Battery-2      	    
device_passcode               	1234           	    
manufacturing_date            	220713         	    
serial_number                 	2012601265     	    
passcode                      	0000           	    
user_data                     	Battery #2a    	  
``` 

**BMS 1**

jkbms -p /dev/ttyUSB1 -P JK232 -n Battery-1 -b 115200 -D

```
2022-09-05 19:37:43,915:INFO:__init__:main@227: Solar Device Command Utility, version: 0.13.2
2022-09-05 19:37:43,916:DEBUG:__init__:main@241: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffffa0547520>, _isConnected=False)
2022-09-05 19:37:43,916:DEBUG:__init__:main@243: udp port 5555
2022-09-05 19:37:43,916:DEBUG:__init__:main@245: Using Postgres None
2022-09-05 19:37:43,917:DEBUG:__init__:main@248: Using Mongo None with mppsolar
2022-09-05 19:37:43,917:INFO:__init__:main@350: Creating device "Battery-1" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK232"
2022-09-05 19:37:43,936:DEBUG:__init__:main@354: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-09-05 19:37:43,936:DEBUG:device:__init__@33: __init__ args ()
2022-09-05 19:37:43,936:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Battery-1', 'port': '/dev/ttyUSB1', 'protocol': 'JK232', 'baud': 115200, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffffa0547520>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-09-05 19:37:43,937:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-09-05 19:37:43,937:INFO:__init__:get_port@107: Using serialio for communications
2022-09-05 19:37:43,997:DEBUG:__init__:get_protocol@12: Protocol JK232
2022-09-05 19:37:44,033:DEBUG:device:__init__@38: __init__ name Battery-1, port <mppsolar.inout.serialio.SerialIO object at 0xffff9fb89c30>, protocol <mppsolar.protocols.jk232.jk232 object at 0xffff9fb89ba0>
2022-09-05 19:37:44,033:DEBUG:__init__:main@403: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xffff9fb89d50>, '', '', 'screen', None, None)]
[... 147 lines cut ...]
06
03
00
00
00
00
00
00
68
01
29
sending command: 4e5700130000000006030000000000006800000129
wrote 21 bytes
Got response: b'NW\x01!\x00\x00\x00\x00\x06\x00\x01y0\x01\x0c\xee\x02\x0c\xf0\x03\x0c\xf3\x04\x0c\xf1\x05\x0c\xf1\x06\x0c\xf0\x07\x0c\xef\x08\x0c\xf0\t\x0c\xf1\n'
Got response: b'\x0c\xf1\x0b\x0c\xef\x0c\x0c\xf0\r\x0c\xf0\x0e\x0c\xeb\x0f\x0c\xf0\x10\x0c\xf1\x80\x00"\x81\x00 \x82\x00\x1f\x83\x14\xb3\x84\x02\xdf\x85F\x86\x02\x87\x00\t\x89\x00\x00\n'
Got response: b'\x00\x8a\x00\x10\x8b\x00\x00\x8c\x00\x03\x8e\x16\x80\x8f\x0f\xf0\x90\x0e\x10\x91\r\xde\x92\x00\x05\x93\t\xf6\x94\n'
Got response: b'Z\x95\x00\x05\x96\x01,\x97\x00\x96\x98\x01,\x99\x00\x96\x9a\x00\x1e\x9b\rH\x9c\x00\x05\x9d\x01\x9e\x00Z\x9f\x00F\xa0\x00d\xa1\x00d\xa2\x00\x14\xa3\x00F\xa4\x00F\xa5\x00\x01\xa6\x00\x05\xa7\xff\xec\xa8\xff\xf6\xa9\x10\xaa\x00\x00\x01\x18\xab\x01\xac\x01\xad\x04\x1a\xae\x01\xaf\x01\xb0\x00\n'
Got response: b'\xb1\x14\xb219qwert79\x00\xb3\x00\xb4Battery \xb52207\xb6\x00\x00\xce\x00\xb710.XW_S10.07___\xb8\x00\xb9\x00\x00\x01\x00\xbaBattery #2\x00\x00Battery-2 \x00\x00\xc0\x01\x00\x00\x00\x00h\x00\x00V\xf3'
```

Then I disconnected the two BMS and measured the resistance between TX and RX.
With the BMS 1, ~ 2.5 Mega Ohm
With the BMS 2, ~ 60 Kilos ohm

Unfortunately I didn't find any information about it. But I think that hardware has a problem with BMS 1.

preussal, 2022-09-06:
BMS 1 maybe with Hardware failure

jkbms -p /dev/ttyUSB1 -P JK485 -n Battery-1 -b 115200 -D
```
2022-09-06 12:47:13,065:INFO:__init__:main@227: Solar Device Command Utility, version: 0.13.2
2022-09-06 12:47:13,065:DEBUG:__init__:main@241: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff81ab7520>, _isConnected=False)
2022-09-06 12:47:13,066:DEBUG:__init__:main@243: udp port 5555
2022-09-06 12:47:13,066:DEBUG:__init__:main@245: Using Postgres None
2022-09-06 12:47:13,066:DEBUG:__init__:main@248: Using Mongo None with mppsolar
2022-09-06 12:47:13,066:INFO:__init__:main@350: Creating device "Battery-1" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK485"
2022-09-06 12:47:13,073:DEBUG:__init__:main@354: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-09-06 12:47:13,073:DEBUG:device:__init__@33: __init__ args ()
2022-09-06 12:47:13,073:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Battery-1', 'port': '/dev/ttyUSB1', 'protocol': 'JK485', 'baud': 115200, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff81ab7520>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-09-06 12:47:13,074:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-09-06 12:47:13,074:INFO:__init__:get_port@107: Using serialio for communications
2022-09-06 12:47:13,084:DEBUG:__init__:get_protocol@12: Protocol JK485
2022-09-06 12:47:13,090:DEBUG:device:__init__@38: __init__ name Battery-1, port <mppsolar.inout.serialio.SerialIO object at 0xffff810f9c30>, protocol <mppsolar.protocols.jk485.jk485 object at 0xffff810f9ba0>
2022-09-06 12:47:13,090:DEBUG:__init__:main@403: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xffff810f9d50>, '', '', 'screen', None, None)]
2022-09-06 12:47:13,090:INFO:__init__:main@408: Looping 1 commands
2022-09-06 12:47:13,090:INFO:__init__:main@418: Getting results from device: jkbms device - name: Battery-1, port: <mppsolar.inout.serialio.SerialIO object at 0xffff810f9c30>, protocol: <mppsolar.protocols.jk485.jk485 object at 0xffff810f9ba0> for command: , tag: , outputs: screen
2022-09-06 12:47:13,091:INFO:device:run_command@274: Running command 
2022-09-06 12:47:13,091:INFO:jk485:get_full_command@97: Using protocol b'JK485' with 1 commands
2022-09-06 12:47:13,091:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-09-06 12:47:13,091:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-09-06 12:47:13,091:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-09-06 12:47:13,092:DEBUG:jk485:get_full_command@111: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2022-09-06 12:47:13,092:DEBUG:jk485:get_full_command@114: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2022-09-06 12:47:13,092:DEBUG:jk485:get_full_command@119: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2022-09-06 12:47:13,092:DEBUG:jk485:get_full_command@122: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2022-09-06 12:47:13,092:INFO:device:run_command@303: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2022-09-06 12:47:13,092:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-09-06 12:47:13,093:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-09-06 12:47:13,093:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-09-06 12:47:13,093:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 115200
2022-09-06 12:47:13,100:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-09-06 12:47:14,203:DEBUG:serialio:send_and_receive@30: serial response was: b''
2022-09-06 12:47:14,207:DEBUG:device:run_command@322: Send and Receive Response b''
2022-09-06 12:47:14,207:INFO:abstractprotocol:decode@196: response passed to decode: b''
2022-09-06 12:47:14,207:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-09-06 12:47:14,208:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-09-06 12:47:14,208:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-09-06 12:47:14,208:INFO:abstractprotocol:decode@236: Processing response of type POSITIONAL
2022-09-06 12:47:14,208:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 2, 'Header', '']
2022-09-06 12:47:14,209:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 1, 'Slave Address', '']
2022-09-06 12:47:14,209:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 1, 'Command Code', '']
2022-09-06 12:47:14,209:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['BigHex2Short:r/100', 2, 'Total Battery Voltage', 'V']
2022-09-06 12:47:14,209:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['BigHex2Short:r/1000', 2, 'Average Cell Voltage', 'V']
2022-09-06 12:47:14,209:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Int', 1, 'Number of Cells', '']
2022-09-06 12:47:14,209:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Int', 1, 'Highest Cell', '']
[... cut ...]
  File "/root/venv/mppsolar/src/mppsolar/mppsolar/protocols/abstractprotocol.py", line 460, in decode
    data_name, value, data_units = self.process_response(
  File "/root/venv/mppsolar/src/mppsolar/mppsolar/protocols/abstractprotocol.py", line 177, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/root/venv/mppsolar/src/mppsolar/mppsolar/protocols/protocol_helpers.py", line 95, in Hex2Int
    answer = hexString[0]
IndexError: index out of range
```

sw25481, 2023-05-14:
Lot of threads on this same subject.  I have a working setup with SolarEdge and the symple Python script from this site https://github.com/fah/jk-bms with the config

    bms = serial.Serial('/dev/ttyUSB1')
    bms.baudrate = 115200
    bms.timeout  = 0.2

but I get

jkbms -p /dev/ttyUSB1 -P JK485 -b 115200 -D 

2023-05-14 21:58:10,627:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.57
2023-05-14 21:58:10,630:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-05-14 21:58:10,635:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2023-05-14 21:58:10,639:DEBUG:__init__:main@241: udp port 5555
2023-05-14 21:58:10,644:DEBUG:__init__:main@243: Using Postgres None
2023-05-14 21:58:10,649:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2023-05-14 21:58:10,653:INFO:__init__:main@347: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK485"
2023-05-14 21:58:10,688:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2023-05-14 21:58:10,691:DEBUG:device:__init__@33: __init__ args ()
2023-05-14 21:58:10,696:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'JK485', 'baud': 115200, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xb5cfff58>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-05-14 21:58:10,700:DEBUG:__init__:get_port_type@73: port matches ttyusb
2023-05-14 21:58:10,705:INFO:__init__:get_port@121: Using serialio for communications
2023-05-14 21:58:10,755:DEBUG:__init__:get_protocol@13: Protocol JK485
2023-05-14 21:58:11,620:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0xb5cfffa0>, protocol JKBMS RS485 serial communication protocol handler
2023-05-14 21:58:11,623:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xb5cfff70>, '', '', 'screen', None, None)]
2023-05-14 21:58:11,626:INFO:__init__:main@407: Looping 1 commands
2023-05-14 21:58:11,629:INFO:__init__:main@417: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0xb5cfffa0>, protocol: JKBMS RS485 serial communication protocol handler for command: , tag: , outputs: screen
2023-05-14 21:58:11,635:INFO:device:run_command@50: Running command
2023-05-14 21:58:11,640:INFO:jk485:get_full_command@100: Using protocol b'JK485' with 1 commands
2023-05-14 21:58:11,644:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-05-14 21:58:11,648:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'getBalancerData'
2023-05-14 21:58:11,654:DEBUG:abstractprotocol:get_command_defn@67: Found command getBalancerData in protocol b'JK485'
2023-05-14 21:58:11,659:DEBUG:jk485:get_full_command@114: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2023-05-14 21:58:11,663:DEBUG:jk485:get_full_command@117: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2023-05-14 21:58:11,667:DEBUG:jk485:get_full_command@122: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2023-05-14 21:58:11,672:DEBUG:jk485:get_full_command@125: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2023-05-14 21:58:11,676:INFO:device:run_command@78: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2023-05-14 21:58:11,680:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-05-14 21:58:11,685:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'getBalancerData'
2023-05-14 21:58:11,689:DEBUG:abstractprotocol:get_command_defn@67: Found command getBalancerData in protocol b'JK485'
2023-05-14 21:58:11,694:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 115200
2023-05-14 21:58:11,703:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-05-14 21:58:12,811:DEBUG:serialio:send_and_receive@30: serial response was: b''
2023-05-14 21:58:12,820:DEBUG:device:run_command@95: Send and Receive Response b''
2023-05-14 21:58:12,826:INFO:abstractprotocol:decode@224: response passed to decode: b''
2023-05-14 21:58:12,832:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-05-14 21:58:12,838:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'getBalancerData'
2023-05-14 21:58:12,844:DEBUG:abstractprotocol:get_command_defn@67: Found command getBalancerData in protocol b'JK485'
2023-05-14 21:58:12,850:INFO:abstractprotocol:decode@268: Processing response of type POSITIONAL
2023-05-14 21:58:12,855:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 2, 'Header', '']
2023-05-14 21:58:12,862:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Slave Address', '']
2023-05-14 21:58:12,868:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Command Code', '']
2023-05-14 21:58:12,874:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/100', 2, 'Total Battery Voltage', 'V']
2023-05-14 21:58:12,880:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Average Cell Voltage', 'V']
2023-05-14 21:58:12,886:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Number of Cells', '']
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 532, in decode
    processed_responses = self.process_response(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 203, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/protocol_helpers.py", line 95, in Hex2Int
    answer = hexString[0]
IndexError: index out of range

lukolszewski, 2024-08-18:
Same here. "jkbms -p XX:XX:XX:XX:XX -P JK02_32 -I -c getCellData" reads fine via bluetooth. but no combination of protocol use works with the serial port. At the same time sending byte strings like '4E 57 00 13 00 00 00 00 06 03 00 00 00 00 00 00 68 00 00 01 29' returns all the data (at 115200 baud).