jblance/mpp-solar #259: wrong Values on JK-B1A20S15P

dakonga, 2022-09-23:
Command: getInfo - BLE Device Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
header                          55aaeb90
record_type                     03
record_counter                  235
device_model                    JK-B1A20S15P
hardware_version                10.XW
software_version                10.09
up_time                         0D4H0M0S
power-on_times                  2
device_name                     JK-B1A20S15P
device_passcode                 1234
manufacturing_date              ÿÿÿÿÿÿ
passcode                        0000
user_data                       Input Userdata
setup_passcode                  123456

Command: getCellData - BLE Cell Data inquiry
------------------------------------------------------------
Parameter                       Value           Unit
header                          55aaeb90
record_type                     02
record_counter                  252
voltage_cell01                  2.8665478285762414e-31  V
voltage_cell02                  2.8511415645154944e-31  V
voltage_cell03                  3.436633906662891e-31   V
voltage_cell04                  3.5136718097349907e-31  V
voltage_cell05                  0.0             V
voltage_cell06                  0.0             V
voltage_cell07                  0.0             V
voltage_cell08                  0.0             V
voltage_cell09                  0.0             V
voltage_cell10                  0.0             V
voltage_cell11                  0.0             V
voltage_cell12                  0.0             V
voltage_cell13                  3.5733110840282835e-43  V
voltage_cell14                  4.504531366809162e-39   V
voltage_cell15                  4.683620113146802e-39   V
voltage_cell16                  4.5918476754197055e-39  V
voltage_cell17                  4.6836817702792324e-39  V
voltage_cell18                  4.591846274121241e-39   V
voltage_cell19                  7.146622168056567e-44   V
voltage_cell20                  0.0             V
voltage_cell21                  0.0             V
voltage_cell22                  0.0             V
voltage_cell23                  0.0             V
voltage_cell24                  0.0             V
resistance_cell01               0.0             Ohm
resistance_cell02               0.0             Ohm
resistance_cell03               0.0             Ohm
resistance_cell04               0.0             Ohm
resistance_cell05               3.673223664534643e-41   Ohm
resistance_cell06               0.0             Ohm
resistance_cell07               0.0             Ohm
resistance_cell08               1.6806155037128292e-38  Ohm
resistance_cell09               3.0828566215145976e-43  Ohm
resistance_cell10               32.25404357910156       Ohm
resistance_cell11               2.1397827550239957e-40  Ohm
resistance_cell12               3.2229864679470793e-40  Ohm
resistance_cell13               0.0             Ohm
resistance_cell14               5.885453550164232e-44   Ohm
resistance_cell15               1.0652891840388056e-33  Ohm
resistance_cell16               2.0398701745176362e-41  Ohm
resistance_cell17               5.98091583095565e-35    Ohm
resistance_cell18               0.0             Ohm
resistance_cell19               0.0             Ohm
resistance_cell20               0.0             Ohm
resistance_cell21               9.184530524724149e-41   Ohm
resistance_cell22               1.1343520485435074e-36  Ohm
resistance_cell23               0.0             Ohm
resistance_cell24               2.982151746749878       Ohm
resistance_cell25               9.367220608115104e-39   Ohm
average_cell_voltage            1.7516230804060213e-42  V
delta_cell_voltage              2.350988701644575e-38   V
highest_cell                    1
lowest_cell                     1
flags                           0000
uptime                          0D0H0M0S
checksum                        eb
highest_cell_voltage            2.8665478285762414e-31  V
lowest_cell_voltage             2.8665478285762414e-31  V

First 8 Cells are connected and have all near 3.23V in real.

syssi, 2022-09-24:
Did you use the `JK02` protocol?

syssi, 2022-09-25:
Since firmware version 10.07 the payload of the `cell info` frame has changed. Instead of 24 cells the payload supports 32 cells now. This leads to an offset for all further values.

dakonga, 2022-09-26:
thanks for the hint. now i get the correct values. 
maybe it should be mentioned on readme.
it was a bit confused because getInfo shows correct values.

closed 2022-09-26 by dakonga

syssi, 2022-09-26:
The `device info` frame hasn't changed between `JK04` and `JK02`.