jblance/mpp-solar #265: mppsolar on a pc

Meganetpr, 2022-10-09:
Will mpp-solar run on an AMD pc running Ubuntu? The command for install mppsolar timeout

Thanks

jblance, 2022-10-09:
yes it does
what install command are you using (there are many ways to install)
what version of python is installed?

Meganetpr, 2022-10-09:
Ubuntu 22.04.1 on AMD E-350 mini pc
Python version 
Python 3.10.6
command
sudo pip install mppsolar

when run the command get a timeout read pipy.org...

jblance, 2022-10-09:
> when run the command get a timeout read pipy.org...

hopefully this is a typo - should be pypi.org

jblance, 2022-10-09:
maybe root cant access pypi.org?
try without sudo?
or `python -m pip install mppsolar`

Meganetpr, 2022-10-09:
similar outcome:
[image]

jblance, 2022-10-09:
its failing on bluepy - which needs to be compiled
you need to install libglib2.0 first:
`sudo apt-get install libglib2.0-dev`
then try the `sudo pip install mppsolar` command

Meganetpr, 2022-10-09:
That's did it
megasolar@4G-MINI:~$ mpp-solar -v
Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes

Thanks a lot

jblance, 2022-10-10:
any reason to install the dev version?
It seems to be missing poetry-core, try `sudo pip install poetry-core` first

Meganetpr, 2022-10-10:
I did that
Requirement already satisfied: poetry-core in /usr/lib/python3/dist-packages (1.                                                                                                             0.7)
Then try:
flamingo@FLAMINGO:~$ pip install mppsolar
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: mppsolar in /usr/local/lib/python3.10/dist-packages/mppsolar-0.13.2-py3.10.egg (0.13.2)
Requirement already satisfied: paho-mqtt in /usr/lib/python3/dist-packages (from mppsolar) (1.5.1)
Requirement already satisfied: psycopg2 in /usr/lib/python3/dist-packages (from mppsolar) (2.9.2)
Requirement already satisfied: pymongo in /usr/local/lib/python3.10/dist-packages/pymongo-4.2.0-py3.10-linux-x86_64.egg (from mppsolar) (4.2.0)
Requirement already satisfied: pyserial in /usr/local/lib/python3.10/dist-packages/pyserial-3.5-py3.10.egg (from mppsolar) (3.5)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (from mppsolar) (5.4.1)

 But get this when inquire version:
flamingo@FLAMINGO:~$ mpp-solar -v
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mppsolar==0.13.2', 'console_scripts', 'mpp-solar')())
  File "/usr/local/bin/mpp-solar", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mppsolar'

   Thanks again , and sorry to bother... just in learning curve here

jblance, 2022-10-10:
Probably a leftover from the failed sudo install
`sudo pip uninstall mppsolar`
`pip uninstall mppsolar`

Then do one of
`sudo pip install mppsolar` (this one installs as root and installs globally) or 
`pip install mppsolar` (this one only installs for the current user)

Meganetpr, 2022-10-10:
got this:

flamingo@FLAMINGO:~$ sudo pip uninstall mppsolar
Found existing installation: mppsolar 0.13.2
Uninstalling mppsolar-0.13.2:
  Would remove:
    /usr/local/bin/jkbms
    /usr/local/bin/mpp-solar
    /usr/local/bin/powermon
    /usr/local/lib/python3.10/dist-packages/mppsolar-0.13.2-py3.10.egg
Proceed (Y/n)? y
  Successfully uninstalled mppsolar-0.13.2
WARNING: Running pip as the 'root' user can result in broken permissions and con                                                                                                             flicting behaviour with the system package manager. It is recommended to use a v                                                                                                             irtual environment instead: https://pip.pypa.io/warnings/venv
flamingo@FLAMINGO:~$ pip uninstall mppsolar
WARNING: Skipping mppsolar as it is not installed.
flamingo@FLAMINGO:~$ sudo pip install mppsolar
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status                                                                                                             =None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='                                                                                                             pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/mppsolar/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status                                                                                                             =None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='                                                                                                             pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/mppsolar/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status                                                                                                             =None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='                                                                                                             pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/mppsolar/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status                                                                                                             =None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='                                                                                                             pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/mppsolar/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status                                                                                                             =None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='                                                                                                             pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/mppsolar/
ERROR: Could not find a version that satisfies the requirement mppsolar (from ve                                                                                                             rsions: none)
ERROR: No matching distribution found for mppsolar
flamingo@FLAMINGO:~$

Meganetpr, 2022-10-10:
still fail:
flamingo@FLAMINGO:~$ pip install mppsolar
Defaulting to user installation because normal site-packages is not writeable
Collecting mppsolar
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out. (read timeout=15)")': /packages/b9/2f/5bae593c720c8871e81fa24ec1723baf1e7bc8c43e4fa8040fde7947e3c8/mppsolar-0.12.5-py3-none-any.whl
  ............
  WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out. (read timeout=15)")': /packages/b9/2f/5bae593c720c8871e81fa24ec1723baf1e7bc8c43e4fa8040fde7947e3c8/mppsolar-0.12.5-py3-none-any.whl

ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/b9/2f/5bae593c720c8871e81fa24ec1723baf1e7bc8c43e4fa8040fde7947e3c8/mppsolar-0.12.5-py3-none-any.whl (Caused by ReadTimeoutError("HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out. (read timeout=15)"))

flamingo@FLAMINGO:~$

jblance, 2022-10-10:
it really looks like a connection problem, try downloading  mppsolar from pypi.org and installing from local archive

Meganetpr, 2022-10-10:
Downloaded-

flamingo@FLAMINGO:~$ python3 -m pip install /media/flamingo/B0D2-8616/mpp-solar-                                                                                                             0.7.76.tar.gz
Defaulting to user installation because normal site-packages is not writeable
Processing /media/flamingo/B0D2-8616/mpp-solar-0.7.76.tar.gz
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [7 lines of output]
      WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None,                                                                                                              status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(                                                                                                             host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/setupt                                                                                                             ools/
      WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None,                                                                                                              status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(                                                                                                             host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/setupt                                                                                                             ools/
      ERROR: Could not find a version that satisfies the requirement setuptools>                                                                                                             =42 (from versions: none)
      ERROR: No matching distribution found for setuptools>=42
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem wit                                                                                                             h pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with                                                                                                              pip.
flamingo@FLAMINGO:~$ sudo pip install setuptools
[sudo] password for flamingo:
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (59.6.0)

-----setuptools is already there

jblance, 2022-10-10:
You are mixing sudo and non sudo commands and the version you downloaded is
the old version (mpp-solar vs mppsolar)

Download the newer version
Try install mppsolar from local with sudo

Meganetpr, 2022-10-10:
still look for some external dependencies packages even when are already in machine but version is newer..

flamingo@FLAMINGO:~$ sudo python3 -m pip install /media/flamingo/B0D2-8616/mppsolar-0.12.5.tar.gz
Processing /media/flamingo/B0D2-8616/mppsolar-0.12.5.tar.gz
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [7 lines of output]
      WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/poetry-core/
      WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/poetry-core/
      WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/poetry-core/
      WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/poetry-core/
      WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/poetry-core/
      ERROR: Could not find a version that satisfies the requirement poetry-core>=1.0.0 (from versions: none)
      ERROR: No matching distribution found for poetry-core>=1.0.0
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
flamingo@FLAMINGO:~$ sudo pip install poetry-core
Requirement already satisfied: poetry-core in /usr/lib/python3/dist-packages (1.0.7)

jblance, 2022-10-11:
youve downloaded the source archive. try with the build archive https://files.pythonhosted.org/packages/b9/2f/5bae593c720c8871e81fa24ec1723baf1e7bc8c43e4fa8040fde7947e3c8/mppsolar-0.12.5-py3-none-any.whl
Otherwise Im not sure why pip cant reach pypi? Are you behind a proxy?

closed 2023-01-08 by jblance