jblance/mpp-solar #266: PV2 readings not showing on protocols PI30 and PI30MAX

iordache-andrei, 2022-10-12:
Hello, 

I have a setup that pulls from 2 sets of 5 panels each, but I can't get a reading from the second set on command QPIGS. On protocol 16 however I do get the readings, but not the other related info, 

Is this intentional? Or am I missing something? 

Thank you in advance.

jblance, 2022-10-12:
There is a QPIGS2 in PI30MAX protocol that gives some more info (for PV2)

iordache-andrei, 2022-10-12:
[image]

Tried it, should mention is have a mpp-solar pip8048max with a regular usb cable

jblance, 2022-10-12:
Thats weird, I personally have the same inverter and use the QPGS0, QPIGS
and QPIGS2 commands for data collection.
Can you run it with debug on and post the result

iordache-andrei, 2022-10-12:
[image]
Debugged output

jblance, 2022-10-12:
i think maybe its a problem with the hidraw io 'chunking'
ill have a look at the code and try to implement a fix/workaround

eliafino, 2022-11-07:
QPISG2h

iordache-andrei, 2022-12-28:
@eliafino Hi and sorry for the very late response. I've tried the command and update the package to the latest version. 
If I run QPISG2h as suggested I get NAK, I tried QPIGS2h, but i get response too short. 
On both commands debug reports "No command_defn found for ... "

Saentist, 2022-12-28:
> QPIGS2h

You mean QPIGS2 without **h**

iordache-andrei, 2022-12-29:
@Saentist  I've mentioned in the original post that QPIGS2 doesn't particularly work for me. I suspect that running the connection between inverter and  raspberryPi with a usb micro-b to usb cable isn't ideal as opposed to the (as far as I read from the docs) RS-232 connection. 

I can (and currently do so) avoid this by calculating the Charging power generated of the second string of panels by simply getting the voltage and current from QPGS0.  However this has two drawbacks: 

- Commands that return lengthy responses take more time to get processed.
- Commands can not be parallelized (as far I can see the inverter can not accept two commands simultaneously without erroring out )

The extra issue with this type of connection, if indeed it is the case for the command, is that a good RS-232 to USB cable can be hard to acquire because vendors sometimes do not list specialized details like the required +-6V (although they are sold as rs-232 to usb cables, the cheap 5V ones do not cover all devices using DB9 ports), or they are special imports which can inquire import tax higher than the cost of the cable itself.

Saentist, 2022-12-29:
@iordache-andrei See this solution
https://github.com/jblance/mpp-solar/discussions/284#discussioncomment-4494084

lordprize, 2023-02-21:
Hi, I have a similar setup to @iordache-andrei consisting of an MPP Solar 8048MAX and an attached Raspberry Pi 3. Connecting to the 8048MAX with WatchPower (even though USB-IP-Passthrough via usbipd) works, but the QPIGS2 command yields NAK. Other commands work flawlessly.
I have attached the debug output of QPIGS (successful) and QPIGS2 (unsuccessful / NAK).
Please let me know if I can support you, e.g. by trying something out. If you want, I can give you shell access to the connected Raspberry Pi.

Output of:
[root@pipraspi:~# /root/venv/mppsolar/bin/mpp-solar  -p /dev/hidraw0 -P PI30MAX -c QPIGS2 -D >/tmp/qpigs2.txt 2>&1](https://github.com/jblance/mpp-solar/files/10793009/qpigs2.txt)
[root@pipraspi:~# /root/venv/mppsolar/bin/mpp-solar  -p /dev/hidraw0 -P PI30MAX -c QPIGS -D >/tmp/qpigs.txt 2>&1](https://github.com/jblance/mpp-solar/files/10793010/qpigs.txt)

jblance, 2023-06-23:
this should work in current release

closed 2023-06-23 by jblance