jblance/mpp-solar #268: PI18: warning0                      	Command XXXX was rejected

Maxym77, 2022-10-26:
hi! 
i have found my inverter is an PI18.
Command as this:

mpp-solar -p /dev/hidraw1 -P PI18 -c FLAG
Command: FLAG - Query enable/disable flag status
------------------------------------------------------------
Parameter                     	Value           Unit
buzzer_beep                   	Enabled        	    
overload_bypass_function      	Enabled        	    
display_back_to_default_page  	Enabled        	    
overload_restart              	Disabled       	    
over_temperature_restart      	Disabled       	    
backlight_on                  	Enabled        	    
alarm_primary_source_interrupt	Enabled        	    
fault_code_record             	Disabled       	    
reserved                      	0              	    
unknown_value_in_response_9   	0              	  

WORK.

But command as this won't work:

 mpp-solar -p /dev/hidraw1 -P PI18 -c GS
Command: GS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
warning0                      	Command GS was rejected	    

I cant get statistics information from PI18.

you can help me ?

jblance, 2022-10-26:
first run the command in debug mode to get extra info (add `-D` to the
command)
also do a `PI` command to confirm the protocol version
and maybe a simpler command like `ID`

Maxym77, 2022-10-27:
ok. my goal is to change the bulk setting.

pi@energiadomestica:~/apps $ mpp-solar -p /dev/hidraw1 -P PI18 -c MCHGV552,536
Command: MCHGV552,536 - Set Battery Bulk,Float charge voltages 		     (Maunal Option 17,18)
------------------------------------------------------------
Parameter                     	Value           Unit
warning0                      	Command MCHGV552,536 was rejected

You can help me for repair thath command ?

DEBUG:

mpp-solar -p /dev/hidraw1 -P PI18 -c MCHGV552,536 -D
2022-10-27 09:36:07,614:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2022-10-27 09:36:07,615:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb668bf40>, _isConnected=False)
2022-10-27 09:36:07,615:DEBUG:__init__:main@234: udp port 5555
2022-10-27 09:36:07,615:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-10-27 09:36:07,615:INFO:__init__:main@337: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw1 (porttype=None)" using protocol "PI18"
2022-10-27 09:36:07,619:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-10-27 09:36:07,619:DEBUG:device:__init__@33: __init__ args ()
2022-10-27 09:36:07,620:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw1', 'protocol': 'PI18', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb668bf40>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-10-27 09:36:07,620:DEBUG:__init__:get_port_type@38: port matches hidraw
2022-10-27 09:36:07,620:INFO:__init__:get_port@89: Using hidrawio for communications
2022-10-27 09:36:07,622:DEBUG:__init__:get_protocol@12: Protocol PI18
2022-10-27 09:36:07,625:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5d2b658>, protocol <mppsolar.protocols.pi18.pi18 object at 0xb5d2b670>
2022-10-27 09:36:07,625:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5d2b5c8>, 'MCHGV552,536', 'MCHGV552,536', 'screen', None, None)]
2022-10-27 09:36:07,626:INFO:__init__:main@395: Looping 1 commands
2022-10-27 09:36:07,626:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5d2b658>, protocol: <mppsolar.protocols.pi18.pi18 object at 0xb5d2b670> for command: MCHGV552,536, tag: MCHGV552,536, outputs: screen
2022-10-27 09:36:07,626:INFO:device:run_command@274: Running command MCHGV552,536
2022-10-27 09:36:07,626:INFO:pi18:get_full_command@512: Using protocol b'PI18' with 22 commands
2022-10-27 09:36:07,626:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'MCHGV552,536'
2022-10-27 09:36:07,626:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-10-27 09:36:07,627:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POP
2022-10-27 09:36:07,627:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSP
2022-10-27 09:36:07,628:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCP
2022-10-27 09:36:07,628:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2022-10-27 09:36:07,629:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2022-10-27 09:36:07,629:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBT
2022-10-27 09:36:07,630:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-10-27 09:36:07,631:DEBUG:abstractprotocol:get_command_defn@70: Matched: MCHGV552,536 to: MCHGV value: 552
2022-10-27 09:36:07,631:DEBUG:pi18:get_full_command@539: _pre_cmd: b'^S015MCHGV552,536'
2022-10-27 09:36:07,631:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'^S015MCHGV552,536'
2022-10-27 09:36:07,631:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x71 0xb9 0x71b9
2022-10-27 09:36:07,631:DEBUG:pi18:get_full_command@547: full command: b'^S015MCHGV552,536q\xb9\r'
2022-10-27 09:36:07,631:INFO:device:run_command@303: full command b'^S015MCHGV552,536q\xb9\r' for command MCHGV552,536
2022-10-27 09:36:07,631:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'MCHGV552,536'
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POP
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSP
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCP
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBT
2022-10-27 09:36:07,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-10-27 09:36:07,633:DEBUG:abstractprotocol:get_command_defn@70: Matched: MCHGV552,536 to: MCHGV value: 552
2022-10-27 09:36:07,699:DEBUG:hidrawio:send_and_receive@29: length of to_send: 20
[... cut ...] 68bf40>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: MCHGV552,536 - Set Battery Bulk,Float charge voltages 		     (Maunal Option 17,18)
------------------------------------------------------------
Parameter                     	Value           Unit
warning0                      	Command MCHGV552,536 was rejected	    
2022-10-27 09:36:09,173:DEBUG:__init__:main@438: Not daemon, so not looping

jblance, 2022-10-27:
I think the problem for this one is in the way the command is broken up - I believe (but arent certain) that the 'chunking' needs to be different somehow 
```
2022-10-27 09:36:07,699:DEBUG:hidrawio:send_and_receive@29: length of to_send: 20
2022-10-27 09:36:07,699:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2022-10-27 09:36:07,699:DEBUG:hidrawio:send_and_receive@50: send: b'^S015MCH', to_send: b'GV552,536q\xb9\r'
2022-10-27 09:36:08,052:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2022-10-27 09:36:08,052:DEBUG:hidrawio:send_and_receive@50: send: b'GV552,53', to_send: b'6q\xb9\r'
2022-10-27 09:36:08,405:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2022-10-27 09:36:08,406:DEBUG:hidrawio:send_and_receive@50: send: b'6q\xb9\r', to_send: b''
2022-10-27 09:36:09,159:DEBUG:hidrawio:send_and_receive@69: usb response was: b'^0\x1b\xe3\r'
```

Maxym77, 2022-10-27:
> I think the problem for this one is in the way the command is broken up - I believe (but arent certain) that the 'chunking' needs to be different somehow
>
> ```
> 2022-10-27 [...]

and how i can  find the correct junk?

Maxym77, 2022-10-28:
the problem is here?

GV552,53  <----

MCHGV552,536

there is a truncated number...

2022-10-27 09:36:08,052:DEBUG:hidrawio:send_and_receive@50: send: b'GV552,53', to_send: b'6q\xb9\r'

jblance, 2022-12-06:
can you try with version 0.14.5 or later and add `--porttype hidfull` to the command

preussal, 2023-02-18:
This comes even if you refrain from the inverter (e.g. Piri / GS / ET) and then send a command at the same time. e.g. 
change the battery voltage. 
You can only do one. 
Either query or change config.