jblance/mpp-solar #269: How to send data as devices/groups to HomeAssistant

Saentist, 2022-11-05:
Currently I use this command 
```bash
mpp-solar -p /dev/hidraw1 -n VII -o hass_mqtt,screen -c QPIGS#QDI#QFLAG#QPIRI \
-q 192.168.1.12 --mqttuser user --mqttpass pass --mqtttopic /PV/status -n SOLAR_INVERTER --daemon
```
But in home assistant i get bunch of sensors no group/source etc.
How to get them in groups?
```
VII
 QPIGS
 - entites
 QDI
 - entites
 QFLAG
 -entites
 QPIRI
 -entites
```

jblance, 2022-11-06:
Im not sure what you are after.
Are you wanting to change how it looks in the configure screen?
If you know what needs to be sent to the broker I can look at changing the message

Saentist, 2022-11-07:
Lets imagine that we have multiple inverters
only difference is in entities name string.
It will be more usefully if grouped by source.

jblance, 2022-11-07:
Im still not clear what you mean by 'group'
When I was using 2 inverters, I was using the 'tag' option to provide
entity names that we easy to select by wildcard

Is this what you mean?

Saentist, 2022-11-07:
I try to use this software
https://github.com/lluiscab/MPQTT
(very limited support to inverters commands response and direct crash in my case)
but perfect organized sending of data
I see Inverter sensors as "Device with sensors" with is more organized, then bunch of sensors
[image]

jblance, 2022-11-07:
Can you grab some of the discovery mqtt messages from mpqtt

Saentist, 2022-11-07:
> Can you grab some of the discovery mqtt messages from mpqtt

Is there any guide how to do it?

jblance, 2022-11-07:
In 'Settings' for MQTT go to Configure
Then you can listen to messages:
(my autoconfigure prefix us homeassistant)
Need to grab the config message related to mpqtt

[image: image.png]

Saentist, 2022-11-07:
[mqtt-1.json.txt](https://github.com/jblance/mpp-solar/files/9955512/mqtt-1.json.txt)
this is mqtt info

jblance, 2022-11-08:
ok, first attempt in version 0.13.7 which uses the name field
[image]

jblance, 2022-11-08:
oh and new output module `hassd_mqtt`

Saentist, 2022-11-08:
Any example of command used?

jblance, 2022-11-08:
Same as your original but change `-o hass_mqtt` to `-o hassd_mqtt`

Saentist, 2022-11-08:
I get some strange results in screen

```shell
 mpp-solar -p /dev/hidraw1 -o hassd_mqtt,screen -c QID#QMOD#QPI#QPIGS#QPIRI -q 192.168.1.12 --mqttuser user --mqttpass 'pass' -n VMIII-5000 --daemon
Service Initializing ...
Getting results from device: mppsolar device - name: VMIII-5000, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7ff065ba5220>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7ff0653f21f0> for command: QID, tag: QID, outputs: hassd_mqtt,screen
{"name": "QID Serial Number", "state_topic": "homeassistant/sensor/mpp_QID_serial_number/state", "unit_of_measurement": "", "unique_id": "mpp_QID_serial_number", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
Parameter                       Value           Unit
serial_number                   55355535553555
Getting results from device: mppsolar device - name: VMIII-5000, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7ff065ba5220>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7ff0653f21f0> for command: QMOD, tag: QMOD, outputs: hassd_mqtt,screen
{"name": "QMOD Device Mode", "state_topic": "homeassistant/sensor/mpp_QMOD_device_mode/state", "unit_of_measurement": "", "unique_id": "mpp_QMOD_device_mode", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
Parameter                       Value           Unit
device_mode                     Line
Getting results from device: mppsolar device - name: VMIII-5000, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7ff065ba5220>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7ff0653f21f0> for command: QPI, tag: QPI, outputs: hassd_mqtt,screen
{"name": "QPI Protocol ID", "state_topic": "homeassistant/sensor/mpp_QPI_protocol_id/state", "unit_of_measurement": "", "unique_id": "mpp_QPI_protocol_id", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
Parameter                       Value           Unit
protocol_id                     PI30
Getting results from device: mppsolar device - name: VMIII-5000, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7ff065ba5220>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7ff0653f21f0> for command: QPIGS, tag: QPIGS, outputs: hassd_mqtt,screen
{"name": "QPIGS AC Input Voltage", "state_topic": "homeassistant/sensor/mpp_QPIGS_ac_input_voltage/state", "unit_of_measurement": "V", "unique_id": "mpp_QPIGS_ac_input_voltage", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS AC Input Frequency", "state_topic": "homeassistant/sensor/mpp_QPIGS_ac_input_frequency/state", "unit_of_measurement": "Hz", "unique_id": "mpp_QPIGS_ac_input_frequency", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS AC Output Voltage", "state_topic": "homeassistant/sensor/mpp_QPIGS_ac_output_voltage/state", "unit_of_measurement": "V", "unique_id": "mpp_QPIGS_ac_output_voltage", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS AC Output Frequency", "state_topic": "homeassistant/sensor/mpp_QPIGS_ac_output_frequency/state", "unit_of_measurement": "Hz", "unique_id": "mpp_QPIGS_ac_output_frequency", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS AC Output Apparent Power", "state_topic": "homeassistant/sensor/mpp_QPIGS_ac_output_apparent_power/state", "unit_of_measurement": "VA", "unique_id": "mpp_QPIGS_ac_output_apparent_power", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS AC Output Active Power", "state_topic": "homeassistant/sensor/mpp_QPIGS_ac_output_active_power/state", "unit_of_measurement": "W", "unique_id": "mpp_QPIGS_ac_output_active_power", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}, "state_class": "measurement", "device_class": "power"}
{"name": "QPIGS AC Output Load", "state_topic": "homeassistant/sensor/mpp_QPIGS_ac_output_load/state", "unit_of_measurement": "%", "unique_id": "mpp_QPIGS_ac_output_load", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS BUS Voltage", "state_topic": "homeassistant/sensor/mpp_QPIGS_bus_voltage/state", "unit_of_measurement": "V", "unique_id": "mpp_QPIGS_bus_voltage", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS Battery Voltage", "state_topic": "homeassistant/sensor/mpp_QPIGS_battery_voltage/state", "unit_of_measurement": "V", "unique_id": "mpp_QPIGS_battery_voltage", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
{"name": "QPIGS Battery Charging Current", "state_topic": "homeassistant/sensor/mpp_QPIGS_battery_charging_current/state", "unit_of_measurement": "A", "unique_id": "mpp_QPIGS_battery_charging_current", "force_update": "true", "device": {"name": "VMIII-5000", "identifiers": ["mppsolar"], "manufacturer": "MPP-Solar"}}
[... cut ...]
charger_source_priority         Only solar charging permitted
max_parallel_units              1               units
machine_type                    Off Grid
topology                        transformerless
output_mode                     single machine output
battery_redischarge_voltage     54.0            V
pv_ok_condition                 As long as one unit of inverters has connect PV, parallel system will consider PV OK
pv_power_balance                PV input max power will be the sum of the max charged power and loads power
Sleeping for 60 sec

```

jblance, 2022-11-08:
oh, I left a print statement uncommented
v 0.13.9 should fix this

Saentist, 2022-11-08:
How can be implement this predefined type of setting icon, 
for sensors to new super module **hassd_mqtt**?
```rs
    // Register QPIGS Sensors
    register_sensor(client, cfg, "qpigs", "grid_voltage", "Grid Voltage", Some("V".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "grid_frequency", "Grid Frequency", Some("Hz".to_string()), "current-ac").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_voltage", "Out Voltage", Some("V".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_frequency", "Out Frequency", Some("Hz".to_string()), "current-ac").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_apparent_power", "Out apparent power", Some("W".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_active_power", "Out active power", Some("W".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "out_load_percent", "Out load percent", Some("%".to_string()), "brightness-percent").await?;
    register_sensor(client, cfg, "qpigs", "bus_voltage", "Bus Voltage", Some("V".to_string()), "details").await?;
    register_sensor(client, cfg, "qpigs", "battery_voltage", "Battery Voltage", Some("V".to_string()), "battery-outline").await?;
    register_sensor(client, cfg, "qpigs", "battery_charge_current", "Battery charge current", Some("A".to_string()), "current-dc").await?;
    register_sensor(client, cfg, "qpigs", "battery_capacity", "Battery capacity", Some("%".to_string()), "battery-outline").await?;
    register_sensor(client, cfg, "qpigs", "inverter_heat_sink_temp", "Heat sink temperature", Some("°C".to_string()), "details").await?;
    register_sensor(client, cfg, "qpigs", "pv_input_current", "PV Input Current", Some("A".to_string()), "solar-power").await?;
    register_sensor(client, cfg, "qpigs", "pv_input_voltage", "PV Input Voltage", Some("V".to_string()), "solar-power").await?;
    register_sensor(client, cfg, "qpigs", "battery_scc_voltage", "Battery SCC Voltage", Some("V".to_string()), "current-dc").await?;
    register_sensor(client, cfg, "qpigs", "battery_discharge_current", "Battery discharge current", Some("A".to_string()), "battery-negative").await?;
    register_sensor(client, cfg, "qpigs", "device_status.charge_status", "Device charge status", None, "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "device_status.active_load", "Active load", None, "power").await?;
```

Saentist, 2022-12-01:
@jblance 
Is value type can be defined?
Aka V, A, W, kWh, Hz, °C/F, % ... etc.
Idea is to be used with Energy monitoring in HA.

RoySalisbury, 2022-12-14:
I'd like to also use this new hassd_mqtt output format.  How do I go about upgrading my install to get the latest version?  I tried:

sudo pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"

But I get errors about modules not installed.

Saentist, 2022-12-14:
> I'd like to also use this new hassd_mqtt output format. How do I go about upgrading my install to get the latest version? I tried:
> 
> sudo pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"
> 
> But I get errors about modules not installed.

I have same problem and need to force update

jblance, 2022-12-14:
The `pip install mppsolar` command should get a suitable version.

Though if you've installed from source you may need to do some cleanup
first

jblance, 2022-12-14:
> No, there is a problem of pip with stuck in some old version in his core
> components.
> There are few lines yellow messages with version conflicts.

Can you post the relevant bits so I can see if I've messed something up

Saentist, 2022-12-14:
@RoySalisbury post complete terminal output whet run update command.

RoySalisbury, 2022-12-17:
pi@workshop-pi:~ $ sudo pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining mppsolar from git+https://github.com/jblance/mpp-solar.git#egg=mppsolar
  Updating ./src/mppsolar clone
  Running command git fetch -q --tags
  Running command git reset --hard -q 6ba5268673280c6c86c6053b01d4d3d86cb2a8df
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: PyYAML in /usr/local/lib/python3.9/dist-packages (from mppsolar) (6.0)
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from mppsolar) (3.5b0)
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.9/dist-packages (from mppsolar) (1.6.1)
Installing collected packages: mppsolar
  Attempting uninstall: mppsolar
    Found existing installation: mppsolar 0.12.5
    Uninstalling mppsolar-0.12.5:
      Successfully uninstalled mppsolar-0.12.5
  Running setup.py develop for mppsolar
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/src/mppsolar/setup.py'"'"'; __file__='"'"'/home/pi/src/mppsolar/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: /home/pi/src/mppsolar/
    Complete output (3 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    ----------------------------------------
  Rolling back uninstall of mppsolar
  Moving to /usr/local/bin/jkbms
   from /tmp/pip-uninstall-dpk4i3gf/jkbms
  Moving to /usr/local/bin/mpp-solar
   from /tmp/pip-uninstall-dpk4i3gf/mpp-solar
  Moving to /usr/local/bin/powermon
   from /tmp/pip-uninstall-dpk4i3gf/powermon
  Moving to /usr/local/lib/python3.9/dist-packages/mppsolar-0.12.5.dist-info/
   from /usr/local/lib/python3.9/dist-packages/~ppsolar-0.12.5.dist-info
  Moving to /usr/local/lib/python3.9/dist-packages/mppsolar/
   from /usr/local/lib/python3.9/dist-packages/~ppsolar
ERROR: Command errored out with exit status 1: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/src/mppsolar/setup.py'"'"'; __file__='"'"'/home/pi/src/mppsolar/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.
pi@workshop-pi:~ $

RoySalisbury, 2022-12-17:
And if just ran as a "upgrade from the original install:

pi@workshop-pi:~ $ sudo pip install mppsolar
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mppsolar in /usr/local/lib/python3.9/dist-packages (0.12.5)
Requirement already satisfied: pymongo in /usr/local/lib/python3.9/dist-packages (from mppsolar) (4.2.0)
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.9/dist-packages (from mppsolar) (1.6.1)
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from mppsolar) (3.5b0)
Requirement already satisfied: bluepy in /usr/local/lib/python3.9/dist-packages (from mppsolar) (1.3.0)
pi@workshop-pi:~ $

jblance, 2022-12-18:
use
`sudo pip install -U mppsolar`
to force update

RoySalisbury, 2022-12-18:
Great.. That works.  The device group seems to work now and I will try with two different inverter sections in the conf file.  I tried with my JK BMS, but the PI bluetooth is so bad that Its just not worth it (will never stay running very log).

jblance, 2022-12-18:
> I tried with my JK BMS, but the PI bluetooth is so bad that Its just not worth it (will never stay running very log).

I have my pi3 doing ble connection to my jkbms working ok
I have started a new approach that will improve on the ble access but it is a complete refactor as some of the 'bad' connectivity is baked in to the current code. 
There is the esp32 esphome solution that i have heard works well for the jkbms https://github.com/syssi/esphome-jk-bms by @syssi

jblance, 2023-01-17:
Home assistant devices added and initial work with icons and device classes added

closed 2023-01-17 by jblance