jblance/mpp-solar #272: Cannot connect to JK-BD6A24S10P

DryZus, 2022-12-05:
Hi,
I am trying to connect to my bms but it seems to give no response.
I can clearly see the device scanning with `sudo hcitool lescan` where it shows as below:
pi@raspberrypivideo:~/Desktop/mppsolar $ sudo hcitool lescan
LE Scan ...
C8:47:8C:E8:44:97 (unknown)
3B:BD:59:BE:64:F6 (unknown)
72:08:3F:83:9D:C2 (unknown)
C8:47:8C:E8:44:97 JK-BD6A24S10P
D9:04:8B:26:6A:4C (unknown)
F4:52:E2:49:16:FE (unknown)

I tried to connect with the command `sudo jkbms -P jk02 -p C8:47:8C:E8:44:97 -c getInfo`, but also `sudo jkbms -p C8:47:8C:E8:44:97 -c getInfo` but in both cases all I get is:

2022-12-05 12:08:23,636:WARNING:jkbleio:ble_connect@61: Cannot connect to mac C8:47:8C:E8:44:97 - exceeded 3 attempts
2022-12-05 12:08:23,638:ERROR:jkbleio:send_and_receive@40: Failed to connect to C8:47:8C:E8:44:97
Parameter                       Value           Unit
error                           No response

I see that it is not connecting but still I don't know what could be the proble.
Below is the info from the device connected to my phone.
[image]

syssi, 2022-12-05:
Which ESPHome version do you use? Did you stick to the ESP-IDF framework? Which branch / version of this external component are use using?

DryZus, 2022-12-05:
I am not using an esp device to run it. Everything is running from a raspberry pi.

DryZus, 2022-12-05:
Apparently I am able to connect once every 5-10-15 tries.
```
(mppsolar) pi@raspberrypivideo:~/Desktop/mppsolar $ jkbms -p C8:47:8C:E8:44:97 -n "JK-BD6A24S10P" -P JK02
Command: getCellData - BLE Cell Data inquiry
------------------------------------------------------------
Parameter                       Value           Unit
header                          55aaeb90
record_type                     02
record_counter                  87
voltage_cell01                  3.245           V
voltage_cell02                  3.241           V
voltage_cell03                  3.242           V
voltage_cell04                  3.242           V
voltage_cell05                  3.242           V
voltage_cell06                  3.244           V
voltage_cell07                  3.241           V
voltage_cell08                  3.245           V
voltage_cell09                  3.242           V
voltage_cell10                  3.245           V
voltage_cell11                  3.244           V
voltage_cell12                  3.245           V
voltage_cell13                  3.244           V
voltage_cell14                  3.244           V
voltage_cell15                  3.245           V
voltage_cell16                  3.245           V
voltage_cell17                  0.0             V
voltage_cell18                  0.0             V
voltage_cell19                  0.0             V
voltage_cell20                  0.0             V
voltage_cell21                  0.0             V
voltage_cell22                  0.0             V
voltage_cell23                  0.0             V
voltage_cell24                  0.0             V
average_cell_voltage            3.244           V
delta_cell_voltage              0.004           V
current_balancer                0.256
resistance_cell01               0.135           Ohm
resistance_cell02               0.139           Ohm
resistance_cell03               0.139           Ohm
resistance_cell04               0.14            Ohm
resistance_cell05               0.138           Ohm
resistance_cell06               0.142           Ohm
resistance_cell07               0.143           Ohm
resistance_cell08               0.142           Ohm
resistance_cell09               0.145           Ohm
resistance_cell10               0.146           Ohm
resistance_cell11               0.145           Ohm
resistance_cell12               0.146           Ohm
resistance_cell13               0.146           Ohm
resistance_cell14               0.143           Ohm
resistance_cell15               0.143           Ohm
resistance_cell16               0.142           Ohm
resistance_cell17               0.0             Ohm
resistance_cell18               0.0             Ohm
resistance_cell19               0.0             Ohm
resistance_cell20               0.0             Ohm
resistance_cell21               0.0             Ohm
resistance_cell22               0.0             Ohm
resistance_cell23               0.0             Ohm
resistance_cell24               0.0             Ohm
battery_voltage                 51.897          V
battery_power                   48368           W
balance_current                 -0.932          A
battery_t1                      15.2            °C
battery_t2                      14.4            °C
mos_temp                        20.3            °C
percent_remain                  45              %
capacity_remain                 104.766         Ah
nominal_capacity                228.0           Ah
cycle_count                     0
cycle_capacity                  224.103         Ah
time                            1D15H20M42S
current_charge                  0.003
current_discharge               1.28
(mppsolar) pi@raspberrypivideo:~/Desktop/mppsolar $ jkbms -p C8:47:8C:E8:44:97 -n "JK-BD6A24S10P" -P JK02
2022-12-05 12:45:42,859:WARNING:jkbleio:ble_connect@61: Cannot connect to mac C8:47:8C:E8:44:97 - exceeded 3 attempts
2022-12-05 12:45:42,860:ERROR:jkbleio:send_and_receive@40: Failed to connect to C8:47:8C:E8:44:97
Parameter                       Value           Unit
error                           No response
(mppsolar) pi@raspberrypivideo:~/Desktop/mppsolar $
```
I don't understand why it is allowing me to connect only once in a lot of tries.

jblance, 2022-12-06:
Is anything else connected to the BMS?
How far apart are the Pi and the BMS?

Can you run it with debug on (`-D`) and paste a failure debug and a success debug

DryZus, 2022-12-06:
There is nothing else connected to the BMS, theoretically, since only my phone has been used in the past days but right now I am 100km+ away. There is another raspberry pi that runs solar assistant and is 1m away from that one, I think it shouldn't interfere tho.
Here is the failed attempt to connect:
`pi@raspberrypivideo:~ $ jkbms -p C8:47:8C:E8:44:97 -n JK-BD6A24S10P -P JK02 -c getCellData -D
2022-12-06 23:46:20,913:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.5
2022-12-06 23:46:20,913:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7fb4d77f70>, _isConnected=False)
2022-12-06 23:46:20,914:DEBUG:__init__:main@240: udp port 5555
2022-12-06 23:46:20,914:DEBUG:__init__:main@242: Using Postgres None
2022-12-06 23:46:20,914:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-06 23:46:20,915:INFO:__init__:main@347: Creating device "JK-BD6A24S10P" (type: "jkbms") on port "C8:47:8C:E8:44:97 (porttype=None)" using protocol "JK02"
2022-12-06 23:46:20,922:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-12-06 23:46:20,922:DEBUG:device:__init__@33: __init__ args ()
2022-12-06 23:46:20,922:DEBUG:device:__init__@34: __init__ kwargs {'name': 'JK-BD6A24S10P', 'port': 'C8:47:8C:E8:44:97', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7fb4d77f70>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-06 23:46:20,923:DEBUG:__init__:get_port_type@54: port matches jkble ':'
2022-12-06 23:46:20,923:INFO:__init__:get_port@107: Using jkbleio for communications
2022-12-06 23:46:20,996:DEBUG:__init__:get_protocol@13: Protocol JK02
2022-12-06 23:46:21,018:DEBUG:device:__init__@38: __init__ name JK-BD6A24S10P, port <mppsolar.inout.jkbleio.JkBleIO object at 0x7fb4564070>, protocol JKBMS BLE 2 byte data communication protocol handler
2022-12-06 23:46:21,019:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7fb454b130>, 'getCellData', 'getCellData', 'screen', None, None)]
2022-12-06 23:46:21,019:INFO:__init__:main@404: Looping 1 commands
2022-12-06 23:46:21,020:INFO:__init__:main@414: Getting results from device: jkbms device - name: JK-BD6A24S10P, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x7fb4564070>, protocol: JKBMS BLE 2 byte data communication protocol handler for command: getCellData, tag: getCellData, outputs: screen
2022-12-06 23:46:21,020:INFO:device:run_command@274: Running command getCellData
2022-12-06 23:46:21,021:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-12-06 23:46:21,021:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-12-06 23:46:21,021:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-12-06 23:46:21,022:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-12-06 23:46:21,022:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getCellData
2022-12-06 23:46:21,022:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-06 23:46:21,023:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-06 23:46:21,023:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2022-12-06 23:46:21,023:INFO:device:run_command@303: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2022-12-06 23:46:21,024:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-12-06 23:46:21,024:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-12-06 23:46:21,024:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-12-06 23:46:21,024:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-12-06 23:46:21,025:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-12-06 23:46:21,025:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-12-06 23:46:21,025:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-12-06 23:46:21,026:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getCellData
2022-12-06 23:46:21,026:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-06 23:46:21,026:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-06 23:46:21,027:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
[... cut ...]
2022-12-06 23:46:22,607:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['No response', '']}, 'tag': 'getCellData', 'name': 'JK-BD6A24S10P', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7fb4d77f70>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           No response
2022-12-06 23:46:22,609:DEBUG:__init__:main@449: Not daemon, so not looping`

DryZus, 2022-12-06:
Here is a succesfull one:
`pi@raspberrypivideo:~ $ jkbms -p C8:47:8C:E8:44:97 -n JK-BD6A24S10P -P JK02 -c getCellData -D
2022-12-07 00:01:15,006:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.5
2022-12-07 00:01:15,006:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7f8f34ff70>, _isConnected=False)
2022-12-07 00:01:15,007:DEBUG:__init__:main@240: udp port 5555
2022-12-07 00:01:15,007:DEBUG:__init__:main@242: Using Postgres None
2022-12-07 00:01:15,007:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-07 00:01:15,008:INFO:__init__:main@347: Creating device "JK-BD6A24S10P" (type: "jkbms") on port "C8:47:8C:E8:44:97 (porttype=None)" using protocol "JK02"
2022-12-07 00:01:15,015:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-12-07 00:01:15,016:DEBUG:device:__init__@33: __init__ args ()
2022-12-07 00:01:15,016:DEBUG:device:__init__@34: __init__ kwargs {'name': 'JK-BD6A24S10P', 'port': 'C8:47:8C:E8:44:97', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7f8f34ff70>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-07 00:01:15,017:DEBUG:__init__:get_port_type@54: port matches jkble ':'
2022-12-07 00:01:15,017:INFO:__init__:get_port@107: Using jkbleio for communications
2022-12-07 00:01:15,092:DEBUG:__init__:get_protocol@13: Protocol JK02
2022-12-07 00:01:15,108:DEBUG:device:__init__@38: __init__ name JK-BD6A24S10P, port <mppsolar.inout.jkbleio.JkBleIO object at 0x7f8eb3c070>, protocol JKBMS BLE 2 byte data communication protocol handler
2022-12-07 00:01:15,108:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7f8eb23130>, 'getCellData', 'getCellData', 'screen', None, None)]
2022-12-07 00:01:15,109:INFO:__init__:main@404: Looping 1 commands
2022-12-07 00:01:15,109:INFO:__init__:main@414: Getting results from device: jkbms device - name: JK-BD6A24S10P, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x7f8eb3c070>, protocol: JKBMS BLE 2 byte data communication protocol handler for command: getCellData, tag: getCellData, outputs: screen
2022-12-07 00:01:15,109:INFO:device:run_command@274: Running command getCellData
2022-12-07 00:01:15,110:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-12-07 00:01:15,110:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-12-07 00:01:15,110:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-12-07 00:01:15,111:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-12-07 00:01:15,111:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getCellData
2022-12-07 00:01:15,111:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-07 00:01:15,112:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-07 00:01:15,112:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2022-12-07 00:01:15,112:INFO:device:run_command@303: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2022-12-07 00:01:15,112:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-12-07 00:01:15,113:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-12-07 00:01:15,113:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-12-07 00:01:15,113:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-12-07 00:01:15,114:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-12-07 00:01:15,114:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-12-07 00:01:15,114:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-12-07 00:01:15,114:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getCellData
2022-12-07 00:01:15,115:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-07 00:01:15,115:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-12-07 00:01:15,115:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2022-12-07 00:01:15,116:INFO:jkbleio:send_and_receive@29: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2022-12-07 00:01:15,116:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-12-07 00:01:15,116:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
[... cut ...]
mos_temp                        19.7            °C
percent_remain                  36              %
capacity_remain                 82.172          Ah
nominal_capacity                228.0           Ah
cycle_count                     1
cycle_capacity                  379.786         Ah
time                            3D2H20M40S
current_charge                  0.003
current_discharge               11.776
2022-12-07 00:01:26,592:DEBUG:__init__:main@449: Not daemon, so not looping
pi@raspberrypivideo:~ $

VKNikov, 2023-08-05:
I can also relate to this problem. I have JK-B2A24S20P with  hardware_version   10.XW and
software_version  10.10 and most of the time the execution of the commands just fails. The bms is inside a metal box but the raspberrypi is right on top of that box, shouldn't be that big of a problem. 

Anyway, even when read the info is totally off. This is a LiFePo4 battery, totally different cell voltage.

`Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Parameter            Value              Unit
header               55aaeb90
record_type          02
record_counter       240
voltage_cell01       3.806418333456994e-31      V
voltage_cell02       3.806417863259254e-31      V
voltage_cell03       3.7756034543467984e-31     V
voltage_cell04       3.83723297746832e-31       V
voltage_cell05       3.8218253028143518e-31     V
voltage_cell06       3.821825773012092e-31      V
voltage_cell07       3.821825537913222e-31      V
voltage_cell08       3.806418333456994e-31      V
voltage_cell09       0.0                V
voltage_cell10       0.0                V
voltage_cell11       0.0                V
voltage_cell12       0.0                V
voltage_cell13       9.183409485952689e-41      V
voltage_cell14       2.8015879937553203e-40     V
voltage_cell15       4.683969036464419e-39      V
voltage_cell16       4.8673555651921434e-39     V
voltage_cell17       4.6836817702792324e-39     V
voltage_cell18       4.6836817702792324e-39     V
voltage_cell19       4.6836817702792324e-39     V
voltage_cell20       4.7755172664372236e-39     V
voltage_cell21       4.683684572876161e-39      V
voltage_cell22       4.6836817702792324e-39     V
voltage_cell23       7.006492321624085e-44      V
voltage_cell24       0.0                V
resistance_cell01    0.0                Ohm
resistance_cell02    0.0                Ohm
resistance_cell03    0.0                Ohm
resistance_cell04    0.0                Ohm
resistance_cell05    7.443837572339861e-41      Ohm
resistance_cell06    8.61994737344768e-41       Ohm
resistance_cell07    nan                Ohm
resistance_cell08    3.214326723697245e-38      Ohm
resistance_cell09    4.680336870844889e-43      Ohm
resistance_cell10    9.223372036854776e+18      Ohm
resistance_cell11    4.0537182365527606e-40     Ohm
resistance_cell12    4.259947331547444e-40      Ohm
resistance_cell13    1.5414283107572988e-44     Ohm
resistance_cell14    4.889627802984122e-39      Ohm
resistance_cell15    2.7330467864372477e-37     Ohm
resistance_cell16    2.5748915333907087e-39     Ohm
resistance_cell17    4.3447008760220265e-35     Ohm
resistance_cell18    0.0                Ohm
resistance_cell19    0.0                Ohm
resistance_cell20    0.0                Ohm
resistance_cell21    9.184530524724149e-41      Ohm
resistance_cell22    1.1578619355599532e-36     Ohm
resistance_cell23    5.510129769479473e-40      Ohm
resistance_cell24    2.9848883152008057 Ohm
resistance_cell25    0.0                Ohm
average_cell_voltage 1.7516230804060213e-42     V
delta_cell_voltage   2.350988701644575e-38      V
highest_cell         1
lowest_cell          1
flags                0000
uptime               0D0H0M0S
checksum             73
highest_cell_voltage 3.806418333456994e-31      V
lowest_cell_voltage  3.806418333456994e-31      V
resistance_cell02    0.0                Ohm
resistance_cell03    0.0                Ohm
resistance_cell04    0.0                Ohm
resistance_cell05    7.443837572339861e-41      Ohm
resistance_cell06    8.61994737344768e-41       Ohm
resistance_cell07    nan                Ohm
resistance_cell08    3.214326723697245e-38      Ohm
resistance_cell09    4.680336870844889e-43      Ohm
resistance_cell10    9.223372036854776e+18      Ohm
resistance_cell11    4.0537182365527606e-40     Ohm
resistance_cell12    4.259947331547444e-40      Ohm
resistance_cell13    1.5414283107572988e-44     Ohm
resistance_cell14    4.889627802984122e-39      Ohm
resistance_cell15    2.7330467864372477e-37     Ohm
resistance_cell16    2.5748915333907087e-39     Ohm
resistance_cell17    4.3447008760220265e-35     Ohm
resistance_cell18    0.0                Ohm
resistance_cell19    0.0                Ohm
resistance_cell20    0.0                Ohm
resistance_cell21    9.184530524724149e-41      Ohm
resistance_cell22    1.1578619355599532e-36     Ohm
resistance_cell23    5.510129769479473e-40      Ohm
resistance_cell24    2.9848883152008057 Ohm
resistance_cell25    0.0                Ohm
average_cell_voltage 1.7516230804060213e-42     V
delta_cell_voltage   2.350988701644575e-38      V
highest_cell         1
lowest_cell          1
flags                0000
uptime               0D0H0M0S
checksum             73
highest_cell_voltage 3.806418333456994e-31      V
lowest_cell_voltage  3.806418333456994e-31      V`

jblance, 2023-08-10:
what protocol are you using?
can you post a debug run (add `-D` to command)

jblance, 2023-08-15:
> Apparently I am able to connect once every 5-10-15 tries.
> 
> 
> I don't understand why it is allowing me to connect only once in a lot of tries.

I dont understand either, should either work or fail consistently
Its either due to the way the code tries to reconnect each time or something wrong with the ble library Im using (or the way Im using it)
I use a JKBMS with BLE (earlier HW version) and it connects and reconnects fine (until the raspberry ble stack crashes)
Ive got a newer JKBMS waiting for me to connect it to some cells for testing etc so this may help me troubleshoot

jblance, 2023-08-15:
> I can also relate to this problem. I have JK-B2A24S20P with hardware_version 10.XW and software_version 10.10 and most of the time the execution of the commands just fails. The bms is inside a metal box but the raspberrypi is right on top of that box, shouldn't be that big of a problem.
> 

Have you tried protocol `JK02` ?

VKNikov, 2023-08-15:
I will try to provide more info later on this week when I have more time. Have to use the raspberryPi for something else right now.

VKNikov, 2023-11-09:
OK, took longer then expected but I tried again. With `JK02` protocol the data seems to be correct, so that at least seems not to be an issue. The intermittent working is still present though.

closed 2024-02-14 by jblance