jblance/mpp-solar #273: --filter --exclfilter re.search - may need addition to documentation

frederickjh, 2022-12-09:
Hi! 

I have been trying to use the `--filter` and `--exclfilter` options (which the documentation says uses `re.search` to include or exclude multiple terms but cannot seem to figure out how to do it. Maybe it is not possible as I see Python also has a `re.findall`. 

Any advice on how this works would be appreciated!
Frederick

jblance, 2022-12-10:
what is the full command you are using

frederickjh, 2022-12-10:
Here is the sanitized terminal session where I tried to filter on both **pv** and **battery**. That way you can see what I tried. At the endI tried to add two '--filter' to the command but then it only seemed to filter on the last one.

```
$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter pv

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
pv1_input_current             	6.3            	A   
pv1_input_voltage             	234.4          	V   
pv1_charging_power            	1482           	W   

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter pv battery

usage: mpp-solar [-h] [-n NAME] [-p PORT] [--porttype PORTTYPE] [-P {PI00,PI16,PI17,PI18,PI30,PI30MAX,PI30REVO,PI41,VED,DALY,DALY40}] [-T TAG] [-b BAUD]
                 [-o [OUTPUT]] [--keepcase] [--filter FILTER] [--exclfilter EXCLFILTER] [-q MQTTBROKER] [--mqttport MQTTPORT] [--mqtttopic MQTTTOPIC]
                 [--mqttuser MQTTUSER] [--mqttpass MQTTPASS] [--udpport UDPPORT] [--mongo_url MONGO_URL] [--mongo_db MONGO_DB] [-c [COMMAND]]
                 [-C [CONFIGFILE]] [--daemon] [--getstatus] [--getsettings] [-v] [-D] [-I]
mpp-solar: error: unrecognized arguments: battery

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter pv --filter battery

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
battery_voltage               	53.6           	V   
battery_charging_current      	27             	A   
battery_capacity              	94             	%   
battery_voltage_from_scc      	0.0            	V   
battery_discharge_current     	0              	A   
is_battery_voltage_to_steady_while_charging	0              	bool
battery_voltage_offset_for_fans_on	0              	10mV

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter "pv | battery"

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter "pv, battery" 

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter "pv.+battery"

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter "pv battery"

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter pv battery

usage: mpp-solar [-h] [-n NAME] [-p PORT] [--porttype PORTTYPE] [-P {PI00,PI16,PI17,PI18,PI30,PI30MAX,PI30REVO,PI41,VED,DALY,DALY40}] [-T TAG] [-b BAUD]
                 [-o [OUTPUT]] [--keepcase] [--filter FILTER] [--exclfilter EXCLFILTER] [-q MQTTBROKER] [--mqttport MQTTPORT] [--mqtttopic MQTTTOPIC]
                 [--mqttuser MQTTUSER] [--mqttpass MQTTPASS] [--udpport UDPPORT] [--mongo_url MONGO_URL] [--mongo_db MONGO_DB] [-c [COMMAND]]
                 [-C [CONFIGFILE]] [--daemon] [--getstatus] [--getsettings] [-v] [-D] [-I]
mpp-solar: error: unrecognized arguments: battery

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter pv --filter battery

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
battery_voltage               	53.4           	V   
battery_charging_current      	16             	A   
battery_capacity              	92             	%   
battery_voltage_from_scc      	0.0            	V   
battery_discharge_current     	0              	A   
is_battery_voltage_to_steady_while_charging	0              	bool
battery_voltage_offset_for_fans_on	0              	10mV

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter battery --filter pv

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
pv1_input_current             	5.0            	A   
pv1_input_voltage             	236.4          	V   
pv1_charging_power            	1182           	W   

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter battery --filter pv

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
pv1_input_current             	4.9            	A   
pv1_input_voltage             	238.1          	V   
pv1_charging_power            	1167           	W
```

jblance, 2022-12-10:
Try
` --filter "pv|battery"`

Ie with out spaces.

frederickjh, 2022-12-10:
Thanks! That worked.

```
$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus --filter "pv|battery"                  4021ms  [Fri Dec  9 10:58:14 AM]
Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
battery_voltage               	52.3           	V   
battery_charging_current      	0              	A   
battery_capacity              	81             	%   
pv1_input_current             	0.0            	A   
pv1_input_voltage             	0.0            	V   
battery_voltage_from_scc      	0.0            	V   
battery_discharge_current     	5              	A   
is_battery_voltage_to_steady_while_charging	0              	bool
battery_voltage_offset_for_fans_on	0              	10mV
pv1_charging_power
```

Might be a good idea to add that to the documentation.

jblance, 2022-12-16:
added an example to usage.md

closed 2022-12-16 by jblance