jblance/mpp-solar #274: Connect to a Remote Serial Port

LichtiMC, 2022-12-10:
Is it possible to connect to a Remote Serial Port in transparent mode over TCP/IP instead of using a physical (local) tty?
I have USR-W610 and tried with socat to create a pseudo tty, but there is a problem, as socat terminates after mpp-solar (or any other serial access) ends its session.

I tried putting socat into a loop like this:
```
#!/bin/bash
# open virtual serial port
while true; do
        sudo socat pty,link=/dev/ttyUSB_WR,waitslave,echo=0,raw tcp:192.168.10.62:8899
done
```
This works with single commands, but when concatenating commands with # then there appears a serial read error or input/output error.
So I guess the socat-restart (reconnection) after termination of socat is not fast enough.

If mpp-solar could natively connect to the remote serial port, these errors wouldn't occur. Hence my question.

jblance, 2022-12-10:
it look like its possible, but would need a new port/inout module - do you know of any python examples of sending commands via a Remote Serial Port in transparent mode over TCP/IP ?

LichtiMC, 2022-12-10:
Here is a very basic example which returns current time of the inverter:
```
#!/usr/bin/env python3
import socket               # Import socket module
s = socket.socket()         # Create a socket object
s.connect(("192.168.10.62", 8899))
s.send("^P002T\r".encode())
print(s.recv(1024).decode())
s.close                     # Close the socket when done
```

jblance, 2023-01-17:
added to version 0.15.5+
needs testing by someone with this setup

closed 2023-01-17 by jblance