jblance/mpp-solar #275: JK-B1A20S15P

podarok, 2022-12-12:
Hi

Trying

```sh
mpp-solar -p 20:21:11:28:15:EF -P JK02 -c getCellData --tag solar/lutsk/jkbms100 -q 127.0.0.0 -o mqtt
```

And getting

```
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/__init__.py", line 417, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/devices/device.py", line 316, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/inout/jkbleio.py", line 37, in send_and_receive
    response = self.ble_get_data(full_command)
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/inout/jkbleio.py", line 88, in ble_get_data
    serviceId = self._device.getServiceByUUID(btle.AssignedNumbers.genericAccess)
  File "/usr/local/lib/python3.10/dist-packages/bluepy/btle.py", line 490, in getServiceByUUID
    raise BTLEGattError("Service %s not found" % (uuid.getCommonName()), rsp)
bluepy.btle.BTLEGattError: Service Generic Access not found
```

Next to it JK-B2A24S20P works well

Sysinfo

```sh
root@metrics:/var/lib/bluetooth# mpp-solar --version
Solar Device Command Utility, version: 0.14.1
```

```sh
root@metrics:/var/lib/bluetooth# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
```

podarok, 2022-12-12:
JK-B1A20S15P
```
hcitool leinfo 20:21:11:28:15:EF
Requesting information ...
	Handle: 16 (0x0010)
	LMP Version: 5.0 (0x9) LMP Subversion: 0x208
	Manufacturer: PHYPLUS Inc (1284)
	Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

JK-B2A24S20P
```
hcitool leinfo C8:47:8C:F7:DB:5A
Requesting information ...
	Handle: 16 (0x0010)
	LMP Version: 4.2 (0x8) LMP Subversion: 0x40b
	Manufacturer: RivieraWaves S.A.S (96)
	Features: 0xfd 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

podarok, 2022-12-12:
Seems like they have different Bluetooth characteristics 

JK-B2A24S20P

[image]

JK-B1A20S15P

[image]

stamina84, 2024-02-04:
I have the BMS model `JK_B2A20S20P` but unable to read proper data, maybe its related to this issue, I can test some code if needed. 
Trying with following command via Bluetooth
`jkbms -p XX:XX:XX -P JK02 -c getCellData`

Got some data, but not matched with real data, I guess its not fit with JK02 protocol (same with 04)

jblance, 2024-02-14:
> I have the BMS model `JK_B2A20S20P` but unable to read proper data, maybe its related to this issue, I can test some code if needed. Trying with following command via Bluetooth `jkbms -p XX:XX:XX -P [...]

post the debug output and I can look to see what is failing

stamina84, 2024-02-15:
> post the debug output and I can look to see what is failing

Thank you for your time, here is the full log with BLE connection:
[JK_B2A20S20P_BLE.log](https://github.com/jblance/mpp-solar/files/14302386/JK_B2A20S20P_BLE.log)

I have a python script that able to read data with serial-usb-rs485 connection, I can provide if it helps, but maybe that is different format from bluetooth connection.

jblance, 2024-02-15:
I tried expanding jk02 to 32 cells and using the response you get it seems to be close
some of the last few look wrong? 
thoughts??
```
$ mppsolar -p test -P jk02_32 
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Parameter            Value              Unit
header               55aaeb90               
record_type          02                     
record_counter       97                     
voltage_cell01       3.126              V   
voltage_cell02       3.128              V   
voltage_cell03       3.113              V   
voltage_cell04       3.13               V   
voltage_cell05       3.125              V   
voltage_cell06       3.13               V   
voltage_cell07       3.128              V   
voltage_cell08       3.13               V   
voltage_cell09       0.0                V   
voltage_cell10       0.0                V   
voltage_cell11       0.0                V   
voltage_cell12       0.0                V   
voltage_cell13       0.0                V   
voltage_cell14       0.0                V   
voltage_cell15       0.0                V   
voltage_cell16       0.0                V   
voltage_cell17       0.0                V   
voltage_cell18       0.0                V   
voltage_cell19       0.0                V   
voltage_cell20       0.0                V   
voltage_cell21       0.0                V   
voltage_cell22       0.0                V   
voltage_cell23       0.0                V   
voltage_cell24       0.0                V   
voltage_cell25       0.0                V   
voltage_cell26       0.0                V   
voltage_cell27       0.0                V   
voltage_cell28       0.0                V   
voltage_cell29       0.0                V   
voltage_cell30       0.0                V   
voltage_cell31       0.0                V   
voltage_cell32       0.0                V   
average_cell_voltage 3.129              V   
delta_cell_voltage   0.028              V   
current_balancer     0.515                  
resistance_cell01    0.056              Ohm 
resistance_cell02    0.057              Ohm 
resistance_cell03    0.056              Ohm 
resistance_cell04    0.056              Ohm 
resistance_cell05    0.056              Ohm 
resistance_cell06    0.056              Ohm 
resistance_cell07    0.055              Ohm 
resistance_cell08    0.056              Ohm 
resistance_cell09    0.0                Ohm 
resistance_cell10    0.0                Ohm 
resistance_cell11    0.0                Ohm 
resistance_cell12    0.0                Ohm 
resistance_cell13    0.0                Ohm 
resistance_cell14    0.0                Ohm 
resistance_cell15    0.0                Ohm 
resistance_cell16    0.0                Ohm 
resistance_cell17    0.0                Ohm 
resistance_cell18    0.0                Ohm 
resistance_cell19    0.0                Ohm 
resistance_cell20    0.0                Ohm 
resistance_cell21    0.0                Ohm 
resistance_cell22    0.0                Ohm 
resistance_cell23    0.0                Ohm 
resistance_cell24    0.0                Ohm 
resistance_cell25    0.0                Ohm 
resistance_cell26    0.0                Ohm 
resistance_cell27    0.0                Ohm 
resistance_cell28    0.0                Ohm 
resistance_cell29    0.0                Ohm 
resistance_cell30    0.0                Ohm 
resistance_cell31    0.0                Ohm 
resistance_cell32    0.0                Ohm 
battery_voltage      25.029             V   
battery_power        214450             W   
balance_current      -8.568             A   
battery_t1           21.2               °C  
battery_t2           20.2               °C  
mos_temp             0.0                °C  
percent_remain       0                  %   
capacity_remain      0.0                Ah  
nominal_capacity     280.0              Ah  
cycle_count          11                     
cycle_capacity       3203.314           Ah  
time                 36D11H29M50S           
current_charge       8.963                  
current_discharge    10.752                
```

stamina84, 2024-02-15:
Yep, it looks kind good, temperatures seems ok, capacity is ok, just mos_temp seems missing, it should around 21-21C too. Charge/discharge could be tricky at this point, there is no really load or charge atm. Battery power should divide by 1000? Its mW? Shall I play around with that config somehow? Then I can confirm with real data and give more accurate result

jblance, 2024-02-15:
can you confirm capacity, battery power, cycle capacity from the app

stamina84, 2024-02-16:
> can you confirm capacity, battery power, cycle capacity from the app

Did another debug run and screened the current state in app, so just a couple of seconds different
[JK_B2A20S20P_BLT.log](https://github.com/jblance/mpp-solar/files/14308654/JK_B2A20S20P_BLT.log)
[image]

jblance, 2024-02-18:
Playing around with the above (using the construct library to try to determine what bytes equal what result), i get a pretty good match up
```
Container: 
    header = b'U\xaa\xeb\x90' (total 4)
    Record_Type = 2
    Record_Counter = 193
    cell_voltage_array = ListContainer: 
        3244
        3245
        3246
        3249
        3249
        3250
        3246
        3246
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
    discard1 = b'\xff\x00\x00\x00' (total 4)
    Average_Cell_Voltage = 3248
    Delta_Cell_Voltage = 7
    Current_Balancer = 3
    cell_resistance_array = ListContainer: 
        56
        57
        56
        56
        56
        56
        55
        56
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
    mos_temp = 226
    discard3 = b'\x00\x00\x00\x00' (total 4)
    battery_voltage = 25985
    battery_power = 424079
    battery_current = 16320
    T2 = 192
    T3 = 190
    balance_current = 0
    discard9 = b'\x00\x00\x00' (total 3)
    Percent_Remain = 9
    Capacity_Remain = 25279
    Nominal_Capacity = 280000
    Cycle_Count = 11
    Cycle_Capacity = 3280276
    discard7 = b'd\x00\x00\x00' (total 4)
    uptime = 3194094
    discard8 = b'\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'... (truncated, total 24)
    rest = b'\x03P\x00\x00\x00\xca\xfa@@\x00\x00\x00\x00&\n\x85'... (truncated, total 79)
```
(ignore the need to /1000 or /10 for some of these numbers)

this is using:
```python
jk02_32_definition = cs.Struct(
    "header" / cs.Bytes(4),
    "Record_Type" / cs.Byte,
    "Record_Counter" / cs.Byte,
    "cell_voltage_array" / cs.Array(32, cs.Int16ul),
    "discard1" / cs.Bytes(4),
    "Average_Cell_Voltage" / cs.Int16ul,
    "Delta_Cell_Voltage" / cs.Int16ul,
    "Current_Balancer" / cs.Int16ul,
    "cell_resistance_array" / cs.Array(32, cs.Int16ul),
    "mos_temp" / cs.Int16ul,
    "discard3" / cs.Bytes(4),
    "battery_voltage" / cs.Int32ul,
    "battery_power" / cs.Int32ul,
    "battery_current" / cs.Int32sl,
    "T2" / cs.Int16ul,
    "T3" / cs.Int16ul,
    "balance_current" / cs.Int32sl,
    "discard9" /  cs.Bytes(3),  # discard of 3 bytes seems wrong
    "Percent_Remain" / cs.Int8ul,
    "Capacity_Remain" / cs.Int32ul,
    "Nominal_Capacity" / cs.Int32ul,
    "Cycle_Count" / cs.Int32ul,
    "Cycle_Capacity" / cs.Int32ul,
    "discard7" / cs.Bytes(4),
    "uptime" / cs.Int24ul,
    "discard8" / cs.Bytes(24),
    "rest" / cs.GreedyBytes
)
```
there are some obvious gaps / possible mistakes
Do you have another seet of data - perhaps with balancing turned on?

stamina84, 2024-02-20:
I tried another connect with Balanced on (so I set lower the start balance voltage), hope you can read something in raw data

[JK_B2A20S20P_BLT.log](https://github.com/jblance/mpp-solar/files/14341673/JK_B2A20S20P_BLT.log)

[image]

jblance, 2024-02-22:
latest version of jk02_32 has as much as i can figure out - still havent found the charge on/off stuff

stamina84, 2024-02-22:
> latest version of jk02_32 has as much as i can figure out - still havent found the charge on/off stuff

thank you so much for your work! I can now read data with that protocol, will play around and try to connect with influxdb, mqtt and grafana and lets see how it works

closed 2024-02-22 by jblance