jblance/mpp-solar #278: No mqtt data.  Please help

3congithub, 2022-12-15:
Hello,

I have spent few days looking to through the discussions and issues but still have not been able to figure out what I have done wrong.  I am using Raspberry 3b connecting via usb/rs232 to EG4 6500ex (LV6548 inverter variation).  Mpp-solar commands seem to communicate with inverters ok.  Below are some of my configuration details.  Thank you in advance for your help.

I have done following:

1. Installed mppsolar clone
2. Installed and started mosquitto client
3. Tested mqtt to see message posted on CMD1 and CMD2
4. Added line * * * * * /usr/local/bin/mpp-solar -q localhost -c QPIGS -i > /home/pi/cron.out 2>&1 on crontab -e
5. Installed and started Influxdb service
6. Installed and started Grafana
7. Created mppsolar database
8. Setup and attempted to read data on grafana ui

Looking forward to hear your input.  Thanks again.

Command output
[image]

[image]

[image]

mpp-solar.conf
[image]

Influx input
[image]

Influx output
[image]

Influx measurements
[image]

Telegraf status
[image]

[image]

jblance, 2022-12-15:
> Hello,
>
> I have spent few days looking to through the discussions and issues but still have not been able to figure out what I have done wrong. I am using Raspberry 3b connecting via usb/rs232 to EG4 [...]

this command will need a `-o mqtt` or better `-o influx2_mqtt`
> 5. Installed and started Influxdb service
> 6. Installed and started Grafana
> 7. Created mppsolar database
> 8. Setup and attempted to read data on grafana ui

Before focusing on grafana make sure mqtt messages are flowing (sub to all topics on the broker)
Then check telegraf for errors and message flow
Then check data is ending up in the db 
Then try to graph it in grafana

3congithub, 2022-12-15:
John,

Thank you for the quick reply.   So just append -o influx2_mqtt to the query schedule line as shown below? 

 * * * * * /usr/local/bin/mpp-solar -q localhost -c QPIGS -i > /home/pi/cron.out 2>&1 -o influx2_mqtt

 Forgive me but I am not very familiar with linux syntax

use mosquitto_sub -h localhost -v -t "#" to check mqtt messages?

telegraf status
 
[image]

3congithub, 2022-12-15:
CMD1
[image]

CMD2
[image]

jblance, 2022-12-15:
> John,
>
> Thank you for the quick reply. So just append -o influx2_mqtt to the query schedule line as shown below?
>
> * * * * * /usr/local/bin/mpp-solar -q localhost -c QPIGS -i > /home/pi/cron.out 2>&1 [...]

No it needs to be
`* * * * * /usr/local/bin/mpp-solar -q localhost -c QPIGS -o influx2_mqtt > /home/pi/cron.out 2>&1 `

The part of the command after the first `>` is redirecting output, first you are redirecting 'normal (stdout or file descriptor 1) output to /home/pi/cron.out then you are redirecting error messages (stderr or descriptor 2) the the same place as 1, so sticking things on the end of that doesnt help

jblance, 2022-12-15:
it also look like telegraf is struggling to write to the influxdb

jblance, 2022-12-15:
I assume youve seen this old doc
https://github.com/jblance/mpp-solar/blob/master/docs/MQTT_Influx_Grafana.md

3congithub, 2022-12-15:
[image]

3congithub, 2022-12-15:
Also got these errors after restarting mpp-solar service

[image]

jblance, 2022-12-15:
you should only use one of the cron entry and the service (service
preferred), because of the way the inverters work running multiple
instances of mppsolar causes problems - even if you are trying to
troubleshoot on the commandline you need to stop the cron and/or service
beforehand
If using the service the config file is the key and yes the output line
should have outputs=influx2_mqtt (the telegraf errors are due to using mqtt
outputter instead.

So update config file
remove the cron entry
restart the mppsolar service
restart telegraf
check telegraf for errors
then check influx for data
profit

3congithub, 2022-12-16:
Ok updated config file
[image]

Commented out cron line
[image]

mppsolar service anabled and started
[image]

restart telegraf and check status
[image]

influx data check
[image]

I must have done something wrong.  Still telegraf errors no influx data.

jblance, 2022-12-16:
Check the mqtt broker messages

3congithub, 2022-12-16:
[image]

mqtt broker messages

3congithub, 2022-12-16:
I used mosquitto_sub -h localhost -v -t "#"

jblance, 2022-12-16:
the mqtt messages above look ok, the telegraf error messages dont look to be related to the same messages
stop mppsolar service and check what messages are being processed by the mqtt broker
also see what is in the mpp-solar measurement in influx
```
use mppsolar
select * from mpp-solar
```

3congithub, 2022-12-16:
[image]

jblance, 2022-12-16:
Try quotes around mpp-solar and maybe a ; at the end
 Note I don't use influx

3congithub, 2022-12-16:
[image]

Also not quite sure what I have done but start to see some data in grafana 

[image]

jblance, 2022-12-16:
Good news

Prolly don't want fill(null)

3congithub, 2022-12-16:
Ok will play around with the options.  So assuming I am good now that data made over to gafana?

if yes would you please provide steps of process and services I need to run/start if PI happen to reboot?  Thank you much for your help and patience.

jblance, 2022-12-16:
If you followed that old doc the services should restart automatically. Need
Mppsolar
Mqtt/ mosquitto
Telefraf
Influx
Grafana

Suggest you do a restart to check before getting too far in

Good luck

3congithub, 2022-12-16:
Yes rebooted the PI.  I had to Restart mppsolar and mqtt.  The rest look like they started on their own.  Grafana has info and seems to be polling fine.  Is there any issue (eg crashing pi or applications) using the fast 5 or 10 seconds refresh rate?

jblance, 2022-12-16:
youll need `sudo systemctl enable mpp-solar` or `systemctl --user enable mpp-solar` depending on whether mpp-solar service is running as root or not (and similar for mqtt)

shouldnt be any problems with a fast refresh (i use 1 sec for my inverter)

closed 2022-12-16 by jblance

3congithub, 2022-12-16:
Also I have 2 inverters in parallel mode and you probably have seen my conf file screenshot above.  It seems most data I see are from the first inverter.  Any change I need to make on content of my second invert in mpp-solar.conf?

jblance, 2022-12-16:
you are separating by tag name atm, so 'command = QPGS0' is inverter 1
changing to tag=invert1 and tag=inverter for the second section would make it easier to distinguish the inverters

3congithub, 2022-12-16:
John, One last question (I hope) - some of the data is not available.  My guess is that this code was designed for LV5048.  I am using LV6548 equivalent inverters.  Would you be able to modify the code to get more out of these inverters?  I added modbus details here for your reference.
https://eg4electronics.com/wp-content/uploads/2022/09/Modbus-protocol-EG4-6500-EX.docx.pdf

Again I appreciate your help with all this.

jblance, 2022-12-16:
Try the pi30max protocol

jblance, 2022-12-16:
Also what data is missing and what commands are you using now?

3congithub, 2022-12-16:
I am looking for PV1 and PV2 data.  The AC load output also not available.  I use QPGS and QPGS1.  Will try PI30max shortly

3congithub, 2022-12-16:
So I changed PI30 in mpp-solar.conf to PI30MAX.  Do I need to update protocol variable anywhere else?

3congithub, 2022-12-16:
[image]

Not sure what I've done but I killed mpp-solar service.

jblance, 2022-12-17:
if in doubt run on the command line using the config file
eg `mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D`

3congithub, 2022-12-17:
[image]

So now looks like I can read inverter 1 but nothing on inverter 2.

jblance, 2022-12-17:
can you post first part of the output (as text if possible)

jblance, 2022-12-17:
> [image]
> 

Dont use `--user` for system services

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D
2022-12-17 04:10:03,152:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-17 04:10:03,153:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9c386f40>, _isConnected=False)
2022-12-17 04:10:03,154:DEBUG:__init__:main@240: udp port 5555
2022-12-17 04:10:03,155:DEBUG:__init__:main@242: Using Postgres None
2022-12-17 04:10:03,156:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-17 04:10:03,189:DEBUG:__init__:main@268: args.configfile is true: /etc/mpp-solar/mpp-solar.conf
2022-12-17 04:10:03,207:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 04:10:03,208:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 04:10:03,209:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_1', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9c386f40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 04:10:03,209:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 04:10:03,210:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 04:10:03,211:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 04:10:03,235:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-17 04:10:03,253:INFO:pi30max:__init__@967: Using protocol b'PI30MAX' with 61 commands
2022-12-17 04:10:03,253:DEBUG:device:__init__@38: __init__ name Inverter_1, port <mppsolar.inout.serialio.SerialIO object at 0x7f9babae20>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-17 04:10:03,254:DEBUG:__init__:main@331: Commands from config file [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f9bab34f0>, 'QPGS0', 'QPGS0', 'influx2_mqtt,screen', None, None)]
2022-12-17 04:10:03,254:INFO:__init__:main@339: Config file: /etc/mpp-solar/mpp-solar.conf
2022-12-17 04:10:03,255:INFO:__init__:main@340: Config setting - pause: 5
2022-12-17 04:10:03,255:INFO:__init__:main@342: Config setting - command sections found: 2
2022-12-17 04:10:03,257:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 04:10:03,258:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 04:10:03,259:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_2', 'port': '/dev/ttyUSB0', 'protocol': None, 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9c386f40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 04:10:03,259:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 04:10:03,260:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 04:10:03,260:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 04:10:03,261:DEBUG:__init__:get_protocol@13: Protocol None
2022-12-17 04:10:03,262:DEBUG:device:__init__@38: __init__ name Inverter_2, port <mppsolar.inout.serialio.SerialIO object at 0x7f9badc1f0>, protocol None
2022-12-17 04:10:03,262:DEBUG:__init__:main@331: Commands from config file [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f9bab34f0>, 'QPGS0', 'QPGS0', 'influx2_mqtt,screen', None, None), (<mppsolar.devices.mppsolar.mppsolar object at 0x7f9baba820>, 'QPGS1', 'QPGS1', 'influx2_mqtt,screen', None, None)]
2022-12-17 04:10:03,263:INFO:__init__:main@339: Config file: /etc/mpp-solar/mpp-solar.conf
2022-12-17 04:10:03,263:INFO:__init__:main@340: Config setting - pause: 5
2022-12-17 04:10:03,263:INFO:__init__:main@342: Config setting - command sections found: 2
2022-12-17 04:10:03,264:INFO:__init__:main@404: Looping 2 commands
2022-12-17 04:10:03,264:INFO:__init__:main@414: Getting results from device: mppsolar device - name: Inverter_1, port: <mppsolar.inout.serialio.SerialIO object at 0x7f9babae20>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: QPGS0, tag: QPGS0, outputs: influx2_mqtt,screen
2022-12-17 04:10:03,265:INFO:device:run_command@274: Running command QPGS0
2022-12-17 04:10:03,265:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30MAX' with 61 commands
2022-12-17 04:10:03,266:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPGS0'
2022-12-17 04:10:03,266:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x3f 0xda 0x3fda
2022-12-17 04:10:03,267:DEBUG:abstractprotocol:get_full_command@56: full command: b'QPGS0?\xda\r'
2022-12-17 04:10:03,267:INFO:device:run_command@303: full command b'QPGS0?\xda\r' for command QPGS0
2022-12-17 04:10:03,268:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'QPGS0'
2022-12-17 04:10:03,268:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: F
2022-12-17 04:10:03,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2022-12-17 04:10:03,272:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MNCHGC
[... cut ...] 3, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9c386f40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Attempted to run command with no protocol defined
2022-12-17 04:10:05,135:DEBUG:__init__:main@449: Not daemon, so not looping
pi@raspberrypi:~ $

jblance, 2022-12-17:
> 2022-12-17 04:10:03,258:DEBUG:device:__init__@33: __init__ args ()
> 2022-12-17 04:10:03,259:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_2', 'port': '/dev/ttyUSB0', 'protocol': None, [...]

For some reason its not getting a protocol for section 2
can you `cat -A /etc/mpp-solar/mpp-solar.conf` and post the result

3congithub, 2022-12-17:
pi@raspberrypi:~ $ cat -A /etc/mpp-solar/mpp-solar.conf
[SETUP]$
#Number of seconds to pause between command execution loop$
# 0 is no pause, greater than 60 will cause service restarts$
pause=5$
mqtt_broker=localhost$
$
# This example would work on 2x PIP4048 installed in parallel$
[Inverter_1]$
protocol=PI30MAX$
port=/dev/ttyUSB0$
porttype=serial$
baud=2400$
command=QPGS0$
tag=QPGS0$
outputs=influx2_mqtt,screen$
$
[Inverter_2]$
port=/dev/ttyUSB0$
porttype=serial$
baud=2400$
command=QPGS1$
tag=QPGS1$
outputs=influx2_mqtt,screen$
pi@raspberrypi:~ $

jblance, 2022-12-17:
oh the `protocol=PI30MAX` line is missing for inverter2

3congithub, 2022-12-17:
Ok I am adding it back

3congithub, 2022-12-17:
pi@raspberrypi:~ $ cat -A /etc/mpp-solar/mpp-solar.conf
[SETUP]$
#Number of seconds to pause between command execution loop$
# 0 is no pause, greater than 60 will cause service restarts$
pause=5$
mqtt_broker=localhost$
$
# This example would work on 2x PIP4048 installed in parallel$
[Inverter_1]$
protocol=PI30MAX$
port=/dev/ttyUSB0$
porttype=serial$
baud=2400$
command=QPGS0$
tag=QPGS0$
outputs=influx2_mqtt,screen$
$
[Inverter_2]$
protocol=PI30MAX$
port=/dev/ttyUSB0$
porttype=serial$
baud=2400$
command=QPGS1$
tag=QPGS1$
outputs=influx2_mqtt,screen$
pi@raspberrypi:~ $

3congithub, 2022-12-17:
Do I need to restart mpp-solar service?

jblance, 2022-12-17:
yes

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo systemctl restart mpp-solar
pi@raspberrypi:~ $ sudo systemctl status mpp-solar
● mpp-solar.service - MPP Solar Service
     Loaded: loaded (/etc/systemd/system/mpp-solar.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-12-17 04:29:12 GMT; 44s ago
    Process: 2129 ExecStart=/usr/bin/python3 /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf --daemon (code=exited, status=1/FAILURE)
   Main PID: 2129 (code=exited, status=1/FAILURE)
        CPU: 1.184s

Dec 17 04:29:12 raspberrypi systemd[1]: mpp-solar.service: Scheduled restart job, restart counter is at 5.
Dec 17 04:29:12 raspberrypi systemd[1]: Stopped MPP Solar Service.
Dec 17 04:29:12 raspberrypi systemd[1]: mpp-solar.service: Consumed 1.184s CPU time.
Dec 17 04:29:12 raspberrypi systemd[1]: mpp-solar.service: Start request repeated too quickly.
Dec 17 04:29:12 raspberrypi systemd[1]: mpp-solar.service: Failed with result 'exit-code'.
Dec 17 04:29:12 raspberrypi systemd[1]: Failed to start MPP Solar Service.
pi@raspberrypi:~ $

3congithub, 2022-12-17:
I had to take that protocol PI30max line out earlier to get it started back up

jblance, 2022-12-17:
test the changes to the config using the commandline first - cant really tell much from the service output
ie
`sudo /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D`

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D
2022-12-17 04:33:42,521:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-17 04:33:42,522:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f8090df40>, _isConnected=False)
2022-12-17 04:33:42,523:DEBUG:__init__:main@240: udp port 5555
2022-12-17 04:33:42,524:DEBUG:__init__:main@242: Using Postgres None
2022-12-17 04:33:42,525:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-17 04:33:42,554:DEBUG:__init__:main@268: args.configfile is true: /etc/mpp-solar/mpp-solar.conf
2022-12-17 04:33:42,573:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 04:33:42,574:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 04:33:42,575:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_1', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f8090df40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 04:33:42,575:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 04:33:42,576:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 04:33:42,577:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 04:33:42,602:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-17 04:33:42,619:INFO:pi30max:__init__@967: Using protocol b'PI30MAX' with 61 commands
2022-12-17 04:33:42,621:DEBUG:device:__init__@38: __init__ name Inverter_1, port <mppsolar.inout.serialio.SerialIO object at 0x7f80041e20>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-17 04:33:42,621:DEBUG:__init__:main@331: Commands from config file [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f8003a4f0>, 'QPGS0', 'QPGS0', 'influx2_mqtt,screen', None, None)]
2022-12-17 04:33:42,622:INFO:__init__:main@339: Config file: /etc/mpp-solar/mpp-solar.conf
2022-12-17 04:33:42,622:INFO:__init__:main@340: Config setting - pause: 5
2022-12-17 04:33:42,623:INFO:__init__:main@342: Config setting - command sections found: 2
2022-12-17 04:33:42,625:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 04:33:42,626:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 04:33:42,627:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_2', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f8090df40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 04:33:42,627:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 04:33:42,628:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 04:33:42,629:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 04:33:42,629:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 313, in main
    device = device_class(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/mppsolar.py", line 7, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 37, in __init__
    self._protocol = get_protocol(get_kwargs(kwargs, "protocol"))
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/__init__.py", line 30, in get_protocol
    return protocol_class()
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/pi30max.py", line 963, in __init__
    self.COMMANDS.pop(item)
KeyError: 'Q1'
pi@raspberrypi:~ $

jblance, 2022-12-17:
ok, thats a bug
try version 0.14.10

3congithub, 2022-12-17:
How do I do that?

jblance, 2022-12-17:
update mpp-solar
how did you install it?
`sudo pip install -U mppsolar`  ?

3congithub, 2022-12-17:
Yes

jblance, 2022-12-17:
need `-U` to force upgrade

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo pip install -U mppsolar
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mppsolar in /usr/local/lib/python3.9/dist-packages (0.14.7)
Collecting mppsolar
  Downloading https://www.piwheels.org/simple/mppsolar/mppsolar-0.14.10-py3-none-any.whl (115 kB)
     |████████████████████████████████| 115 kB 23 kB/s
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.9/dist-packages (from mppsolar) (1.6.1)
Requirement already satisfied: pyserial in /usr/local/lib/python3.9/dist-packages (from mppsolar) (3.5)
Requirement already satisfied: PyYAML in /usr/local/lib/python3.9/dist-packages (from mppsolar) (6.0)
Installing collected packages: mppsolar
  Attempting uninstall: mppsolar
    Found existing installation: mppsolar 0.14.7
    Uninstalling mppsolar-0.14.7:
      Successfully uninstalled mppsolar-0.14.7
Successfully installed mppsolar-0.14.10
pi@raspberrypi:~ $

Got the update.  Thank you.  I will restart mpp-solar

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo systemctl status mpp-solar
● mpp-solar.service - MPP Solar Service
     Loaded: loaded (/etc/systemd/system/mpp-solar.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-12-17 04:56:41 GMT; 3min 18s ago
   Main PID: 2343 (python3)
      Tasks: 1 (limit: 779)
        CPU: 3.750s
     CGroup: /system.slice/mpp-solar.service
             └─2343 /usr/bin/python3 /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf --daemon

Dec 17 04:59:56 raspberrypi python3[2343]: max_ac_charger_current                30                     A
Dec 17 04:59:56 raspberrypi python3[2343]: pv1_input_current                     0                      A
Dec 17 04:59:56 raspberrypi python3[2343]: battery_discharge_current             0                      A
Dec 17 04:59:56 raspberrypi python3[2343]: pv2_input_voltage                     63.4                   V
Dec 17 04:59:56 raspberrypi python3[2343]: pv2_input_current                     0                      A
Dec 17 04:59:56 raspberrypi python3[2343]: Getting results from device: mppsolar device - name: Inverter_2, port: <mppsolar.inout.serialio.SerialIO object at 0x7f9cee4370>, protocol: PI30 protocol han>
Dec 17 04:59:56 raspberrypi python3[2343]: 2022-12-17 04:59:56,397:WARNING:serialio:send_and_receive@33: Serial read error: device reports readiness to read but returned no data (device disconnected o>
Dec 17 04:59:56 raspberrypi python3[2343]: Parameter                             Value           Unit
Dec 17 04:59:56 raspberrypi python3[2343]: error                                 Serial command execution failed
Dec 17 04:59:56 raspberrypi python3[2343]: Sleeping for 5 sec
lines 1-19/19 (END)

jblance, 2022-12-17:
> test the changes to the config using the commandline first - cant really tell much from the service output ie `sudo /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D`

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D
2022-12-17 05:04:49,646:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.10
2022-12-17 05:04:49,647:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9660cf40>, _isConnected=False)
2022-12-17 05:04:49,647:DEBUG:__init__:main@240: udp port 5555
2022-12-17 05:04:49,648:DEBUG:__init__:main@242: Using Postgres None
2022-12-17 05:04:49,649:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-17 05:04:49,677:DEBUG:__init__:main@268: args.configfile is true: /etc/mpp-solar/mpp-solar.conf
2022-12-17 05:04:49,696:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 05:04:49,696:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 05:04:49,697:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_1', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9660cf40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 05:04:49,698:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 05:04:49,699:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 05:04:49,699:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 05:04:49,723:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-17 05:04:49,740:INFO:pi30max:__init__@968: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:04:49,741:DEBUG:device:__init__@38: __init__ name Inverter_1, port <mppsolar.inout.serialio.SerialIO object at 0x7f95d40e20>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-17 05:04:49,742:DEBUG:__init__:main@331: Commands from config file [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f95d394f0>, 'QPGS0', 'QPGS0', 'influx2_mqtt,screen', None, None)]
2022-12-17 05:04:49,742:INFO:__init__:main@339: Config file: /etc/mpp-solar/mpp-solar.conf
2022-12-17 05:04:49,743:INFO:__init__:main@340: Config setting - pause: 5
2022-12-17 05:04:49,743:INFO:__init__:main@342: Config setting - command sections found: 2
2022-12-17 05:04:49,745:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 05:04:49,746:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 05:04:49,747:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_2', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9660cf40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 05:04:49,747:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 05:04:49,748:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 05:04:49,748:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 05:04:49,749:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-17 05:04:49,750:INFO:pi30max:__init__@968: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:04:49,750:DEBUG:device:__init__@38: __init__ name Inverter_2, port <mppsolar.inout.serialio.SerialIO object at 0x7f95d621f0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-17 05:04:49,751:DEBUG:__init__:main@331: Commands from config file [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f95d394f0>, 'QPGS0', 'QPGS0', 'influx2_mqtt,screen', None, None), (<mppsolar.devices.mppsolar.mppsolar object at 0x7f95d40820>, 'QPGS1', 'QPGS1', 'influx2_mqtt,screen', None, None)]
2022-12-17 05:04:49,751:INFO:__init__:main@339: Config file: /etc/mpp-solar/mpp-solar.conf
2022-12-17 05:04:49,752:INFO:__init__:main@340: Config setting - pause: 5
2022-12-17 05:04:49,752:INFO:__init__:main@342: Config setting - command sections found: 2
2022-12-17 05:04:49,753:INFO:__init__:main@404: Looping 2 commands
2022-12-17 05:04:49,753:INFO:__init__:main@414: Getting results from device: mppsolar device - name: Inverter_1, port: <mppsolar.inout.serialio.SerialIO object at 0x7f95d40e20>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: QPGS0, tag: QPGS0, outputs: influx2_mqtt,screen
2022-12-17 05:04:49,754:INFO:device:run_command@274: Running command QPGS0
2022-12-17 05:04:49,754:INFO:abstractprotocol:get_full_command@55: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:04:49,754:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPGS0'
2022-12-17 05:04:49,755:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x3f 0xda 0x3fda
2022-12-17 05:04:49,756:DEBUG:abstractprotocol:get_full_command@63: full command: b'QPGS0?\xda\r'
2022-12-17 05:04:49,756:INFO:device:run_command@303: full command b'QPGS0?\xda\r' for command QPGS0
2022-12-17 05:04:49,757:DEBUG:abstractprotocol:get_command_defn@67: Processing command 'QPGS0'
2022-12-17 05:04:49,757:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: F
2022-12-17 05:04:49,759:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: MCHGC
[... cut ...]
2022-12-17 05:04:51,753:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Serial command execution failed', '']}, 'tag': 'QPGS1', 'name': 'Inverter_2', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9660cf40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Serial command execution failed
2022-12-17 05:04:51,754:DEBUG:__init__:main@449: Not daemon, so not looping
pi@raspberrypi:~ $

jblance, 2022-12-17:
is the service stopped?
try (after ensuring the service is stopped and there is no user version of the service)
`sudo mpp-solar -p /dev/ttyUSB0 -P PI30MAX -c QPGS1 -D`

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo systemctl stop mpp-solar
pi@raspberrypi:~ $ sudo mpp-solar -p /dev/ttyUSB0 -P PI30MAX -c QPGS1 -D
2022-12-17 05:11:45,213:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.10
2022-12-17 05:11:45,214:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7fa1402f40>, _isConnected=False)
2022-12-17 05:11:45,215:DEBUG:__init__:main@240: udp port 5555
2022-12-17 05:11:45,215:DEBUG:__init__:main@242: Using Postgres None
2022-12-17 05:11:45,216:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-17 05:11:45,217:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30MAX"
2022-12-17 05:11:45,230:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 05:11:45,230:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 05:11:45,231:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7fa1402f40>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-17 05:11:45,232:DEBUG:__init__:get_port_type@69: port matches ttyusb
2022-12-17 05:11:45,233:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 05:11:45,258:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-17 05:11:45,277:INFO:pi30max:__init__@968: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:11:45,277:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7fa0b2ffa0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-17 05:11:45,278:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fa0b2ff70>, 'QPGS1', 'QPGS1', 'screen', None, None)]
2022-12-17 05:11:45,279:INFO:__init__:main@404: Looping 1 commands
2022-12-17 05:11:45,279:INFO:__init__:main@414: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7fa0b2ffa0>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: QPGS1, tag: QPGS1, outputs: screen
2022-12-17 05:11:45,280:INFO:device:run_command@274: Running command QPGS1
2022-12-17 05:11:45,281:INFO:abstractprotocol:get_full_command@55: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:11:45,281:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPGS1'
2022-12-17 05:11:45,282:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x2f 0xfb 0x2ffb
2022-12-17 05:11:45,283:DEBUG:abstractprotocol:get_full_command@63: full command: b'QPGS1/\xfb\r'
2022-12-17 05:11:45,283:INFO:device:run_command@303: full command b'QPGS1/\xfb\r' for command QPGS1
2022-12-17 05:11:45,284:DEBUG:abstractprotocol:get_command_defn@67: Processing command 'QPGS1'
2022-12-17 05:11:45,284:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: F
2022-12-17 05:11:45,289:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: MCHGC
2022-12-17 05:11:45,291:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: MNCHGC
2022-12-17 05:11:45,294:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: MUCHGC
2022-12-17 05:11:45,296:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PBCV
2022-12-17 05:11:45,298:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PBDV
2022-12-17 05:11:45,300:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PBFT
2022-12-17 05:11:45,302:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PBT
2022-12-17 05:11:45,304:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PCP
2022-12-17 05:11:45,306:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PCVV
2022-12-17 05:11:45,308:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PE
2022-12-17 05:11:45,309:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PD
2022-12-17 05:11:45,311:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PGR
2022-12-17 05:11:45,313:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: POP
2022-12-17 05:11:45,315:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: POPLG
2022-12-17 05:11:45,318:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: POPM
2022-12-17 05:11:45,320:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PPCP
2022-12-17 05:11:45,322:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PPVOKC
2022-12-17 05:11:45,337:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PSDV
2022-12-17 05:11:45,339:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PSPB
2022-12-17 05:11:45,341:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PBATCD
2022-12-17 05:11:45,344:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: DAT
2022-12-17 05:11:45,347:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: PBATMAXDISC
2022-12-17 05:11:45,350:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: QPGS
[... cut ...]
output_mode                     Phase 1 of 2 phase output
charger_source_priority         Solar only
max_charger_current             100             A
max_charger_range               120             A
max_ac_charger_current          30              A
pv1_input_current               0               A
battery_discharge_current       0               A
pv2_input_voltage               77.7            V
pv2_input_current               0               A
2022-12-17 05:11:47,110:DEBUG:__init__:main@449: Not daemon, so not looping
pi@raspberrypi:~ $

jblance, 2022-12-17:
ok (with service still stopped) try
`sudo /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D`

3congithub, 2022-12-17:
pi@raspberrypi:~ $ sudo /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D
2022-12-17 05:17:48,858:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.10
2022-12-17 05:17:48,859:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84104f40>, _isConnected=False)
2022-12-17 05:17:48,860:DEBUG:__init__:main@240: udp port 5555
2022-12-17 05:17:48,861:DEBUG:__init__:main@242: Using Postgres None
2022-12-17 05:17:48,861:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-17 05:17:48,891:DEBUG:__init__:main@268: args.configfile is true: /etc/mpp-solar/mpp-solar.conf
2022-12-17 05:17:48,912:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 05:17:48,913:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 05:17:48,914:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_1', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84104f40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 05:17:48,915:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 05:17:48,915:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 05:17:48,916:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 05:17:48,943:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-17 05:17:48,961:INFO:pi30max:__init__@968: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:17:48,962:DEBUG:device:__init__@38: __init__ name Inverter_1, port <mppsolar.inout.serialio.SerialIO object at 0x7f83838e20>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-17 05:17:48,963:DEBUG:__init__:main@331: Commands from config file [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f838314f0>, 'QPGS0', 'QPGS0', 'influx2_mqtt,screen', None, None)]
2022-12-17 05:17:48,964:INFO:__init__:main@339: Config file: /etc/mpp-solar/mpp-solar.conf
2022-12-17 05:17:48,964:INFO:__init__:main@340: Config setting - pause: 5
2022-12-17 05:17:48,965:INFO:__init__:main@342: Config setting - command sections found: 2
2022-12-17 05:17:48,967:DEBUG:__init__:main@311: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-17 05:17:48,968:DEBUG:device:__init__@33: __init__ args ()
2022-12-17 05:17:48,968:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Inverter_2', 'port': '/dev/ttyUSB0', 'protocol': 'PI30MAX', 'outputs': 'influx2_mqtt,screen', 'baud': '2400', 'porttype': 'serial', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84104f40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None}
2022-12-17 05:17:48,969:INFO:__init__:get_port@81: Port overide - using port 'serial'
2022-12-17 05:17:48,969:DEBUG:__init__:get_port_type@66: port matches serial
2022-12-17 05:17:48,970:INFO:__init__:get_port@112: Using serialio for communications
2022-12-17 05:17:48,971:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-17 05:17:48,971:INFO:pi30max:__init__@968: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:17:48,972:DEBUG:device:__init__@38: __init__ name Inverter_2, port <mppsolar.inout.serialio.SerialIO object at 0x7f8385a1f0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-17 05:17:48,973:DEBUG:__init__:main@331: Commands from config file [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f838314f0>, 'QPGS0', 'QPGS0', 'influx2_mqtt,screen', None, None), (<mppsolar.devices.mppsolar.mppsolar object at 0x7f83838820>, 'QPGS1', 'QPGS1', 'influx2_mqtt,screen', None, None)]
2022-12-17 05:17:48,973:INFO:__init__:main@339: Config file: /etc/mpp-solar/mpp-solar.conf
2022-12-17 05:17:48,974:INFO:__init__:main@340: Config setting - pause: 5
2022-12-17 05:17:48,974:INFO:__init__:main@342: Config setting - command sections found: 2
2022-12-17 05:17:48,975:INFO:__init__:main@404: Looping 2 commands
2022-12-17 05:17:48,975:INFO:__init__:main@414: Getting results from device: mppsolar device - name: Inverter_1, port: <mppsolar.inout.serialio.SerialIO object at 0x7f83838e20>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: QPGS0, tag: QPGS0, outputs: influx2_mqtt,screen
2022-12-17 05:17:48,976:INFO:device:run_command@274: Running command QPGS0
2022-12-17 05:17:48,976:INFO:abstractprotocol:get_full_command@55: Using protocol b'PI30MAX' with 61 commands
2022-12-17 05:17:48,977:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPGS0'
2022-12-17 05:17:48,978:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x3f 0xda 0x3fda
2022-12-17 05:17:48,978:DEBUG:abstractprotocol:get_full_command@63: full command: b'QPGS0?\xda\r'
2022-12-17 05:17:48,979:INFO:device:run_command@303: full command b'QPGS0?\xda\r' for command QPGS0
2022-12-17 05:17:48,980:DEBUG:abstractprotocol:get_command_defn@67: Processing command 'QPGS0'
2022-12-17 05:17:48,980:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: F
2022-12-17 05:17:48,982:DEBUG:abstractprotocol:get_command_defn@73: Regex commands _command: MCHGC
[... cut ...]
2022-12-17 05:17:49,329:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Serial command execution failed', '']}, 'tag': 'QPGS1', 'name': 'Inverter_2', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84104f40>, _isConnected=False), 'udp_port': None, 'postgres_url': None, 'mongo_url': None, 'mongo_db': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Serial command execution failed
2022-12-17 05:17:49,330:DEBUG:__init__:main@449: Not daemon, so not looping
pi@raspberrypi:~ $

jblance, 2022-12-17:
`sudo systemctl status mpp-solar`
and
`systemctl --user status mpp-solar`

[9 more replies]