jblance/mpp-solar #279: how to use hassd_mqtt

3congithub, 2022-12-17:
John,

I can't stay away long!  So I re-installed mpp-solar on a new sd.  I did everything but installing and setting up mqtt, infux and grafana as I want to read the data on Homeassistant in another PI.  Ran this deamon command below and was able to see all data on HA.  Is this the right way to do it?  Or I need to do something with mpp-solar service so it will run without issuing this daemon command?

  mpp-solar -p /dev/ttyUSB0 -n inveter1 -o hassd_mqtt,screen -c QPGS0#QPGS1 \
-q 192.168.50.184 --mqttuser mqtt-user --mqttpass pass --mqtttopic /PV/status -n SOLAR_INVERTER --daemon

jblance, 2022-12-18:
you just need to change your config file to match your command line
something like
```
[SETUP]
## Number of seconds to pause between command executions
pause=1
mqtt_broker=192.168.50.184
mqtt_user=mqtt-user 
mqtt_pass=pass 

[inveter1 ]
protocol=PI30MAX
port=/dev/ttyUSB0
command=QPGS0#QPGS1
outputs=hassd_mqtt
```

and then check from command line with `mpp-solar -C /etc/mpp-solar/mpp-solar.conf` (or whatever the config file is)
if that works, then the service can be used instead of your command line
otherwise add `-D` to help diagnose

3congithub, 2022-12-18:
That makes perfect sense sir.  Thank you.

closed 2022-12-19 by 3congithub