jblance/mpp-solar #280: PI17 Command ACCT2200-0259 was rejected

drbacke, 2022-12-20:
Hi, 
I have a Mppsolar MPI5k and almost all PI17 / PI17INFINI command (read) work. But with the write commands there is often a Command Rejected error. Is there any solution for this?
Here is an example (with PI17 the same error):  
`mpp-solar -p /dev/hidraw0  -P PI17INFINI -o screen -c ACCT2200-0259 -D`

```
2022-12-20 16:17:20,679:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-20 16:17:20,680:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6607210>, _isConnected=False)
2022-12-20 16:17:20,680:DEBUG:__init__:main@240: udp port 5555
2022-12-20 16:17:20,680:DEBUG:__init__:main@242: Using Postgres None
2022-12-20 16:17:20,681:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-20 16:17:20,681:INFO:__init__:main@348: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17INFINI"
2022-12-20 16:17:20,687:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-20 16:17:20,688:DEBUG:device:__init__@33: __init__ args ()
2022-12-20 16:17:20,688:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17INFINI', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6607210>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-20 16:17:20,689:DEBUG:__init__:get_port_type@39: port matches hidraw
2022-12-20 16:17:20,689:INFO:__init__:get_port@92: Using hidrawio for communications
2022-12-20 16:17:20,692:DEBUG:__init__:get_protocol@13: Protocol PI17INFINI
2022-12-20 16:17:20,702:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5fd14f0>, protocol PI17INFINI protocol handler
2022-12-20 16:17:20,702:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5fd1470>, 'ACCT2200-0259', 'ACCT2200-0259', 'screen', None, None)]
2022-12-20 16:17:20,703:INFO:__init__:main@404: Looping 1 commands
2022-12-20 16:17:20,703:INFO:__init__:main@415: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5fd14f0>, protocol: PI17INFINI protocol handler for command: ACCT2200-0259, tag: ACCT2200-0259, outputs: screen
2022-12-20 16:17:20,703:INFO:device:run_command@274: Running command ACCT2200-0259
2022-12-20 16:17:20,704:INFO:pi17infini:get_full_command@802: Using protocol b'PI17INFINI' with 42 commands
2022-12-20 16:17:20,704:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT2200-0259'
2022-12-20 16:17:20,704:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-20 16:17:20,705:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-20 16:17:20,706:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
2022-12-20 16:17:20,708:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EH
2022-12-20 16:17:20,709:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LON
2022-12-20 16:17:20,710:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PA
2022-12-20 16:17:20,711:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PB
2022-12-20 16:17:20,712:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PC
2022-12-20 16:17:20,712:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-20 16:17:20,713:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-20 16:17:20,714:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PF
2022-12-20 16:17:20,715:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-12-20 16:17:20,716:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LST
2022-12-20 16:17:20,718:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDA
2022-12-20 16:17:20,719:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDB
2022-12-20 16:17:20,719:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDC
2022-12-20 16:17:20,720:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
2022-12-20 16:17:20,721:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDE
2022-12-20 16:17:20,722:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDF
2022-12-20 16:17:20,723:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDG
2022-12-20 16:17:20,724:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BT
2022-12-20 16:17:20,725:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-12-20 16:17:20,726:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCT
2022-12-20 16:17:20,729:DEBUG:abstractprotocol:get_command_defn@71: Matched: ACCT2200-0259 to: ACCT value: 22
2022-12-20 16:17:20,729:DEBUG:pi17infini:get_full_command@844: _pre_cmd: b'^S014ACCT2200-0259'
2022-12-20 16:17:20,729:DEBUG:pi17infini:get_full_command@851: full command: b'^S014ACCT2200-0259\r'
2022-12-20 16:17:20,730:INFO:device:run_command@303: full command b'^S014ACCT2200-0259\r' for command ACCT2200-0259
2022-12-20 16:17:20,730:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT2200-0259'
2022-12-20 16:17:20,730:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-20 16:17:20,730:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-20 16:17:20,731:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
2022-12-20 16:17:20,731:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EH
2022-12-20 16:17:20,731:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LON
2022-12-20 16:17:20,731:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PA
2022-12-20 16:17:20,732:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PB
2022-12-20 16:17:20,732:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PC
2022-12-20 16:17:20,732:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-20 16:17:20,733:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-20 16:17:20,733:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PF
2022-12-20 16:17:20,733:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-12-20 16:17:20,733:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LST
2022-12-20 16:17:20,734:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDA
2022-12-20 16:17:20,734:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDB
2022-12-20 16:17:20,734:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDC
2022-12-20 16:17:20,734:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
[... 34 lines cut ...]
2022-12-20 16:17:22,205:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
2022-12-20 16:17:22,205:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDE
2022-12-20 16:17:22,206:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDF
2022-12-20 16:17:22,206:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDG
2022-12-20 16:17:22,206:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BT
2022-12-20 16:17:22,207:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-12-20 16:17:22,207:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCT
2022-12-20 16:17:22,207:DEBUG:abstractprotocol:get_command_defn@71: Matched: ACCT2200-0259 to: ACCT value: 22
2022-12-20 16:17:22,207:INFO:abstractprotocol:decode@236: Processing response of type DEFAULT
2022-12-20 16:17:22,208:DEBUG:abstractprotocol:decode@240: trimmed and split responses: ['NAK']
2022-12-20 16:17:22,208:INFO:abstractprotocol:decode@247: Processing DEFAULT type responses
Processing DEFAULT type responses
[... cut ...]  object at 0xb6607210>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: ACCT2200-0259 - Set AC charge time range
------------------------------------------------------------
Parameter                     	Value           Unit
warning0                      	Command ACCT2200-0259 was rejected	    
2022-12-20 16:17:22,214:DEBUG:__init__:main@449: Not daemon, so not looping
```

jblance, 2022-12-20:
> 2022-12-20 ***@***.***: Solar Device Command Utility, version: 0.14.7
> 2022-12-20 ***@***.***: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6607210>, _isConnected=False)
> 2022-12-20 ***@***.***: udp port 5555
> 2022-12-20 ***@***.***: Using Postgres None
> 2022-12-20 ***@***.***: Using Mongo None with mppsolar
> 2022-12-20 ***@***.***: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17INFINI"
> 2022-12-20 ***@***.***: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
> 2022-12-20 ***@***.***: __init__ args ()
> 2022-12-20 ***@***.***: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17INFINI', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6607210>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
> 2022-12-20 ***@***.***: port matches hidraw
> 2022-12-20 ***@***.***: Using hidrawio for communications
> 2022-12-20 ***@***.***: Protocol PI17INFINI
> 2022-12-20 ***@***.***: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5fd14f0>, protocol PI17INFINI protocol handler
> 2022-12-20 ***@***.***: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5fd1470>, 'ACCT2200-0259', 'ACCT2200-0259', 'screen', None, None)]
> 2022-12-20 ***@***.***: Looping 1 commands
> [... cut ...] None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6607210>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
> 2022-12-20 ***@***.***: Not daemon, so not looping

Can you try with a comma instead of the dash between the 2 values

drbacke, 2022-12-20:
Thank you for your help. I changed the regex in the pi17.py but it doesn't help:
```
(mppsolar) pi@raspberrypi:~/MQTT_MPPSOLAR $ mpp-solar -p /dev/hidraw0  -P PI17 -o screen -c ACCT2200,0259 -D
2022-12-20 19:29:52,218:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-20 19:29:52,219:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6612250>, _isConnected=False)
2022-12-20 19:29:52,219:DEBUG:__init__:main@240: udp port 5555
2022-12-20 19:29:52,219:DEBUG:__init__:main@242: Using Postgres None
2022-12-20 19:29:52,220:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-20 19:29:52,220:INFO:__init__:main@348: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2022-12-20 19:29:52,226:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-20 19:29:52,227:DEBUG:device:__init__@33: __init__ args ()
2022-12-20 19:29:52,227:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6612250>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-20 19:29:52,228:DEBUG:__init__:get_port_type@39: port matches hidraw
2022-12-20 19:29:52,228:INFO:__init__:get_port@92: Using hidrawio for communications
2022-12-20 19:29:52,231:DEBUG:__init__:get_protocol@13: Protocol PI17
2022-12-20 19:29:52,278:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5fdd510>, protocol PI17 protocol handler
2022-12-20 19:29:52,278:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5fdd490>, 'ACCT2200,0259', 'ACCT2200,0259', 'screen', None, None)]
2022-12-20 19:29:52,279:INFO:__init__:main@404: Looping 1 commands
2022-12-20 19:29:52,279:INFO:__init__:main@415: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5fdd510>, protocol: PI17 protocol handler for command: ACCT2200,0259, tag: ACCT2200,0259, outputs: screen
2022-12-20 19:29:52,280:INFO:device:run_command@274: Running command ACCT2200,0259
2022-12-20 19:29:52,280:INFO:pi17:get_full_command@905: Using protocol b'PI17' with 47 commands
2022-12-20 19:29:52,280:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT2200,0259'
2022-12-20 19:29:52,281:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-20 19:29:52,282:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-20 19:29:52,283:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
2022-12-20 19:29:52,285:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EH
2022-12-20 19:29:52,286:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LON
2022-12-20 19:29:52,288:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PA
2022-12-20 19:29:52,289:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PB
2022-12-20 19:29:52,290:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PC
2022-12-20 19:29:52,291:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-20 19:29:52,292:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-20 19:29:52,293:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PF
2022-12-20 19:29:52,295:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-12-20 19:29:52,296:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LST
2022-12-20 19:29:52,298:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDA
2022-12-20 19:29:52,300:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDB
2022-12-20 19:29:52,301:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDC
2022-12-20 19:29:52,302:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
2022-12-20 19:29:52,303:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDE
2022-12-20 19:29:52,304:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDF
2022-12-20 19:29:52,304:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDG
2022-12-20 19:29:52,305:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BT
2022-12-20 19:29:52,306:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-12-20 19:29:52,307:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCT
2022-12-20 19:29:52,311:DEBUG:abstractprotocol:get_command_defn@71: Matched: ACCT2200,0259 to: ACCT value: 22
2022-12-20 19:29:52,312:DEBUG:pi17:get_full_command@959: _pre_cmd: b'^S014ACCT2200,0259'
2022-12-20 19:29:52,312:DEBUG:pi17:get_full_command@966: full command: b'^S014ACCT2200,0259\r'
2022-12-20 19:29:52,313:INFO:device:run_command@303: full command b'^S014ACCT2200,0259\r' for command ACCT2200,0259
2022-12-20 19:29:52,313:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT2200,0259'
2022-12-20 19:29:52,314:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
[... cut ...] object at 0xb6612250>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: ACCT2200,0259 - Set AC charge time range
------------------------------------------------------------
Parameter                     	Value           Unit
warning0                      	Command ACCT2200,0259 was rejected	    
2022-12-20 19:29:53,796:DEBUG:__init__:main@449: Not daemon, so not looping

```

jblance, 2022-12-20:
> (mppsolar) ***@***.***:~/MQTT_MPPSOLAR $ mpp-solar -p /dev/hidraw0  -P PI17 -o screen -c ACCT2200,0259 -D
> 2022-12-20 ***@***.***: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6612250>, _isConnected=False)
> 2022-12-20 ***@***.***: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6612250>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
> 2022-12-20 ***@***.***: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5fdd510>, protocol PI17 protocol handler
> 2022-12-20 ***@***.***: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5fdd490>, 'ACCT2200,0259', 'ACCT2200,0259', 'screen', None, None)]
> 2022-12-20 ***@***.***: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5fdd510>, protocol: PI17 protocol handler for command: ACCT2200,0259, tag: ACCT2200,0259, outputs: screen
> 2022-12-20 ***@***.***: Using protocol b'PI17' with 47 commands
> 2022-12-20 ***@***.***: kwargs {'data': {'raw_response': ['^0\x1bã\r', ''], '_command': 'ACCT2200,0259', '_command_description': 'Set AC charge time range', 'WARNING0': ['Command ACCT2200,0259 was rejected', '']}, 'tag': 'ACCT2200,0259', 'name': 'unnamed', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6612250>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}

Can you try adding `--porttype hidfull` to the command.

drbacke, 2022-12-20:
Of course:
```
(mppsolar) pi@raspberrypi:~/MQTT_MPPSOLAR $ mpp-solar -p /dev/hidraw0  -P PI17 -o screen -c ACCT2200-0259 --porttype hidfull -D
2022-12-20 20:22:49,784:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-20 20:22:49,785:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb662e210>, _isConnected=False)
2022-12-20 20:22:49,785:DEBUG:__init__:main@240: udp port 5555
2022-12-20 20:22:49,785:DEBUG:__init__:main@242: Using Postgres None
2022-12-20 20:22:49,785:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-20 20:22:49,786:INFO:__init__:main@348: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=hidfull)" using protocol "PI17"
2022-12-20 20:22:49,789:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-20 20:22:49,789:DEBUG:device:__init__@33: __init__ args ()
2022-12-20 20:22:49,789:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': 'hidfull', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb662e210>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-20 20:22:49,789:INFO:__init__:get_port@81: Port overide - using port 'hidfull'
2022-12-20 20:22:49,790:DEBUG:__init__:get_port_type@45: port matches hidfull
2022-12-20 20:22:49,790:INFO:__init__:get_port@97: Using hidfullio for communications
2022-12-20 20:22:49,791:DEBUG:__init__:get_protocol@13: Protocol PI17
2022-12-20 20:22:49,821:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidfullio.HIDFullIO object at 0xb5ff84d0>, protocol PI17 protocol handler
2022-12-20 20:22:49,822:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5ff8450>, 'ACCT2200-0259', 'ACCT2200-0259', 'screen', None, None)]
2022-12-20 20:22:49,822:INFO:__init__:main@404: Looping 1 commands
2022-12-20 20:22:49,823:INFO:__init__:main@415: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidfullio.HIDFullIO object at 0xb5ff84d0>, protocol: PI17 protocol handler for command: ACCT2200-0259, tag: ACCT2200-0259, outputs: screen
2022-12-20 20:22:49,823:INFO:device:run_command@274: Running command ACCT2200-0259
2022-12-20 20:22:49,823:INFO:pi17:get_full_command@905: Using protocol b'PI17' with 47 commands
2022-12-20 20:22:49,824:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT2200-0259'
2022-12-20 20:22:49,824:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-20 20:22:49,825:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-20 20:22:49,827:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
2022-12-20 20:22:49,829:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EH
2022-12-20 20:22:49,831:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LON
2022-12-20 20:22:49,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PA
2022-12-20 20:22:49,833:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PB
2022-12-20 20:22:49,833:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PC
2022-12-20 20:22:49,835:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-20 20:22:49,836:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-20 20:22:49,837:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PF
2022-12-20 20:22:49,838:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-12-20 20:22:49,839:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LST
2022-12-20 20:22:49,841:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDA
2022-12-20 20:22:49,842:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDB
2022-12-20 20:22:49,843:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDC
2022-12-20 20:22:49,844:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
2022-12-20 20:22:49,845:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDE
2022-12-20 20:22:49,846:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDF
2022-12-20 20:22:49,847:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDG
2022-12-20 20:22:49,847:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BT
2022-12-20 20:22:49,848:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-12-20 20:22:49,850:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCT
2022-12-20 20:22:49,852:DEBUG:abstractprotocol:get_command_defn@71: Matched: ACCT2200-0259 to: ACCT value: 22
2022-12-20 20:22:49,853:DEBUG:pi17:get_full_command@959: _pre_cmd: b'^S014ACCT2200-0259'
2022-12-20 20:22:49,853:DEBUG:pi17:get_full_command@966: full command: b'^S014ACCT2200-0259\r'
2022-12-20 20:22:49,854:INFO:device:run_command@303: full command b'^S014ACCT2200-0259\r' for command ACCT2200-0259
2022-12-20 20:22:49,854:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT2200-0259'
[... cut ...]
  File "/home/pi/MQTT_MPPSOLAR/venv/mppsolar/lib/python3.7/site-packages/mppsolar/__init__.py", line 417, in main
    results = _device.run_command(command=_command)
  File "/home/pi/MQTT_MPPSOLAR/venv/mppsolar/lib/python3.7/site-packages/mppsolar/devices/device.py", line 320, in run_command
    command_defn=self._protocol.get_command_defn(command),
  File "/home/pi/MQTT_MPPSOLAR/venv/mppsolar/lib/python3.7/site-packages/mppsolar/inout/hidfullio.py", line 38, in send_and_receive
    os.write(usb0, to_send)
TimeoutError: [Errno 110] Connection timed out

```

jblance, 2022-12-20:
ok, so looks like i need to work on the hidraw interface a bit
doesnt explain why the queries work and the sets dont

jblance, 2022-12-20:
> (mppsolar) pi@raspberrypi:~/MQTT_MPPSOLAR $ mpp-solar -p /dev/hidraw0  -P PI17 -o screen -c ACCT2200-0259 

not that it is relevant for the error here, but the command format (according to the doc i have seen) is to use a comma
what does the `-c ACCT` query command return

drbacke, 2022-12-21:
Once again, but it's the same error:
Do you think it's about the protocol itself or something about the communication? 
Maybe I can help a little.
```
(mppsolar) pi@raspberrypi:~/MQTT_MPPSOLAR $ mpp-solar -p /dev/hidraw0  -P PI17 -o screen -c ACCT2200,0259 --porttype hidfull -D
2022-12-21 09:06:32,434:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-21 09:06:32,434:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6628270>, _isConnected=False)
2022-12-21 09:06:32,435:DEBUG:__init__:main@240: udp port 5555
2022-12-21 09:06:32,435:DEBUG:__init__:main@242: Using Postgres None
2022-12-21 09:06:32,435:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-21 09:06:32,436:INFO:__init__:main@348: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=hidfull)" using protocol "PI17"
2022-12-21 09:06:32,442:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-21 09:06:32,442:DEBUG:device:__init__@33: __init__ args ()
2022-12-21 09:06:32,443:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': 'hidfull', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb6628270>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-21 09:06:32,443:INFO:__init__:get_port@81: Port overide - using port 'hidfull'
2022-12-21 09:06:32,443:DEBUG:__init__:get_port_type@45: port matches hidfull
2022-12-21 09:06:32,444:INFO:__init__:get_port@97: Using hidfullio for communications
2022-12-21 09:06:32,447:DEBUG:__init__:get_protocol@13: Protocol PI17
2022-12-21 09:06:32,457:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidfullio.HIDFullIO object at 0xb5ff34f0>, protocol PI17 protocol handler
2022-12-21 09:06:32,457:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5ff3470>, 'ACCT2200,0259', 'ACCT2200,0259', 'screen', None, None)]
2022-12-21 09:06:32,458:INFO:__init__:main@404: Looping 1 commands
2022-12-21 09:06:32,458:INFO:__init__:main@415: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidfullio.HIDFullIO object at 0xb5ff34f0>, protocol: PI17 protocol handler for command: ACCT2200,0259, tag: ACCT2200,0259, outputs: screen
2022-12-21 09:06:32,459:INFO:device:run_command@274: Running command ACCT2200,0259
2022-12-21 09:06:32,459:INFO:pi17:get_full_command@905: Using protocol b'PI17' with 47 commands
2022-12-21 09:06:32,459:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT2200,0259'
2022-12-21 09:06:32,459:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-21 09:06:32,460:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-21 09:06:32,461:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
2022-12-21 09:06:32,463:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EH
2022-12-21 09:06:32,464:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LON
2022-12-21 09:06:32,465:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PA
2022-12-21 09:06:32,465:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PB
2022-12-21 09:06:32,466:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PC
2022-12-21 09:06:32,467:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-21 09:06:32,468:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-21 09:06:32,469:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PF
2022-12-21 09:06:32,470:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-12-21 09:06:32,471:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LST
2022-12-21 09:06:32,473:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDA
2022-12-21 09:06:32,474:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDB
2022-12-21 09:06:32,475:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDC
2022-12-21 09:06:32,476:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
2022-12-21 09:06:32,476:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDE
2022-12-21 09:06:32,477:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDF
2022-12-21 09:06:32,478:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDG
2022-12-21 09:06:32,479:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BT
2022-12-21 09:06:32,480:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-12-21 09:06:32,481:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCT
2022-12-21 09:06:32,485:DEBUG:abstractprotocol:get_command_defn@71: Matched: ACCT2200,0259 to: ACCT value: 22
2022-12-21 09:06:32,485:DEBUG:pi17:get_full_command@959: _pre_cmd: b'^S014ACCT2200,0259'
2022-12-21 09:06:32,486:DEBUG:pi17:get_full_command@966: full command: b'^S014ACCT2200,0259\r'
2022-12-21 09:06:32,486:INFO:device:run_command@303: full command b'^S014ACCT2200,0259\r' for command ACCT2200,0259
[... cut ...]
  File "/home/pi/MQTT_MPPSOLAR/venv/mppsolar/lib/python3.7/site-packages/mppsolar/__init__.py", line 417, in main
    results = _device.run_command(command=_command)
  File "/home/pi/MQTT_MPPSOLAR/venv/mppsolar/lib/python3.7/site-packages/mppsolar/devices/device.py", line 320, in run_command
    command_defn=self._protocol.get_command_defn(command),
  File "/home/pi/MQTT_MPPSOLAR/venv/mppsolar/lib/python3.7/site-packages/mppsolar/inout/hidfullio.py", line 38, in send_and_receive
    os.write(usb0, to_send)
TimeoutError: [Errno 110] Connection timed out

```

jblance, 2022-12-21:
Can yoy try `mpp-solar -p /dev/hidraw0  -P PI17 -o screen -c ACCT -D`

drbacke, 2022-12-21:
```
(mppsolar) pi@raspberrypi:~/MQTT_MPPSOLAR $ mpp-solar -p /dev/hidraw0  -P PI17 -o screen -c ACCT -D
2022-12-21 11:12:40,559:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-21 11:12:40,560:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb65ac210>, _isConnected=False)
2022-12-21 11:12:40,560:DEBUG:__init__:main@240: udp port 5555
2022-12-21 11:12:40,560:DEBUG:__init__:main@242: Using Postgres None
2022-12-21 11:12:40,561:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-21 11:12:40,561:INFO:__init__:main@348: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2022-12-21 11:12:40,567:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-21 11:12:40,568:DEBUG:device:__init__@33: __init__ args ()
2022-12-21 11:12:40,568:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb65ac210>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-21 11:12:40,569:DEBUG:__init__:get_port_type@39: port matches hidraw
2022-12-21 11:12:40,569:INFO:__init__:get_port@92: Using hidrawio for communications
2022-12-21 11:12:40,572:DEBUG:__init__:get_protocol@13: Protocol PI17
2022-12-21 11:12:40,618:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5f764d0>, protocol PI17 protocol handler
2022-12-21 11:12:40,618:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5f76450>, 'ACCT', 'ACCT', 'screen', None, None)]
2022-12-21 11:12:40,618:INFO:__init__:main@404: Looping 1 commands
2022-12-21 11:12:40,619:INFO:__init__:main@415: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xb5f764d0>, protocol: PI17 protocol handler for command: ACCT, tag: ACCT, outputs: screen
2022-12-21 11:12:40,619:INFO:device:run_command@274: Running command ACCT
2022-12-21 11:12:40,619:INFO:pi17:get_full_command@905: Using protocol b'PI17' with 47 commands
2022-12-21 11:12:40,620:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT'
2022-12-21 11:12:40,620:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-21 11:12:40,621:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-21 11:12:40,622:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
2022-12-21 11:12:40,623:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EH
2022-12-21 11:12:40,624:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LON
2022-12-21 11:12:40,625:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PA
2022-12-21 11:12:40,626:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PB
2022-12-21 11:12:40,627:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PC
2022-12-21 11:12:40,628:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-21 11:12:40,629:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-21 11:12:40,630:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PF
2022-12-21 11:12:40,631:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-12-21 11:12:40,632:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LST
2022-12-21 11:12:40,633:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDA
2022-12-21 11:12:40,634:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDB
2022-12-21 11:12:40,635:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDC
2022-12-21 11:12:40,636:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
2022-12-21 11:12:40,637:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDE
2022-12-21 11:12:40,638:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDF
2022-12-21 11:12:40,639:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDG
2022-12-21 11:12:40,640:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BT
2022-12-21 11:12:40,641:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-12-21 11:12:40,642:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCT
2022-12-21 11:12:40,643:DEBUG:abstractprotocol:get_command_defn@71: Matched: ACCT to: ACCT value: 
2022-12-21 11:12:40,643:DEBUG:pi17:get_full_command@959: _pre_cmd: b'^S005ACCT'
2022-12-21 11:12:40,644:DEBUG:pi17:get_full_command@966: full command: b'^S005ACCT\r'
2022-12-21 11:12:40,644:INFO:device:run_command@303: full command b'^S005ACCT\r' for command ACCT
2022-12-21 11:12:40,644:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT'
2022-12-21 11:12:40,645:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-21 11:12:40,645:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-21 11:12:40,645:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
[... cut ...] qtt.client.Client object at 0xb65ac210>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: ACCT - Set AC charge time range
------------------------------------------------------------
Parameter                     	Value           Unit
warning0                      	Command ACCT was rejected	    
2022-12-21 11:12:41,777:DEBUG:__init__:main@449: Not daemon, so not looping

```

jblance, 2022-12-21:
interesting - seems the wrong command is 'found'

drbacke, 2022-12-22:
```
(mppsolar) pi@raspberrypi:~/MQTT_MPPSOLAR $ mpp-solar -p /dev/hidraw0  -P PI17infini -o screen -c ACCT -D
2022-12-22 09:43:34,258:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-22 09:43:34,259:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb663f210>, _isConnected=False)
2022-12-22 09:43:34,259:DEBUG:__init__:main@240: udp port 5555
2022-12-22 09:43:34,260:DEBUG:__init__:main@242: Using Postgres None
2022-12-22 09:43:34,260:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-22 09:43:34,261:INFO:__init__:main@348: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17infini"
2022-12-22 09:43:34,270:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-22 09:43:34,270:DEBUG:device:__init__@33: __init__ args ()
2022-12-22 09:43:34,271:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17infini', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb663f210>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-22 09:43:34,271:DEBUG:__init__:get_port_type@39: port matches hidraw
2022-12-22 09:43:34,272:INFO:__init__:get_port@92: Using hidrawio for communications
2022-12-22 09:43:34,276:DEBUG:__init__:get_protocol@13: Protocol PI17infini
2022-12-22 09:43:34,286:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xb60094b0>, protocol PI17INFINI protocol handler
2022-12-22 09:43:34,287:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb6009430>, 'ACCT', 'ACCT', 'screen', None, None)]
2022-12-22 09:43:34,288:INFO:__init__:main@404: Looping 1 commands
2022-12-22 09:43:34,288:INFO:__init__:main@415: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xb60094b0>, protocol: PI17INFINI protocol handler for command: ACCT, tag: ACCT, outputs: screen
2022-12-22 09:43:34,289:INFO:device:run_command@274: Running command ACCT
2022-12-22 09:43:34,289:INFO:pi17infini:get_full_command@802: Using protocol b'PI17INFINI' with 42 commands
2022-12-22 09:43:34,290:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'ACCT'
2022-12-22 09:43:34,290:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EY
2022-12-22 09:43:34,291:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EM
2022-12-22 09:43:34,293:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ED
2022-12-22 09:43:34,305:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EH
2022-12-22 09:43:34,309:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LON
2022-12-22 09:43:34,310:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PA
2022-12-22 09:43:34,313:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PB
2022-12-22 09:43:34,318:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PC
2022-12-22 09:43:34,321:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-22 09:43:34,326:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-22 09:43:34,327:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PF
2022-12-22 09:43:34,329:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-12-22 09:43:34,331:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: LST
2022-12-22 09:43:34,333:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDA
2022-12-22 09:43:34,334:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDB
2022-12-22 09:43:34,336:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDC
2022-12-22 09:43:34,337:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDD
2022-12-22 09:43:34,339:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDE
2022-12-22 09:43:34,340:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDF
2022-12-22 09:43:34,341:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: EDG
2022-12-22 09:43:34,342:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BT
2022-12-22 09:43:34,343:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGV
2022-12-22 09:43:34,345:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCT
2022-12-22 09:43:34,350:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: ACCB
2022-12-22 09:43:34,352:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2022-12-22 09:43:34,354:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: SEP
2022-12-22 09:43:34,355:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: BDCM
2022-12-22 09:43:34,358:INFO:abstractprotocol:get_command_defn@75: No command_defn found for ACCT
2022-12-22 09:43:34,359:INFO:device:run_command@303: full command None for command ACCT
2022-12-22 09:43:34,359:ERROR:device:run_command@306: full_command not found for ACCT in protocol b'PI17INFINI'
[... cut ...] username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xb663f210>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	full_command not found for ACCT in protocol b'PI17INFINI'	    
2022-12-22 09:43:34,366:DEBUG:__init__:main@449: Not daemon, so not looping

```

jblance, 2023-07-27:
should be resolved by 0.16.10

closed 2023-07-27 by jblance