jblance/mpp-solar #281: Trying to get  generated energy in KWH on an LV5048

multilectical, 2022-12-21:
Hi, I've got things running nicely here for my setup with an MPP LV5048 and thanks a ton for this btw, really appreciate the work that went into building it. 

My one thing I've been trying to figure out is get is if I can get daily or periodic kwh from charging and/or discharging from the inverter? Grepping around the repo, I see it in other protocols, but not in PI41. Is there  a way to do this here?

jblance, 2022-12-22:
Not according to the protocol documentation I have, you could try some of the other protocol commands and see if any respond (the docs never seem completely correct)
eg try `-c QET` or `-c QEY2022` but I dont think these are supported on the LV5048
Alternative is to use a downstream solution (software like: influx/grafana or home assistant, or hardware like victron smartshunt) to collect this info

multilectical, 2022-12-22:
No luck with QET or QEY2022.  Also tried QED20221220 (and variations) as well as some other commands I found looking around these tickets. 

I do have grafana set up, but not sure how to do the calculation with constantly varying voltage and amperage. I'm thinking breaking it up into chunks by minute, and then average the volts and amps before multiplying those by time? I feel like doing calculations like this could get way off and am sure there is a better way. Maybe even a native thing built into grafana that someone out there knows about?

my best leads so far:
https://community.grafana.com/t/daily-power-consumption-from-watts/16196/
https://community.grafana.com/t/display-daily-consumption-in-kwh/38393/4
https://www.reddit.com/r/homelab/comments/65yn4p/calculating_power_consumption_costs_in_grafana/

jblance, 2023-01-31:
Maybe I can add a power calculated field to PI41, then grafana can aggregate (i think - I know home assistant can do this aggregation)