jblance/mpp-solar #286: Return units within the JSON response

ziadakiki, 2022-12-23:
Hello, is there a way to add a new output perhaps that returns the unit with the value in json format?

Specifically, I would like `data[key][1]` to be returned within the JSON.

Here's the snipped I'm talking about but I'm not sure what would be the best solution regarding specifying a certain flag or a new output type in order to not cause any regressions.

https://github.com/jblance/mpp-solar/blob/6ba5268673280c6c86c6053b01d4d3d86cb2a8df/mppsolar/outputs/__init__.py#L84-L99

In other words, I'm looking for an output similar to this:
```
{ "ac_output_voltage": { "value": 220, "unit": "V" }, ... }
```
Instead of this:
```
{ "ac_output_voltage": 220, ... }
```

Saentist, 2022-12-23:
I suggest something similar

```rs
    // Register QPIGS Sensors
    register_sensor(client, cfg, "qpigs", "grid_voltage", "Grid Voltage", Some("V".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "grid_frequency", "Grid Frequency", Some("Hz".to_string()), "current-ac").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_voltage", "Out Voltage", Some("V".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_frequency", "Out Frequency", Some("Hz".to_string()), "current-ac").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_apparent_power", "Out apparent power", Some("W".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "ac_out_active_power", "Out active power", Some("W".to_string()), "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "out_load_percent", "Out load percent", Some("%".to_string()), "brightness-percent").await?;
    register_sensor(client, cfg, "qpigs", "bus_voltage", "Bus Voltage", Some("V".to_string()), "details").await?;
    register_sensor(client, cfg, "qpigs", "battery_voltage", "Battery Voltage", Some("V".to_string()), "battery-outline").await?;
    register_sensor(client, cfg, "qpigs", "battery_charge_current", "Battery charge current", Some("A".to_string()), "current-dc").await?;
    register_sensor(client, cfg, "qpigs", "battery_capacity", "Battery capacity", Some("%".to_string()), "battery-outline").await?;
    register_sensor(client, cfg, "qpigs", "inverter_heat_sink_temp", "Heat sink temperature", Some("°C".to_string()), "details").await?;
    register_sensor(client, cfg, "qpigs", "pv_input_current", "PV Input Current", Some("A".to_string()), "solar-power").await?;
    register_sensor(client, cfg, "qpigs", "pv_input_voltage", "PV Input Voltage", Some("V".to_string()), "solar-power").await?;
    register_sensor(client, cfg, "qpigs", "battery_scc_voltage", "Battery SCC Voltage", Some("V".to_string()), "current-dc").await?;
    register_sensor(client, cfg, "qpigs", "battery_discharge_current", "Battery discharge current", Some("A".to_string()), "battery-negative").await?;
    register_sensor(client, cfg, "qpigs", "device_status.charge_status", "Device charge status", None, "power-plug").await?;
    register_sensor(client, cfg, "qpigs", "device_status.active_load", "Active load", None, "power").await?;
```
https://github.com/jblance/mpp-solar/issues/269#issuecomment-1307901311

jblance, 2022-12-23:
something like
```
{"_command": "Q1", "_command_description": "Q1 query",
"time_until_the_end_of_absorb_charging": {"value": 0, "unit": "sec"},
"time_until_the_end_of_float_charging": {"value": 0, "unit": "sec"},
"scc_flag": {"value": "SCC is powered and communicating", "unit": ""},
"allowscconflag": {"value": "01", "unit": ""}, "chargeaveragecurrent":
{"value": "00", "unit": ""}, "scc_pwm_temperature": {"value": 59, "unit":
"\u00b0C"}, "inverter_temperature": {"value": 45, "unit": "\u00b0C"},
"battery_temperature": {"value": 53, "unit": "\u00b0C"},
"transformer_temperature": {"value": 68, "unit": "\u00b0C"}, "gpio13":
{"value": 0, "unit": ""}, "fan_lock_status": {"value": "Not locked",
"unit": ""}, "not_used": {"value": "000", "unit": ""}, "fan_pwm_speed":
{"value": 40, "unit": "Percent"}, "scc_charge_power": {"value": 580,
"unit": "W"}, "parallel_warning??": {"value": "0000", "unit": ""},
"sync_frequency": {"value": 50.0, "unit": ""}, "inverter_charge_status":
{"value": "float", "unit": ""}}
```

?

jblance, 2022-12-23:
> I suggest something similar
> 
> ```rust
>     // Register QPIGS Sensors
>     register_sensor(client, cfg, "qpigs", "grid_voltage", "Grid Voltage", Some("V".to_string()), "power-plug").await?;
>     register_sensor(client, cfg, "qpigs", "grid_frequency", "Grid Frequency", Some("Hz".to_string()), "current-ac").await?;
>     register_sensor(client, cfg, "qpigs", "ac_out_voltage", "Out Voltage", Some("V".to_string()), "power-plug").await?;
>     register_sensor(client, cfg, "qpigs", "ac_out_frequency", "Out Frequency", Some("Hz".to_string()), "current-ac").await?;
>     register_sensor(client, cfg, "qpigs", "ac_out_apparent_power", "Out apparent power", Some("W".to_string()), "power-plug").await?;
>     register_sensor(client, cfg, "qpigs", "ac_out_active_power", "Out active power", Some("W".to_string()), "power-plug").await?;
>     register_sensor(client, cfg, "qpigs", "out_load_percent", "Out load percent", Some("%".to_string()), "brightness-percent").await?;
>     register_sensor(client, cfg, "qpigs", "bus_voltage", "Bus Voltage", Some("V".to_string()), "details").await?;
>     register_sensor(client, cfg, "qpigs", "battery_voltage", "Battery Voltage", Some("V".to_string()), "battery-outline").await?;
>     register_sensor(client, cfg, "qpigs", "battery_charge_current", "Battery charge current", Some("A".to_string()), "current-dc").await?;
>     register_sensor(client, cfg, "qpigs", "battery_capacity", "Battery capacity", Some("%".to_string()), "battery-outline").await?;
>     register_sensor(client, cfg, "qpigs", "inverter_heat_sink_temp", "Heat sink temperature", Some("°C".to_string()), "details").await?;
>     register_sensor(client, cfg, "qpigs", "pv_input_current", "PV Input Current", Some("A".to_string()), "solar-power").await?;
>     register_sensor(client, cfg, "qpigs", "pv_input_voltage", "PV Input Voltage", Some("V".to_string()), "solar-power").await?;
>     register_sensor(client, cfg, "qpigs", "battery_scc_voltage", "Battery SCC Voltage", Some("V".to_string()), "current-dc").await?;
>     register_sensor(client, cfg, "qpigs", "battery_discharge_current", "Battery discharge current", Some("A".to_string()), "battery-negative").await?;
>     register_sensor(client, cfg, "qpigs", "device_status.charge_status", "Device charge status", None, "power-plug").await?;
>     register_sensor(client, cfg, "qpigs", "device_status.active_load", "Active load", None, "power").await?;
> ```
> 
> [#269 (comment)](https://github.com/jblance/mpp-solar/issues/269#issuecomment-1307901311)

this one is a bit harder as i need to refactor to core code and expand the command definitions - working on it though

ziadakiki, 2022-12-23:
Another suggestion might be to return an array:
```
"fan_pwm_speed": [40, "%"]
```
To be honest anything works, whatever you feel more suitable as long as I'm able to get the units with the values in the JSON.

jblance, 2022-12-23:
can do either, just thinking about the info without units (like command and command description) 
and whatever is easiest for you to process

jblance, 2022-12-23:
v0.14.12 has new output json_units which works as per first post (hopefully)
let me know if needs changes

ziadakiki, 2022-12-24:
I guess an output called `json_mqtt_units` or something similar should be added as well. Same goes for UDP (for completeness).

jblance, 2024-05-13:
moved to powermon code

closed 2024-05-13 by jblance