jblance/mpp-solar #287: [FR] Smart automation for correct port detection

Saentist, 2022-12-23:
On my PC there is a few USB devices which are HIDRAW(0-5)
but after restart they changed ordering.
If inverter was "hidraw1" next time is "hidraw5"

there is a simple bash script for detection

```bash
#!/bin/bash

FILES=/dev/hidraw*
for f in $FILES
do
  FILE=${f##*/}
  DEVICE="$(cat /sys/class/hidraw/${FILE}/device/uevent | grep HID_NAME | cut -d '=' -f2)"
  printf "%s \t %s\n" $FILE "$DEVICE"
done
```
In my case result is:
```shell
hidraw0          American Power Conversion  Smart-UPS C 1500 FW:UPS 10.0 / ID
hidraw1          STMicroelectronics HID in FS Mode   <<< Inverter 0665:5161 Cypress Semiconductor USB to Serial
hidraw2          HID TOUCH HID Touch Panel
hidraw3          SIGMACHIP USB Keyboard
hidraw4          SIGMACHIP USB Keyboard

```

How can make automatic hidraw detection and selection, 
without hard selecting and avoid after restart non working results?

jblance, 2022-12-23:
bit out of scope of my code
the way to do it is with udev rules - once done correctly they will allow permanent names for the devices even with reboots
eg https://unix.stackexchange.com/questions/66901/how-to-bind-usb-device-under-a-static-name

Saentist, 2022-12-24:
Sorry maybe stupid question but:
This will not work or even will destroy communication, 
if there is more then one inverter connected to PC.
Just because inverters use same communication chip aka VID:PID.

Isn't more logical to have some "in start" code, 
with to check all hidrawX/ttyUSBx with QID#QPI#QMN#QVFW#QVFW2#QVFW3 command 

```shell
mpp-solar -p /dev/hidraw1 -o raw -c QID#QPI#QMN#QVFW#QVFW2#QVFW3
Processing DEFAULT type responses
Command: QID - Device Serial Number inquiry
------------------------------------------------------------
raw_response                    '(553555355535552\xfd\r'
Processing DEFAULT type responses
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
raw_response                    '(PI30\x9a\x0b\r'
Command: QMN - No description found     <<<< Inverter Model
------------------------------------------------------------
raw_response                    "(VMIII-5000\xa8'\r"
Processing DEFAULT type responses
Command: QVFW - Main CPU firmware version inquiry
------------------------------------------------------------
raw_response                    '(VERFW:00091.20\xf8&\r'
Command: QVFW3 - No description found     <<<< Bluetooth Firmware
------------------------------------------------------------
raw_response                    '(VERFW:00002.61\x17c\r'
```

And if there is no problem with "Device Serial Number" aka 55355535553555 or device cloning
this information is enough to match "path to device"

----

This can be new functionality
```shell
mpp-solar -detect-inverter 
```
with will display inverter hardware information and eventually suggest "install as service"

jblance, 2022-12-24:
Not really ideal to spam data at a serial port - if it's in use for other
things it could cause problems

The udev naming can use a script as well, or your script could just be a
startup script thay creates symlinks as necessary
The udev naming can also use the physical port as part of the selection

I am thinking of a similar comand to your suggestion to help with protocol
selection though

Maybe the refactored code could have a pre loop command for info gathering
that then selects which commands or setup is used

Saentist, 2023-06-07:
Still interested to see some automation in detection...

```python
import glob
import os

for f in glob.glob("/dev/hidraw*"):
    with open(os.path.join("/sys/class/hidraw", os.path.basename(f), "device/uevent")) as uevent_file:
        device_name = None
        for line in uevent_file:
            if line.startswith("HID_NAME="):
                device_name = line[len("HID_NAME="):].strip()
                break
        print(f"{os.path.basename(f)}\t{device_name}")
```
now I have this result
```
hidraw2 STMicroelectronics HID in FS Mode
hidraw1 HID TOUCH HID Touch Panel
hidraw0 American Power Conversion  Smart-UPS C 1500 FW:UPS 10.0 / ID=1005
```
Question is how software to use automatically `hidrawX` `STMicroelectronics HID in FS Mode`  (0665:5161 Cypress Semiconductor USB to Serial),
when usb cable used?

jblance, 2023-06-07:
I still think this is the wrong approach - you are better to use udev rules (and base it off physical path if you have multiple)
If you are worried about multiple inverters causing issues with udev, you will have the same problem with the above giving duplicate results

Saentist, 2023-06-08:
maybe I need to make video
same usb port
simple restart
Result hidraw are rearranged but APC is always hidraw0
hidraw1 and 2 are randomly assigned to Inverter and AIO PC Touchscreen

jblance, 2023-06-08:
Yes hidraw will be random. But the actual physical port that the cable is
on is the same and you can write udev rules to create a new device based on
the physical port.

https://unix.stackexchange.com/questions/443250/udev-rule-to-alias-device-depending-on-usb-port-number
is an example (though it's not correct for your situation)

michas79de, 2023-08-12:
I also prefer udev rules isntead any automation.
Then I hope my Home Assistant can use command switches to the proper ports.

With two same cables, I get the same VID:PID combinations. But with their serial numbers I could use a rule.
Now my inverter is at /dev/inverter:

```
$ ls -l /dev/inverter
lrwxrwxrwx 1 root root 7 12. Aug 21:48 /dev/inverter -> ttyUSB0
```

But it doesn't seem to work. Though it points to the correct port, mpp-solar doesn't understand it:

```
$ mpp-solar -p /dev/inverter -c QPI
2023-08-12 22:07:58,834:ERROR:device:run_command@56: No communications port defined - unable to run command QPI
Command: Unknown command - No description found
--------------------------------------------------------------------------------
Parameter Value                 Unit
error     No communications port defined - unable to run command QPI
```

But with the direkt port it works:

```
$ mpp-solar -p /dev/ttyUSB0 -c QPI
Command: QPI - Protocol ID inquiry
--------------------------------------------------------------------------------
Parameter   Value               Unit
protocol_id PI30
```

So how do I have to use this?

michas79de, 2023-08-12:
In the meantime I flew over your code and found some kind of filter check.
After renaming to /dev/ttyusbinverter it fulfils the condition

```
    elif "ttyusb" in port:
        log.debug("port matches ttyusb")
        return PortType.SERIAL
```
and works nice now:

```
$ mpp-solar -p /dev/ttyusbinverter -c QPI
Command: QPI - Protocol ID inquiry
--------------------------------------------------------------------------------
Parameter   Value               Unit
protocol_id PI30
```

So for me it's solved. But I let it here, maybe it's helpful for others.

jblance, 2023-08-13:
alternatively you need to tell it what type of port you are using, eg
`$ mpp-solar -p /dev/inverter -c QPI --porttype serial`

jblance, 2024-05-13:
in powermon code

closed 2024-05-13 by jblance