jblance/mpp-solar #288: Sending commands via MQTT

EmanuelLoos, 2022-12-24:
Is there a way to change the inverters settings from HomeAssistant over MQTT in order to charge/discharge based on electricity prices? If not implemented yet, this would be a great feature as there are electricity providers like [aWATTar](https://www.awattar.at/) where prices are different for each hour.

michas79de, 2023-01-08:
I doubt it, because mpp-solar isn't listening for any MQTT massages, I think.

But after a fast first overflying of this, I think you can use Home Assistant to trigger a shell command via SSH.
https://community.home-assistant.io/t/running-a-shell-command-from-home-assistant-to-remote-linux-pc/135221
So HA should be able to change settings directly via mpp-solar without MQTT.

I think I will try this in a near future...

EmanuelLoos, 2023-01-08:
I'd very much prefer a MQTT listener as mpp-solar currently only runs as root with my setup and I really don't want to give HA root access to that machine.

jblance, 2023-01-08:
My dev code has this working now. Still testing it though. And there are a bunch of things to port

jblance, 2023-01-17:
working in powermon command of mppsolar - requires a different config and only supports serial devices at the moment

closed 2023-01-17 by jblance

robwolff3, 2023-03-08:
> working in powermon command of mppsolar - requires a different config and only supports serial devices at the moment

Is there a way to use the beta powermon version or do we have to wait for the next release? I didn't see a branch or beta release for it.

jblance, 2023-03-09:
The command is included in the current mppsolar package. Config file is
changing and broken in latest version. I'm on holiday atm but will get it
minimally working next week

robwolff3, 2023-03-21:
Very good to hear! I have many uses cases for sending commands to my inverter and I'm excited to use this new feature when its ready.

omarghader, 2023-04-24:
Hello, has the MQTT listener been implemented?
I can't find any doc about it. Thank you for this project, it's really awesome. Good job guys !