jblance/mpp-solar #291: record_type not present

mcnalu, 2022-12-26:
BLE comms with DALY BMS fails because there is no record_type set. It looks like this only exists for JK protocols.

$ jkbms -p 96:69:08:03:0C:5F -P DALY -c SOC -D
2022-12-26 16:17:52,257:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.5
2022-12-26 16:17:52,258:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7f8bb6fdb3a0>, _isConnected=False)
2022-12-26 16:17:52,258:DEBUG:__init__:main@240: udp port 5555
2022-12-26 16:17:52,258:DEBUG:__init__:main@242: Using Postgres None
2022-12-26 16:17:52,258:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-26 16:17:52,258:INFO:__init__:main@347: Creating device "unnamed" (type: "jkbms") on port "96:69:08:03:0C:5F (porttype=None)" using protocol "DALY"
2022-12-26 16:17:52,261:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-12-26 16:17:52,261:DEBUG:device:__init__@33: __init__ args ()
2022-12-26 16:17:52,261:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '96:69:08:03:0C:5F', 'protocol': 'DALY', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7f8bb6fdb3a0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-26 16:17:52,261:DEBUG:__init__:get_port_type@54: port matches jkble ':'
2022-12-26 16:17:52,261:INFO:__init__:get_port@107: Using jkbleio for communications
2022-12-26 16:17:52,293:DEBUG:__init__:get_protocol@13: Protocol DALY
2022-12-26 16:17:52,295:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0x7f8bb664e830>, protocol Daly BMS protocol handler
2022-12-26 16:17:52,295:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7f8bb664e0b0>, 'SOC', 'SOC', 'screen', None, None)]
2022-12-26 16:17:52,295:INFO:__init__:main@404: Looping 1 commands
2022-12-26 16:17:52,295:INFO:__init__:main@414: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x7f8bb664e830>, protocol: Daly BMS protocol handler for command: SOC, tag: SOC, outputs: screen
2022-12-26 16:17:52,295:INFO:device:run_command@274: Running command SOC
2022-12-26 16:17:52,295:INFO:daly:get_full_command@213: Using protocol b'DALY' with 7 commands
2022-12-26 16:17:52,296:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'SOC'
2022-12-26 16:17:52,296:DEBUG:abstractprotocol:get_command_defn@62: Found command SOC in protocol b'DALY'
2022-12-26 16:17:52,296:DEBUG:daly:get_full_command@230: full command: b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n'
2022-12-26 16:17:52,296:INFO:device:run_command@303: full command b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n' for command SOC
2022-12-26 16:17:52,296:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'SOC'
2022-12-26 16:17:52,296:DEBUG:abstractprotocol:get_command_defn@62: Found command SOC in protocol b'DALY'
2022-12-26 16:17:52,296:INFO:daly:get_full_command@213: Using protocol b'DALY' with 7 commands
2022-12-26 16:17:52,296:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'SOC'
2022-12-26 16:17:52,296:DEBUG:abstractprotocol:get_command_defn@62: Found command SOC in protocol b'DALY'
2022-12-26 16:17:52,296:DEBUG:daly:get_full_command@230: full command: b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n'
2022-12-26 16:17:52,297:INFO:jkbleio:send_and_receive@29: full command b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n' for command SOC
2022-12-26 16:17:52,297:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'SOC'
2022-12-26 16:17:52,297:DEBUG:abstractprotocol:get_command_defn@62: Found command SOC in protocol b'DALY'
Traceback (most recent call last):
  File "/usr/local/bin/jkbms", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/__init__.py", line 417, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/devices/device.py", line 316, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/inout/jkbleio.py", line 32, in send_and_receive
    record_type = command_defn["record_type"]
KeyError: 'record_type'

jblance, 2023-01-08:
Daly is not supported via bluetooth (yet), I now have a bluetooth daly bms that i will build into my next battery, so this can be worked on

mcnalu, 2023-01-08:
Ah ok. I've managed to write my own code in python to talk to the daly bms
now and am working on an ESP32 arduino sketch also. Let me know of i can
assist in any way.

jblance, 2023-01-08:
nice. if you can provide a link to your code others can use it too and/or i can try to incorporate it

hasselk, 2023-09-07:
> Ah ok. I've managed to write my own code in python to talk to the daly bms now and am working on an ESP32 arduino sketch also. Let me know of i can assist in any way.
> […](#)

@mcnalu  Do you have a link to your code? This would be very nice to have, since i have now an Daly Smart Equalizer ;-)

jblance, 2023-09-07:
Yes I agree, can you link to your code?

closed 2024-05-07 by jblance