jblance/mpp-solar #293: [FR] Voltronic /Axpert Invertors Bluetooth support

Saentist, 2022-12-31:
Is there any information how to read status over Bluetooth in Voltronics / Axpert and clones
All they are controlled by WatchPower app on android
Mostly all Inverters with this control panel.
[image] 
[image]

jblance, 2023-01-08:
Do you have this setup working (with the app)?
Have you logged the bluetooth traffic from the app?

Saentist, 2023-01-08:
> Do you have this setup working (with the app)? Have you logged the bluetooth traffic from the app?

Yes I have Bluetooth compatible inverter.
But for second part accept any suggestions how to do it.
I can connect to inverter with standard PIN 123456 under Linux/Windows, 
but no COM port emulation is enabled on it as on phones.

jblance, 2023-01-08:
Yep thats the way - then post the log and/or try to decode it yourself 
It takes a lot of trial and error to reverse engineer the ble log, but I am building a new battery at the moment and have a similar inverter, so will take a look once i get it running

Saentist, 2023-01-09:
[btsnoop_hci(1).zip](https://github.com/jblance/mpp-solar/files/10371745/btsnoop_hci.1.zip)

Its not so human readable, but `ID55355535553555` is inverter

jblance, 2023-01-09:
Great. Did you do anything in the app or is it just receiving data?
Can you grab a screen shot of the data

Ps you need wireshark to be able to read it and even then it takes a lot of
guess work

Saentist, 2023-01-09:
I try to analyse via Wireshark but no success yet not see values.
capture is made with very low spec android 6 phone (Alcatel Pixi 4)
Will try to make new capture with screen recording, 
to be more easy to compare actions and responses.

Saentist, 2023-01-09:
there is some long repeating string in peyload with lenght 671
````
806001060002428000000000059cbd357d0000000000000000007776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db9cbd357d0000000000000000007776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db9cbd357d0000000000000000007776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db9cbd357d0000000000000000007776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db9cbd357d0000000000000000007776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db7776db6eedb6dbbbb6db776db6ddddb6dbbb6db6eeedb6dddb6db
````

guessing by repeating logic can be something as this
```
806001060002428000000000059
cbd357d000000000000000000
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db9
cbd357d000000000000000000
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db9
cbd357d000000000000000000
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db9
cbd357d000000000000000000
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db9
cbd357d000000000000000000
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
7776db6eedb6dbbbb6db
776db6ddddb6dbbb6db6eeedb6dddb6db
```
repeating segments aka possibly not changed setting from report

first 8 symbols are growing ... guess time or timer.

Saentist, 2023-01-09:
one more log with video for comparsion
[hci_snoop20230109214525.zip](https://github.com/jblance/mpp-solar/files/10376606/hci_snoop20230109214525.zip)

https://user-images.githubusercontent.com/16975883/211401729-532a9128-f45e-4b7a-b061-9992afe42156.mp4

jblance, 2023-01-09:
it looks like an encrypted setup
[image]
but some info is clearly visible in the trace, eg product info in frame 396
[image]
[image]

Saentist, 2023-01-09:
There is a encryption key after connecting.

jblance, 2023-01-09:
response on handle 0x001d might have basic info
[image]
[image]
[image]
[image]

jblance, 2023-01-09:
> There is a encryption key after connecting.

yes, but the encryption will make connecting with python harder

Saentist, 2023-01-09:
> [image]

No battery connected to inverter.

jblance, 2023-01-09:
> No battery connected to inverter.

yes, just trying to see the same values (after decode) in the log file

Saentist, 2023-01-09:
first log have PV voltage, MPPT make it flat ~120Vdc

jblance, 2023-01-10:
handle 0x0023 rated info
[image]
[image]
handle 0x0023
`fc08d900fc08f401d90088138813e00101000000`
```
fc08 2300
d900 217
fc08 2300
f401 500
d900 217
8813 5000
8813 5000
e001 480
0100 1
0000 0
```

Saentist, 2023-01-10:
0x0023 inverter rated specs
0x0017 firmware versions
0x001a serial number
0x001d ???
0x0020 ???
0x0026 ???
0x0029 ??? only zeros
0x002c ???
0x002f ???
0x0035 ???
0x0038 ??? * small changes 
0x003b ??? *
0x003e ??? zeros
0x0042 ???
0x0045 ???
0x0048 ??? 
0x004b ???

* respond on value changing in settings.

Saentist, 2023-01-11:
I try some ble scan tool and this is strange results.
[image]

https://user-images.githubusercontent.com/16975883/211927065-7b1ca994-561f-4f12-b960-22e0e9e3afcc.mp4

[hci_snoop20230109214525 (2).zip](https://github.com/jblance/mpp-solar/files/10396293/hci_snoop20230109214525.2.zip)
[hci_snoop20230109214525 (2)_.zip](https://github.com/jblance/mpp-solar/files/10396296/hci_snoop20230109214525.2._.zip)

sniffing is larger and in two parts

Interesting if inverter s paired via android /not app/ it cannot be scanned

jblance, 2023-01-11:
> Interesting if inverter s paired via android /not app/ it cannot be scanned

Wonder if this the encryption? 

I thought the inverter I just got was one of the ble ones, turns out it's just wifi
[image]

jblance, 2023-01-11:
On the git repo there is a ble-scan.py can you try running that somewhere in range of the inverter (you'll need `pip install bleak` if you don't have it)

Saentist, 2023-01-12:
Yes will try

----
TI recommend this tool
https://www.ti.com/lit/an/swpa234/swpa234.pdf?ts=1673270553469
https://github.com/Hari-Nagalla/BT-HCI-Logging

----
Wondering how many time will take to "close source payed monitoring" from "free software" to monetize your code
Enable donations in repo ;)

jblance, 2023-01-12:
The log capture is the easy part, the hard bit is figuring out what the data corresponds to.

I don't think there's money in this code. Helpful for some maybe but that about it

Saentist, 2023-01-15:
> ble-scan.py

Can you post link to this file?

jblance, 2023-01-15:
https://github.com/jblance/mpp-solar/blob/master/ble-scan.py

DonDavici, 2024-07-13:
Maybe this helps you.

https://github.com/DonDavici/OutbackSPC/blob/main/outbackbt.py

Saentist, 2025-03-29:
@jblance @DonDavici 
can we have some test version for testing

Saentist, 2025-04-21:
https://github.com/syssi/esphome-votronic/blob/main/votronic_bluetooth.md
Some more code examples.