jblance/mpp-solar #298: data sent to iobroker mqtt server

ts1cos, 2023-01-09:
I'm trying to send my data to the iobroker mqtt server but I can't get a connection. what am I doing wrong?

`2023-01-09 19:30:22,651:WARNING:mqtt:output@95: Error publishing MQTT messages to broker '192.168.178.20' on port '1886' with auth '{'username': 'user1', 'password': 'user'}'
2023-01-09 19:30:22,651:WARNING:mqtt:output@98: '<=' not supported between instances of 'str' and 'int'`

jblance, 2023-01-09:
> 2023-01-09 ***@***.***: Error publishing MQTT

What is the command you are using?

ts1cos, 2023-01-09:
mpp-solar -n 'FirstInverter' -P pi30revo -o screen -c QALL -p /dev/ttyUSB1 -o mqtt,screen -q 192.168.178.20 --mqttport 1886 --mqttuser user1 --mqttpass user --mqtttopic /pv/status

jblance, 2023-01-09:
did you change the port of your broker from the default?
what version of mpp-solar are you using?
can you post a full debug output

ts1cos, 2023-01-09:
yes i have change the port to 1886 in iobroker mqttserver.
Solar Device Command Utility, version: 0.14.12

`2023-01-09 20:18:38,146:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.12
2023-01-09 20:18:38,146:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2023-01-09 20:18:38,147:INFO:__init__:get_port@112: Using serialio for communications
2023-01-09 20:18:38,150:INFO:__init__:main@404: Looping 1 commands
2023-01-09 20:18:38,150:INFO:__init__:main@414: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7f8f383f5f90>, protocol: PI30 protocol handler for command: , tag: , outputs: screen
2023-01-09 20:18:38,150:INFO:device:run_command@274: Running command 
2023-01-09 20:18:38,150:INFO:abstractprotocol:get_full_command@55: Using protocol b'PI30' with 40 commands
2023-01-09 20:18:38,150:INFO:device:run_command@303: full command b'QPI\xbe\xac\r' for command QPI
2023-01-09 20:18:38,150:WARNING:serialio:send_and_receive@33: Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2023-01-09 20:18:38,151:INFO:serialio:send_and_receive@34: Command execution failed
2023-01-09 20:18:38,151:INFO:__init__:get_output@38: attempting to create output processor: screen
2023-01-09 20:18:38,151:INFO:screen:output@19: Using output processor: screen
Parameter                       Value           Unit
error                           Serial command execution failed     
root@influx2:/dev# mpp-solar -v
Solar Device Command Utility, version: 0.14.12
root@influx2:/dev# mpp-solar -n 'FirstInverter' -P pi30revo -o screen -c QALL -p /dev/ttyUSB1 -o domoticz_mqtt,screen -q 192.168.178.20 --mqttport 1886 --mqttuser user1 --mqttpass user -D
2023-01-09 20:20:15,956:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.12
2023-01-09 20:20:15,956:DEBUG:__init__:main@238: MqttBroker(name='192.168.178.20', port='1886', username='user1', password='user', results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f6da526e8f0>, _isConnected=False)
2023-01-09 20:20:15,957:DEBUG:__init__:main@240: udp port 5555
2023-01-09 20:20:15,957:DEBUG:__init__:main@242: Using Postgres None
2023-01-09 20:20:15,957:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2023-01-09 20:20:15,957:INFO:__init__:main@347: Creating device "FirstInverter" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=None)" using protocol "pi30revo"
2023-01-09 20:20:15,958:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-01-09 20:20:15,958:DEBUG:device:__init__@33: __init__ args ()
2023-01-09 20:20:15,958:DEBUG:device:__init__@34: __init__ kwargs {'name': 'FirstInverter', 'port': '/dev/ttyUSB1', 'protocol': 'pi30revo', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='192.168.178.20', port='1886', username='user1', password='user', results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f6da526e8f0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-01-09 20:20:15,958:DEBUG:__init__:get_port_type@69: port matches ttyusb
2023-01-09 20:20:15,958:INFO:__init__:get_port@112: Using serialio for communications
2023-01-09 20:20:15,960:DEBUG:__init__:get_protocol@13: Protocol pi30revo
2023-01-09 20:20:15,961:DEBUG:device:__init__@38: __init__ name FirstInverter, port <mppsolar.inout.serialio.SerialIO object at 0x7f6da48e5f90>, protocol PI30 protocol handler for REVO and similar inverters
2023-01-09 20:20:15,961:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f6da48e60b0>, 'QALL', 'QALL', 'domoticz_mqtt,screen', None, None)]
2023-01-09 20:20:15,961:INFO:__init__:main@404: Looping 1 commands
2023-01-09 20:20:15,961:INFO:__init__:main@414: Getting results from device: mppsolar device - name: FirstInverter, port: <mppsolar.inout.serialio.SerialIO object at 0x7f6da48e5f90>, protocol: PI30 protocol handler for REVO and similar inverters for command: QALL, tag: QALL, outputs: domoticz_mqtt,screen
2023-01-09 20:20:15,961:INFO:device:run_command@274: Running command QALL
2023-01-09 20:20:15,961:INFO:pi30revo:get_full_command@397: sing protocol b'PI30REVO' with 6 commands
2023-01-09 20:20:15,961:DEBUG:abstractprotocol:get_command_defn@67: Processing command 'QALL'
2023-01-09 20:20:15,962:DEBUG:abstractprotocol:get_command_defn@69: Found command QALL in protocol b'PI30REVO'
2023-01-09 20:20:15,962:DEBUG:pi30revo:get_full_command@409: Using CHK checksum approach for command QALL
2023-01-09 20:20:15,962:DEBUG:pi30revo:get_full_command@411: checksum 43
2023-01-09 20:20:15,962:DEBUG:pi30revo:get_full_command@420: full command: b'QALL+\r'
2023-01-09 20:20:15,962:INFO:device:run_command@303: full command b'QALL+\r' for command QALL
2023-01-09 20:20:15,962:DEBUG:abstractprotocol:get_command_defn@67: Processing command 'QALL'
2023-01-09 20:20:15,962:DEBUG:abstractprotocol:get_command_defn@69: Found command QALL in protocol b'PI30REVO'
2023-01-09 20:20:15,962:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 2400
2023-01-09 20:20:15,963:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-01-09 20:20:16,349:DEBUG:serialio:send_and_receive@30: serial response was: b'(230 49.9 230 49.9 0644 011 52.1 085 000 000 000 00.0 0000 276403 000276 L 00 00}\r'
[... cut ...]
pv_input_voltage                0               V   
pv_input_current                0.0             A   
pv_power                        0               W   
daily_pv_power_generated        276403          WH  
total_pv_power_generated        276             kWH 
device_mode                     Line                
warning_code                    No warning          
fault_code                      No fault            
2023-01-09 20:20:17,437:DEBUG:__init__:main@449: Not daemon, so not looping`

ts1cos, 2023-01-09:
it works with new instance of mqtt server without password on port 1883. but username password and port is correct

jblance, 2023-01-09:
> it works with new instance of mqtt server without password on port 1883. but username password and port is correct

Was this a domoticz server as well?

ts1cos, 2023-01-09:
it works with mqtt and with domoticz_mqtt i think domoticz is better. but how can i send the data every 10 sec to the mqtt server? then I just have to run 2 mqtt servers.

jblance, 2023-01-09:
ok, i was wondering if it was that the code didnt work with domoticz 
so its that the auth code doesnt work (with domoticz)?
the auth bit is pretty simplistic, so it is possible that its wrong

Saentist, 2023-01-09:
Why in command line there is two time output to `screen`?

```bash 
mpp-solar -n 'FirstInverter' -P pi30revo -o screen -c QALL -p /dev/ttyUSB1 -o domoticz_mqtt,screen -q 192.168.178.20 --mqttport 1886 --mqttuser user1 --mqttpass user -D
```
just 
```bash
mpp-solar -n 'FirstInverter' -P pi30revo -c QALL -p /dev/ttyUSB1 -o domoticz_mqtt,screen -q 192.168.178.20 --mqttport 1886 --mqttuser user1 --mqttpass user -D
```
is enough.

ts1cos, 2023-01-09:
now i use this code  `mpp-solar -n 'FirstInverter' -P pi30revo -c QALL -p /dev/ttyUSB1 -o domoticz_mqtt,screen -q 192.168.178.20` to the 2. mqttserver in iobroker. and it works. how can i run the code every 10 sec?

jblance, 2023-01-09:
either use cron or setup mpp-solar as a service

Saentist, 2023-01-09:
Add `--daemon`
but will loop every 60 seconds with is enough.
or setup as service

closed 2023-01-31 by jblance