jblance/mpp-solar #304: This entity is no longer being provided by the mqtt integration. If the entity is no longer in use, delete it in settings.

Saentist, 2023-01-29:
latest
```bash
mpp-solar -p /dev/hidraw2 -o screen -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG
Command: QID - Device Serial Number inquiry
--------------------------------------------------------------------------------
Parameter     Value             Unit
serial_number 55355535553555
Command: QDI - Default Settings inquiry
--------------------------------------------------------------------------------
Parameter                      Value            Unit
ac_output_voltage              230.0            V
ac_output_frequency            50.0             Hz
max_ac_charging_current        30               A
battery_under_voltage          42.0             V
battery_float_charge_voltage   54.0             V
battery_bulk_charge_voltage    56.4             V
battery_recharge_voltage       46.0             V
max_charging_current           60               A
input_voltage_range            Appliance
output_source_priority         Utility first
charger_source_priority        Solar + Utility
battery_type                   AGM
buzzer                         enabled
power_saving                   disabled
overload_restart               disabled
over_temperature_restart       disabled
lcd_backlight                  enabled
primary_source_interrupt_alarm enabled
record_fault_code              enabled
overload_bypass                disabled
lcd_reset_to_default           enabled
output_mode                    single machine output
battery_redischarge_voltage    54.0             V
pv_ok_condition                As long as one unit of inverters has connect PV, parallel system will consider PV OK
pv_power_balance               PV input max power will be the sum of the max charged power and loads power
Command: QMOD - Mode inquiry
--------------------------------------------------------------------------------
Parameter   Value               Unit
device_mode Line
Command: QPI - Protocol ID inquiry
--------------------------------------------------------------------------------
Parameter   Value               Unit
protocol_id PI30
Command: QPIGS - General Status Parameters inquiry
--------------------------------------------------------------------------------
Parameter                                   Value               Unit
ac_input_voltage                            214.3               V       {'icon': 'lightning'}
ac_input_frequency                          50.0                Hz
ac_output_voltage                           214.3               V
ac_output_frequency                         50.0                Hz
ac_output_apparent_power                    0                   VA
ac_output_active_power                      0                   W
ac_output_load                              0                   %
bus_voltage                                 362                 V
battery_voltage                             1.0                 V
battery_charging_current                    0                   A
battery_capacity                            0                   %
inverter_heat_sink_temperature              28                  °C
pv_input_current_for_battery                0.0                 A
pv_input_voltage                            0.0                 V
battery_voltage_from_scc                    0.0                 V
battery_discharge_current                   0                   A
is_sbu_priority_version_added               0                   bool
is_configuration_changed                    0                   bool
is_scc_firmware_updated                     0                   bool
is_load_on                                  1                   bool
is_battery_voltage_to_steady_while_charging 0                   bool
is_charging_on                              0                   bool
is_scc_charging_on                          0                   bool
is_ac_charging_on                           0                   bool
[... 148 lines cut ...]
QPIRI AC Output Active Power = 5,000 W
QPIRI AC Output Apparent Power = 5,000 VA
QPIRI AC Output Current = 21.7 A
QPIRI AC Output Frequency = 50.0 Hz
QPIRI AC Output Voltage = 230.0 V
QPIRI Battery Bulk Charge Voltage = 56.4 V
QPIRI Battery Float Charge Voltage = 54.0 V
QPIRI Battery Recharge Voltage = 46.0 V
QPIRI Battery Redischarge Voltage = 54.0 V
QPIRI Battery Type = AGM
QPIRI Battery Under Voltage = 42.0 V
QPIRI Battery Voltage = 48.0 V
QPIRI Charger Source Priority = Solar first
QPIRI Input Voltage Range = UPS
QPIRI Machine Type = Off Grid
QPIRI Max AC Charging Current = 2 A
QPIRI Max Charging Current = 10 A
QPIRI Max Parallel Units = 1 units
QPIRI Output Mode = single machine output
QPIRI Output Source Priority = Utility first
QPIRI PV OK Condition = As long as one unit of inverters has connect PV, parallel system will consider PV OK
QPIRI PV Power Balance = PV input max power will be the sum of the max charged power and loads power
QPIRI Topology = transformerless
```
No firmware or etc. changed.

jblance, 2023-01-29:
> QDI Over temperature restart = Unavailable
> QDI Over temperature restart = Off

I assume this is in home assistant - the entities look to be duplicated, this might be the first update youve done since the change to move boolean variable to 'binary sensors' can you check the details of an unavailable entity and one that is available with the same name?

If one is a sensor and the other a binary sensor, you will either need to delete the unavailable sensor entities or roll back to an earlier version and remove the binary sensors

jblance, 2023-01-29:
> QPIGS Is AC Charging On = Unavailable
> QPIGS Is AC Charging On = Off
> QPIGS Is Battery Voltage to Steady While Charging = Unavailable
> QPIGS Is Battery Voltage to Steady While Charging = Off
> QPIGS Is [...]

Also note that some of these might change names in the future as I get a better understanding of the inverters (particularly in PI30 and related protocols

eg `Is SCC Charging On = Off` would be better as `SCC Charging = Off` or `Solar Charge Controller Charging = Off` in my opinion

the code is still very beta (hence the 0.xx.y version numbering) and so is likely to have breaking changes from time to time

Im relying on pip/pypi for package distribution currently so have not really gone with release notes or the like

Saentist, 2023-01-29:
There is a logic, 
but what about `QPIGS AC Input Voltage`

```json
mpp-solar -p /dev/hidraw2 -o screen -c QPIGS
Command: QPIGS - General Status Parameters inquiry
--------------------------------------------------------------------------------
Parameter                                   Value               Unit
ac_input_voltage                            220.8               V       {'icon': 'lightning'}
ac_input_frequency                          50.0                Hz
ac_output_voltage                           220.8               V
ac_output_frequency                         50.0                Hz
ac_output_apparent_power                    0                   VA
ac_output_active_power                      0                   W
ac_output_load                              0                   %
bus_voltage                                 362                 V
battery_voltage                             0.9                 V
battery_charging_current                    0                   A
battery_capacity                            0                   %
inverter_heat_sink_temperature              37                  °C
pv_input_current_for_battery                0.0                 A
pv_input_voltage                            0.0                 V
battery_voltage_from_scc                    0.0                 V
battery_discharge_current                   0                   A
is_sbu_priority_version_added               0                   bool
is_configuration_changed                    0                   bool
is_scc_firmware_updated                     0                   bool
is_load_on                                  1                   bool
is_battery_voltage_to_steady_while_charging 0                   bool
is_charging_on                              0                   bool
is_scc_charging_on                          0                   bool
is_ac_charging_on                           0                   bool
rsv1                                        0                   A
rsv2                                        0                   A
pv_input_power                              0                   W
is_charging_to_float                        0                   bool
is_switched_on                              1                   bool
is_reserved                                 0                   bool
```

jblance, 2023-01-29:
what about it?
has it disappeared?

Saentist, 2023-01-29:
Yes, all with return "icon"

jblance, 2023-01-29:
yes thats new development, device classes and icons are being added to protocols to improve home assistant support - its a work in progress and will likely change
Has it caused a problem?

Saentist, 2023-01-29:
Minimal, but is bug, hope fixed soon.

jblance, 2023-01-29:
how so? the print to screen is used as debug - do you have a use case for the screen output that is needed to not show the extra info?

Saentist, 2023-01-29:
PI30MAX work but not correctly with all commands.

```bash
Command: QID - No description found
--------------------------------------------------------------------------------
Parameter Value                 Unit
warning   No definition for command QID in protocol b'PI30MAX'
response    53555355535552ý
```
Is there a way no mater protocol, commands response values to be decoded?

jblance, 2023-01-29:
oh thats a bug
I'll get a fix in the next release (its in github now)
Also try `QSID`

> Is there a way no mater protocol, commands response values to be decoded?

No, because the protocol 'defines' how to decode, and it differs by inverter and command

Saentist, 2023-01-29:
```
 mpp-solar -p /dev/hidraw2 -c QSID -P PI30MAX
Command: QSID - Device Serial Number inquiry
--------------------------------------------------------------------------------
Parameter     Value             Unit
serial_number 1455355535553555355535
```
by spec this is:
```
QSID<cr>: The device serial number inquiry (the length is more than 14)
Computer: QSID<CRC><cr>
Device: (NNXXXXXXXXXXXXXXXXXXXX <CRC><cr>
NN: Serial number valid length, X: Serial number, invalid part is filled as ‘0’, total X is 20. 
```
[MAX Communication Protocol for HV7.2k LV5k V00 20200717.pdf](https://github.com/jblance/mpp-solar/files/10531016/MAX.Communication.Protocol.for.HV7.2k.LV5k.V00.20200717.pdf)

there mus be a table of devices and it's command capability's 
something as 
```bash
mpp-solar -p /dev/hidraw2 -o screen -c QMN#QGMN
Command: QMN - Model Name Inquiry
--------------------------------------------------------------------------------
Parameter  Value                Unit
model_name VMIII-5000
Command: QGMN - General Model Name Inquiry
--------------------------------------------------------------------------------
Parameter          Value                Unit
general_model_name 037
```
[Axpert KS&MKS&V&KING RS232 Protocol 20180104.pdf](https://github.com/jblance/mpp-solar/files/10531029/Axpert.KS.MKS.V.KING.RS232.Protocol.20180104.pdf)
page 22  QGMN
With model and sub model can have nice database of supported protocol and commands.

jblance, 2023-01-30:
yes, although need to get all that info and determine which protocol is relevant
also this only works for PI30 protocol

jblance, 2023-01-30:
> yes, although need to get all that info and determine which protocol is relevant also this only works for PI30 protocol

My inverter has
```
QMN: MKS2-8000
QGMN: 044
```

Maybe QPI and QGMN together works, eg `PI30-044` for my inverter and `PI30-037` for yours
The page 22 you linked to only has values to 024

Saentist, 2023-01-30:
> > yes, although need to get all that info and determine which protocol is relevant also this only works for PI30 protocol
> Maybe QPI and QGMN together works, eg `PI30-044` for my inverter and `PI30-037` for yours The page 22 you linked to only has values to 024

Document is older (year 2018) and not provided to end users.
QPI#QMN#QGMN 
Enable Wiki in repo to make this database, and move documentation there.

jblance, 2023-01-30:
Good idea
Wiki enabled 
Doc move a really good idea, will take me some time tho

Saentist, 2023-01-30:
QMN: VMIII
QGMN
023 PIP-5048MK PIP-5048MK
024 PIP-3024MK PIP-3024MK

jblance, 2023-01-30:
Can you still not get into the wiki?
Ive added some pages including the start of a protocol recommendations etc page
https://github.com/jblance/mpp-solar/wiki/Protocol-Recommendations
I will add a command to build an identifier across a bunch of different protocols this built up of a few different commands to make it easier to build the database

jblance, 2023-01-30:
> Inverter: (MMMM-NNNN-BB-VV-SSSS<CRC><cr> if Inverter accepts this command, otherwise,
> responds (NAK<cr>
> MMMM: model name, NNNN: Rated output VA, BB: Battery voltage, VV: HV/LV, SSSS:
> Special flag

I love how none of the items in the table actually follow this spec

jblance, 2023-01-31:
Are there any outstanding issues for this item now?
Do you need a 'screen' output without the extra info (like icons)?

Saentist, 2023-01-31:
after run one with P30MAX protocol all icons was set and data function with regular P30

Saentist, 2023-02-01:
Reopening 
After restart of HA, again do not want to receive values with icon on P30 protocol.
In terminal with `screen` output all seems OK, but in `hassd_mqtt` output there is something weird.

jblance, 2023-02-01:
the icons are in pi30max only atm - is your inverter not best with pi30max?

Saentist, 2023-02-01:
there is in PI30
https://github.com/jblance/mpp-solar/blob/5a90dd874fe137956f7493b0a5721f73eae4c600/mppsolar/protocols/pi30.py#L328
https://github.com/jblance/mpp-solar/blob/5a90dd874fe137956f7493b0a5721f73eae4c600/mppsolar/protocols/pi30.py#L336

jblance, 2023-02-01:
yes but way fewer that pi30max - do you use the Q1 command? is the icons working there?

Saentist, 2023-02-01:
`QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG`
just this one.

jblance, 2023-04-11:
completed i think - let me know if there is still an issue outstanding

closed 2023-04-11 by jblance