jblance/mpp-solar #306: add `--getDeviceId`

jblance, 2023-01-30:


jblance, 2023-02-01:
done for pi30

jblance, 2023-02-01:
need :
- [ ] pi16, 
- [ ] pi17, 
- [ ] pi18, 
- [ ] pi41

preussal, 2023-02-21:
PI18 not have a Query for the model or any other Device.
I can Query the Protocol and the Serial number

jblance, 2023-02-23:
the purpose of this command is to distinguish between implementations of the protocol - ie for pi30 there are many different versions / ways to decode, so we need something more than "its PI30" 
for pi18 i think the protocol is enough

RoySalisbury, 2023-02-27:
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value          	Unit
deviceid  PI30:045:MKS2-6500	    

For this inverter (EG4-6500EX) the protocol should be PI30MAX .. This is because it has two PV inputs and the QPIGS/QPIGS2 commands are needed.

Saentist, 2023-02-27:
> #### Command: Get Device ID - Generate a device id
> Parameter Value Unit deviceid PI30:045:MKS2-6500
> 
> For this inverter (EG4-6500EX) the protocol should be PI30MAX .. This is because it has two PV inputs and the QPIGS/QPIGS2 commands are needed.

There is no inverter command for correct exact identification.
There is no public database of model/sub-model.
help accepted https://github.com/jblance/mpp-solar/wiki/Protocol-Recommendations#protocol-map

RoySalisbury, 2023-02-27:
> There is no inverter command for correct exact identification. There is no public database of model/sub-model. help accepted [...]

That is why I posted the output here.. so the wiki could be updated for MKS2-6500.

jblance, 2023-03-15:
> so the wiki could be updated for MKS2-6500.

wiki updated

closed 2023-04-03 by jblance