jblance/mpp-solar #307: option mqtttopic does not seem to work

alithechemist, 2023-01-30:
Hi, 
I have several devices that return the same fields, so i need to use a different mqtt topic so i am able to distinguish them.

My command looks like this:

`/usr/local/bin/mpp-solar --porttype DALY -o influx2_mqtt,screen -q 192.168.1.2 --mqtttopic BSM1_cellVoltages -p /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -b 9600 -P DALY40 -c cellVoltages;`

But if i look into mosquitto_sub -v -t "#" i see

```
mpp-solar mpp-solar,command=cellVoltages cell_01_voltage=3.642
mpp-solar mpp-solar,command=cellVoltages cell_02_voltage=3.694
...
etc
```

Is there something I'm missing from the documentation?
IIRC  I had this in the past but i can't recall how it was solved.
Thanks in advance.

jblance, 2023-01-31:
That is the correct way, looks like I have introduced a bug
I'll get a change in when I get back to my dev environment

jblance, 2023-02-01:
fix should be in mppsolar-0.15.22

alithechemist, 2023-02-01:
Indeed, thanks!

closed 2023-02-01 by alithechemist