jblance/mpp-solar #308: Can you add SNMP output?

man55, 2023-02-03:
Great program. Only it works easily for my Chinese inverter.

It would be really cool to have output support in SNMP for easy integration with NUT (networkupstool). 
At least read-only for the most basic variables.

my output:
```
Command: QPIGS - General Status Parameters inquiry
--------------------------------------------------------------------------------
Parameter                                   Value          	Unit
ac_input_voltage                            239.0          	V   	{'icon': 'lightning'}
ac_input_frequency                          50.0           	Hz  
ac_output_voltage                           238.0          	V   
ac_output_frequency                         50.0           	Hz  
ac_output_apparent_power                    134            	VA  
ac_output_active_power                      138            	W   
ac_output_load                              5              	%   
bus_voltage                                 430            	V   
battery_voltage                             27.29          	V   
battery_charging_current                    0              	A   
battery_capacity                            100            	%   
inverter_heat_sink_temperature              423            	°C  
pv_input_current_for_battery                0.0            	A   
pv_input_voltage                            0.0            	V   
battery_voltage_from_scc                    0.0            	V   
battery_discharge_current                   0              	A   
is_sbu_priority_version_added               1              	bool
is_configuration_changed                    0              	bool
is_scc_firmware_updated                     0              	bool
is_load_on                                  1              	bool
is_battery_voltage_to_steady_while_charging 1              	bool
is_charging_on                              1              	bool
is_scc_charging_on                          0              	bool
is_ac_charging_on                           1              	bool
rsv1                                        64             	A   
rsv2                                        8              	A   
pv_input_power                              0              	W   
is_charging_to_float                        1              	bool
is_switched_on                              0              	bool
is_reserved                                 0              	bool
```

jblance, 2023-02-07:
To work the way NUT intends you would need to add to NUT so that it could talk direct to the inverter
SNMP is mostly a 'query / response' protocol, while mpp-solar is a 'run command / return result' - so it doesnt really fit with SNMP. You could try to send unsolicited SNMP messages as an output, but would need to figure out what MIB / IDs to use
Not sure this is a good or easy fit

Saentist, 2023-02-07:
Technically [Voltronic](https://voltronicpower.com) is company with also make UPS's long time ago.
Even Watchpower have SNMP options.

jblance, 2023-02-07:
how does watchpower snmp function?
Mind you Watchpower maintains state, this code doesnt

man55, 2023-02-07:
I already did it. I just set up the snmpd server, looked at the OIDs from the APC and wrote down the necessary ones in the snmpd config, creating the appropriate scripts that take data from the inverter using this script. It turned out not to be difficult.

Saentist, 2023-02-07:
> how does watchpower snmp function? Mind you Watchpower maintains state, this code doesnt

SNMP is simple protocol with represent content in a tree format
aka
group of values are represented as tree on with each brand represent specific value
values tree are represented in device MIB format with SNMPWALK 

with other words 
CSV result is represented in snmp mib oid's 

https://gist.github.com/nickma82/f2efdb9cd37f1d3aab28
https://oidref.com/1.3.6.1.4.1.318
https://voltronicpower.com/content/download/Brochure/SNMP_DS.pdf
http://www.sklep.orvaldi.pl/data/include/cms/Manual_SNMP_for_ORVALDI_Solar_Infini.pdf
SNMP card are supported by Axpert KS/ MKS/ EX/ MEX 1K~5K /Plus Duo 1-3K & Alfa 3K

example 
https://voltaconsolar.com/snmp-card.html
on bottom documents in PDF

In SNMP also can GET as all Query commands
and SET as Program commands, possible in MPP-Solar

man55, 2023-02-08:
I have Tasmota on HA connected via MQTT, but sorry, I haven't figured out how to use mqtt with mpp-solar to get regular data on HA. To do this, I need step-by-step precise instructions. I'm not a linux expert.

man55, 2023-02-08:
Okay, after a lot of thinking, I got it. But it was somehow not quite obvious))
Used this command:
`mpp-solar -p /dev/ttyUSB0  -q 192.168.1.XXX -o hassd_mqtt --mqtttopic inverter --mqttuser XXXXXXX --mqttpass XXXXXXXXX -c QPIG`
Now I probably need to make a service for it or put it in cron.

man55, 2023-02-08:
It didn't work out very well.
First, I lost the most important parameter: ac_input_voltage.
I don't understand where he disappeared to.
[image]

It is present in the mpp-solar output, but lost through mqqt ((
```
Command: QPIGS - General Status Parameters inquiry
--------------------------------------------------------------------------------
Parameter                                   Value               Unit
ac_input_voltage                            238.0               V       {'icon': 'lightning'}
ac_input_frequency                          50.0                Hz
ac_output_voltage                           238.0               V
ac_output_frequency                         50.0                Hz
ac_output_apparent_power                    189                 VA
ac_output_active_power                      184                 W
ac_output_load                              7                   %
bus_voltage                                 435                 V
battery_voltage                             27.29               V
battery_charging_current                    0                   A
battery_capacity                            100                 %
inverter_heat_sink_temperature              515                 °C
pv_input_current_for_battery                0.0                 A
pv_input_voltage                            0.0                 V
battery_voltage_from_scc                    0.0                 V
battery_discharge_current                   0                   A
is_sbu_priority_version_added               1                   bool
is_configuration_changed                    0                   bool
is_scc_firmware_updated                     0                   bool
is_load_on                                  1                   bool
is_battery_voltage_to_steady_while_charging 1                   bool
is_charging_on                              1                   bool
is_scc_charging_on                          0                   bool
is_ac_charging_on                           1                   bool
rsv1                                        64                  A
rsv2                                        8                   A
pv_input_power                              0                   W
is_charging_to_float                        1                   bool
is_switched_on                              0                   bool
is_reserved                                 0                   bool

```
Without this parameter, everything becomes useless. The inverter does not know how to transmit the status "on line or on battery" and the input voltage is an indication of the status. And in general it is important.

man55, 2023-02-08:
Need to use `hass_mqtt` instead of `hassd_mqtt` and then ac_input_voltage was found )

jblance, 2023-02-08:
Its strange that it didnt work with hassd_mqtt - this output module works well in my situation
If you have a chance to diagnose can you post the full command you are using now (looks like you are using `--getstatus` as part of the command)
Also can you run the same command but with `-q screen` and post the output

Saentist, 2023-02-08:
@man55 
I use this command
`screen mpp-solar -p /dev/hidraw2 -o hassd_mqtt,screen -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG -q 192.168.1.12 --mqttuser username--mqttpass password -n INVERTER-NAME --daemon`

`--mqtttopic` is useless in HA

go to MQTT device and post it to dashboard.

man55, 2023-02-08:
As you can see above, there is one extra field in the line ac_input_voltage: {'icon': 'lightning'}. 
Perhaps this is the reason. But the fact is that `hass_mqtt` works great for me. There is no need to change anything. TNX

jblance, 2023-02-08:
ok

closed 2023-02-08 by jblance