jblance/mpp-solar #313: mqtttopic seems to not be working again

lombaardcj, 2023-02-26:
Related to https://github.com/jblance/mpp-solar/issues/183

Hi, I'm using v 0.15.24 and the --mqtttopic seems to not be used when publishing output of command to MQTT.

The command I'm using specifically is: `mpp-solar -p /dev/ttyUSB0 --protocol PI30 --baud 2400 -o mqtt --mqttbroker 192.168.1.105 --mqttuser <username> --mqttpass <password> -c QPIGS --mqtttopic /home/inverter`

The only result I'm seeing is the MQTT gets published but at `QPIGS/status/...` for the respective output fields.

Am I doign something wrong, or using an outdated version?

Saentist, 2023-02-26:
To what software you send data?

lombaardcj, 2023-02-26:
I send it to home assistant mosquitto broker on a seperate device to where i'm running the mpp-solar command.  The problem is that the root of the topic is always the command, and not prepended with the mqtttopic provided.

I have narrowed it down to a change made to json_mqtt.py
https://github.com/jblance/mpp-solar/blob/master/mppsolar/outputs/json_mqtt.py#L26

It seems the condition is not met that the broker is not defined, and only then is the topic added, otherwise it is not added.

My only other thought is the `mqtt_topic` variable is not properly defined at this point.

Saentist, 2023-02-26:
Why not use `-o hassd_mqtt` with send organised data to HA, instead `-o mqtt` with send bunch of values?

`mpp-solar -p /dev/hidraw1 -o hassd_mqtt -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG -q HA.IP.OR.DNS --mqttuser USER --mqttpass 'PASS' -n VMIII-5000 --daemon
`

lombaardcj, 2023-02-26:
Thank you.  I've tried the new output processor, and it's working well.  Is there a reason why the mqtttopic is not adding a root topic if the mqtt output processor is chosen?

A side question, how ofter is the information polled if the --daemon option is chosen.  I'm not seeing documentation around it.

Saentist, 2023-02-26:
`--daemon` set to 60 seconds pull, with is enough.

`mqtttopic` ... is it supported by HA?

lombaardcj, 2023-02-27:
Thank you for your explanation around the --daemon interval.  That is actually a veruy convenient way to run it as a service.  So the output processor for mqtt is by design not taking the mqtttopic into consideration.  I would have hoped for a slight adjustment to allow the mqtttopic to be left empty by default, and if a user choses to use the mqtttopic, it will provide the ability to specify the root topic.

jblance, 2023-02-27:
Yeah mqtttopic isn't implemented into the mqtt output mostly as this output
uses the tag as the base of the topic and if I changed it to support
mqtttopic it would break for those that use the tag...

You could supply a tag to get it to work as you are thinking.

A lot of problems arise from the way the code has evolved over time to
support many different ways of working.

I'm slowly reworking into a new command where the configuration won't be
overloaded and I can fix a number of issues, but this takes time (and you
know you get what you pay for)

lombaardcj, 2023-02-27:
Sorry yes, I realise after looking carefully at the documentation, the {tag} is clearly documented for the mqtt option.  I noticed it is also true for json_mqtt.  This solution will work perfectly.  As an aside consider a way for supporters to contribute towards your efforts by showing their appreciation.  The product in it's current form is working very well otherwise.

jblance, 2023-03-26:
buyacoffee added :-)

closed 2023-03-26 by jblance