jblance/mpp-solar #316: Need help with Docker setup

frederickjh, 2023-02-28:
Hi!
I followed the instruction in the [docker.md](https://github.com/jblance/mpp-solar/blob/master/docs/docker.md) file but have not been able to get it working correctly. The container builds and starts but the log has errors.

<details><summary>CLICK ME to see collapsed docker container log</summary>
<p>

```bash
Service Initializing ...

2023-02-28 08:23:03,128:CRITICAL:helpers:get_device_class@78: Error loading device mppsolar                    # required - type of device (default: mppsolar): No module named 'mppsolar.devices.mppsolar                    # required - type of device (default: mppsolar)'

Traceback (most recent call last):

  File "/usr/local/bin/mpp-solar", line 8, in <module>

    sys.exit(main())

  File "/usr/local/lib/python3.10/site-packages/mppsolar/__init__.py", line 312, in main

    device = device_class(

TypeError: 'NoneType' object is not callable

2023-02-28 08:23:54,953:CRITICAL:helpers:get_device_class@78: Error loading device mppsolar                    # required - type of device (default: mppsolar): No module named 'mppsolar.devices.mppsolar                    # required - type of device (default: mppsolar)'

Traceback (most recent call last):

  File "/usr/local/bin/mpp-solar", line 8, in <module>

    sys.exit(main())

  File "/usr/local/lib/python3.10/site-packages/mppsolar/__init__.py", line 312, in main

Service Initializing ...

    device = device_class(

TypeError: 'NoneType' object is not callable
```

</p>
</details>

Errors during the successful `docker build` command (which I considered not really issues):

`debconf: delaying package configuration, since apt-utils is not installe`

`WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv`

`WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual enviro`

`WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv`

I had to change the second line of the Dockerfile to:
`RUN apt-get install -y pkg-config libsystemd-dev gcc linux-libc-dev libc6-dev`
adding the last two packages to get pass some errors regarding missing headers.

I added a line between lines two and three to update `pip` as I was getting errors about it being out of date and aborting the build.
`RUN /usr/local/bin/python -m pip install --upgrade pip`

Any help you can give in getting this up and running would be appreciated!
Frederick

frederickjh, 2023-03-02:
Versions:
`pip`  23.0.1
`mpp-solar` Solar Device Command Utility, version: 0.15.24
`docker` Docker version 20.10.22, build 3a2c30b

robwolff3, 2023-03-09:
I wrote that docker file. It builds just fine for me right now, building for my amd64 architecture I'm running docker on. I would worry about warnings pip throws about being out of data. That doesn't break anything. Not a lot of context to go on here otherwise. My money is on the container running errors are because the the serial/usb device is not being passed through to the container properly or a misconfiguration in the config file.

frederickjh, 2023-03-12:
@robwolff3 Thanks for getting back to me. Turns out that the issue is not with the docker container but with the configuration file. Not sure what is going on there exactly but it seems that it does not ignore comments properly. I started out with the commented configuration file and modified it for my needs. After removing both commented lines and end commented lines, it is now working. I will file issues for this.

@robwolff3 since you were the one that wrote the docker file maybe you know if it will support `udev`?  This is where a USB device can be assigned to a specific device path and name when it is attached. I have not gotten this working. If you have any information that would help in this regard it would be appreciated.

frederickjh, 2023-03-12:
#320

robwolff3, 2023-03-21:
@frederickjh I am also using udev to assign usb devices to explicit mount points on my host machine. And i'm having no issue using them with my container. I mount them under devices just like in the documentation: https://github.com/jblance/mpp-solar/blob/master/docs/docker.md

I named my rules file 99-usb-serial.rules and used the following rule:
`SUBSYSTEM=="tty", ATTRS{idVendor}=="067b", ATTRS{idProduct}=="2303", SYMLINK+="mpp"`

frederickjh, 2023-03-23:
@robwolff3 are you connected directly to the micro USB port on the inverter or to the serial port via a serial to USB adapter? 

Just wondering about the `SUBSYSTEM=="tty"` part of your rule. I do not have a SUBSYSTEM section. I am connected directly to the micro USB port on the inverter.

robwolff3, 2023-03-23:
Yes, I am using the serial port with a serial to usb adapter. There are some commands that this project can run over the usb interface due to bugs I think.

frederickjh, 2023-03-24:
> Yes, I am using the serial port with a serial to usb adapter. There are some commands that this project can't run over the usb interface due to a bug I think.

Any idea where I could find information about that @robwolff3 ? I would hate to get this setup and working with the USB when I could do it with a USB serial adapter. Maybe @jblance knows where to find that information.

robwolff3, 2023-03-24:
https://github.com/jblance/mpp-solar/issues/280
https://github.com/jblance/mpp-solar/issues/268