jblance/mpp-solar #319: PI17 Protocal Modifications

robwolff3, 2023-03-09:
I'm using the PI17 protocol with my MPP LVX6048WP. It working quite well! but I noticed unknown_value_in_response for many of the values in the HECS command. I changed the HECS potion of the protocol to the following and its working now for me:

```
    "HECS": {
        "name": "HECS",
        "prefix": "^P005",
        "description": "Query energy control status",
        "help": " -- queries the device energy distribution",
        "type": "QUERY",
        "response": [
            [
                "keyed",
                "Solar Energy Distribution Priority",
                {
                    "00": "Battery-Load-Grid",
                    "01": "Load-Battery-Grid",
                    "02": "Load-Grid-Battery",
                },
            ],
            ["option", "Solar charge battery", ["disabled", "enabled"]],
            ["option", "AC charge battery", ["disabled", "enabled"]],
            ["option", "Feed power to utility", ["disabled", "enabled"]],
            ["option", "Battery discharge to loads when solar input normal", ["disabled", "enabled"]],
            ["option", "Battery discharge to loads when solar input loss", ["disabled", "enabled"]],
            ["option", "Battery discharge to feed grid when solar input normal", ["disabled", "enabled"]],
            ["option", "Battery discharge to feed grid when solar input loss", ["disabled", "enabled"]],
            ["option", "Reserved", ["disabled", "enabled"]],
        ],
        "test_responses": [
            b"^D019\xd9\x9f,0,0,0,0,0,0,0,0\r",
        ],
    },
```

I didn't want to assume this was a needed modification beyond my inverter model so I'm posting it here. I am more then happy to submit this as a pull request instead if that's appropriate.

Saentist, 2023-03-10:
ProtocOl or ProtocAl ;)

jblance, 2023-03-15:
hi, i think this is a safe change for all
can you do a PR for it please.

thanks

JakobTewes, 2023-06-16:
Heyhey @jblance,

I think, it might be a good idea, having a list of supported features/commands for the PI17 devices. 
If I look from my perspective, i found out, that GS and PS work but am uncertain if control also works and got no real overview.

What do you think?

jblance, 2023-06-16:
If you run mppsolar command with no command after `-c` but still supplying
the protocol you get a list of implemented commands eg  `mpp-solar -p test
-P PI17 -c`

closed 2023-07-22 by jblance