jblance/mpp-solar #320: Comments in configuration file do not seem to be ignored properly

frederickjh, 2023-03-12:
Not sure what is going on there exactly but it seems that mpp-solar does not ignore comments in the configuration file properly. 

I started out with the commented configuration file and modified it for my needs, but ran into a bunch of errors.. After removing both commented lines and end commented lines, it is now working.

Thanks for looking into this.

jblance, 2023-03-12:
End commented lines dont work
Did the config file work on a non-docker install?
Can you post the failing config?

The config file is processed by a library (ConfigParser) so I dont have a lot of scope for changing the functionality

frederickjh, 2023-03-13:
@jblance  I have no problem if the end commented lines do not work, but then they should be removed from [configfile.md](https://github.com/jblance/mpp-solar/blob/master/docs/configfile.md) and it should be also noted there that end commented lines do not work.

I lost quite a few hours trying to troubleshoot this. I finally realized that it works fine on my laptop when run from the command line but in the docker container with the configuration file not. So, started to try and eliminate some possibilities. I tried the configuration file on my laptop and it also failed there too. I double checked it but could not see anything that differed from the example configuration file that looked like it might be the issue. Finally something in the error messages said to me that I should try removing comments to see if that was the issues and then it worked both on the laptop and in the docker container.

Below is the text of the failing configuration file before I made a copy and removed all comments.

<details><summary>CLICK ME to see collapsed text of the failing configuration fill with comments</summary>
<p>

```bash
[SETUP]
# Number of seconds to pause between loops of processing the sections
# i.e. the pause at the end of an entire run through the config file
# default is 60
pause=5
# ipaddress or hostname of the mqtt broker, default is 'localhost'
mqtt_broker=homeassistant
# mqtt broker port number, default is 1883
mqtt_port=1883
# username and password (if required) for mqtt broker, default to None and not used unless defined
mqtt_user=mqtt_user
mqtt_pass=hthkMm3iC4xWwn-JyS7K

# The section name needs to be unique
# There can be multiple sections which are processed sequentially without pause
# The pause occurs after all sections are processed, before the next loop
# The name is used for:
#   client_id in MQTTIO (using in the command and response topics)
#
[PIP8048MAX]
#
#protocol=PI30MAX                 # required - protocol to use to decode command and response (default: PI30)
protocol=pi30
type=mppsolar                    # required - type of device (default: mppsolar)
#port=/dev/usb/hiddev0            # required - port used to communicate with device (default: /dev/ttyUSB0)
port=/dev/mppsolarusb/direct/0
#baud=2400                        # optional - baud rate of port communications (default: 2400)
command=QPIGS#QLT#QET           # required - hash separated list of commands to execute
tag=PIP8048MAX                          # optional - used in various ways in the outputs (see output list)
name=PIP8048MAX
outputs=screen               # required - comma separated list of outputs (default: screen)
#outputs=hassd_mqtt
#porttype=serial                  # optional - used to override the automatic port type determination
#filter=^voltage                  # optional - if defined only field names that match the filter will be output (uses python re format)
#excl_filter=test                 # optional - if defined any field names that match the filter will be excluded from the output (uses python re format)
#udpport=5555
```

</p>
</details>

frederickjh, 2023-03-13:
I checked out `ConfigParser`  docs and they say:

> \# By default only in an empty line.
> \# Inline comments can be harmful because they prevent users
> \# from using the delimiting characters as parts of values.
> \# That being said, this can be customized.

However since the `mpp-solar` commands are separated with the **#** symbol it does not make much sense to enable them.

jblance, 2023-03-13:
> @jblance <https://github.com/jblance> Just a quick question, do the
> commands need to be separated with the *#* symbol or could the be
> separated with commas?

I switched to # as some commands use the comma

PR #321 (2023-03-15): Added information about not using end of line comments and moved all …
…end of line comments above the lines they refer to and added empty lines to make it easier to read. Fixes #320
commit ce6ea78d (2023-03-15): Added information about not using end of line comments and moved all … (#321)

* Added information about not using end of line comments and moved all end of line comments above the lines they refer to and added empty lines to make it easier to read. Fixes #320

* Removed information that the comment needs to start at the beginning of a line as they can be indented.
closed 2023-03-15 by jblance