jblance/mpp-solar #326: Problem setting up mpp-solar

geofreybingachie, 2023-03-30:
Hello, I am not very good with Linux.Please help me make use of /dev/hidraw0 instead of the default /dev/ttyUSB0
Below is the error I am getting and I don't know what to edit.

2023-03-30 14:18:39,193:WARNING:serialio:send_and_receive@33: Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
Command: QID - Device Serial Number inquiry
--------------------------------------------------------------------------------
Parameter      Value            Unit
error          Serial command execution failed
validity_check Error: incorrect response format

geofreybingachie, 2023-03-30:
> 

I managed to set hidraw0 as the default device by editing .py file

Saentist, 2023-03-30:
No need to change anything in code
```shell
mpp-solar -p /dev/hidraw1 -o screen -c QID
```

geofreybingachie, 2023-03-30:
> 

Kindly help me setup the remain part .Please

Saentist, 2023-03-30:
https://github.com/jblance/mpp-solar/blob/master/docs/MQTT_Influx_Grafana.md

geofreybingachie, 2023-03-30:
> https://github.com/jblance/mpp-solar/blob/master/docs/MQTT_Influx_Grafana.md

I gone through all this but i can seem to find where i and going wrong .I can't get measurements into the DB for forward processing to grafana . Also my mpp-solar.service won't run.I started installing this setup yesterday morning and have spent countless hours trying to make it work but with no success. I need help please.

Saentist, 2023-03-30:
I'm use Home Assistant 

what say 
`service mpp-solar status`
and what is config file content

jblance, 2023-03-30:
You need to supply more information 
- what are you trying to do
- what have you done
- what results did you get
Post the commands you use and the output
Also check through the troubleshooting document

geofreybingachie, 2023-04-01:
> You need to supply more information
> 
> * what are you trying to do
> * what have you done
> * what results did you get
>   Post the commands you use and the output
>   Also check through the troubleshooting document

I have formatted my pc and installed ubuntu 22.04 afresh to try and see if i will manage to suceed this time round.

geofreybingachie, 2023-04-03:
I managed to setup and all is well now .Thank you @jblance and @Saentist .
I would like to have more data on my grafana dashboard specifically device mode and fan speed but that data is not available using QPIGS query .However its available using other queries i.e Q1 and QMOD and --getstatus .
Is it possible to add the queries on my config file ?

geofreybingachie, 2023-04-10:
> I managed to setup and all is well now .Thank you @jblance and @Saentist . I would like to have more data on my grafana dashboard specifically device mode and fan speed but that data is not [...]

I managed to query QPIGS#QMOD at the same time though I have noted some delay in communication and it seems the inverter is the bottle neck.

[image]

jblance, 2023-04-12:
Nice graph
I'll close this, open a new issue if you have other problems

closed 2023-04-12 by jblance