jblance/mpp-solar #328: root@raspberrypi:/usr/src/mpp-solar# mpp-solar -c QPIGS -P PI16 -p /dev/hidraw0

kitos23, 2023-03-31:
root@raspberrypi:/usr/src/mpp-solar# mpp-solar -c QPIGS -P PI16 -p /dev/hidraw0
Parameter                       Value           Unit
grid_voltage                    221.8           V
output_power                    0               W
grid_frequency                  50.0            Hz
output_current                  1.8             A
ac_output_voltage_r             233.4           V
ac_output_power_r               349             W
ac_output_frequency             50.1            Hz
ac_output_current_r             1.4             A
output_load_percent             7               %
pbus_voltage                    409.1           V
sbus_voltage                    409.1           V
positive_battery_voltage        56.7            V
negative_battery_voltage        0.0             V
battery_capacity                93              %
pv1_input_power                 0               W
pv2_input_power                 0               W
pv3_input_power                 0               W
pv1_input_voltage               0.0             V
pv2_input_voltage               0.0             V
pv3_input_voltage               0.0             V
max_temperature                 33.0            °C
status_todo                     D---110001

Perfect my friend!

_Originally posted by @eliafino in https://github.com/jblance/mpp-solar/issues/33#issuecomment-661726447_
            Hello, I have an E 5.5 infinisolar inverter, and I would like to extract all that data with the raspberry. I barely know anything about python code. Could you tell me where I can find it? Yes, I have seen your website with all the data in graphics, it is amazing. But that is too far for me. Thank you so much

Saentist, 2023-03-31:
`mpp-solar -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG -P PI16 -p /dev/hidraw0`

kitos23, 2023-03-31:
Thanks for answering, but I don't know what you've put what it is. My level with raspberry is very very low. I understand that the inverter protocol is PI16, but what I'm asking is where to get the x.py file that can read the inverter data

kitos23, 2023-03-31:
I will try it and comment the result. Thank you very much for this contribution.

kitos23, 2023-04-01:
Last night I was tinkering around a bit, with no results so far. I think the first problem is reading Bus 001 Device 004: ID 0665:5161 Cypress Semiconductor USB to Serial, I imagine because mpp-solar is configured for the PI30 protocol and I think the infinisolar is PI16. As test mpp-solar -p /dev/ttyUSB0 –getDeviceId and the response is WARNING:serialio:send_and_receive@33: Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'.
I also saw in more installation options venv Install – recommended if testing new features/release, and I didn't think this would help me right now, maybe later.
I also tried the command line that you gave me mpp-solar -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG -P PI16 -p /dev/hidraw0 and it wrote me this Command: QID - No description found
-------------------------------------------------- ------------------------------
Parameter Value Unit
warning No definition for command QID in protocol b'PI16'
response ERROR
Command: QDI - No description found
-------------------------------------------------- ------------------------------
Parameter Value Unit
warning No definition for command QDI in protocol b'PI16'
response ERROR
Traceback (most recent call last):
   File "/usr/local/bin/mpp-solar", line 33, in <module>
     sys.exit(load_entry_point('mppsolar==0.15.27', 'console_scripts', 'mpp-solar')())
   File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 419, in main
     results = _device.run_command(command=_command)
   File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 112, in run_command
     decoded_response = self._protocol.decode(raw_response, command)
   File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 264, in decode
     responses = self.get_responses(response)
   File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 81, in get_responses
     return response[1:-3].split(b" ")
TypeError: unhashable type: 'slice' . It would have been magical if it had worked the first time lol.
So I don't know if I have to go somewhere, to modify mppsolar so that it can read the data from the inverter as a first step and then continue advancing. Let's see if you can guide me, thanks.

jblance, 2023-04-01:
> `mpp-solar -c QPIGS -P PI16 -p /dev/hidraw0`

If this worked for you

You will need to use `-p /dev/hidraw0 ` in any of the commands you use as this is the port that your inverter is on. 
You don't need to change any python code or access x.py, you just need to use the correct options to the command
You will need to use `-P PI16` in all commands to tell mppsolar to use this protocol and not the default
Protocol pi16 has different set of commands, you can list them with `mpp-solar -c  -P PI16 -p /dev/hidraw0`

jblance, 2023-04-01:
Also `mpp-solar-h` will provide a description of the various options

kitos23, 2023-04-01:
Ok, I would need to see some practical examples, but I'm already seeing how it works. Forgive my ignorance. This command line (mpp-solar -c -P PI16 -p /dev/hidraw0) I will always have to use. Now, I've gone to the end of the help and for Get Inverter Status I should add to that line -C –getstatus ???. I don't know if it is done that way, or I have misinterpreted it. Without examples it's hard for me, sorry. And thanks jblance for the help

jblance, 2023-04-01:
It would be
`mpp-solar -P PI16 -p /dev/hidraw0 --getstatus`
Or
`mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS`

kitos23, 2023-04-01:
no this doesn't work  mpp-solar -P PI16 -p /dev/hidraw0 --getstatus
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mppsolar==0.15.27', 'console_scripts', 'mpp-solar')())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 419, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 67, in run_command
    return self.get_status()
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 121, in get_status
    data.update(self.run_command(command))
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 112, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 264, in decode
    responses = self.get_responses(response)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 81, in get_responses
    return response[1:-3].split(b" ")
TypeError: unhashable type: 'slice'
marcos@raspberrypi:~ $ mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mppsolar==0.15.27', 'console_scripts', 'mpp-solar')())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 419, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 112, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 264, in decode
    responses = self.get_responses(response)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 81, in get_responses
    return response[1:-3].split(b" ")
TypeError: unhashable type: 'slice'

jblance, 2023-04-01:
Add `-D` to one of the commands and post the output

kitos23, 2023-04-01:
mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS -D
2023-04-01 12:11:03,044:INFO:__init__:main@213: Solar Device Command Utility, version: 0.15.24
2023-04-01 12:11:03,045:DEBUG:mqttbrokerc:__init__@32: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-04-01 12:11:03,046:DEBUG:__init__:main@238: MqttBroker name: localhost, port: 1883, user: None
2023-04-01 12:11:03,047:DEBUG:__init__:main@240: udp port 5555
2023-04-01 12:11:03,047:DEBUG:__init__:main@242: Using Postgres None
2023-04-01 12:11:03,048:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2023-04-01 12:11:03,048:INFO:__init__:main@346: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI16"
2023-04-01 12:11:03,058:DEBUG:__init__:main@350: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-04-01 12:11:03,059:DEBUG:device:__init__@33: __init__ args ()
2023-04-01 12:11:03,060:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI16', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x75febb50>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-04-01 12:11:03,060:DEBUG:__init__:get_port_type@40: port matches hidraw
2023-04-01 12:11:03,061:INFO:__init__:get_port@97: Using hidrawio for communications
2023-04-01 12:11:03,066:DEBUG:__init__:get_protocol@13: Protocol PI16
2023-04-01 12:11:03,075:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x75feba60>, protocol PI16 protocol handler
2023-04-01 12:11:03,076:DEBUG:__init__:main@401: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75febdc0>, 'QPIGS', 'QPIGS', 'screen', None, None)]
2023-04-01 12:11:03,076:INFO:__init__:main@406: Looping 1 commands
2023-04-01 12:11:03,077:INFO:__init__:main@416: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x75feba60>, protocol: PI16 protocol handler for command: QPIGS, tag: QPIGS, outputs: screen
2023-04-01 12:11:03,077:INFO:device:run_command@50: Running command QPIGS
2023-04-01 12:11:03,078:INFO:pi16:get_full_command@183: Using protocol b'PI16' with 6 commands
2023-04-01 12:11:03,078:DEBUG:abstractprotocol:get_command_defn@58: Processing command 'QPIGS'
2023-04-01 12:11:03,079:DEBUG:abstractprotocol:get_command_defn@60: Found command QPIGS in protocol b'PI16'
2023-04-01 12:11:03,079:DEBUG:pi16:get_full_command@201: full command: b'QPIGS\r'
2023-04-01 12:11:03,080:INFO:device:run_command@78: full command b'QPIGS\r' for command QPIGS
2023-04-01 12:11:03,080:DEBUG:abstractprotocol:get_command_defn@58: Processing command 'QPIGS'
2023-04-01 12:11:03,081:DEBUG:abstractprotocol:get_command_defn@60: Found command QPIGS in protocol b'PI16'
2023-04-01 12:11:03,081:DEBUG:hidrawio:send_and_receive@24: USB open error: [Errno 13] Permission denied: '/dev/hidraw0'
2023-04-01 12:11:03,082:DEBUG:device:run_command@95: Send and Receive Response {'ERROR': ["USB open error: [Errno 13] Permission denied: '/dev/hidraw0'", '']}
2023-04-01 12:11:03,083:INFO:abstractprotocol:decode@217: response passed to decode: {'ERROR': ["USB open error: [Errno 13] Permission denied: '/dev/hidraw0'", '']}
2023-04-01 12:11:03,083:DEBUG:abstractprotocol:get_command_defn@58: Processing command 'QPIGS'
2023-04-01 12:11:03,083:DEBUG:abstractprotocol:get_command_defn@60: Found command QPIGS in protocol b'PI16'
2023-04-01 12:11:03,084:INFO:abstractprotocol:decode@261: Processing response of type SEQUENTIAL
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mppsolar==0.15.27', 'console_scripts', 'mpp-solar')())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 419, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 112, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 264, in decode
    responses = self.get_responses(response)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 81, in get_responses
    return response[1:-3].split(b" ")
TypeError: unhashable type: 'slice'

Saentist, 2023-04-01:
@kitos23 
can you make file
`check-hidraw.sh`
with contain this
```bash
#!/bin/bash

FILES=/dev/hidraw*
for f in $FILES
do
  FILE=${f##*/}
  DEVICE="$(cat /sys/class/hidraw/${FILE}/device/uevent | grep HID_NAME | cut -d '=' -f2)"
  printf "%s \t %s\n" $FILE "$DEVICE"
done
```
then run command
```shell
bash ./check-hidraw.sh
```
and post result
My guess is that inverter is not only usb device in system.

kitos23, 2023-04-01:
only the inverter is connected by USB to the raspberry 
bash ./check-hidraw.sh
hidraw0 	 HID 0665:5161

jblance, 2023-04-01:
> 2023-04-01 12:11:03,083:INFO:abstractprotocol:decode@217: response passed to decode: {'ERROR': ["USB open error: [Errno 13] Permission denied: '/dev/hidraw0'", '']}

This is the problem. Need to either use `sudo` or allow the current user access to the hidraw device (there is a doc somewhere that details how)

kitos23, 2023-04-01:
these hidraw0 HID 0665:5161 are the id of the vendor and the product. Could it be that the inverter is blocked from reading without those 2 keys?

Saentist, 2023-04-01:
No `0665:5161` is a VID:PID (Vendor IDentification : Product IDentification)

or simpler
```shell 
$bash ./check-hidraw.sh
hidraw0          American Power Conversion  Smart-UPS C 1500 FW:UPS 10.0 / ID
hidraw1          STMicroelectronics HID in FS Mode   <<<<<<<<<<<<
hidraw2          HID TOUCH HID Touch Panel
$ lsusb
Bus 001 Device 007: ID 13d3:5129 IMC Networks USB2.0 UVC VGA WebCam
Bus 001 Device 005: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 001 Device 008: ID 04ca:f001 Lite-On Technology Corp. TvTUNER
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 1bfd:1688 TouchPack Resistive Touch Screen
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 004: ID 0665:5161 Cypress Semiconductor USB to Serial <<<<<<<<<<<
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 051d:0003 American Power Conversion UPS
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

jblance, 2023-04-01:
Try
`sudo mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS`

Saentist, 2023-04-01:
or to be more sure check is any other process not using it
`lsof /dev/hidraw0`

kitos23, 2023-04-01:
> Try `sudo mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS`

sudo mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS
Command: QPIGS - General status query
--------------------------------------------------------------------------------
Parameter    Value          	Unit
grid_voltage 0              	V

kitos23, 2023-04-01:
> or to be more sure check is any other process not using it
> 
> ```shell
> lsof /dev/hidraw0
> ```

nothing

Saentist, 2023-04-01:
Ask distributor of inverter to give you a "Communication Protocol" PDF
It contain all command with inverter support
something as this
https://watts247.com/manuals/mpp/PIP-HS_MS_MSX_HYBRIDV%20RS232%20Protocol-C_20170822_.pdf
This is a most relevant way to communicate with inverter

until then
post output of 
```
sudo mpp-solar -p /dev/hidraw0 -P PI16-o raw -c QID#QPI#QMN#QVFW#QVFW2#QVFW3
```
for update wiki data with your inverter

kitos23, 2023-04-01:
mpp-solar: error: unrecognized arguments: raw

Saentist, 2023-04-01:
> mpp-solar: error: unrecognized arguments: raw

```
sudo mpp-solar -p /dev/hidraw0 -P PI16 -o raw -c QID#QPI#QMN#QVFW#QVFW2#QVFW3`
#space before -
```

kitos23, 2023-04-01:
Command: QID - No description found
------------------------------------------------------------
raw_response                  	'(96131801100057\x93\xa5\r'
Command: QPI - Device Protocol Version inquiry
------------------------------------------------------------
raw_response                  	'(PI16\x9c\xaf\r'
Command: QMN - No description found
------------------------------------------------------------
raw_response                  	'(NAKss\r'     
Command: QVFW - No description found
------------------------------------------------------------
raw_response                  	'(VERFW:00000.27V\t\r'
Command: QVFW2 - No description found
------------------------------------------------------------
raw_response                  	'(VERFW2:00000.30\xe3\x88\r'
Command: QVFW3 - No description found
------------------------------------------------------------
raw_response                  	'(NAKss\r'

kitos23, 2023-04-01:
sorry for the copy paste. I don't know how to do it better

Saentist, 2023-04-01:
> sorry for the copy paste. I don't know how to do it better

select pasted text and on top there is `<>` button or press "Ctrl+E" it do rest

kitos23, 2023-04-01:
ok changed

Saentist, 2023-04-01:
> ok changed
You can do for all your posts will be more easy for all with read them.

p.s. forget to write QGMN /subversion of inverter/
```shell
sudo mpp-solar -p /dev/hidraw0 -P PI16 -o raw -c QID#QPI#QMN#QGMN#QVFW#QVFW2#QVFW3
```
@jblance need to update
https://github.com/jblance/mpp-solar/wiki/Protocol-Recommendations

kitos23, 2023-04-01:
sudo mpp-solar -p /dev/hidraw0 -P PI16 -o raw -c QID#QPI#QMN#QGMN#QVFW#QVFW2#QVFW3
```
Command: QID - No description found
------------------------------------------------------------
raw_response                  	'(96131801100057\x93\xa5\r'
Command: QPI - Device Protocol Version inquiry
------------------------------------------------------------
raw_response                  	'(PI16\x9c\xaf\r'
Command: QMN - No description found
------------------------------------------------------------
raw_response                  	'(NAKss\r'     
Command: QGMN - No description found
------------------------------------------------------------
raw_response                  	'(000D\xee\r'  
Command: QVFW - No description found
------------------------------------------------------------
raw_response                  	'(VERFW:00000.27V\t\r'
Command: QVFW2 - No description found
------------------------------------------------------------
raw_response                  	'(VERFW2:00000.30\xe3\x88\r'
Command: QVFW3 - No description found
------------------------------------------------------------
raw_response                  	'(NAKss\r'     
```

kitos23, 2023-04-01:
My investor is harder to tame than a wild horse jaja

kitos23, 2023-04-01:
I tried sudo mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS -D and it created a bunch of lines and at the end this:
```
Command: QPIGS - General status query
--------------------------------------------------------------------------------
Parameter                Value          	Unit
grid_voltage             225.1          	V   
output_power             100013         	W   
grid_frequency           50.0           	Hz  
output_current           2.5            	A   
ac_output_voltage_r      227.2          	V   
ac_output_power_r        556            	W   
ac_output_frequency      49.9           	Hz  
ac_output_current_r      2.4            	A   
output_load_percent      10             	%   
pbus_voltage             430.1          	V   
sbus_voltage             430.1          	V   
positive_battery_voltage 55.5           	V   
negative_battery_voltage 0              	V   
battery_capacity         73             	%   
pv1_input_power          1512           	W   
pv2_input_power          2183           	W   
pv3_input_power          0              	W   
pv1_input_voltage        378.0          	V   
pv2_input_voltage        263.1          	V   
pv3_input_voltage        0              	V   
max_temperature          37.0           	°C  
status_todo              B---101001     	    
2023-04-01 16:36:35,033:DEBUG:__init__:main@451: Not daemon, so not looping
```

kitos23, 2023-04-01:
Well, first of all, thank you both, John Blance and Saentist, for the help to get and understand a little how mpp solar is going.
The second thing is the next step, I've already seen that all this data can be sent to the web with mpp solar, but I don't think it's as “easy” as this. Is there a link where this topic is answered, so I can see how it can be done?
Thank you very much, I owe you an invitation.

Saentist, 2023-04-01:
Depend on where do you want to send it.
[image]

kitos23, 2023-04-01:
If you wanted me to drool, you have achieved it. Beautiful

jblance, 2023-04-02:
@Saentist nice grafana setup!!
As mentioned, it all depends on what your desired end result is
My dashboard isnt as pretty and uses Home Assistant (which allows more automations)
[image]

jblance, 2023-04-02:
> I tried sudo mpp-solar -P PI16 -p /dev/hidraw0 -c QPIGS -D and it created a bunch of lines and at the end this:
> 
> ```
> Command: QPIGS - General status query
> --------------------------------------------------------------------------------
> Parameter                Value          	Unit
> grid_voltage             225.1          	V   
> output_power             100013         	W   
> grid_frequency           50.0           	Hz  
> output_current           2.5            	A   
> ac_output_voltage_r      227.2          	V   
> ac_output_power_r        556            	W   
> ac_output_frequency      49.9           	Hz  
> ac_output_current_r      2.4            	A   
> output_load_percent      10             	%   
> pbus_voltage             430.1          	V   
> sbus_voltage             430.1          	V   
> positive_battery_voltage 55.5           	V   
> negative_battery_voltage 0              	V   
> battery_capacity         73             	%   
> pv1_input_power          1512           	W   
> pv2_input_power          2183           	W   
> pv3_input_power          0              	W   
> pv1_input_voltage        378.0          	V   
> pv2_input_voltage        263.1          	V   
> pv3_input_voltage        0              	V   
> max_temperature          37.0           	°C  
> status_todo              B---101001     	    
> 2023-04-01 16:36:35,033:DEBUG:__init__:main@451: Not daemon, so not looping
> ```
> 
> ps ok, I already removed -D and it only gives the table of values of the inverter

output_power looks wrong

jblance, 2023-04-02:
> You can do for all your posts will be more easy for all with read them.
>
> p.s. forget to write QGMN /subversion of inverter/
>
> ```shell
> sudo mpp-solar -p /dev/hidraw0 -P PI16 -o raw -c [...]

Ive updated PI16, now 
`mpp-solar -p test -P PI16 --getDeviceId`
```
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value                 Unit
deviceid  PI16:000:00000.27         
```

Saentist, 2023-04-02:
> Ive updated PI16, now `mpp-solar -p test -P PI16 --getDeviceId`
> 
> ```
> Command: Get Device ID - Generate a device id
> --------------------------------------------------------------------------------
> Parameter Value                 Unit
> deviceid  PI16:000:00000.27         
> ```
@jblance 
this cannot be real string
`00000.27` is a firmware version

set global description of commands if possible,
its strange is some protocols to existing and on other not

```
mpp-solar -p /dev/hidraw1 -o raw -c QID#QPI#QMN#QGMN#QVFW#QVFW2#QVFW3
Command: QID - Device Serial Number inquiry
------------------------------------------------------------
raw_response                    '(553555355535552\xfd\r'

Command: QPI - Protocol ID inquiry
------------------------------------------------------------
raw_response                    '(PI30\x9a\x0b\r'

Command: QMN - Model Name Inquiry
------------------------------------------------------------
raw_response                    "(VMIII-5000\xa8'\r"

Command: QGMN - General Model Name Inquiry
------------------------------------------------------------
raw_response                    '(037aZ\r'

Command: QVFW - Main CPU firmware version inquiry
------------------------------------------------------------
raw_response                    '(VERFW:00091.20\xf8&\r'

Command: QVFW2 - Secondary CPU firmware version inquiry
------------------------------------------------------------

Command: QVFW3 - No description found        <<<<<<<< PI30 Bluetooth Firmware
------------------------------------------------------------
raw_response                    '(VERFW:00002.61\x17c\r'

```

Grafana template from this [link](https://github.com/TaiyoMatsumoto/VoltronicAxpertSolarMonitorBMS/blob/main/grafana/AXPERT.json), from [TaiyoMatsumoto](https://github.com/TaiyoMatsumoto/VoltronicAxpertSolarMonitorBMS) repo (too complicated for my needs), 
it also take values from Home Assistant in my case ;)
HA with plotly-graph https://github.com/dbuezas/lovelace-plotly-graph-card easy history scrolling.
[image]

kitos23, 2023-04-02:
> > 
> output_power looks wrong

yes sir. Yesterday I saw that value that didn't make much sense, unless my house was a nuclear power plant. I love your graph. I have already seen that the battery and its cells are also represented. I also have an LFP battery that I made with 32x70 mm size batteries, a total of 768 and about 15 kwh approx with a 250A Daly BMS and 2 QNBBM active balancers of 8s each.

jblance, 2023-04-02:
> @jblance
> this cannot be real string
> `00000.27` is a firmware version

yes, but there are limited identifying results for the PI16 protocol - so was adding something that should provide an indication of differences

jblance, 2023-04-02:
> Command: QVFW3 - No description found        <<<<<<<< PI30 Bluetooth Firmware
> ------------------------------------------------------------
> raw_response                    '(VERFW:00002.61\x17c\r'

The base PI30 doesnt support bluetooth, well at least thats the though behind not having this defined in PI30 but defining it in the PI30 based protocols

Saentist, 2023-04-02:
> > Command: QVFW3 - No description found        <<<<<<<< PI30 Bluetooth Firmware
> > raw_response                    '(VERFW:00002.61\x17c\r'
> 
> The base PI30 doesnt support bluetooth, well at least thats the though behind not having this defined in PI30 but defining it in the PI30 based protocols

This is a just firmware version installed, 2.61 in this case.
This can be read by USB and RS232 interfaces.
not connected to https://github.com/jblance/mpp-solar/issues/293

kitos23, 2023-04-04:
I have to take the next step and I think it will not be a road of roses for me. I have been studying the wiki a bit and it is not clear to me, if to upload data to the grafana web, everything can be done from mpp solar, with all its commands or I will need something else. At least I have already taken the first step, which is to register in grafana haha.
I don't know the alternatives either, I have seen that you do it via home assistant, I don't know if it will be easier than publishing in grafana. Anyway, I don't know much and it's going to cost me a world to get it, due to my almost total ignorance.
Let's see if you can give me a hand, and at least tell me where to start or which path to take and if there was a step-by-step tutorial for newbies like me. Thank you.

jblance, 2023-04-04:
It depends on what you are trying to do (and what skills/knowledge you have)

Saentist, 2023-04-04:
@kitos23 
Create [Discussion](https://github.com/jblance/mpp-solar/discussions) Topic.
Maby more people are interested and can join with comments

kitos23, 2023-04-04:
> @kitos23 Create [Discussion](https://github.com/jblance/mpp-solar/discussions) Topic. Maby more people are interested and can join with comments

ok, so i will

> It depends on what you are trying to do (and what skills/knowledge you have)

knowledge practically nothing. very very basic

kitos23, 2023-04-04:
Yes, basically the QPIGS values show them on a web. I don't care about grafana or home assistant. If it's easier with a home assistant, then this one.

closed 2023-04-19 by jblance