jblance/mpp-solar #340: Screen output visual optimization.

Saentist, 2023-04-22:
This is not a bug but some optimization of visualization needed.
What I mean.
When multiple commands send response is some messy.

`--------------------------------------------------------------------------------`
is not on place
If there is some  separation between command outputs also will be good

```
+----=---------------=----+
| poor man's draw is good |
\-------------------------/
```

jblance, 2023-05-25:
added footer to latest branch
```
$ mpp-solar -p test -c QID#QPI
Command: QID - Device Serial Number inquiry
--------------------------------------------------------------------------------
Parameter     Value             Unit
serial_number 9293333010501         
--------------------------------------------------------------------------------

Command: QPI - Protocol ID inquiry
--------------------------------------------------------------------------------
Parameter   Value               Unit
protocol_id PI30                    
--------------------------------------------------------------------------------
```

closed 2023-05-25 by jblance

Saentist, 2023-05-25:
Simple example

$ mpp-solar -p test -c QID#QPI
Command: QID - Device Serial Number inquiry
--------------------------------------------------------------------------------
Parameter     Value             Unit
serial_number 9293333010501         
--------------------------------------------------------------------------------

Command: QPI - Protocol ID inquiry
--------------------------------------------------------------------------------
Parameter   Value               Unit
protocol_id PI30                    
--------------------------------------------------------------------------------

Versus

$ mpp-solar -p test -c QID#QPI
|------------------------------------------------------------------------------\
Command: QID - Device Serial Number inquiry
+-----------+------------------------------------------------------------------|
Parameter     Value             Unit
serial_number 9293333010501         
|___________________________________________________________________________/

|------------------------------------------------------------------------------\
Command: QPI - Protocol ID inquiry
+-----------+------------------------------------------------------------------|
Parameter   Value               Unit
protocol_id PI30                    
|___________________________________________________________________________/

```
$ mpp-solar -p test -c QID#QPI
|------------------------------------------------------------------------------\
Command: QID - Device Serial Number inquiry
+-----------+------------------------------------------------------------------|
Parameter     Value             Unit
serial_number 9293333010501         
|_____________________________________________________________________________/

|------------------------------------------------------------------------------\
Command: QPI - Protocol ID inquiry
+-----------+------------------------------------------------------------------|
Parameter   Value               Unit
protocol_id PI30                    
|_____________________________________________________________________________/
```

----
p.s. 
formatting not applied for visual reason
4 minus symbols is BB code
ASCII art is very interesting
line draw double line draw, good old DOS times ;)

```
███╗   ███╗██████╗ ██████╗       ███████╗ ██████╗ ██╗      █████╗ ██████╗ 
████╗ ████║██╔══██╗██╔══██╗      ██╔════╝██╔═══██╗██║     ██╔══██╗██╔══██╗
██╔████╔██║██████╔╝██████╔╝█████╗███████╗██║   ██║██║     ███████║██████╔╝
██║╚██╔╝██║██╔═══╝ ██╔═══╝ ╚════╝╚════██║██║   ██║██║     ██╔══██║██╔══██╗
██║ ╚═╝ ██║██║     ██║           ███████║╚██████╔╝███████╗██║  ██║██║  ██║
╚═╝     ╚═╝╚═╝     ╚═╝           ╚══════╝ ╚═════╝ ╚══════╝╚═╝  ╚═╝╚═╝  ╚═╝
```

jblance, 2023-05-25:
Not sure what you are asking / saying here?
You want + symbols as well?

jblance, 2023-05-28:
added a boxdraw output
```
$ mpp-solar -p test -o boxdraw -c QPIGS
╔══════════════════════════════════════════════════════════════╗
║ Command: QPIGS - General Status Parameters inquiry           ║
╠═════════════════════════════════════════════╤════════╤═══════╣
║ Parameter                                   │ Value  │ Unit  ║
╟─────────────────────────────────────────────┼────────┼───────╢
║ ac_input_voltage                            │ 0.0    │ V     ║
║ ac_input_frequency                          │ 0.0    │ Hz    ║
║ ac_output_voltage                           │ 230.0  │ V     ║
║ ac_output_frequency                         │ 49.9   │ Hz    ║
║ ac_output_apparent_power                    │ 161    │ VA    ║
║ ac_output_active_power                      │ 119    │ W     ║
║ ac_output_load                              │ 3      │ %     ║
║ bus_voltage                                 │ 460    │ V     ║
║ battery_voltage                             │ 57.5   │ V     ║
║ battery_charging_current                    │ 12     │ A     ║
║ battery_capacity                            │ 100    │ %     ║
║ inverter_heat_sink_temperature              │ 69     │ °C    ║
║ pv_input_current_for_battery                │ 14.0   │ A     ║
║ pv_input_voltage                            │ 103.8  │ V     ║
║ battery_voltage_from_scc                    │ 57.45  │ V     ║
║ battery_discharge_current                   │ 0      │ A     ║
║ is_sbu_priority_version_added               │ 0      │ bool  ║
║ is_configuration_changed                    │ 0      │ bool  ║
║ is_scc_firmware_updated                     │ 1      │ bool  ║
║ is_load_on                                  │ 1      │ bool  ║
║ is_battery_voltage_to_steady_while_charging │ 0      │ bool  ║
║ is_charging_on                              │ 1      │ bool  ║
║ is_scc_charging_on                          │ 1      │ bool  ║
║ is_ac_charging_on                           │ 0      │ bool  ║
║ rsv1                                        │ 0      │ A     ║
║ rsv2                                        │ 0      │ A     ║
║ pv_input_power                              │ 856    │ W     ║
║ is_charging_to_float                        │ 0      │ bool  ║
║ is_switched_on                              │ 1      │ bool  ║
║ is_reserved                                 │ 0      │ bool  ║
╚═════════════════════════════════════════════╧════════╧═══════╝
```