jblance/mpp-solar #342: jkbms connect via rs485

krime777, 2023-04-26:
I'm querying the jkbms via bluetooth every minute and besides being slow sometimes it gets stuck, so I tried to switch to the solution presented by the solar assistant
[image]

lsusb:
```
Bus 001 Device 007: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
Bus 001 Device 004: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 005: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC)
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dmesg:
```
[814179.432398] usb 1-1.2: new full-speed USB device number 8 using dwc_otg
[814179.564650] usb 1-1.2: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 3.00
[814179.564682] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[814179.564698] usb 1-1.2: Product: USB-Serial Controller
[814179.564711] usb 1-1.2: Manufacturer: Prolific Technology Inc.
[814179.565697] pl2303 1-1.2:1.0: pl2303 converter detected
[814179.569828] usb 1-1.2: pl2303 converter now attached to ttyUSB0
```

jkbms:
```
Solar Device Command Utility, version: 0.15.28
```

tried with -P JK485 and JK232, tried with -b 115200 and 9600, serial response is always b''

not sure what else i can try :(

krime777, 2023-04-26:
Info via bluetooth about the bms:

```
jkbms -p C8:47:8C:E1:E4:72 -P JK02 -c getInfo
Command: getInfo - BLE Device Information inquiry
--------------------------------------------------------------------------------
Parameter          Value                Unit
header             55aaeb90
record_type        03
record_counter     107
device_model       JK-B1A24S15P
hardware_version   10.XG
software_version   10.07
up_time            373D8H30M0S
power-on_times     4
device_name        JK-B1A24S15P
device_passcode    1234
manufacturing_date 220411
serial_number      2012601118
passcode           0000
user_data          Input Userdata
setup_passcode     123456
```

jblance, 2023-04-26:
ive not had an opportunity to debug this so far (not had someone with the rs485 connection have the time to see what is going on)
im pretty sure it doesnt work yet - but if you have time for testing (and testing again...) we can try to work on it
adding `-D` and posting all the output is key to help see what is and isnt working and trying to fix the problems

note: others have used their own code and suggested that not everything works via rs485

krime777, 2023-04-27:
```
root@raspberrypi:/home/pi# jkbms -p /dev/ttyUSB0 -b 9600 -P JK485 -D
2023-04-27 06:44:43,725:INFO:__init__:main@213: Solar Device Command Utility, version: 0.15.28
2023-04-27 06:44:43,726:DEBUG:mqttbrokerc:__init__@32: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-04-27 06:44:43,726:DEBUG:__init__:main@238: MqttBroker name: localhost, port: 1883, user: None
2023-04-27 06:44:43,726:DEBUG:__init__:main@240: udp port 5555
2023-04-27 06:44:43,727:DEBUG:__init__:main@242: Using Postgres None
2023-04-27 06:44:43,727:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2023-04-27 06:44:43,727:INFO:__init__:main@346: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK485"
2023-04-27 06:44:43,732:DEBUG:__init__:main@350: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2023-04-27 06:44:43,732:DEBUG:device:__init__@33: __init__ args ()
2023-04-27 06:44:43,733:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK485', 'baud': 9600, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x76099bc8>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-04-27 06:44:43,733:DEBUG:__init__:get_port_type@73: port matches ttyusb
2023-04-27 06:44:43,733:INFO:__init__:get_port@121: Using serialio for communications
2023-04-27 06:44:43,747:DEBUG:__init__:get_protocol@13: Protocol JK485
2023-04-27 06:44:43,753:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x76099cd0>, protocol JKBMS RS485 serial communication protocol handler
2023-04-27 06:44:43,753:DEBUG:__init__:main@401: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x76099ad8>, '', '', 'screen', None, None)]
2023-04-27 06:44:43,753:INFO:__init__:main@406: Looping 1 commands
2023-04-27 06:44:43,754:INFO:__init__:main@416: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x76099cd0>, protocol: JKBMS RS485 serial communication protocol handler for command: , tag: , outputs: screen
2023-04-27 06:44:43,754:INFO:device:run_command@50: Running command
2023-04-27 06:44:43,754:INFO:jk485:get_full_command@100: Using protocol b'JK485' with 1 commands
2023-04-27 06:44:43,755:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-04-27 06:44:43,755:DEBUG:abstractprotocol:get_command_defn@59: Processing command 'getBalancerData'
2023-04-27 06:44:43,755:DEBUG:abstractprotocol:get_command_defn@61: Found command getBalancerData in protocol b'JK485'
2023-04-27 06:44:43,755:DEBUG:jk485:get_full_command@114: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2023-04-27 06:44:43,756:DEBUG:jk485:get_full_command@117: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2023-04-27 06:44:43,756:DEBUG:jk485:get_full_command@122: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2023-04-27 06:44:43,756:DEBUG:jk485:get_full_command@125: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2023-04-27 06:44:43,756:INFO:device:run_command@78: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2023-04-27 06:44:43,757:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-04-27 06:44:43,757:DEBUG:abstractprotocol:get_command_defn@59: Processing command 'getBalancerData'
2023-04-27 06:44:43,757:DEBUG:abstractprotocol:get_command_defn@61: Found command getBalancerData in protocol b'JK485'
2023-04-27 06:44:43,757:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 9600
2023-04-27 06:44:43,762:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-04-27 06:44:44,864:DEBUG:serialio:send_and_receive@30: serial response was: b''
2023-04-27 06:44:45,135:DEBUG:device:run_command@95: Send and Receive Response b''
2023-04-27 06:44:45,135:INFO:abstractprotocol:decode@218: response passed to decode: b''
2023-04-27 06:44:45,136:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-04-27 06:44:45,136:DEBUG:abstractprotocol:get_command_defn@59: Processing command 'getBalancerData'
2023-04-27 06:44:45,137:DEBUG:abstractprotocol:get_command_defn@61: Found command getBalancerData in protocol b'JK485'
2023-04-27 06:44:45,137:INFO:abstractprotocol:decode@262: Processing response of type POSITIONAL
2023-04-27 06:44:45,138:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 2, 'Header', '']
2023-04-27 06:44:45,138:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Slave Address', '']
2023-04-27 06:44:45,139:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Command Code', '']
2023-04-27 06:44:45,139:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/100', 2, 'Total Battery Voltage', 'V']
2023-04-27 06:44:45,140:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Average Cell Voltage', 'V']
2023-04-27 06:44:45,140:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Number of Cells', '']
2023-04-27 06:44:45,141:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Highest Cell', '']
2023-04-27 06:44:45,141:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Lowest Cell', '']
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 526, in decode
    processed_responses = self.process_response(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 197, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/protocol_helpers.py", line 95, in Hex2Int
    answer = hexString[0]
IndexError: index out of range
```

krime777, 2023-05-07:
How can we help?

sugar012, 2023-05-08:
> ive not had an opportunity to debug this so far (not had someone with the rs485 connection have the time to see what is going on) im pretty sure it doesnt work yet - but if you have time for testing [...]

That's a bummer. I manager tontake data from bluetooth and send It to mysql but since i'm waiting to install a second JK i can't use bluetooth for both in continuous polling. Hope we can arrange something

sw25481, 2023-05-14:
I have the same setup.  First I installed Solar Assistant and confirmed it could communicate with my JK BMS over RS485, so I know my connection is good.  YOu can see form the debug I do get some data back

root@pizero:/etc/mbusd# jkbms -p /dev/ttyUSB1 -P JK485 -c
Command: command help - List available commands for protocol JK485
--------------------------------------------------------------------------------
Parameter       Value           Unit
getBalancerData Get Balancer Data -- Get Balancer Data
root@pizero:/etc/mbusd# jkbms -p /dev/ttyUSB1 -P JK485 -c getBalancerData
Traceback (most recent call last):
  File "/usr/local/bin/jkbms", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 420, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 112, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 532, in decode
    processed_responses = self.process_response(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 203, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/protocol_helpers.py", line 95, in Hex2Int
    answer = hexString[0]
IndexError: index out of range
root@pizero:/etc/mbusd# jkbms -p /dev/ttyUSB1 -P JK485 -c getBalancerData -D
2023-05-14 19:28:29,092:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.57
2023-05-14 19:28:29,098:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-05-14 19:28:29,104:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2023-05-14 19:28:29,107:DEBUG:__init__:main@241: udp port 5555
2023-05-14 19:28:29,111:DEBUG:__init__:main@243: Using Postgres None
2023-05-14 19:28:29,115:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2023-05-14 19:28:29,120:INFO:__init__:main@347: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK485"
2023-05-14 19:28:29,156:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2023-05-14 19:28:29,160:DEBUG:device:__init__@33: __init__ args ()
2023-05-14 19:28:29,163:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'JK485', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xb5cdaf10>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-05-14 19:28:29,169:DEBUG:__init__:get_port_type@73: port matches ttyusb
2023-05-14 19:28:29,172:INFO:__init__:get_port@121: Using serialio for communications
2023-05-14 19:28:29,225:DEBUG:__init__:get_protocol@13: Protocol JK485
2023-05-14 19:28:30,119:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0xb5cdafe8>, protocol JKBMS RS485 serial communication protocol handler
2023-05-14 19:28:30,122:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xb5cdaf28>, 'getBalancerData', 'getBalancerData', 'screen', None, None)]
2023-05-14 19:28:30,126:INFO:__init__:main@407: Looping 1 commands
2023-05-14 19:28:30,130:INFO:__init__:main@417: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0xb5cdafe8>, protocol: JKBMS RS485 serial communication protocol handler for command: getBalancerData, tag: getBalancerData, outputs: screen
2023-05-14 19:28:30,135:INFO:device:run_command@50: Running command getBalancerData
2023-05-14 19:28:30,140:INFO:jk485:get_full_command@100: Using protocol b'JK485' with 1 commands
2023-05-14 19:28:30,143:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-05-14 19:28:30,149:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'getBalancerData'
2023-05-14 19:28:30,152:DEBUG:abstractprotocol:get_command_defn@67: Found command getBalancerData in protocol b'JK485'
2023-05-14 19:28:30,157:DEBUG:jk485:get_full_command@114: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2023-05-14 19:28:30,161:DEBUG:jk485:get_full_command@117: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2023-05-14 19:28:30,165:DEBUG:jk485:get_full_command@122: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2023-05-14 19:28:30,168:DEBUG:jk485:get_full_command@125: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2023-05-14 19:28:30,175:INFO:device:run_command@78: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2023-05-14 19:28:30,179:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2023-05-14 19:28:30,184:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'getBalancerData'
2023-05-14 19:28:30,188:DEBUG:abstractprotocol:get_command_defn@67: Found command getBalancerData in protocol b'JK485'
2023-05-14 19:28:30,192:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 2400
2023-05-14 19:28:30,202:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-05-14 19:28:31,311:DEBUG:serialio:send_and_receive@30: serial response was: b''
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 532, in decode
    processed_responses = self.process_response(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 203, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/protocol_helpers.py", line 95, in Hex2Int
    answer = hexString[0]
IndexError: index out of range

sw25481, 2023-05-14:
Also printbms.py from https://github.com/fah/jk-bms works so this is possible

$ python3 printbms.py                                                        
Cell 1 : 3.297 V
Cell 2 : 3.289 V
Cell 3 : 3.295 V
Cell 4 : 3.295 V
Cell 5 : 3.295 V
Cell 6 : 3.293 V
Cell 7 : 3.29 V
Cell 8 : 3.294 V
Cell 9 : 3.29 V
Cell 10 : 3.289 V
Cell 11 : 3.292 V
Cell 12 : 3.295 V
Cell 13 : 3.295 V
Cell 14 : 3.297 V
Cell 15 : 3.29 V
Cell 16 : 3.294 V
Temp1:  20 ?C
Temp2:  21 ?C
Battery voltage:  52.68 V
Current:  24.05 A
Remaining capacity:  75 %

sugar012, 2023-05-14:
> Also printbms.py from https://github.com/fah/jk-bms works so this is possible
>
> $ python3 printbms.py Cell 1 : 3.297 V Cell 2 : 3.289 V Cell 3 : 3.295 V Cell 4 : 3.295 V Cell 5 : 3.295 V Cell 6 : [...]

How Is connected to the bms?

sw25481, 2023-05-17:
Just like the picture at the top of this thread which comes from here https://solar-assistant.io/help/battery/jk-bms

A USB RS485 adapter using the FTDI chip bought from Amazon, plugged into the JK BMS RS485 module.  The JK BMS RS485 module connected to the BMS using the GPS port.  I use a Pi Zero W (the first one) which does not have much power for the USB ports so I use a small powered USB hub.

>

sugar012, 2023-05-17:
> Just like the picture at the top of this thread which comes from here https://solar-assistant.io/help/battery/jk-bms
>
> A USB RS485 adapter using the FTDI chip bought from Amazon, plugged into the JK [...]

i have the exact same setup as the photo in #1 but it doesen't work for me. I already tried that

sugar012, 2023-06-28:
Hello there. @jblance do you have any news for the 485 support! I'm stuck with bluetooth and is very bad

nfaction, 2023-07-08:
Hello all! I've been very interested in this option via RS485 due to Bluetooth instabilities people are talking about.

I was able to confirm that I can get data using the JKBMS RS485 adapter connected to a USB adapter.

Once connected to a raspberry pi zero w via USB, I was able to get data from USB using https://github.com/fah/jk-bms as well.

```shell
# Git diff (changing to USB2)
$ git diff
diff --git a/printbms.py b/printbms.py
index 01dd162..9f68c62 100644
--- a/printbms.py
+++ b/printbms.py
@@ -14,7 +14,7 @@ import serial
 sleepTime = 10

 try:
-    bms = serial.Serial('/dev/ttyUSB0')
+    bms = serial.Serial('/dev/ttyUSB2')
     bms.baudrate = 115200
     bms.timeout  = 0.2
 except:

# Output
~/jk-bms $ python3 printbms.py
Cell 1 : 3.305 V
Cell 2 : 3.299 V
Cell 3 : 3.3 V
Cell 4 : 3.3 V
Cell 5 : 3.301 V
Cell 6 : 3.32 V
Cell 7 : 3.311 V
Cell 8 : 3.305 V
Cell 9 : 3.3 V
Cell 10 : 3.3 V
Cell 11 : 3.297 V
Cell 12 : 3.3 V
Cell 13 : 3.299 V
Cell 14 : 3.3 V
Cell 15 : 3.3 V
Cell 16 : 3.293 V
Temp1:  21 ?C
Temp2:  21 ?C
Battery voltage:  52.78 V
Current:  0.72 A
Remaining capacity:  14 %
```

As many others have mentioned, I get :

```
$ jkbms -b 115200 -P JK485 -D -p /dev/ttyUSB2
...
2023-07-08 03:03:56,258:DEBUG:abstractprotocol:process_response@117: Processing data_type: Hex2Int for data_name: Number of Cells, raw_value b''
2023-07-08 03:03:56,262:DEBUG:abstractprotocol:process_response@206: Processing format string Hex2Int(raw_value)
...
    answer = hexString[0]
IndexError: index out of range
```

As you can see, same exact setup and USB, but the jkbms command via mpp-solar does not work. Any ideas?

As a somewhat related note, the TTL connection (RS485 connection via GPS) does NOT work. I only have success with the JK BMS dongle. I have two JK BMS connected to two separate 16s LifePo4 batteries, one using dongle (ttyUSB2), the other using TTL cable (ttyUSB1) and USB1 does not work.

I was successful in getting data via ESP32 bluetooth [esphome-jk-bms](https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml) but it's pretty unreliable. Also I was hoping to emulate Pylontec protocol for my MPP inverter. I get data from both JK BMS. I'd much prefer to use USB.

torndar, 2023-11-23:
As others have said I can get data via printbms.py from https://github.com/fah/jk-bms . Using the Waveshare USB to RS458 and the command `jkbms -P jkserial -b 115000 -I` 

The first run I got this output 

```
$ jkbms -P jkserial -b 115200 -I
2023-11-23 15:03:30,347:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.13, python version: 3.10.12
2023-11-23 15:03:30,347:INFO:__init__:main@359: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "jkserial"
2023-11-23 15:03:30,348:INFO:__init__:get_port@111: Using serialio for communications
2023-11-23 15:03:30,350:INFO:__init__:main@413: Looping 1 commands
2023-11-23 15:03:30,350:INFO:__init__:main@423: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7ff1d06126b0>, protocol: JKBMS TTL serial communication protocol handler for command: , tag: , outputs: screen
2023-11-23 15:03:30,350:INFO:device:run_command@51: Running command 
2023-11-23 15:03:30,350:INFO:device:run_command@81: full command bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&') for command getBalancerData
2023-11-23 15:03:31,460:INFO:abstractprotocol:decode@246: response passed to decode: b"]\x05\xda\x02\x04\x04\x04h\x08PP/\xa5@(\x96\x90\xe85\xec\xa8\x9f\x90\xa8V\x90\xa8\x96\x90\x88\xd6\x90\xa8\x13\xc8hS\xc8\x88\xdf\xc8(\xd6\xc8(\xc5\xc8\xa8V\xc8\xa8\xdf\xc8\xa8\x1f\x180\xa20@!0`h\x18\x91#\x18 \xa0a'\x98\xa0\x180\xa0s@\xc0\x80\x80\x8a\x00\x0e\x8b\x00\x00\x8c\x00\x03\x8e\x16\xea\x8f\x10\xf4\x90\x10^\x91\x10@\x92\x00\x05\x93\x0c\x1c\x94\x0c\x80\x95\x00\x05\x96\x01,\x97\x00*\x98\x00\x1e\x99\x00\x19\x9a\x00\x1e\x9b\x0fn\x9c\x00\n\x9d\x01\x9e\x00F\x9f\x00<\xa0\x00d\xa1\x00d\xa2\x00\x14\xa3\x002\xa4\x00<\xa5\x00\x01\xa6\x00\x05\xa7\xff\xec\xa8\xff\xf6\xa9\x0e\xaa\x00\x00\x00\x15\xab\x01\xac\x01\xad\x03I\xae\x01\xaf\x01\xb0\x00\n\xb1\x14\xb2631790\x00\x00\x00\x00\xb3\x00\xb4Input Us\xb52310\xb6\x00\x00S\xbc\xb711.XW_S11.271__\xb8\x00\xb9\x00\x00\x00\x15\xbaInput UserdaTestStandBMS\xc0\x01\x00\x00\x00\x00h\x00\x00J\x81"
2023-11-23 15:03:31,461:INFO:abstractprotocol:decode@290: Processing response of type POSITIONAL
2023-11-23 15:03:31,465:INFO:device:run_command@116: Decoded response {'_command': 'getBalancerData', '_command_description': 'Get Balancer Data', 'raw_response': ["]\x05Ú\x02\x04\x04\x04h\x08PP/¥@(\x96\x90è5ì¨\x9f\x90¨V\x90¨\x96\x90\x88Ö\x90¨\x13ÈhSÈ\x88ßÈ(ÖÈ(ÅÈ¨VÈ¨ßÈ¨\x1f\x180¢0@!0`h\x18\x91#\x18 \xa0a'\x98\xa0\x180\xa0s@À\x80\x80\x8a\x00\x0e\x8b\x00\x00\x8c\x00\x03\x8e\x16ê\x8f\x10ô\x90\x10^\x91\x10@\x92\x00\x05\x93\x0c\x1c\x94\x0c\x80\x95\x00\x05\x96\x01,\x97\x00*\x98\x00\x1e\x99\x00\x19\x9a\x00\x1e\x9b\x0fn\x9c\x00\n\x9d\x01\x9e\x00F\x9f\x00<\xa0\x00d¡\x00d¢\x00\x14£\x002¤\x00<¥\x00\x01¦\x00\x05§ÿì¨ÿö©\x0eª\x00\x00\x00\x15«\x01¬\x01\xad\x03I®\x01¯\x01°\x00\n±\x14²631790\x00\x00\x00\x00³\x00´Input Usµ2310¶\x00\x00S¼·11.XW_S11.271__¸\x00¹\x00\x00\x00\x15ºInput UserdaTestStandBMSÀ\x01\x00\x00\x00\x00h\x00\x00J\x81", ''], 'Cells_connected': [55.0, ''], 'Voltage_Cell01': [10.39, 'V'], 'Voltage_Cell02': [-6.091, 'V'], 'Voltage_Cell03': [-22.369, 'V'], 'Voltage_Cell04': [-22.442, 'V'], 'Voltage_Cell05': [-22.378, 'V'], 'Voltage_Cell06': [-30.506, 'V'], 'Voltage_Cell07': [-22.509, 'V'], 'Voltage_Cell08': [26.707, 'V'], 'Voltage_Cell09': [-30.497, 'V'], 'Voltage_Cell10': [10.454, 'V'], 'Voltage_Cell11': [10.437, 'V'], 'Voltage_Cell12': [-22.442, 'V'], 'Voltage_Cell13': [-22.305, 'V'], 'Voltage_Cell14': [-22.497, 'V'], 'Voltage_Cell15': [12.45, 'V'], 'Voltage_Cell16': [16.417, 'V'], 'MOS_Temp': [24680, '°C'], 'Battery_T1': [-28381, '°C'], 'Battery_T2': [8352, '°C'], 'Battery_Voltage': [101.36, 'V'], 'Battery_Current': [-61.92, 'A'], 'Percent_Remain': [115, '%'], 'Cycle_Count': [-32630, ''], 'Total_capacity': [243990528, 'Ahr'], 'Battery Warning Message': ['16ea', ''], 'Battery status information': ['10f4', ''], 'Balancer Active': [60, ''], 'Capacity Setting': [2885790979, 'Ahr'], 'Charge Enabled': [174, ''], 'Discharge Enabled': [175, '']}
2023-11-23 15:03:31,465:INFO:__init__:get_output@40: attempting to create output processor: screen
2023-11-23 15:03:31,466:INFO:screen:output@21: Using output processor: screen
Command: getBalancerData - Get Balancer Data
--------------------------------------------------------------------------------
Parameter                  Value          	Unit
cells_connected            55.0           	    
voltage_cell01             10.39          	V   
voltage_cell02             -6.091         	V   
voltage_cell03             -22.369        	V   
voltage_cell04             -22.442        	V   
voltage_cell05             -22.378        	V   
voltage_cell06             -30.506        	V   
voltage_cell07             -22.509        	V   
voltage_cell08             26.707         	V   
voltage_cell09             -30.497        	V   
voltage_cell10             10.454         	V   
voltage_cell11             10.437         	V   
voltage_cell12             -22.442        	V   
voltage_cell13             -22.305        	V   
voltage_cell14             -22.497        	V   
voltage_cell15             12.45          	V   
voltage_cell16             16.417         	V   
mos_temp                   24680          	°C  
battery_t1                 -28381         	°C  
battery_t2                 8352           	°C  
battery_voltage            101.36         	V   
battery_current            -61.92         	A   
[... cut ...]
Powering the BMS off with the button and turning it back on I get the initial response and then everything after that is the second example response. 

The Victron BMS driver from https://github.com/Louisvdw/dbus-serialbattery/issues/29 seems to be working but the BMS is blocking charge and discharge. Not by disabling a Victron charger but in the BMS itself as it won't take anything from a bench power supply even when not connected to the RPI. Everything looks fine in the app so I figured we'd try a different piece of software to troubleshoot.

jblance, 2024-02-14:
you are getting incomplete results from the BMS - likely due to the serial input/output logic
you'd need to use 
`jkbms -P jkserial -b 115000 -I --porttype dalyserial`

geobitus, 2024-03-03:
Hardware: R_pi4 with CAN/RS485 hut connected trough 485 bus to a JK_B2A8S20P hw V11.XW sw V11.261H using CAN/RS485 port configured for CAN protocol (the RS485 protocol doesn't return anything, maybe the JK messed up the CAN/RS485 ticks).
 
On my system the hut native port is: /dev/ttyS0 wich is not recognized by the script so for be abble to access it I do a dirty but fast " mv /dev/ttyS0 /dev/ttyUSB0 "
 #mppsolar -p /dev/ttyUSB0 -P jkserial -D

2024-03-03 06:29:07,754:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.27-dev, python version: 3.11.2
2024-03-03 06:29:07,755:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-03-03 06:29:07,755:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-03-03 06:29:07,755:DEBUG:__init__:main@242: udp port 5555
2024-03-03 06:29:07,755:DEBUG:__init__:main@244: Using Postgres None
2024-03-03 06:29:07,755:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-03-03 06:29:07,755:INFO:__init__:main@358: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "jkserial"
2024-03-03 06:29:07,758:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-03-03 06:29:07,758:DEBUG:device:__init__@34: __init__ args ()
2024-03-03 06:29:07,759:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'jkserial', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7f933c1dd0>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-03-03 06:29:07,759:DEBUG:__init__:get_port_type@69: port matches ttyusb
2024-03-03 06:29:07,759:INFO:__init__:get_port@111: Using serialio for communications
2024-03-03 06:29:07,766:DEBUG:__init__:get_protocol@13: Protocol jkserial
2024-03-03 06:29:08,037:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7f9382a6d0>, protocol JKBMS TTL serial communication protocol handler
2024-03-03 06:29:08,037:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f92c57490>, '', '', 'screen', None, None)]
2024-03-03 06:29:08,037:INFO:__init__:main@413: Looping 1 commands
2024-03-03 06:29:08,037:INFO:__init__:main@423: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7f9382a6d0>, protocol: JKBMS TTL serial communication protocol handler for command: , tag: , outputs: screen
2024-03-03 06:29:08,037:INFO:device:run_command@51: Running command 
2024-03-03 06:29:08,037:DEBUG:jkserial:get_full_command@117: Using protocol b'JKSERIAL' with 1 commands
2024-03-03 06:29:08,037:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2024-03-03 06:29:08,037:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2024-03-03 06:29:08,038:DEBUG:jkserial:get_full_command@159: cmd with crc: bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&')
2024-03-03 06:29:08,038:INFO:device:run_command@81: full command bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&') for command getBalancerData
2024-03-03 06:29:08,038:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2024-03-03 06:29:08,038:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2024-03-03 06:29:08,038:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2024-03-03 06:29:08,038:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2024-03-03 06:29:09,141:DEBUG:serialio:send_and_receive@30: serial response was: b'\xfc\xff\x94GBT\\\xceD\xc7U\\\xcfDG\x0eT\\\xceD\xc7U\\\xcfDG\x86T^\xceD\xc7U\\\xcfD\xc7\x86T^\xceD\xc7U\\\xcfD\xc7LD\xcfDFBU\\\xcfD\xc7LL\xcfDFBU\\\xcfD\xc7LL\xcfDGBT\\\xcfD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT\\\xceD\xc7U\\\xcfDG\x86T\\\xceD\xc7U\\\xcfDG\x86T^\xceD\xc7U\\\xcfD\xc7\x86T^\xceD\xc7U\\\xcfD\xc7LD\xcfDFBU\\\xcfD\xc7LL\xcfDFBU\\\xcfD\xc7LL\xcfDGBT\\\xcfD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT'
2024-03-03 06:29:09,141:DEBUG:device:run_command@98: Send and Receive Response b'\xfc\xff\x94GBT\\\xceD\xc7U\\\xcfDG\x0eT\\\xceD\xc7U\\\xcfDG\x86T^\xceD\xc7U\\\xcfD\xc7\x86T^\xceD\xc7U\\\xcfD\xc7LD\xcfDFBU\\\xcfD\xc7LL\xcfDFBU\\\xcfD\xc7LL\xcfDGBT\\\xcfD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT\\\xceD\xc7U\\\xcfDG\x86T\\\xceD\xc7U\\\xcfDG\x86T^\xceD\xc7U\\\xcfD\xc7\x86T^\xceD\xc7U\\\xcfD\xc7LD\xcfDFBU\\\xcfD\xc7LL\xcfDFBU\\\xcfD\xc7LL\xcfDGBT\\\xcfD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT'
2024-03-03 06:29:09,142:INFO:abstractprotocol:decode@236: response passed to decode: b'\xfc\xff\x94GBT\\\xceD\xc7U\\\xcfDG\x0eT\\\xceD\xc7U\\\xcfDG\x86T^\xceD\xc7U\\\xcfD\xc7\x86T^\xceD\xc7U\\\xcfD\xc7LD\xcfDFBU\\\xcfD\xc7LL\xcfDFBU\\\xcfD\xc7LL\xcfDGBT\\\xcfD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT\\\xceD\xc7U\\\xcfDG\x86T\\\xceD\xc7U\\\xcfDG\x86T^\xceD\xc7U\\\xcfD\xc7\x86T^\xceD\xc7U\\\xcfD\xc7LD\xcfDFBU\\\xcfD\xc7LL\xcfDFBU\\\xcfD\xc7LL\xcfDGBT\\\xcfD\xc7LL\xcfDGBT\\\xceD\xc7LL\xcfDGBT'
2024-03-03 06:29:09,142:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
[... cut ...]
percent_remain             92             	%   
cycle_count                19532          	    
total_capacity             1145520724     	Ahr 
battery_warning_message    4c4c           	    
battery_status_information 4447           	    
balancer_active            94             	    
capacity_setting           1415368516     	Ahr 
charge_enabled             76             	    
discharge_enabled          207            	    
--------------------------------------------------------------------------------

jblance, 2024-03-03:
> 2024-03-03 ***@***.***: Solar Device Command
> 2024-03-03 ***@***.***: udp port 5555
> 2024-03-03 ***@***.***: Using Postgres None
> 2024-03-03 ***@***.***: Using Mongo None with
> 2024-03-03 ***@***.***: Creating device "unnamed"
> 2024-03-03 ***@***.***: device_class <class
> 2024-03-03 ***@***.***: port matches ttyusb
> 2024-03-03 ***@***.***: Using serialio for
> 2024-03-03 ***@***.***: Protocol jkserial
> 2024-03-03 ***@***.***: Commands
> 2024-03-03 ***@***.***: Looping 1 commands
> 2024-03-03 ***@***.***: Getting results from
> 2024-03-03 ***@***.***: Running command
> 2024-03-03 ***@***.***: Using
> 2024-03-03 ***@***.***: Found
> 2024-03-03 ***@***.***: cmd with
> 2024-03-03 ***@***.***: full command
> 2024-03-03 ***@***.***: port
> 2024-03-03 ***@***.***: Executing
> 2024-03-03 ***@***.***: serial
> 2024-03-03 ***@***.***: Send and Receive
> 2024-03-03 ***@***.***: response passed
> 2024-03-03 ***@***.***: Processing
> 2024-03-03 ***@***.***: trimmed and
> 2024-03-03 ***@***.***: Got defn
> 2024-03-03 ***@***.***: Got
> 2024-03-03 ***@***.***: Hex b'\xcf'
> 2024-03-03 ***@***.***: Hex
> 2024-03-03 ***@***.***: Hex b'\'
> 2024-03-03 ***@***.***: Hex b'LL'
> 2024-03-03 ***@***.***: Hex b'DG'
> 2024-03-03 ***@***.***: Hex b'^'
> 2024-03-03 ***@***.***: Hex b'L'
> 2024-03-03 ***@***.***: Decoded response
> 2024-03-03 ***@***.***: attempting to create
> 2024-03-03 ***@***.***: Using output filter: None
> 2024-03-03 ***@***.***: Not daemon, so not looping

Use `--porttype serial`

geobitus, 2024-03-03:
Hardware: R_pi4 with CAN/RS485 hut connected trough 485 bus to a JK_B2A8S20P hw V11.XW sw V11.261H using CAN/RS485 port configured for CAN protocol (the RS485 protocol doesn't return anything, maybe the JK messed up the CAN/RS485 ticks).
 
On my system the hut native port is: /dev/ttyS0 wich is not recognized by the script so for be abble to access it I do a dirty but fast " mv /dev/ttyS0 /dev/ttyUSB0 "
 #mppsolar -p /dev/ttyUSB0 -P jkserial -D
[jkserial.txt](https://github.com/jblance/mpp-solar/files/14472690/jkserial.txt)

jblance, 2024-03-03:
looks like that baud rate is wrong, try 9600 or 115200

geobitus, 2024-03-04:
9600 doesn't return any data.
# jkbms -b 115200 -P jkserial --porttype serial >>115200_no_eol.txt (without 120 ohm terminator)
[115200_no_eol.txt](https://github.com/jblance/mpp-solar/files/14482112/115200_no_eol.txt)
# jkbms -b 115200 -P jkserial --porttype serial >>115200_eol.txt (with 120 ohm terminator)
[115200_eol.txt](https://github.com/jblance/mpp-solar/files/14482139/115200_eol.txt)
any further test idea will be much appreciated

jblance, 2024-03-04:
oh, wait you are using the CAN protocol. I dont have any support for CAN
(yet)
maybe try rs485 again at different baud rates (and dont mv the port, just
use the system location and specify the port type)
also can you add the debug flag to the command `-D` and post the debug up
to where is shows the response (or all of it)

geobitus, 2024-03-08:
Back on business.
Now just for test the new hardware  (jikong RS485 adapter connected to GPS port) using the printbms.py script I get:
(mppsolar) geo@solarpi:~ $ python ./printbms.py
Cell 1 : 3.358 V
Cell 2 : 3.359 V
Cell 3 : 3.359 V
Cell 4 : 3.35 V
Cell 5 : 3.357 V
Cell 6 : 3.358 V
Cell 7 : 3.353 V
Cell 8 : 3.359 V
Temp1:  14 ?C
Temp2:  14 ?C
Battery voltage:  26.85 V
Current:  2.55 A
Remaining capacity:  99 %
mppsolar debug below:
(mppsolar) geo@solarpi:~ $ mppsolar -p /dev/ttyS0 -P jkserial -b 115000 --porttype serial -D 2> dbg.txt
[dbg.txt](https://github.com/jblance/mpp-solar/files/14540300/dbg.txt)

jblance, 2024-03-10:
> (mppsolar) ***@***.***:~ $ python ./printbms.py
> (mppsolar) ***@***.***:~ $ mppsolar -p /dev/ttyS0 -P jkserial -b 115000

try `--porttype dalyserial` instead of serial (as the data is getting
truncated

geobitus, 2024-03-31:
I encountered a new issue that I don't succeed to cleanly address.
The  dalyserialio.py, in my case, seems that don't/can't access  the port unless is opened in the background by an app let say by minicom.
The first workaround was to open one of the two ports I have ttySC0 & ttySC1 with "sudo minicom -D /dev/ttySCx " and put it to sleep "^Aj", and that is working fine and logically correct until the rpi is rebooted (I use sudo to stay resident even after I kill the putty terminal)
The second workaround was to add in the dalyserialio.py after line 21 (try:)" s = serial.Serial(self._serial_port)" and that seem to work fine for the last few days.
I will attach the must modify files in my case.
[jkserial.py.txt](https://github.com/jblance/mpp-solar/files/14814509/jkserial.py.txt)
[dalyserialio.py.txt](https://github.com/jblance/mpp-solar/files/14814512/dalyserialio.py.txt)