jblance/mpp-solar #351: Inaccurate Reading jkbms Bluetooth

sugar012, 2023-05-05:
Hello, i'm tring to retreive data from my JK but i have inaccurate reading suddenly. It worked before but now all i have Is this output:

```
pi@raspberrypi:~ $ jkbms -p C8:47:8C:EC:B7:28 -b 115200 -P JK02 -c getCellData
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------Parameter            Value              Unit
header               55aaeb90
record_type          02
record_counter       153
voltage_cell01       3.519              V
voltage_cell02       3.493              V
voltage_cell03       3.493              V
voltage_cell04       3.493              V
voltage_cell05       3.639              V
voltage_cell06       3.626              V
voltage_cell07       3.64               V
voltage_cell08       3.639              V
voltage_cell09       3.638              V
voltage_cell10       3.64               V
voltage_cell11       3.64               V
voltage_cell12       3.64               V
voltage_cell13       3.641              V
voltage_cell14       3.64               V
voltage_cell15       3.639              V
voltage_cell16       3.641              V
voltage_cell17       0.0                V
voltage_cell18       0.0                V
voltage_cell19       0.0                V
voltage_cell20       0.0                V
voltage_cell21       0.0                V
voltage_cell22       0.0                V
voltage_cell23       0.0                V
voltage_cell24       0.0                V
average_cell_voltage 0.0                V
delta_cell_voltage   0.0                V
current_balancer     0.0
resistance_cell01    0.0                Ohm
resistance_cell02    0.0                Ohm
resistance_cell03    0.0                Ohm
resistance_cell04    -0.001             Ohm
resistance_cell05    0.0                Ohm
resistance_cell06    3.604              Ohm
resistance_cell07    0.148              Ohm
resistance_cell08    0.26               Ohm
resistance_cell09    0.063              Ohm
resistance_cell10    0.063              Ohm
resistance_cell11    0.061              Ohm
resistance_cell12    0.061              Ohm
resistance_cell13    0.059              Ohm
resistance_cell14    0.06               Ohm
resistance_cell15    0.059              Ohm
resistance_cell16    0.059              Ohm
resistance_cell17    0.062              Ohm
resistance_cell18    0.063              Ohm
resistance_cell19    0.058              Ohm
resistance_cell20    0.059              Ohm
resistance_cell21    0.06               Ohm
resistance_cell22    0.06               Ohm
resistance_cell23    0.06               Ohm
resistance_cell24    0.061              Ohm
battery_voltage      0.0                V
battery_power        0                  W
balance_current      0.0                A
battery_t1           0.0                °C
battery_t2           0.0                °C
mos_temp             0.0                °C
percent_remain       0                  %
capacity_remain      18153.472          Ah
nominal_capacity     0.0                Ah
cycle_count          57660
cycle_capacity       0.0                Ah
time                 13D15H45M29S
current_charge       25.6
current_discharge    0.0
```

```
pi@raspberrypi:~ $ jkbms -p C8:47:8C:EC:B7:28 -b 115200 -P JK02 -c getInfo
Command: getInfo - BLE Device Information inquiry
--------------------------------------------------------------------------------Parameter          Value                Unit
header             55aaeb90
record_type        03
record_counter     159
device_model       JK_B2A24S15P
hardware_version   11.XW
software_version   11.26
up_time            26D4H30M12S
power-on_times     3
device_name        JK_B2A24S15P
device_passcode    1234
manufacturing_date 230409
serial_number      3010546682
passcode           0000
user_data          Input Userdata
setup_passcode     1234
```

reset12, 2023-05-08:
Hello,
i think to have the same Problem.

sudo hcitool lescan
LE Scan ...
C8:47:8C:E8:D8:05 (unknown)
C8:47:8C:E8:D8:05 JK_B2A24S20P

[image]

[image]

sudo mpp-solar -D -p C8:47:8C:E8:D8:05 -P JK04 2>&1 | tee -a /tmp/trace.txt

[trace.txt](https://github.com/jblance/mpp-solar/files/11420011/trace.txt)

I hope this will help to solve the error.
with regards and thanks for your work
Bruno

sugar012, 2023-05-08:
> Hello, i think to have the same Problem.
>
> sudo hcitool lescan LE Scan ... C8:47:8C:E8:D8:05 (unknown) C8:47:8C:E8:D8:05 JK_B2A24S20P
>
> [image]
>
> [image]
>
> sudo mpp-solar -D -p C8:47:8C:E8:D8:05 -P JK04 [...]

Try `-P JK02`

reset12, 2023-05-08:
With -P JK02 it is better, but not OK.
Resitance Cell 6 Wrong

Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Parameter            Value          	Unit
header               55aaeb90       	    
record_type          02             	    
record_counter       234            	    
voltage_cell01       3.363          	V   
voltage_cell02       3.36           	V   
voltage_cell03       3.36           	V   
voltage_cell04       3.356          	V   
voltage_cell05       3.363          	V   
voltage_cell06       3.36           	V   
voltage_cell07       3.361          	V   
voltage_cell08       3.359          	V   
voltage_cell09       3.359          	V   
voltage_cell10       3.363          	V   
voltage_cell11       3.361          	V   
voltage_cell12       3.36           	V   
voltage_cell13       3.36           	V   
voltage_cell14       3.361          	V   
voltage_cell15       3.36           	V   
voltage_cell16       3.359          	V   
voltage_cell17       0.0            	V   
voltage_cell18       0.0            	V   
voltage_cell19       0.0            	V   
voltage_cell20       0.0            	V   
voltage_cell21       0.0            	V   
voltage_cell22       0.0            	V   
voltage_cell23       0.0            	V   
voltage_cell24       0.0            	V   
**average_cell_voltage** 0.0            	V   
delta_cell_voltage   0.0            	V   
current_balancer     0.0            	    
resistance_cell01    0.0            	Ohm 
resistance_cell02    0.0            	Ohm 
resistance_cell03    0.0            	Ohm 
resistance_cell04    -0.001         	Ohm 
resistance_cell05    0.0            	Ohm 
**resistance_cell06    3.36**           	Ohm 
resistance_cell07    0.005          	Ohm 
resistance_cell08    0.768          	Ohm 
resistance_cell09    0.05           	Ohm 
resistance_cell10    0.05           	Ohm 
resistance_cell11    0.05           	Ohm 
resistance_cell12    0.05           	Ohm 
resistance_cell13    0.049          	Ohm 
resistance_cell14    0.05           	Ohm 
resistance_cell15    0.049          	Ohm 
resistance_cell16    0.049          	Ohm 
resistance_cell17    0.049          	Ohm 
resistance_cell18    0.049          	Ohm 
resistance_cell19    0.049          	Ohm 
resistance_cell20    0.049          	Ohm 
resistance_cell21    0.049          	Ohm 
resistance_cell22    0.049          	Ohm 
resistance_cell23    0.049          	Ohm 
resistance_cell24    0.049          	Ohm 
**battery_voltage**      0.0            	V   
battery_power        0              	W   
balance_current      0.0            	A   
battery_t1           0.0            	°C  
battery_t2           0.0            	°C  
mos_temp             0.0            	°C  
**percent_remain**       0              	%   
capacity_remain      16449.536      	Ah  
**nominal_capacity**     0.0            	Ah  
cycle_count          53758          	    
cycle_capacity       657.78         	Ah  
time                 156D6H10M14S   	    
current_charge       25.6           	    
current_discharge    0.0            	    
2023-05-08 13:03:06,208:DEBUG:__init__:main@452: Not daemon, so not looping

sugar012, 2023-05-08:
paste a `-getInfo`

If you have 11. like me you have to modify the jk02 protocol file in your system starting from line 44 and adding lines until 32. same for the resistance lines

https://github.com/jblance/mpp-solar/blob/master/mppsolar/protocols/jk02.py

reset12, 2023-05-08:
With the Updates it works very well.
Please can you update your install document to install pip3 install bluepy after mppsolar.

Thank you for your help.
with regards
Bruno

jblance, 2023-05-08:
> paste a `-getInfo`
> 
> If you have 11. like me you have to modify the jk02 protocol file in your system starting from line 44 and adding lines until 32. same for the resistance lines
> 
> https://github.com/jblance/mpp-solar/blob/master/mppsolar/protocols/jk02.py

this protocol `jkv11`

jblance, 2023-05-08:
updated wiki for protocols and pypi install info

closed 2023-05-08 by jblance

sugar012, 2023-05-09:
> 

didn't see that! My bad
Can confirm it works OK with JKV11

reset12, 2023-05-09:
Yes , it works very well with jkv11

Low characters !