jblance/mpp-solar #352: PI18: Attempted to run command with no protocol defined

sycophantic, 2023-05-05:
Updating from 0.15.24 to 0.15.56 I get the following error when using `-P PI18`

```Attempted to run command with no protocol defined```

I see commits updating the PI18 code, but don't know why this is occurring.

jblance, 2023-05-05:
Can you post the debug output of a command run

jmck82, 2023-05-16:
Not sure if this is the same issue but I just upgraded to 15.57 and got this

/usr/local/bin/mpp-solar -D  -P PI18 -c GS -p /dev/ttyUSB1
2023-05-15 21:43:43,235:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.57
2023-05-15 21:43:43,235:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-05-15 21:43:43,235:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2023-05-15 21:43:43,235:DEBUG:__init__:main@241: udp port 5555
2023-05-15 21:43:43,235:DEBUG:__init__:main@243: Using Postgres None
2023-05-15 21:43:43,235:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2023-05-15 21:43:43,235:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=None)" using protocol "PI18"
2023-05-15 21:43:43,236:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-05-15 21:43:43,236:DEBUG:device:__init__@33: __init__ args ()
2023-05-15 21:43:43,236:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'PI18', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fc0f3f60250>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-05-15 21:43:43,236:DEBUG:__init__:get_port_type@73: port matches ttyusb
2023-05-15 21:43:43,236:INFO:__init__:get_port@121: Using serialio for communications
2023-05-15 21:43:43,239:DEBUG:__init__:get_protocol@13: Protocol PI18
2023-05-15 21:43:43,277:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7fc0f3f60490>, protocol PI18 protocol handler
2023-05-15 21:43:43,277:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fc0f3f60280>, 'GS', 'GS', 'screen', None, None)]
2023-05-15 21:43:43,277:INFO:__init__:main@407: Looping 1 commands
2023-05-15 21:43:43,277:INFO:__init__:main@417: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7fc0f3f60490>, protocol: PI18 protocol handler for command: GS, tag: GS, outputs: screen
2023-05-15 21:43:43,277:INFO:device:run_command@50: Running command GS
2023-05-15 21:43:43,277:INFO:pi18:get_full_command@583: Using protocol b'PI18' with 24 commands
2023-05-15 21:43:43,277:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'GS'
2023-05-15 21:43:43,277:DEBUG:abstractprotocol:get_command_defn@67: Found command GS in protocol b'PI18'
2023-05-15 21:43:43,277:DEBUG:pi18:get_full_command@593: _cmd is: b'GS'
2023-05-15 21:43:43,277:DEBUG:pi18:get_full_command@596: _type is: QUERY
2023-05-15 21:43:43,277:DEBUG:pi18:get_full_command@600: _prefix: ^P005
2023-05-15 21:43:43,278:DEBUG:pi18:get_full_command@610: auto_prefix: ^P005
2023-05-15 21:43:43,278:DEBUG:pi18:get_full_command@614: _pre_cmd: b'^P005GS'
2023-05-15 21:43:43,278:DEBUG:pi18:get_full_command@627: CRC: True
2023-05-15 21:43:43,278:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'^P005GS'
2023-05-15 21:43:43,278:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x58 0x14 0x5814
2023-05-15 21:43:43,278:DEBUG:pi18:get_full_command@637: full command: b'^P005GSX\x14\r'
2023-05-15 21:43:43,278:INFO:device:run_command@78: full command b'^P005GSX\x14\r' for command GS
2023-05-15 21:43:43,278:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'GS'
2023-05-15 21:43:43,278:DEBUG:abstractprotocol:get_command_defn@67: Found command GS in protocol b'PI18'
2023-05-15 21:43:43,278:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 2400
2023-05-15 21:43:43,281:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-05-15 21:43:44,383:DEBUG:serialio:send_and_receive@30: serial response was: b''
2023-05-15 21:43:44,437:DEBUG:device:run_command@95: Send and Receive Response b''
2023-05-15 21:43:44,437:INFO:abstractprotocol:decode@224: response passed to decode: b''
2023-05-15 21:43:44,438:DEBUG:abstractprotocol:get_command_defn@65: Processing command 'GS'
2023-05-15 21:43:44,438:DEBUG:abstractprotocol:get_command_defn@67: Found command GS in protocol b'PI18'
2023-05-15 21:43:44,438:INFO:abstractprotocol:decode@268: Processing response of type DEFAULT
2023-05-15 21:43:44,438:DEBUG:abstractprotocol:decode@272: trimmed and split responses: [b'']
2023-05-15 21:43:44,438:INFO:abstractprotocol:decode@279: Processing DEFAULT type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/__init__.py", line 420, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/devices/device.py", line 112, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/protocols/abstractprotocol.py", line 318, in decode
    msgs[key] = [float(result) / 10, resp_format[2]]
ValueError: could not convert string to float: ''

jblance, 2023-05-16:
there was no response in that call (which should be caught better) are all commands getting no result now? (I assume they worked before?)

jmck82, 2023-05-16:
Actually, I think something is going on with the comms.    I was on a 1 year + old version, and when in upgraded it now it seems to send to the MQTT about 1 out of 60 times.    I will go check connections, but that seems weird that it did it after an update and was fine prior.   I honestly only updated as the old version had very limited commands on PI18

jblance, 2023-05-16:
there has been a heap of changes so its probable that bugs have been added, let me know if you find any and ill try to fix them

closed 2023-05-25 by jblance