jblance/mpp-solar #355: MPP MPI 15KW WP Supported?

JakobTewes, 2023-05-21:
Heyhey team,

I got two MPP MPI 15KW and currently trying to connect to the units via mpp-solar. Unfortunately I got no lock with both (via MicroUSB and RS232). 

`
root@bpvraspi:~# mpp-solar -p /dev/hidraw0 --getDeviceId
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value          	Unit
deviceid  0:0:0
`

`
root@bpvraspi:~#  mpp-solar -p /dev/hidraw0 -c QPI
Command: QPI - Protocol ID inquiry
--------------------------------------------------------------------------------
Parameter   Value          	Unit
protocol_id 0
`

`
root@bpvraspi:~# mpp-solar -p /dev/hidraw0 -c QPIGS
Command: QPIGS - General Status Parameters inquiry
--------------------------------------------------------------------------------
Parameter        Value          	Unit
ac_input_voltage 0.0            	V   	{'icon': 'mdi:lightning'}
`

tbh. I don´t know, if I´m a complete idiot, but is it correct, the unit just does not report anything back?
Connection and parametrising via SolarPower manufacturer app works

Thanks and regards

Jakob

JakobTewes, 2023-05-21:
Just tried with serial:

Plugged to "RS232" Port
`root@bpvraspi:~# mpp-solar -p /dev/hidraw0 -c QPIGS`
`Command: QPIGS - General Status Parameters inquiry`
`--------------------------------------------------------------------------------`
`Parameter        Value          	Unit`
`ac_input_voltage 0.0            	V   	{'icon': 'mdi:lightning'}`

Plugged to "BMS" Port / not connected on Inverter Side
`root@bpvraspi:~# mpp-solar -p /dev/ttyUSB0 -c QPIGS`
`Command: QPIGS - General Status Parameters inquiry`
`--------------------------------------------------------------------------------`
`Parameter      Value          	Unit`
`validity_check Error: Response to short`

Do I read that correct, that data is replied, but not encoded in the expected way?

Saentist, 2023-05-21:
Are you sure dat use correct port?
```bash
ls /dev/tty*
ls /dev/hid*
```
----
p.s.
instead to put each line to code
select citation and mark it as code

jblance, 2023-05-21:
Also check `dmesg` just after connecting to the inverter

jblance, 2023-05-23:
it looks like it is currently on /dev/hidraw1
try: `mpp-solar -p /dev/hidraw1 -c QPI` and/or `mpp-solar -p /dev/hidraw0 --getDeviceId`

JakobTewes, 2023-06-03:
Got the protocol specifications from MPP.
https://drive.google.com/file/d/14B_5zj4YBTdXVWDj3iMzYnDPIQ4TgfxK/view?usp=drivesdk
Does this help? Do the named inverters work different?

jblance, 2023-06-03:
That's the pi17 protocol. 

> it looks like it is currently on /dev/hidraw1 try: `mpp-solar -p /dev/hidraw1 -P PI17 -c PI` and/or `mpp-solar -p /dev/hidraw1 -P PI17 --getDeviceId`

JakobTewes, 2023-06-06:
hmmm...looking good? 

```shell
root@bpvraspi:/dev# mpp-solar -p /dev/hidraw0 -P PI17 -c PI
Command: PI - Device Protocol Version inquiry
--------------------------------------------------------------------------------
Parameter        Value          	Unit
protocol_version 17
root@bpvraspi:/dev# mpp-solar -p /dev/hidraw0 -P PI17 --getDeviceId
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value          	Unit
deviceid  17:058
```

closed 2023-06-23 by jblance