jblance/mpp-solar #357: Feature Request: Consider making mpp-solar version available via MQTT

frederickjh, 2023-05-24:
I run mpp-solar in a Docker container and when I go upgrade to the newest version it seems that sometimes the build uses the cache and the version does not upgrade mpp-solar to the newest version. To figure out what version it is I have to edit the Dockerfile to run `CMD mpp-solar -v` and rebuild the container so I can see what version is running.

If the mpp-solar version was available via MQTT this could be seen in my Home Assistant dashboard.

Thanks for considering this!
Frederick

jblance, 2023-05-24:
Hmm thay should be doable. Also there is now a mppsolar image on docker hub
jblance/mppsolar. Try it out and let me know if it needs improvement- the wiki has some further details

jblance, 2023-05-25:
added `--getVersion` to latest branch this will add the version to the command stack and output to the supplied outputs

closed 2023-05-25 by jblance

frederickjh, 2023-05-25:
> . . . Also there is now a mppsolar image on docker hub jblance/mppsolar. Try it out and let me know if it needs improvement- the wiki has some further details

I switched over to the Docker Hub image using my configuration file and every thing seems to be running OK.