jblance/mpp-solar #358: JK_B1A24S15P Data not Right

Smoeller0, 2023-05-28:
Hello 
I have an Problem in the Data of the JK_B1A24S15P.  The voltage_Cell seems Okay but all other data are not right. 
I already have some JK_B2A24S15P and the work fine but the 1A Version not.

mpp-solar -p C8:47:8C:ED:22:52 -P JK02 -c getCellData
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Parameter            Value          	Unit
header               55aaeb90       	    
record_type          02             	    
record_counter       182            	    
voltage_cell01       3.64           	V   
voltage_cell02       3.693          	V   
voltage_cell03       3.407          	V   
voltage_cell04       3.627          	V   
voltage_cell05       3.671          	V   
voltage_cell06       3.848          	V   
voltage_cell07       3.677          	V   
voltage_cell08       3.821          	V   
voltage_cell09       3.712          	V   
voltage_cell10       3.727          	V   
voltage_cell11       3.4            	V   
voltage_cell12       3.376          	V   
voltage_cell13       3.547          	V   
voltage_cell14       3.53           	V   
voltage_cell15       0.0            	V   
voltage_cell16       0.0            	V   
voltage_cell17       0.0            	V   
voltage_cell18       0.0            	V   
voltage_cell19       0.0            	V   
voltage_cell20       0.0            	V   
voltage_cell21       0.0            	V   
voltage_cell22       0.0            	V   
voltage_cell23       0.0            	V   
voltage_cell24       0.0            	V   
average_cell_voltage 0.0            	V   
delta_cell_voltage   0.0            	V   
current_balancer     0.0            	    
resistance_cell01    0.0            	Ohm 
resistance_cell02    0.0            	Ohm 
resistance_cell03    0.0            	Ohm 
resistance_cell04    16.383         	Ohm 
resistance_cell05    0.0            	Ohm 
resistance_cell06    3.62           	Ohm 
resistance_cell07    0.473          	Ohm 
resistance_cell08    2.821          	Ohm 
resistance_cell09    0.057          	Ohm 
resistance_cell10    0.06           	Ohm 
resistance_cell11    0.062          	Ohm 
resistance_cell12    0.059          	Ohm 
resistance_cell13    0.06           	Ohm 
resistance_cell14    0.06           	Ohm 
resistance_cell15    0.055          	Ohm 
resistance_cell16    0.053          	Ohm 
resistance_cell17    0.053          	Ohm 
resistance_cell18    0.056          	Ohm 
resistance_cell19    0.057          	Ohm 
resistance_cell20    0.055          	Ohm 
resistance_cell21    0.053          	Ohm 
resistance_cell22    0.055          	Ohm 
resistance_cell23    0.0            	Ohm 
resistance_cell24    0.0            	Ohm 
battery_voltage      0.0            	V   
battery_power        0              	W   
balance_current      0.0            	A   
battery_t1           0.0            	°C  
battery_t2           0.0            	°C  
mos_temp             0.0            	°C  
percent_remain       0              	%   
capacity_remain      13500.416      	Ah  
nominal_capacity     0.0            	Ah  
cycle_count          50679          	    
cycle_capacity       0.0            	Ah  
time                 37D3H27M44S    	    
current_charge       25.6           	    
current_discharge    0.0 

[image]

Smoeller0, 2023-05-28:
mpp-solar -p C8:47:8C:ED:22:52 -P JK02 -c getCellData -D
2023-05-28 16:53:36,889:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.59
2023-05-28 16:53:36,890:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-05-28 16:53:36,890:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2023-05-28 16:53:36,890:DEBUG:__init__:main@241: udp port 5555
2023-05-28 16:53:36,890:DEBUG:__init__:main@243: Using Postgres None
2023-05-28 16:53:36,890:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2023-05-28 16:53:36,891:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "C8:47:8C:ED:22:52 (porttype=None)" using protocol "JK02"
2023-05-28 16:53:36,894:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-05-28 16:53:36,894:DEBUG:device:__init__@33: __init__ args ()
2023-05-28 16:53:36,894:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:8C:ED:22:52', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xb5dc3b50>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-05-28 16:53:36,894:DEBUG:__init__:get_port_type@51: port matches jkble ':'
2023-05-28 16:53:36,894:INFO:__init__:get_port@105: Using jkbleio for communications
2023-05-28 16:53:36,919:DEBUG:__init__:get_protocol@13: Protocol JK02
2023-05-28 16:53:37,021:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0xb5dc3c58>, protocol JK02 - JKBMS BLE 2 byte data communication protocol handler
2023-05-28 16:53:37,022:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5dc3b68>, 'getCellData', 'getCellData', 'screen', None, None)]
2023-05-28 16:53:37,022:INFO:__init__:main@407: Looping 1 commands
2023-05-28 16:53:37,022:INFO:__init__:main@417: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0xb5dc3c58>, protocol: JK02 - JKBMS BLE 2 byte data communication protocol handler for command: getCellData, tag: getCellData, outputs: screen
2023-05-28 16:53:37,022:INFO:device:run_command@50: Running command getCellData
2023-05-28 16:53:37,022:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK02' with 3 commands
2023-05-28 16:53:37,022:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2023-05-28 16:53:37,023:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getCellData'
2023-05-28 16:53:37,023:DEBUG:abstractprotocol:get_command_defn@72: Found command getCellData in protocol b'JK02'
2023-05-28 16:53:37,023:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2023-05-28 16:53:37,023:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-05-28 16:53:37,023:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-05-28 16:53:37,023:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2023-05-28 16:53:37,023:INFO:device:run_command@78: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2023-05-28 16:53:37,023:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2023-05-28 16:53:37,024:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getCellData'
2023-05-28 16:53:37,024:DEBUG:abstractprotocol:get_command_defn@72: Found command getCellData in protocol b'JK02'
2023-05-28 16:53:37,024:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK02' with 3 commands
2023-05-28 16:53:37,024:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2023-05-28 16:53:37,024:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getCellData'
2023-05-28 16:53:37,024:DEBUG:abstractprotocol:get_command_defn@72: Found command getCellData in protocol b'JK02'
2023-05-28 16:53:37,024:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2023-05-28 16:53:37,024:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-05-28 16:53:37,025:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-05-28 16:53:37,025:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2023-05-28 16:53:37,025:INFO:jkbleio:send_and_receive@29: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2023-05-28 16:53:37,025:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2023-05-28 16:53:37,025:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getCellData'
2023-05-28 16:53:37,025:DEBUG:abstractprotocol:get_command_defn@72: Found command getCellData in protocol b'JK02'
[... cut ...]
battery_t1           0.0            	°C  
battery_t2           0.0            	°C  
mos_temp             0.0            	°C  
percent_remain       0              	%   
capacity_remain      13500.416      	Ah  
nominal_capacity     0.0            	Ah  
cycle_count          50679          	    
cycle_capacity       0.0            	Ah  
time                 37D3H27M44S    	    
current_charge       25.6           	    
current_discharge    0.0            	    
2023-05-28 16:53:48,678:DEBUG:__init__:main@452: Not daemon, so not looping

jblance, 2023-05-28:
Try with `-P jkv11`
This is the protocol for hardware/software version 11

On Mon, 29 May 2023, 3:17 am Smoeller0, ***@***.***> wrote:

> mpp-solar -p C8:47:8C:ED:22:52 -P JK02 -c getCellData -D
> 2023-05-28 ***@***.***: Solar Device Command
> Utility, version: 0.15.59
> 2023-05-28 ***@***.***
> <https://github.com/29>: mqttbroker config: {'name': 'localhost', 'port':
> 1883, 'user': None, 'pass': None}
> 2023-05-28 ***@***.***: MqttBroker name:
> localhost, port: 1883, user: None
> 2023-05-28 ***@***.***: udp port 5555
> 2023-05-28 ***@***.***: Using Postgres None
> 2023-05-28 ***@***.***: Using Mongo None with
> mppsolar
> 2023-05-28 ***@***.***: Creating device "unnamed"
> (type: "mppsolar") on port "C8:47:8C:ED:22:52 (porttype=None)" using
> protocol "JK02"
> 2023-05-28 ***@***.***: device_class <class
> 'mppsolar.devices.mppsolar.mppsolar'>
> 2023-05-28 ***@***.*** <https://github.com/33>:
> *init* args ()
> 2023-05-28 ***@***.*** <https://github.com/34>:
> *init* kwargs {'name': 'unnamed', 'port': 'C8:47:8C:ED:22:52',
> 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker':
> <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xb5dc3b50>, 'udp_port':
> '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
> 2023-05-28 ***@***.***: port matches jkble
> ':'
> 2023-05-28 ***@***.***: Using jkbleio for
> communications
> 2023-05-28 ***@***.***: Protocol JK02
> 2023-05-28 ***@***.*** <https://github.com/38>:
> *init* name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at
> 0xb5dc3c58>, protocol JK02 - JKBMS BLE 2 byte data communication protocol
> handler
> 2023-05-28 ***@***.***: Commands
> [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5dc3b68>, 'getCellData',
> 'getCellData', 'screen', None, None)]
> 2023-05-28 ***@***.***: Looping 1 commands
> 2023-05-28 ***@***.***: Getting results from
> device: mppsolar device - name: unnamed, port:
> <mppsolar.inout.jkbleio.JkBleIO object at 0xb5dc3c58>, protocol: JK02 -
> JKBMS BLE 2 byte data communication protocol handler for command:
> getCellData, tag: getCellData, outputs: screen
> 2023-05-28 ***@***.***: Running command
> getCellData
> 2023-05-28 ***@***.***:
> Using protocol b'JK02' with 3 commands
> 2023-05-28 ***@***.***:
> get_command_defn for: getCellData
> 2023-05-28 ***@***.***:
> Processing command 'getCellData'
> 2023-05-28 ***@***.***: Found
> command getCellData in protocol b'JK02'
> 2023-05-28 ***@***.***:
> self._command = getCellData
> 2023-05-28 ***@***.***: cmd
> with SOR:
> bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
> 2023-05-28 ***@***.***: cmd
> with command code:
> bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
> 2023-05-28 ***@***.***:
> cmd with crc:
> bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
> 2023-05-28 ***@***.***: full command
> bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
[... 565 lines cut ...]
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell06', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell07', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell08', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell09', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell10', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell11', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell12', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell13', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell14', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell15', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell16', 'V']
> 2023-05-28 ***@***.***: Got
> defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell17', 'V']
> 202

Smoeller0, 2023-05-29:
Oh wow thank you. this works fine.

closed 2023-06-06 by jblance