jblance/mpp-solar #363: mpp-solar error with influxdb2_mqtt parameter

rstoffel, 2023-06-13:
After updating to the latest version of mppsolar (using pip install) I get get an errormessage:

### command:
mpp-solar -c QPGS0 -p /dev/hidraw1 -q localhost --mqttport 1883 -o influx2_mqtt,screen --tag=QPGS0 --mqtttopic mpp-solar

### error:
_Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/__init__.py", line 428, in main
    op.output(
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/outputs/mqtt.py", line 113, in output
    mqtt_broker.publishMultiple(msgs)
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/libs/mqttbrokerc.py", line 137, in publishMultiple
    self.publish(msg["topic"], msg["payload"])
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/libs/mqttbrokerc.py", line 147, in publish
    self.connect()
  File "/usr/local/lib/python3.10/dist-packages/mppsolar/libs/mqttbrokerc.py", line 91, in connect
    self.mqttc.connect(self.name, port=self.port, keepalive=60)
  File "/home/rstoffel/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 912, in connect
    self.connect_async(host, port, keepalive,
  File "/home/rstoffel/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 979, in connect_async
    if port <= 0:
TypeError: '<=' not supported between instances of 'str' and 'int'_

this is consistent on several installations! ANy help woulb be appreciated!

jblance, 2023-06-13:
> 8, in sys.exit(main()) File

Can you post the debug output

rstoffel, 2023-06-14:
I did send the trackback info. if you tell me what else you need, I will
send it to you.
In the mean time I found out that the problem disappears, when I change the
command line as follows:

*mpp-solar -c QPGS0 -p /dev/hidraw1 -o influx2_mqtt,screen --tag=QPGS0
--mqtttopic mpp-solar*

by just removing the mqtt broker address and the port I get the result I
want on the local mosquitto boker.

Best regards
Reto

jblance, 2023-06-14:
Sorry I ment add `-D` to the command and post the output

It looks like I've added a bug - probably related to the port number option
but the debug output helps figure out where the problem is.
For the default port it isn't necessary to supply the info, but it's good
to fix any bugs that creep in

rstoffel, 2023-06-14:
[20230614_mppsolar_err.txt](https://github.com/jblance/mpp-solar/files/11743161/20230614_mppsolar_err.txt)

I put the output into the attached file.

Thank you for your fast reply!
Reto

jblance, 2023-06-14:
Great thanks. It does look like I've messed up the broker port handling.
I'll have a look and try to fix it tomorrow

jblance, 2023-06-15:
should be fixed in latest version

closed 2023-06-23 by jblance