jblance/mpp-solar #364: Example of using in python script

duculete, 2023-06-20:
Can you create a simple example of how can we use it in python script?

jblance, 2023-06-20:
Not really intended to be used in a python script but it could be
What are you trying to do?

duculete, 2023-06-20:
i have a python script that pull the data and push it to blynk.io 
currently i have the mpp-solar command (via contrab) that pull data from inverter and push it via mqtt to influxdb to see it in grafana.
a python script listens to mqtt topic and get that data and push it to blynk.io (where i have some nice dashboards and also mobile push notifications, etc).

I want to create one python app that pulls data directly from invertor, push the data to influxdb and blynk and i want also to schedule  update of output_source_priority (Solar first @ 9pm for example and SBU @9am) + other parameters set via blynk (ui mobile interface).

I can do it using os.popen("/usr/local/bin/mpp-solar -c %s" % mode) or other method that invoke shell but it will be more elegant if i can have all the code and modules in python and don't use sell for that.

thanks

jblance, 2023-06-20:
I'd suggest adding an outputter to mppsolar for blynk as the easiest way.
Check the git repo or share the key parts of your mqtt to blynk code and I
can have a look

The powermon command is under development and adds adhoc commands (so you
can send a one off command) this is intended for better control (powermon
is also adding a web api for similar reasons)

duculete, 2023-06-20:
thanks,

let me check and i will come back with more info/solution

closed 2023-07-28 by jblance