jblance/mpp-solar #368: Rich Solar/MPP Green LV2424

vknmakes, 2023-07-04:
I have something that is probably simple...
I've got an MPP LV2424 which is actually branded as a Rich Solar.  The SolarPower java crapplette will talk to it via either the built in USB or via the serial.  I cannot get mpp-solar to talk to it with either serial or the USB.  When I send a --getstatus or a -c QPI, -c QID I get a 0 for the output.

What am I missing?  This is great work BTW!

jblance, 2023-07-04:
Start by adding debug to the command and posting the response

jblance, 2023-07-04:
also try with a `-P PI18` added to the command

vknmakes, 2023-07-05:
Thank you very much.  Did I misread that the default for the mpp's was PI30?  Foolish of me not to have run through the other protocols..

Aha!  -PI18 was it:
mpp-solar -p /dev/hidraw0 -P PI18 --getstatus
Command: ET - Total Generated Energy query
------------------------------------------------------------
Parameter                     	Value           Unit
working_mode                  	Battery mode   	    
grid_voltage                  	0.0            	V   
grid_frequency                	0.0            	Hz  
ac_output_voltage             	120.0          	V   
ac_output_frequency           	59.9           	Hz  
ac_output_apparent_power      	24             	VA  
ac_output_active_power        	10             	W   
output_load_percent           	0              	%   
battery_voltage               	26.0           	V   
battery_voltage_from_scc      	26.0           	V   
battery_voltage_from_scc2     	0.0            	V   
battery_discharge_current     	0              	A   
battery_charging_current      	0              	A   
battery_capacity              	98             	%   
inverter_heat_sink_temperature	25             	°C  
mppt1_charger_temperature     	32             	°C  
mppt2_charger_temperature     	0              	°C  
pv1_input_power               	21             	W   
pv2_input_power               	0              	W   
pv1_input_voltage             	32.0           	V   
pv2_input_voltage             	0.0            	V   
setting_value_configuration_state	Nothing changed	    
mppt1_charger_status          	charging       	    
mppt2_charger_status          	abnormal       	    
load_connection               	connect        	    
battery_power_direction       	discharge      	    
dc/ac_power_direction         	DC-AC          	    
line_power_direction          	donothing      	    
local_parallel_id             	0              	    
total_generated_energy        	24628          	Wh  

But for completeness, here is the debug on QID with default protocol of PI30, which, clearly isn't it..
mpp-solar -p /dev/hidraw0 -c QID -D
2023-07-05 05:40:22,925:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2023-07-05 05:40:22,926:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
2023-07-05 05:40:22,928:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-07-05 05:40:22,928:DEBUG:device:__init__@31: __init__ args ()
2023-07-05 05:40:22,928:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2023-07-05 05:40:22,928:DEBUG:device:get_port_type@59: port matches hidraw
2023-07-05 05:40:22,928:INFO:device:set_port@142: Using hidrawio for communications
2023-07-05 05:40:22,930:DEBUG:device:set_protocol@96: Protocol PI30
2023-07-05 05:40:22,948:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x7f88a10040>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7f88a10070>
2023-07-05 05:40:22,948:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f88a05f70>, 'QID', 'QID', 'screen', None, None)]
2023-07-05 05:40:22,948:INFO:__init__:main@395: Looping 1 commands
2023-07-05 05:40:22,948:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x7f88a10040>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7f88a10070> for command: QID, tag: QID, outputs: screen
2023-07-05 05:40:22,948:INFO:device:run_command@270: Running command QID
2023-07-05 05:40:22,949:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2023-07-05 05:40:22,949:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QID'
2023-07-05 05:40:22,949:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xd6 0xea 0xd6ea
2023-07-05 05:40:22,949:DEBUG:abstractprotocol:get_full_command@38: full command: b'QID\xd6\xea\r'
2023-07-05 05:40:22,949:INFO:device:run_command@296: full command b'QID\xd6\xea\r' for command QID
2023-07-05 05:40:22,949:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'QID'
2023-07-05 05:40:22,949:DEBUG:abstractprotocol:get_command_defn@44: Found command QID in protocol b'PI30'
2023-07-05 05:40:23,008:DEBUG:hidrawio:send_and_receive@29: length of to_send: 6
2023-07-05 05:40:23,008:DEBUG:hidrawio:send_and_receive@36: 1 chunk send
2023-07-05 05:40:23,760:DEBUG:hidrawio:send_and_receive@69: usb response was: b'^0\x1b\xe3\r'
2023-07-05 05:40:23,761:DEBUG:device:run_command@315: Send and Receive Response b'^0\x1b\xe3\r'
2023-07-05 05:40:23,761:INFO:abstractprotocol:decode@175: response passed to decode: b'^0\x1b\xe3\r'
2023-07-05 05:40:23,761:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'^0'
2023-07-05 05:40:23,761:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x1b 0xe3 0x1be3
2023-07-05 05:40:23,761:DEBUG:pi30:check_response_valid@861: CRCs match
2023-07-05 05:40:23,761:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'QID'
2023-07-05 05:40:23,761:DEBUG:abstractprotocol:get_command_defn@44: Found command QID in protocol b'PI30'
2023-07-05 05:40:23,761:INFO:abstractprotocol:decode@215: Processing response of type DEFAULT
[... cut ...]
2023-07-05 05:40:23,764:DEBUG:screen:output@20: kwargs {'data': {'raw_response': ['^0\x1bã\r', ''], '_command': 'QID', '_command_description': 'Device Serial Number inquiry', 'Serial Number': ['0', '']}, 'tag': 'QID', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: QID - Device Serial Number inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
serial_number                 	0              	    
2023-07-05 05:40:23,764:DEBUG:__init__:main@435: Not daemon, so not looping

jblance, 2023-07-06:
You are also using the old version (current is [0.15.62](https://pypi.org/manage/project/mppsolar/release/0.15.62/)
Note mppsolar vs mpp-solar

closed 2023-07-07 by jblance

vknmakes, 2023-07-07:
Aha, thank you.  Running the latest from git at the moment on an rpi.

mpp-solar -p /dev/hidraw0 -P PI18 -c GS
Command: GS - General status query
--------------------------------------------------------------------------------
Parameter                         Value          	Unit
grid_voltage                      0.0            	V   
grid_frequency                    0.0            	Hz  
ac_output_voltage                 120.1          	V   
ac_output_frequency               59.9           	Hz  
ac_output_apparent_power          108            	VA  
ac_output_active_power            91             	W   
output_load_percent               3              	%   
battery_voltage                   26.0           	V   
battery_voltage_from_scc          26.0           	V   
battery_voltage_from_scc2         0.0            	V   
battery_discharge_current         3              	A   
battery_charging_current          0              	A   
battery_capacity                  98             	%   
inverter_heat_sink_temperature    22             	°C  
mppt1_charger_temperature         29             	°C  
mppt2_charger_temperature         0              	°C  
pv1_input_power                   16             	W   
pv2_input_power                   0              	W   
pv1_input_voltage                 30.5           	V   
pv2_input_voltage                 0.0            	V   
setting_value_configuration_state Nothing changed	    
mppt1_charger_status              charging       	    
mppt2_charger_status              abnormal       	    
load_connection                   connect        	    
battery_power_direction           discharge      	    
dc/ac_power_direction             DC-AC          	    
line_power_direction              donothing      	    
local_parallel_id                 0              	    
--------------------------------------------------------------------------------

 $ mpp-solar -v
Solar Device Command Utility, version: 0.16.01