jblance/mpp-solar #369: PI17 Protocol: Problem with BATS

JakobTewes, 2023-07-08:
Heyhey @jblance,

with current version 0.16.01 I got problems with BATS command:

```
# mpp-solar -v
Solar Device Command Utility, version: 0.16.01
# mpp-solar -p /dev/hidraw0 -n 'BPVInverter2' -T 'BPVInverter2' -P PI17 -o screen -c BATS
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/__init__.py", line 424, in main
    results = _device.run_command(command=_command)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/devices/device.py", line 115, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/protocols/abstractprotocol.py", line 342, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
                 ~~~~~~~~~~~~~~^^^^^^^^^^^^^
IndexError: list index out of range
```

Can you please check, if there is an issue with the BATS command (or am I stupid and broke my python)?

Thanks so much

Jakob

jblance, 2023-07-08:
Can you post a debug output (add -D to command)

JakobTewes, 2023-07-09:
Here we go :-)

```
root@bpvraspi:~# mpp-solar -p /dev/hidraw0 -n 'BPVInverter2' -T 'BPVInverter2' -P PI17 -o screen -c BATS -D
2023-07-09 18:25:19,521:INFO:__init__:main@215: Solar Device Command Utility, version: 0.16.01
2023-07-09 18:25:19,522:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-07-09 18:25:19,523:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2023-07-09 18:25:19,523:DEBUG:__init__:main@242: udp port 5555
2023-07-09 18:25:19,523:DEBUG:__init__:main@244: Using Postgres None
2023-07-09 18:25:19,524:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2023-07-09 18:25:19,524:INFO:__init__:main@348: Creating device "BPVInverter2" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2023-07-09 18:25:19,531:DEBUG:__init__:main@352: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-07-09 18:25:19,531:DEBUG:device:__init__@34: __init__ args ()
2023-07-09 18:25:19,532:DEBUG:device:__init__@35: __init__ kwargs {'name': 'BPVInverter2', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7faff47a90>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-07-09 18:25:19,532:DEBUG:__init__:get_port_type@39: port matches hidraw
2023-07-09 18:25:19,533:INFO:__init__:get_port@93: Using hidrawio for communications
2023-07-09 18:25:19,536:DEBUG:__init__:get_protocol@13: Protocol PI17
2023-07-09 18:25:19,769:DEBUG:device:__init__@39: __init__ name BPVInverter2, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7faffa7410>, protocol PI17 protocol handler
2023-07-09 18:25:19,770:DEBUG:__init__:main@406: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fafea1b10>, 'BATS', 'BPVInverter2', 'screen', None, None)]
2023-07-09 18:25:19,771:INFO:__init__:main@411: Looping 1 commands
2023-07-09 18:25:19,771:INFO:__init__:main@421: Getting results from device: mppsolar device - name: BPVInverter2, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7faffa7410>, protocol: PI17 protocol handler for command: BATS, tag: BPVInverter2, outputs: screen
2023-07-09 18:25:19,771:INFO:device:run_command@51: Running command BATS
2023-07-09 18:25:19,772:INFO:pi17:get_full_command@901: Using protocol b'PI17' with 47 commands
2023-07-09 18:25:19,772:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'BATS'
2023-07-09 18:25:19,772:DEBUG:abstractprotocol:get_command_defn@72: Found command BATS in protocol b'PI17'
2023-07-09 18:25:19,773:DEBUG:pi17:get_full_command@916: _pre_cmd: b'^P005BATS'
2023-07-09 18:25:19,773:DEBUG:pi17:get_full_command@917: _prefix: ^P005
2023-07-09 18:25:19,773:DEBUG:pi17:get_full_command@924: full command: b'^P005BATS\r'
2023-07-09 18:25:19,774:INFO:device:run_command@81: full command b'^P005BATS\r' for command BATS
2023-07-09 18:25:19,774:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'BATS'
2023-07-09 18:25:19,774:DEBUG:abstractprotocol:get_command_defn@72: Found command BATS in protocol b'PI17'
2023-07-09 18:25:19,833:DEBUG:hidrawio:send_and_receive@28: length of to_send: 10
2023-07-09 18:25:19,833:DEBUG:hidrawio:send_and_receive@37: multiple chunk send
2023-07-09 18:25:19,833:DEBUG:hidrawio:send_and_receive@44: sending chunk: b'^P005BAT'
2023-07-09 18:25:19,886:DEBUG:hidrawio:send_and_receive@44: sending chunk: b'S\r\x00\x00\x00\x00\x00\x00'
2023-07-09 18:25:22,139:DEBUG:hidrawio:send_and_receive@63: usb response was: b'^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,010,020,020,080,0mr\r'
2023-07-09 18:25:22,140:DEBUG:device:run_command@98: Send and Receive Response b'^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,010,020,020,080,0mr\r'
2023-07-09 18:25:22,140:INFO:abstractprotocol:decode@245: response passed to decode: b'^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,010,020,020,080,0mr\r'
2023-07-09 18:25:22,141:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'BATS'
2023-07-09 18:25:22,141:DEBUG:abstractprotocol:get_command_defn@72: Found command BATS in protocol b'PI17'
2023-07-09 18:25:22,142:INFO:abstractprotocol:decode@289: Processing response of type DEFAULT
2023-07-09 18:25:22,142:DEBUG:abstractprotocol:decode@293: trimmed and split responses: [b'1750', b'0560', b'0540', b'0000', b'060', b'0530', b'0420', b'0480', b'0480', b'0540', b'0', b'', b'', b'0', b'0480', b'000', b'0100', b'0175', b'010', b'020', b'020', b'080', b'0']
2023-07-09 18:25:22,142:INFO:abstractprotocol:decode@300: Processing DEFAULT type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/__init__.py", line 424, in main
    results = _device.run_command(command=_command)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/devices/device.py", line 115, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/protocols/abstractprotocol.py", line 342, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
                 ~~~~~~~~~~~~~~^^^^^^^^^^^^^
IndexError: list index out of range
```

JakobTewes, 2023-07-09:
and (only every third trial)
```
 mpp-solar -p /dev/hidraw0 -n 'BPVInverter2' -T 'BPVInverter2' -P PI17 -o screen -c BATS -D
2023-07-09 18:27:17,968:INFO:__init__:main@215: Solar Device Command Utility, version: 0.16.01
2023-07-09 18:27:17,969:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-07-09 18:27:17,969:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2023-07-09 18:27:17,970:DEBUG:__init__:main@242: udp port 5555
2023-07-09 18:27:17,970:DEBUG:__init__:main@244: Using Postgres None
2023-07-09 18:27:17,970:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2023-07-09 18:27:17,971:INFO:__init__:main@348: Creating device "BPVInverter2" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2023-07-09 18:27:17,977:DEBUG:__init__:main@352: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-07-09 18:27:17,978:DEBUG:device:__init__@34: __init__ args ()
2023-07-09 18:27:17,978:DEBUG:device:__init__@35: __init__ kwargs {'name': 'BPVInverter2', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fb32e9c50>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-07-09 18:27:17,978:DEBUG:__init__:get_port_type@39: port matches hidraw
2023-07-09 18:27:17,979:INFO:__init__:get_port@93: Using hidrawio for communications
2023-07-09 18:27:17,982:DEBUG:__init__:get_protocol@13: Protocol PI17
2023-07-09 18:27:18,214:DEBUG:device:__init__@39: __init__ name BPVInverter2, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fb31d5890>, protocol PI17 protocol handler
2023-07-09 18:27:18,215:DEBUG:__init__:main@406: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fb31d5710>, 'BATS', 'BPVInverter2', 'screen', None, None)]
2023-07-09 18:27:18,215:INFO:__init__:main@411: Looping 1 commands
2023-07-09 18:27:18,215:INFO:__init__:main@421: Getting results from device: mppsolar device - name: BPVInverter2, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fb31d5890>, protocol: PI17 protocol handler for command: BATS, tag: BPVInverter2, outputs: screen
2023-07-09 18:27:18,216:INFO:device:run_command@51: Running command BATS
2023-07-09 18:27:18,216:INFO:pi17:get_full_command@901: Using protocol b'PI17' with 47 commands
2023-07-09 18:27:18,216:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'BATS'
2023-07-09 18:27:18,217:DEBUG:abstractprotocol:get_command_defn@72: Found command BATS in protocol b'PI17'
2023-07-09 18:27:18,217:DEBUG:pi17:get_full_command@916: _pre_cmd: b'^P005BATS'
2023-07-09 18:27:18,217:DEBUG:pi17:get_full_command@917: _prefix: ^P005
2023-07-09 18:27:18,217:DEBUG:pi17:get_full_command@924: full command: b'^P005BATS\r'
2023-07-09 18:27:18,218:INFO:device:run_command@81: full command b'^P005BATS\r' for command BATS
2023-07-09 18:27:18,218:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'BATS'
2023-07-09 18:27:18,218:DEBUG:abstractprotocol:get_command_defn@72: Found command BATS in protocol b'PI17'
2023-07-09 18:27:18,277:DEBUG:hidrawio:send_and_receive@28: length of to_send: 10
2023-07-09 18:27:18,277:DEBUG:hidrawio:send_and_receive@37: multiple chunk send
2023-07-09 18:27:18,277:DEBUG:hidrawio:send_and_receive@44: sending chunk: b'^P005BAT'
2023-07-09 18:27:18,329:DEBUG:hidrawio:send_and_receive@44: sending chunk: b'S\r\x00\x00\x00\x00\x00\x00'
2023-07-09 18:27:20,583:DEBUG:hidrawio:send_and_receive@63: usb response was: b'^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,000,000,000,000,0\xc9\xd9\r'
2023-07-09 18:27:20,584:DEBUG:device:run_command@98: Send and Receive Response b'^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,000,000,000,000,0\xc9\xd9\r'
2023-07-09 18:27:20,584:INFO:abstractprotocol:decode@245: response passed to decode: b'^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,000,000,000,000,0\xc9\xd9\r'
2023-07-09 18:27:20,584:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'BATS'
2023-07-09 18:27:20,585:DEBUG:abstractprotocol:get_command_defn@72: Found command BATS in protocol b'PI17'
2023-07-09 18:27:20,585:INFO:abstractprotocol:decode@289: Processing response of type DEFAULT
2023-07-09 18:27:20,585:DEBUG:abstractprotocol:decode@293: trimmed and split responses: [b'1750', b'0560', b'0540', b'0000', b'060', b'0530', b'0420', b'0480', b'0480', b'0540', b'0', b'', b'', b'0', b'0480', b'000', b'0100', b'0175', b'000', b'000', b'000', b'000', b'0']
2023-07-09 18:27:20,586:INFO:abstractprotocol:decode@300: Processing DEFAULT type responses
2023-07-09 18:27:20,586:INFO:device:run_command@116: Decoded response {'_command': 'BATS', '_command_description': 'Query battery setting', 'raw_response': ['^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,000,000,000,000,0ÉÙ\r', ''], 'Battery maximum charge current': [1750, '0.1A'], 'Battery constant charge voltage(C.V.)': [560, '0.1V'], 'Battery floating charge voltage': [540, '0.1V'], 'Battery stop charger current level in floating charging': [0, '0.1A'], 'Keep charged time of battery catch stopped charging current level': [60, 'Minutes'], 'Battery voltage of recover to charge when battery stop charger in floating charging': [530, '0.1V'], 'Battery under voltage': [420, '0.1V'], 'Battery under voltage release': [480, '0.1V'], 'Battery weak voltag [... cut ...]
battery_voltage_of_cut-off_main_output_in_battery_mode(                             0              	0.1V
battery_voltage_of_re-connecting_main_output_in_battery_mode                        0              	0.1V
unknown_value_in_response_21                                                        000
unknown_value_in_response_22                                                        0
--------------------------------------------------------------------------------

2023-07-09 18:27:20,594:DEBUG:__init__:main@456: Not daemon, so not looping
```

jblance, 2023-07-09:
So the response is b'^D0941750,0560,0540,0000,060,0530,0420,0480,0480,0540,0,,,0,0480,000,0100,0175,**010**,020,020,080,0mr\r'
And the protocol definition (that I have / are using has
[image] which expects a 0 or 1
It _works_ when the response has `000` response for this parameter
So the protocol I have is incorrect for this command on your inverter (also explains the 2 unknown responses)
I have no idea what `000` and `010` could signify or relate to. (parameter or value decode)
What is the model / brand etc of your inverter?
What does the `--getDeviceId` command return?

Any ideas?

JakobTewes, 2023-07-11:
Thanks for elaborating and quite strange @jblance :-). Inverter is made by MPP Solar and called "MPI 12KW WP" respective "MPI 15KW WP", in my case its a "MPI 15KW WP".

Maybe I´d ask MPP Solar for this fields function with MPI 12KW WP and MPI 15KW WP inverters?

Here comes the DeviceID

```
# mpp-solar -p /dev/hidraw0 -n 'BPVInverter2' -T 'BPVInverter2' -P PI17 -o screen --getDeviceId
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value          	Unit
deviceid  17:058
--------------------------------------------------------------------------------
```

riogrande75, 2023-07-12:
The WP versions of the 10/12/15k models run a slightly modified PI17.
The BATS command looks like that:
[image]
Maybe this helps.

I uploaded the latest PI17 protocol spec from 2022-09-22 [here](https://www.mediafire.com/file/jugrzapeh8gocha/Infini-Solar_WP_LV6K_3P_10-15K_and_TWIN_Series_protocol_20220921.xlsx/file).

jblance, 2023-07-13:
Ok, thanks for the protocol doc
What does "Battery under SOC" even mean. 000 and 010 seem to be the 2 values you have mentioned - any thoughts

Will look at adding a new protocol to cover this when i get a chance

jblance, 2023-07-27:
added new protocol `pi17m058` with a modified BATS in v 0.16.10 or late

closed 2023-07-27 by jblance

JakobTewes, 2023-07-28:
Works like a charm! Thanks so much @jblance!