jblance/mpp-solar #389: JK-BMS over TTL

proxer05, 2023-08-14:
Hi, I implemented logic to read data over ttl(https://github.com/proxer05/mpp-solar/commit/666c5c69873df93a9705fd79b4cf2a9a364c0821) but i have problem with decoding. Response length depends on cell count so its not possible to use static "addresses" of data(atleast for reading all data). 
Tested on B1A20S15P(v11 hw)
<details>
<summary>Debug info if needed</summary>

jkbms -b 115200 -P JKSERIAL -D -p /dev/ttyUSB0 
2023-08-14 13:36:08,725:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.10, python version: 3.11.2
2023-08-14 13:36:08,726:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-08-14 13:36:08,726:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2023-08-14 13:36:08,726:DEBUG:__init__:main@253: udp port 5555
2023-08-14 13:36:08,726:DEBUG:__init__:main@255: Using Postgres None
2023-08-14 13:36:08,726:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2023-08-14 13:36:08,726:INFO:__init__:main@359: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JKSERIAL"
2023-08-14 13:36:08,729:DEBUG:__init__:main@363: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2023-08-14 13:36:08,729:DEBUG:device:__init__@34: __init__ args ()
2023-08-14 13:36:08,729:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JKSERIAL', 'baud': 115200, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fb24a1cdf10>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-08-14 13:36:08,729:DEBUG:__init__:get_port_type@69: port matches ttyusb
2023-08-14 13:36:08,729:INFO:__init__:get_port@111: Using serialio for communications
2023-08-14 13:36:08,737:DEBUG:__init__:get_protocol@13: Protocol JKSERIAL
2023-08-14 13:36:08,839:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7fb24a238b50>, protocol JKBMS TTL serial communication protocol handler
2023-08-14 13:36:08,840:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7fb24a238850>, '', '', 'screen', None, None)]
2023-08-14 13:36:08,840:INFO:__init__:main@413: Looping 1 commands
2023-08-14 13:36:08,840:INFO:__init__:main@423: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7fb24a238b50>, protocol: JKBMS TTL serial communication protocol handler for command: , tag: , outputs: screen
2023-08-14 13:36:08,840:INFO:device:run_command@51: Running command 
2023-08-14 13:36:08,840:INFO:jkserial:get_full_command@49: Using protocol b'JKSERIAL' with 1 commands
2023-08-14 13:36:08,840:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2023-08-14 13:36:08,840:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2023-08-14 13:36:08,840:DEBUG:jkserial:get_full_command@91: cmd with crc: bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&')
2023-08-14 13:36:08,840:INFO:device:run_command@81: full command bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&') for command getBalancerData
2023-08-14 13:36:08,840:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2023-08-14 13:36:08,840:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2023-08-14 13:36:08,841:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2023-08-14 13:36:08,842:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-08-14 13:36:08,944:DEBUG:serialio:send_and_receive@30: serial response was: b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r'
2023-08-14 13:36:09,059:DEBUG:device:run_command@98: Send and Receive Response b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r'
2023-08-14 13:36:09,060:INFO:abstractprotocol:decode@246: response passed to decode: b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r'
2023-08-14 13:36:09,060:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2023-08-14 13:36:09,060:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2023-08-14 13:36:09,060:INFO:abstractprotocol:decode@290: Processing response of type POSITIONAL
2023-08-14 13:36:09,061:DEBUG:jkserial:get_responses@119: get_responses: responses [b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r']
2023-08-14 13:36:09,061:DEBUG:abstractprotocol:decode@294: trimmed and split responses: [b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r']
2023-08-14 13:36:09,061:DEBUG:abstractprotocol:decode@495: Processing POSITIONAL type responses
2023-08-14 13:36:09,061:DEBUG:abstractprotocol:decode@522: Got defn ['Hex2Str', 50, 'response', '']
2023-08-14 13:36:09,061:DEBUG:abstractprotocol:process_response@118: Processing data_type: Hex2Str for data_name: response, raw_value b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r'
2023-08-14 13:36:09,062:DEBUG:abstractprotocol:process_response@207: Processing format string Hex2Str(raw_value)
2023-08-14 13:36:09,062:DEBUG:protocol_helpers:Hex2Str@105: Hex b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r' decoded to 4e57011b00000000030001792a010f90020f92030f97040f90050f93060f93070f92080f90090f940a0f900b0f920c0f910d
2023-08-14 13:36:09,062:INFO:device:run_command@116: Decoded response {'_command': 'getBalancerData', '_command_description': 'Get Balancer Data', 'raw_response': ['NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r', ''], 'response': ['4e57011b00000000030001792a010f90020f92030f97040f90050f93060f93070f92080f90090f940a0f900b0f920c0f910d', '']}
2023-08-14 13:36:09,062:DEBUG:__init__:main@427: results: {'_command': 'getBalancerData', '_command_description': 'Get Balancer Data', 'raw_response': ['NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r', ''], 'response': ['4e57011b00000000030001792a010f90020f92030f97040f90050f93060f93070f92080f90090f940a0f900b0f920c0f910d', '']}
2023-08-14 13:36:09,063:INFO:__init__:get_output@40: attempting to create output processor: screen
2023-08-14 13:36:09,065:DEBUG:screen:__init__@15: processor.screen __init__ kwargs {}
2023-08-14 13:36:09,065:DEBUG:__init__:main@433: Using output filter: None
2023-08-14 13:36:09,065:INFO:screen:output@21: Using output processor: screen
2023-08-14 13:36:09,065:DEBUG:screen:output@22: kwargs {'data': {'_command': 'getBalancerData', '_command_description': 'Get Balancer Data', 'raw_response': ['NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x92\x03\x0f\x97\x04\x0f\x90\x05\x0f\x93\x06\x0f\x93\x07\x0f\x92\x08\x0f\x90\t\x0f\x94\n\x0f\x90\x0b\x0f\x92\x0c\x0f\x91\r', ''], 'response': ['4e57011b00000000030001792a010f90020f92030f97040f90050f93060f93070f92080f90090f940a0f900b0f920c0f910d', '']}, 'tag': '', 'name': 'unnamed', 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fb24a1cdf10>, 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'mqtt_topic': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
[... cut ...] 60f93070f92080f90090f940a0f900b0f920c0f910d', '']}
Command: getBalancerData - Get Balancer Data
--------------------------------------------------------------------------------
Parameter Value                 Unit
response  4e57011b00000000030001792a010f90020f92030f97040f90050f93060f93070f92080f90090f940a0f900b0f920c0f910d      
--------------------------------------------------------------------------------

2023-08-14 13:36:09,066:DEBUG:__init__:main@458: Not daemon, so not looping

</details>

jblance, 2023-08-14:
i think the data is including a `\r` so the standard serial reader is truncating the data (its a readuntil you find \r)
can you try 
`jkbms -b 115200 -P JKSERIAL -D -p /dev/ttyUSB0 --porttype daly`

jblance, 2023-08-15:
yep definitely a premature data termination due to the serial input read logic, cell number 13 (`0d` in hex) is interpreted as end of data by the serial readline logic (porttype daly doesnt do this)

partial decode below seem to work (assuming there are 14 cells in your setup)
```
4e57011b00000000030001792a010f90020f92030f97040f90050f93060f93070f92080f90090f940a0f900b0f920c0f910d
bytes returned = 50 

1 start (2) 4e57
2 length (2) 01 1b -> 283
3 terminal #(4) 00 00 00 00
4 command word (1) 03 -> read the data
5 frame source (1) 00 -> 0.BMS,1. Bluetooth,2.GPS,3. PC
6 transport type (1) 01 -> 0 for request frame, 1 for reply frame. 2 for active reporting 
7 data (xx) 79 2a 010f90 020f92 030f97 040f90 050f93 060f93 070f92 080f90 090f94 0a0f90 0b0f92 0c0f91 0d
8 record # (4)  
9 end of data (1 - 0x68) 
10 checksum (4)  

Data decode
79 2a 010f90 020f92 030f97 040f90 050f93 060f93 070f92 080f90 090f94 0a0f90 0b0f92 0c0f91 0d
1 0x79 'Single battery voltage'
2 2a byte length = 42 or 14 cells?
3 #cells*3 = cell#, voltage, eg
  01 0f90  02 0f92  03 0f97
  #1 3984  #2 3986  #3 3991
```

proxer05, 2023-08-15:
Yeh, now it works.  Do you have idea how to adjust response addresses based on cell count as currently my setup will only work on 14 cells setup? Thanks anyway.

<details>
<summary>Logs</summary>
(current and capacity needs other formula not important atm)
jkbms -b 115200 -P JKSERIAL -D -p /dev/ttyUSB0 --porttype daly
2023-08-15 12:02:55,863:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.10, python version: 3.11.2
2023-08-15 12:02:55,864:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-08-15 12:02:55,864:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2023-08-15 12:02:55,864:DEBUG:__init__:main@253: udp port 5555
2023-08-15 12:02:55,864:DEBUG:__init__:main@255: Using Postgres None
2023-08-15 12:02:55,864:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2023-08-15 12:02:55,864:INFO:__init__:main@359: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=daly)" using protocol "JKSERIAL"
2023-08-15 12:02:55,867:DEBUG:__init__:main@363: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2023-08-15 12:02:55,867:DEBUG:device:__init__@34: __init__ args ()
2023-08-15 12:02:55,867:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JKSERIAL', 'baud': 115200, 'porttype': 'daly', 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fd8223c9d50>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-08-15 12:02:55,867:INFO:__init__:get_port@81: Port overide - using port 'daly'
2023-08-15 12:02:55,867:DEBUG:__init__:get_port_type@57: port matches daly
2023-08-15 12:02:55,867:INFO:__init__:get_port@117: Using dalyserialio for communications
2023-08-15 12:02:55,873:DEBUG:__init__:get_protocol@13: Protocol JKSERIAL
2023-08-15 12:02:55,978:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.dalyserialio.DalySerialIO object at 0x7fd8223c9f50>, protocol JKBMS TTL serial communication protocol handler
2023-08-15 12:02:55,978:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7fd822434a90>, '', '', 'screen', None, None)]
2023-08-15 12:02:55,978:INFO:__init__:main@413: Looping 1 commands
2023-08-15 12:02:55,978:INFO:__init__:main@423: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.dalyserialio.DalySerialIO object at 0x7fd8223c9f50>, protocol: JKBMS TTL serial communication protocol handler for command: , tag: , outputs: screen
2023-08-15 12:02:55,978:INFO:device:run_command@51: Running command 
2023-08-15 12:02:55,978:INFO:jkserial:get_full_command@103: Using protocol b'JKSERIAL' with 1 commands
2023-08-15 12:02:55,978:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2023-08-15 12:02:55,978:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2023-08-15 12:02:55,979:DEBUG:jkserial:get_full_command@145: cmd with crc: bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&')
2023-08-15 12:02:55,979:INFO:device:run_command@81: full command bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&') for command getBalancerData
2023-08-15 12:02:55,979:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2023-08-15 12:02:55,979:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2023-08-15 12:02:55,979:DEBUG:dalyserialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2023-08-15 12:02:55,981:DEBUG:dalyserialio:send_and_receive@22: Executing command via dalyserialio...
2023-08-15 12:02:56,481:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 285
2023-08-15 12:02:56,982:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 0
2023-08-15 12:02:56,982:DEBUG:dalyserialio:send_and_receive@38: serial response was: b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x91\x03\x0f\x94\x04\x0f\x8e\x05\x0f\x92\x06\x0f\x91\x07\x0f\x91\x08\x0f\x91\t\x0f\x93\n\x0f\x8e\x0b\x0f\x91\x0c\x0f\x90\r\x0f\x90\x0e\x0f\x8d\x80\x00!\x81\x00\x1c\x82\x00\x1e\x83\x15\xca\x84\x81\xc5\x85d\x86\x02\x87\x00\x19\x89\x00\x00\x16\xda\x8a\x00\x0e\x8b\x00\x00\x8c\x00\x03\x8e\x16\xb2\x8f\x10\xf4\x90\x106\x91\x10\x04\x92\x00\x05\x93\x0c\x1c\x94\x0c\x80\x95\x00\x05\x96\x01,\x97\x00n\x98\x01,\x99\x00U\x9a\x00\x1e\x9b\x0b\xb8\x9c\x002\x9d\x01\x9e\x00Z\x9f\x00F\xa0\x00d\xa1\x00d\xa2\x00\x14\xa3\x00<\xa4\x00<\xa5\x00\x01\xa6\x00\x03\xa7\xff\xec\xa8\xff\xf6\xa9\x0e\xaa\x00\x00\x00\xea\xab\x01\xac\x01\xad\x047\xae\x01\xaf\x01\xb0\x00\n\xb1\x14\xb2123456\x00\x00\x00\x00\xb3\x00\xb4Input Us\xb52306\xb6\x00\x01\x82\xe3\xb711.XW_S11.261__\xb8\x00\xb9\x00\x00\x00\xea\xbaInput UserdaJK_B1A20S15P\xc0\x01\x00\x00\x00\x00h\x00\x00Q\xd6'
2023-08-15 12:02:57,095:DEBUG:device:run_command@98: Send and Receive Response b'NW\x01\x1b\x00\x00\x00\x00\x03\x00\x01y*\x01\x0f\x90\x02\x0f\x91\x03\x0f\x94\x04\x0f\x8e\x05\x0f\x92\x06\x0f\x91\x07\x0f\x91\x08\x0f\x91\t\x0f\x93\n\x0f\x8e\x0b\x0f\x91\x0c\x0f\x90\r\x0f\x90\x0e\x0f\x8d\x80\x00!\x81\x00\x1c\x82\x00\x1e\x83\x15\xca\x84\x81\xc5\x85d\x86\x02\x87\x00\x19\x89\x00\x00\x16\xda\x8a\x00\x0e\x8b\x00\x00\x8c\x00\x03\x8e\x16\xb2\x8f\x10\xf4\x90\x106\x91\x10\x04\x92\x00\x05\x93\x0c\x1c\x94\x0c\x80\x95\x00\x05\x96\x01,\x97\x00n\x98\x01,\x99\x00U\x9a\x00\x1e\x9b\x0b\xb8\x9c\x002\x9d\x01\x9e\x00Z\x9f\x00F\xa0\x00d\xa1\x00d\xa2\x00\x [... cut ...]
balance_current                 -32.315         A   
percent_remaining               100             %   
cycle_count                     25                  
total_capacity                  000016da            
total_number_of_battery_strings 14                  
battery_warning_message         0000                
battery_status_information      0003                
--------------------------------------------------------------------------------

2023-08-15 12:02:57,154:DEBUG:__init__:main@458: Not daemon, so not looping

</details>

geofreybingachie, 2023-08-17:
I am very happy with this USB TTL option .I managed to connect my JK BMS but i am getting some data that is wrong .

Command: getBalancerData - Get Balancer Data
--------------------------------------------------------------------------------
Parameter                  Value                Unit
cells_connected            8.0
voltage_cell01             3.412                V
voltage_cell02             3.416                V
voltage_cell03             3.416                V
voltage_cell04             3.416                V
voltage_cell05             3.414                V
voltage_cell06             3.417                V
voltage_cell07             3.416                V
voltage_cell08             3.417                V
voltage_cell09             0.029                V
voltage_cell10             0.026                V
voltage_cell11             0.026                V
voltage_cell12             2.732                V
voltage_cell13             -32.759              V
voltage_cell14             25.478               V
mos_temp                   -30976               °C
battery_t1                 -30464               °C
battery_t2                 3698                 °C
battery_voltage            0.08                 V
balance_current            0.0                  A
percent_remain             0                    %
cycle_count                26767
total_capacity             d0900e42
battery_warning_message    0005
battery_status_information 09c4

proxer05, 2023-08-17:
> I am very happy with this USB TTL option .I managed to connect my JK BMS but i am getting some data that is wrong .
>
> Command: getBalancerData - Get Balancer Data [...]

As I mentioned previously currently there is no cell count detection. You need to adjust it manually in protocol file.
@jblance can you add comment with this information in commit?

geofreybingachie, 2023-08-17:
> mos_temp -30976 °C
> battery_t1 -30464 °C
> battery_t2 3698 °C
> battery_voltage 0.08 V
> balance_current 0.0 A
> percent_remain 0 %
> cycle_count 26767
> total_capacity d0900e42

The above are what i am interested to get correct

proxer05, 2023-08-17:
Clone this repo and edit mppsolar/protocols/jkserial.py(delete cell9-14) then install by running pip install -e . in mpp-solar folder.

jblance, 2023-08-17:
Can you post a debug run (then I'll 2 different cell count examples to work
from)

proxer05, 2023-08-17:
Also for current there is formula needed (1000-BigHex2Short)*0.01

Apprisco, 2023-08-18:
I don't mean to hijack the channel, but is it possible to directly grab data off of the JKBMS using python code using this library, from TTL to USB?
Not just solar assistant or whatever.
I have the B2A24S20P

jblance, 2023-08-18:
Yes. There are a variety of output methods including ones that just print
the results (actually thebl default does this)

proxer05, 2023-08-18:
> I don't mean to hijack the channel, but is it possible to directly grab data off of the JKBMS using python code using this library, from TTL to USB?
> Not just solar assistant or whatever.
> I have the B2A24S20P

Here you have full list of output options https://github.com/jblance/mpp-solar/wiki/Detailed-Usage#list-available-output-processors

proxer05, 2023-08-18:
> I don't mean to hijack the channel, but is it possible to directly grab data off of the JKBMS using python code using this library, from TTL to USB? Not just solar assistant or whatever. I have the B2A24S20P

Can you post a debug run too? It would help in development as currently only 14 cell setup report correct values.

Apprisco, 2023-08-19:
I should just try this out, but is mpp-solar and jkbms a python importable library now?
And, is it possible to communicate over USB to TTL? Or which connection protocol do you recommend? (Clearly TTL has some issues, hence the creation of this issue.)

jblance, 2024-02-22:
logic to address this is implemented in powermon jkserial

closed 2024-02-22 by jblance