jblance/mpp-solar #392: Home Assistant Discovered entities with a name that starts with the device name

frederickjh, 2023-08-22:
After updating Home Assistant to version 2023.8.3, I noticed the following notification in Home Assistant:

> Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.
> 
> https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

The bar at the top of the notification says: " This stops working in version 2024.2.0. Please address before upgrading.

Here is the list of affected entitites from the notification:

binary_sensor.pip8048max_is_load_on
binary_sensor.pip8048max_is_dustproof_installed
sensor.pip8048max_pv1_input_current
sensor.pip8048max_battery_charging_current
binary_sensor.pip8048max_is_scc_firmware_updated
sensor.pip8048max_ac_output_apparent_power
sensor.pip8048max_ac_output_frequency
sensor.pip8048max_battery_voltage_from_scc
sensor.pip8048max_pv1_input_voltage
binary_sensor.pip8048max_is_configuration_changed
sensor.pip8048max_ac_output_voltage
sensor.pip8048max_ac_input_voltage
sensor.pip8048max_ac_output_load
sensor.pip8048max_total_output_load_energy
sensor.pip8048max_eeprom_version
sensor.pip8048max_ac_input_frequency
sensor.pip8048max_ac_output_active_power
binary_sensor.pip8048max_is_battery_voltage_to_steady_while_charging
sensor.pip8048max_battery_voltage_offset_for_fans_on
binary_sensor.pip8048max_is_charging_to_float
sensor.pip8048max_total_pv_generated_energy
binary_sensor.pip8048max_is_ac_charging_on
sensor.pip8048max_pv1_charging_power
binary_sensor.pip8048max_is_switched_on
binary_sensor.pip8048max_is_charging_on
binary_sensor.pip8048max_is_sbu_priority_version_added
sensor.pip8048max_battery_discharge_current
binary_sensor.pip8048max_is_scc_charging_on
sensor.pip8048max_inverter_heat_sink_temperature
sensor.pip8048max_battery_capacity
sensor.pip8048max_bus_voltage
sensor.pip8048max_battery_voltage

Saentist, 2023-08-22:
Use **hassd_mqtt** output

```
mpp-solar -p /dev/hidraw1 -o hassd_mqtt -c QID#QDI#QMOD#QPI#QPIGS#QPIRI#QFLAG -q 192.168.1.12 --mqttuser user --mqttpass 'pass' -n pip8048max
```

frederickjh, 2023-08-22:
I am using `hassd_mqtt` output. `config/mpp-solar.conf` for my docker container looks like this:

```
[SETUP]
pause=15
mqtt_broker=192.168.7.100
mqtt_port=1883
mqtt_user=mqtt_user
mqtt_pass=xxxxxxxxxxxx

[PIP8048MAX]
protocol=PI30MAX
type=mppsolar
port=/dev/hidraw1
baud=2400
command=QPIGS#QLT#QET
tag=PIP8048MAX
name=PIP8048MAX
outputs=hassd_mqtt
```

Solar Device Command Utility, version: 0.15.57

pveredas, 2023-10-19:
Hi [frederickjh](https://github.com/frederickjh), is the integration with homeassistant working? 
I have the same outputs configured as you but does not any sensor registered

frederickjh, 2023-10-21:
@pveredas yes it is working, this is off topic, but try the following. 

* Are you using the USB port?
  * If so is `/dev/hidraw1` the correct port? ( could be hidraw0, hidraw2, etc. )
  * If you are using the serial port the port will be different. Something like `/dev/USBtty0`.
* Are you connecting to the same inverter, PIP8048MAX, that I am?
  * If not then then you may need a different protocol.
* The mqtt settings should match you mqtt broker's settings.

pveredas, 2023-10-21:
> @pveredas yes it is working, this is off topic, but try the following.
>
>
>
> * Are you using the USB port?
>
>   * If so is `/dev/hidraw1` the correct port? ( could be hidraw0, hidraw2, etc. )
>
>   * If [...]

Thanks for your response. All the settings where good as I saw them in screen output and mqtt server. It took a day for the sensors to register in HA. Now it's all good.

jblance, 2024-02-22:
sorry, just looking into this further, the hassd output uses the tag value as part of the name, so if you change it to something other than the device name (from the name `-n` value) it will fix this error