jblance/mpp-solar #394: Documentation for powermon in container is blank

frederickjh, 2023-08-22:
Hello, 
I am interest to switch from using the `mpp-solar` command to the `powermon` in my docker container. Looking at the available options for `powermon`, it looks like I could monitor and also setup morning and evening routines to be run.

However, currently in the documentation there is a title for "powermon under docker" but no text following it.

If this is currently possible any advice / documentation you can provide would be appreciated. 
Thanks!

frederickjh, 2023-08-23:
I took a crack at trying to put together a powermon.yaml configuration file, however I am noticing that the documentation does not always agree with the files in `mpp-solar/powermon/config`.  `powermon -h` does not offer any help on the configuration file either. Still not sure if I have this correct. Just wondering if `powermon` is a work in progress at this point.

jblance, 2023-08-23:
Yes it's very much a wip and is changing frequently

There are some ssmple yamls on github if you want to try but caution is
advised (and limited protocols etc have been ported)

frederickjh, 2023-08-24:
Okay understood. I am guessing that the sample configuration files are more up to date than the `powermon` config documentation.

frederickjh, 2023-08-24:
I was hopeful as the pi30max protocol that my inverter uses has been implemented. However I got an error:

```
$ powermon -C powermon-test.yaml 
Traceback (most recent call last):
  File "/usr/local/bin/powermon", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/powermon/__init__.py", line 152, in main
    schedule = Schedule.parseScheduleConfig(scheduling_config, device, mqtt_broker)
  File "/usr/local/lib/python3.10/dist-packages/powermon/libs/schedule.py", line 129, in parseScheduleConfig
    for schedule in config["schedules"]:
KeyError: 'schedules'
```

Sanitized `powermon-test.yaml ` that I was used:

```
device:
  name: Inverter_1
  id: 92932108104635
  model: PIP8048MAX
  manufacturer: MPP-Solar
  port:
    type: usb
    path: /dev/hidraw1
    baud: 2400
    protocol: PI30MAX
commands:
- command: QPIGS
  trigger:
    every: 15
  outputs:
  - type: screen
    format: hass
- command: QLT
  trigger:
    every: 15
  outputs:
  - type: screen
    format: hass
- command: QET
  trigger:
    every: 15
  outputs:
  - type: screen
    format: hass
mqttbroker:
  name: 192.168.7.100
  port: 1883
  username: username
  password: xxxxxxxxxxxxxxxxxxxxx
  adhoc_commands:
    topic: Inverter_1/commands
    outputs:
    - name: screen
      tag: Inverter_1
```

Any ideas as to what I did not get correct in the config?

frederickjh, 2023-08-25:
@jblance Thanks for asking. I was testing on the host and forgot to update so I was using 0.15.57. Updated to 0.16.10 and the output to the screen seems to be working as I am seeing what look like data from the inverter but I am seeing an error show up  in the output:

```
ERROR:usbport:send_and_receive@58: USB open error: 'USBPort' object has no attribute '_device'
```
I need to test more.

closed 2023-10-19 by jblance

frederickjh, 2023-10-23:
@jblance I see that you closed this but I do not find any documentation for `powermon`. I updated to the most recent version of mppsolar, 0.16.11-dev,  that was just released and it looks like the configuration file syntax for `powermon` has changed a lot. I end up with a lot of **type=missing** errors.

```
powermon -D -V -C config/powermon-test.yaml                                                                                       418ms  [Sat Oct 21 2023 21:35:05 HST]
2023-10-21 21:35:31,805:INFO:__init__:main@116: Power Device Monitoring Utility, version: 0.16.11-dev, python version: 3.10.12
2023-10-21 21:35:31,805:INFO:__init__:main@122: Using config file: config/powermon-test.yaml
Config validation failed
config={'device': {'name': 'Inverter_1', 'id': '92932108104635', 'model': 'PIP8048MAX', 'manufacturer': 'MPP-Solar', 'port': {'type': 'usb', 'path': '/dev/hidraw1', 'baud': 2400, 'protocol': 'PI30MAX'}}, 'commands': [{'command': 'QPIGS', 'type': 'basic', 'trigger': {'every': 15}, 'outputs': [{'type': 'screen', 'format': {'type': 'hass'}}]}], 'mqttbroker': {'name': '192.168.7.100', 'port': 1883, 'username': 'mqtt_user', 'password': 'hthkMm3iC4xWwn-JyS7K'}, 'debuglevel': 10}
3 validation errors for ConfigModel
config.api
  Field required [type=missing, input_value={'device': {'name': 'Inve...S7K'}, 'debuglevel': 10}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.4/v/missing
config.daemon
  Field required [type=missing, input_value={'device': {'name': 'Inve...S7K'}, 'debuglevel': 10}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.4/v/missing
config.loop
  Field required [type=missing, input_value={'device': {'name': 'Inve...S7K'}, 'debuglevel': 10}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.4/v/missing
```
### config/powermon-test.yaml 
```
device:
  name: Inverter_1
  id: "245897456325784"
  model: PIP8048MAX
  manufacturer: MPP-Solar
  port:
    type: usb
    path: /dev/hidraw1
    baud: 2400
    protocol: PI30MAX
commands:
- command: QPIGS
  type: basic
  trigger:
    every: 15
  outputs:
  - type: screen
    format:
      type: hass
mqttbroker:
  name: 192.168.70.10
  port: 1883
  username: username
  password: xxxxxxxx
```

I had more "missing" errors when I started but managed to take care of some of them.

I also came across something you wrote in a [different issue](https://github.com/jblance/mpp-solar/issues/288) back in the middle of January

>[ working in powermon command of mppsolar - requires a different config and only supports serial devices at the moment](https://github.com/jblance/mpp-solar/issues/288#issuecomment-1384768506)

Is it still the case that only serial connections are supported in `powermon`, as that maybe the problem I am facing as I was using a USB connection?

rossandrews, 2023-10-23:
I use USB without issues. Pydantic is a data validation library and I think something changed with the latest version since I have also found it be far more strict recently. I will add some sensible defaults for the api, daemon and loop config so they don't need to be included to pass validation.

rossandrews, 2023-10-23:
Here is my PR to help fix the issues your are having. https://github.com/jblance/mpp-solar/pull/416

jblance, 2023-10-23:
Also powermon is still under heavy development, so isnt supported or recommended as yet

frederickjh, 2023-10-24:
Thanks for the replies. I will have another look at `powermon` after it is a bit more fully baked. 

For now I have my own solution to monitoring and comands using `mpp-solar`. I have a docker container doing monitoring via the USB port on the inverter.

For commands, I have created configuration files for the commands that I want to run. This would be via the serial port on the inverter. With ssh key login setup from Home Assistant to the computer connected to the serial port so it can login and run commands. I have setup the commands that I want to run as shell scripts in HA. It then provides these as services that can be run from automations, etc.