jblance/mpp-solar #397: Egg version install on Ubuntu 20.04

Saentist, 2023-09-04:
Is there some dependency or python version problem?

```bash
pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"

Obtaining mppsolar from git+https://github.com/jblance/mpp-solar.git#egg=mppsolar
  Updating ./src/mppsolar clone
  Running command git fetch -q --tags
  Running command git reset --hard -q 2d1020e1b34ad05aaf6f2a07549fda2900c3499e
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting PyYAML>=6 (from mppsolar)
  Obtaining dependency information for PyYAML>=6 from https://files.pythonhosted.org/packages/7b/5e/efd033ab7199a0b2044dab3b9f7a4f6670e6a52c089de572e928d2873b06/PyYAML-6.0.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata
  Using cached PyYAML-6.0.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.1 kB)
Collecting fastapi>=0.68.0 (from mppsolar)
  Obtaining dependency information for fastapi>=0.68.0 from https://files.pythonhosted.org/packages/76/e5/ca411b260caa4e72f9ac5482f331fe74fd4eb5b97aa74d1d2806ccf07e2c/fastapi-0.103.1-py3-none-any.whl.metadata
  Downloading fastapi-0.103.1-py3-none-any.whl.metadata (24 kB)
Collecting fastapi-mqtt>=1.1 (from mppsolar)
  Obtaining dependency information for fastapi-mqtt>=1.1 from https://files.pythonhosted.org/packages/39/77/357a0e23abce66eea546b2cefd24134de39f5810892e54217cfcc20deb29/fastapi_mqtt-1.1.1-py3-none-any.whl.metadata
  Using cached fastapi_mqtt-1.1.1-py3-none-any.whl.metadata (5.0 kB)
Collecting jinja2>=3.0.0 (from mppsolar)
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting paho-mqtt>=1.6 (from mppsolar)
  Using cached paho-mqtt-1.6.1.tar.gz (99 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      ERROR: Can not execute `setup.py` since setuptools is not available in the build environment.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

jblance, 2023-09-12:
yes - I have moved to poetry for dependancy and build management (and removed the setup.py)

jblance, 2023-09-13:
hmm, it might be that pip needs an upgrade? what version do you have?
My test in a codespace seems to allow the install
```bash
[notice] A new release of pip available: 22.2.2 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
(venv) @jblance ➜ /workspaces/mpp-solar (master) $ pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"
Obtaining mppsolar from git+https://github.com/jblance/mpp-solar.git#egg=mppsolar
  Cloning https://github.com/jblance/mpp-solar.git to ./venv/src/mppsolar
  Running command git clone --filter=blob:none --quiet https://github.com/jblance/mpp-solar.git /workspaces/mpp-solar/venv/src/mppsolar
  Resolved https://github.com/jblance/mpp-solar.git to commit 0a6d84befb4300b5b2487475bf4daf8a9da50c77
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting paho-mqtt>=1.6
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 99.4/99.4 kB 3.8 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting uvicorn>=0.15.0
  Downloading uvicorn-0.23.2-py3-none-any.whl (59 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 59.5/59.5 kB 2.5 MB/s eta 0:00:00
Collecting PyYAML>=6
  Downloading PyYAML-6.0.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (705 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 705.5/705.5 kB 18.6 MB/s eta 0:00:00
Collecting jinja2>=3.0.0
  Downloading Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 kB 6.3 MB/s eta 0:00:00
Collecting sqlalchemy<3.0.0,>=2.0.15
  Downloading SQLAlchemy-2.0.20-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.0/3.0 MB 47.6 MB/s eta 0:00:00
Collecting strenum<0.5.0,>=0.4.10
  Downloading StrEnum-0.4.15-py3-none-any.whl (8.9 kB)
Collecting fastapi-mqtt>=1.1
  Downloading fastapi_mqtt-1.1.1-py3-none-any.whl (8.3 kB)
Collecting pydantic>=1.8.0
  Downloading pydantic-2.3.0-py3-none-any.whl (374 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 374.5/374.5 kB 14.7 MB/s eta 0:00:00
Collecting pyserial>3
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 4.5 MB/s eta 0:00:00
Collecting fastapi>=0.68.0
  Downloading fastapi-0.103.1-py3-none-any.whl (66 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 66.2/66.2 kB 3.2 MB/s eta 0:00:00
Collecting starlette<0.28.0,>=0.27.0
  Downloading starlette-0.27.0-py3-none-any.whl (66 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 67.0/67.0 kB 3.3 MB/s eta 0:00:00
Collecting typing-extensions>=4.5.0
  Downloading typing_extensions-4.7.1-py3-none-any.whl (33 kB)
Collecting anyio<4.0.0,>=3.7.1
  Downloading anyio-3.7.1-py3-none-any.whl (80 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 80.9/80.9 kB 3.9 MB/s eta 0:00:00
Collecting pydantic>=1.8.0
  Downloading pydantic-1.10.12-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.1/3.1 MB 60.2 MB/s eta 0:00:00
Collecting black<24.0.0,>=23.3.0
  Downloading black-23.9.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (1.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.6/1.6 MB 46.1 MB/s eta 0:00:00
Collecting gmqtt<0.7.0,>=0.6.11
  Downloading gmqtt-0.6.12-py3-none-any.whl (30 kB)
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (25 kB)
Collecting greenlet!=0.4.17
  Downloading greenlet-2.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (613 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 613.7/613.7 kB 24.6 MB/s eta 0:00:00
Collecting h11>=0.8
  Downloading h11-0.14.0-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.3/58.3 kB 2.4 MB/s eta 0:00:00
Collecting click>=7.0
  Downloading click-8.1.7-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.9/97.9 kB 4.9 MB/s eta 0:00:00
[... 2 lines cut ...]
Collecting idna>=2.8
  Downloading idna-3.4-py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.5/61.5 kB 3.0 MB/s eta 0:00:00
Collecting sniffio>=1.1
  Downloading sniffio-1.3.0-py3-none-any.whl (10 kB)
Collecting pathspec>=0.9.0
  Downloading pathspec-0.11.2-py3-none-any.whl (29 kB)
Collecting mypy-extensions>=0.4.3
  Downloading mypy_extensions-1.0.0-py3-none-any.whl (4.7 kB)
Collecting platformdirs>=2
  Downloading platformdirs-3.10.0-py3-none-any.whl (17 kB)
Collecting packaging>=22.0
  Using cached packaging-23.1-py3-none-any.whl (48 kB)
Collecting tomli>=1.1.0
  Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
Using legacy 'setup.py install' for paho-mqtt, since package 'wheel' is not installed.
Building wheels for collected packages: mppsolar
  Building editable for mppsolar (pyproject.toml) ... done
  Created wheel for mppsolar: filename=mppsolar-0.16.10-py3-none-any.whl size=2949 sha256=499edb21c9cf3c25c9eaaeae7f8eae7f735a5c9cda633aa55bfba925795ba365
  Stored in directory: /tmp/pip-ephem-wheel-cache-tpgemp7j/wheels/54/d7/96/0eb85c933282893e539679dc49df59837bc006928b79f2fadf
Successfully built mppsolar
Installing collected packages: strenum, pyserial, paho-mqtt, typing-extensions, tomli, sniffio, PyYAML, platformdirs, pathspec, packaging, mypy-extensions, MarkupSafe, idna, h11, greenlet, gmqtt, exceptio [... cut ...]
  Running setup.py install for paho-mqtt ... done
Successfully installed MarkupSafe-2.1.3 PyYAML-6.0.1 anyio-3.7.1 black-23.9.1 click-8.1.7 exceptiongroup-1.1.3 fastapi-0.103.1 fastapi-mqtt-1.1.1 gmqtt-0.6.12 greenlet-2.0.2 h11-0.14.0 idna-3.4 jinja2-3.1.2 mppsolar-0.16.10 mypy-extensions-1.0.0 packaging-23.1 paho-mqtt-1.6.1 pathspec-0.11.2 platformdirs-3.10.0 pydantic-1.10.12 pyserial-3.5 sniffio-1.3.0 sqlalchemy-2.0.20 starlette-0.27.0 strenum-0.4.15 tomli-2.0.1 typing-extensions-4.7.1 uvicorn-0.23.2
```

Saentist, 2023-09-14:
```shell
pip -V
pip 23.2.1 from /usr/local/lib/python3.11/dist-packages/pip (python 3.11)

mpp-solar -v
Solar Device Command Utility, version: 0.16.11-dev, python version: 3.8.10
```

jblance, 2023-10-19:
thats the latest version number, are you still having a problem?

Saentist, 2023-10-20:
Yes

```shell
# pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"
Obtaining mppsolar from git+https://github.com/jblance/mpp-solar.git#egg=mppsolar
  Updating ./src/mppsolar clone
  Running command git fetch -q --tags
  Running command git reset --hard -q c3708f7b320dc8e90c933f1c4243acbdf26a4d16
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting PyYAML>=6 (from mppsolar)
  Downloading PyYAML-6.0.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.1 kB)
Collecting fastapi>=0.68.0 (from mppsolar)
  Downloading fastapi-0.104.0-py3-none-any.whl.metadata (24 kB)
Collecting fastapi-mqtt>=1.1 (from mppsolar)
  Downloading fastapi_mqtt-2.0.0-py3-none-any.whl.metadata (5.1 kB)
Collecting jinja2>=3.0.0 (from mppsolar)
  Downloading Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 kB 1.2 MB/s eta 0:00:00
Collecting paho-mqtt>=1.6 (from mppsolar)
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 99.4/99.4 kB 2.8 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      ERROR: Can not execute `setup.py` since setuptools is not available in the build environment.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```

jblance, 2023-12-13:
> 
> ```shell
> # pip -V
> pip 23.3 from /usr/local/lib/python3.11/dist-packages/pip (python 3.11)
> # ip -V
> pip 23.3 from /usr/local/lib/python3.11/dist-packages/pip (python 3.11)
> # mpp-solar -v
> Solar Device Command Utility, version: 0.16.11-dev, python version: 3.8.10
> ```
> 
> Why **mpp-solar** show python version **3.8.10**?

this I dont know - it shouldnt, the code is using `platform.python_version` which should give the version of python that the code is using, the pyproject.toml has a version >=3.10 (due to some functions only avail in 3.10) so not sure why it is getting installed/using a lower version

jblance, 2024-02-22:
does it work if you install the regular package from pypi?

Saentist, 2024-02-23:
> does it work if you install the regular package from pypi?

I updated Ubuntu to 22.04 to solve python dependencies
with is not good from practical point of view i use 

```sh
            .-/+oossssoo+/-.               root@ET1610PT
        `:+ssssssssssssssssss+:`           -------------
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 22.04.3 LTS x86_64
    .ossssssssssssssssssdMMMNysssso.       Host: ET1610PT x.x
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 5.15.0-94-generic
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 8 days, 21 hours, 33 mins
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 3150 (dpkg), 7 (snap)
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: bash 5.1.16
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Resolution: 1366x768
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   CPU: Intel Atom D410 (2) @ 1.666GHz
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   GPU: Intel Atom Processor D4xx/D5xx/N4xx/N5xx
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Memory: 841MiB / 1954MiB
.ssssssssdMMMNhsssssssssshNMMMdssssssss.
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/
  +sssssssssdmydMMMMMMMMddddyssssssss+
   /ssssssssssshdmNNNNmyNMMMMhssssss/
    .ossssssssssssssssssdMMMNysssso.
      -+sssssssssssssssssyyyssss+-
        `:+ssssssssssssssssss+:`
            .-/+oossssoo+/-.
```

Solution can be some bundled binary release without dependencies,
also support for micropython inside ESP not as remote reader
https://github.com/jblance/micropython-esp_mqtt_slave

Small challenge ;)
https://www.lilygo.cc/products/t-can485
AIO solution with mmp-solar rs485 CANbus Bluetooth /sd card storage/

jblance, 2024-02-24:
> Small challenge ;)
> https://www.lilygo.cc/products/t-can485
> AIO solution with mmp-solar rs485 CANbus Bluetooth /sd card storage/

That looks interesting - might try to source some and try them out

Saentist, 2024-02-24:
> > Solution can be some bundled binary release without dependencies,
> > also support for micropython inside ESP not as remote reader
> > https://github.com/jblance/micropython-esp_mqtt_slave
> 
> So you mean to have a full solution on the ESP? The main problem is that mppsolar/powermon are growing bigger with more flexibility - I guess the solution would be more a build environment that created a minimal bui;ld based on actual inverter/hardware (but I think that is doen with the ESPHome solutions?

ESPHome is very good in integration on Home assistant but not practical in solution with some specific customization's.
just compare  webUI of ESPHome and [ESPEasy ](https://github.com/letscontrolit/ESPEasy) to get basic idea what can be done with 180 mhz cpu 2m ram and 4m storage
Windows 3 work on lower specs ;) With GUI