jblance/mpp-solar #398: Trying to use powermon over serial but no luck so far

hasselk, 2023-09-08:
Hello There,
Is powermon in the moment even working over serial connections or ist it only working with test connections?

I cant find anywhere that the Serial Port connect Funtion is even called. But that does not have to mean too much ;)

but i then only get an error on linux and windows with the current release:
2023-09-08 12:16:59,505:DEBUG:serialport:send_and_receive@57: port None
2023-09-08 12:16:59,505:ERROR:serialport:send_and_receive@59: Port not available

Testconfig used:
device:
  name: VMIII-5k
  id: VMIII-5k
  model: VMIII-5k
  manufacturer: Masterpower
  port:
    type: serial
    path: "COM5"
    baud: 2400
    protocol: PI30

commands:
    - command: QPIGS #Must be in the protocol command list
      #type: poll
      #device: VMIII-5k
      outputs:
        - type: screen
          format: table
#        - type: screen
#          #device: VMIII-5k
#          format: 
#            type: hass
#            entity_id_prefix: VMIII-5k
#            discovery_prefix: home

do i have to do something differently?
Thats for checking.

jblance, 2023-09-08:
> 2023-09-08 ***@***.***: port None
> 2023-09-08 ***@***.***: Port not

I'm not sure it works on windows (haven't ever tried to get it working) for
Linux the path will not be com5, but more like /dev/ttyUSB0 etc.
Will need more details on your physical setup and the output of a debug run
to help more

Mppsolar might be easier place to start (especially as powermon is wip)

hasselk, 2023-09-08:
MPP-solar is working. But id like to control the Inverter thats why i need Powermon running :-)
i got the Same error on linux with  /dev/ttyUSB0.
Let me try the linux container with debug output and i will post it here. Thanks a lot

jblance, 2023-09-08:
Post the mppsolar command and that will help translate to powermon config

hasselk, 2023-09-08:
its all running in the Docker container, just built it myself because the official container seems to be missing the Docker startup Command.
by the way is there an easy way in mpp-solar to change the mqtt topic it is using without changing the code itself?

working mpp-solar config:
`[SETUP]
pause=1
mqtt_broker=hass.xxxx.xxx
mqtttopic=home
#mqtt_user=
#mqtt_pass=

[VMIII-5k]
port=/dev/ttyUSB0
type=mppsolar
protocol=PI30
 #PI30:037:VMIII-5000
command=QPIGS
#QFLAG
#,QMOD
#,QPIWS,QOPM,QFLAG,QMOD
tag=mpp
outputs=screen,hass_mqtt`

powermon_config on linux:
`device:
  id: VMIII-5k
  model: VMIII-5k
  manufacturer: Masterpower
  port:
    type: serial
    path: /dev/ttyUSB0
    protocol: PI30

commands:
- command: QPIGS #Must be in the protocol command list
  #  type: poll
  outputs:
    - type: screen
      format: table
    #- type: mqtt
    #format:  
    # type: hass
    # entity_id_prefix: VMIII-5k 
    # discovery_prefix: home`

powermon_debug_output:
`2023-09-08 10:52:10,578:INFO:__init__:main@114: Power Device Monitoring Utility, version: 0.16.10
2023-09-08 10:52:10,578:INFO:__init__:main@120: Using config file: powermon.yaml
2023-09-08 10:52:10,584:INFO:__init__:main@130: config=BaseConfig(device=DeviceConfig(name=None, id='VMIII-5k', model='VMIII-5k', manufacturer='Masterpower', port=SerialPortConfig(type='serial', path='/dev/ttyUSB0', baud=None, protocol='PI30')), commands=[CommandConfig(command='QPIGS', type=None, trigger=None, outputs=[OutputConfig(type='screen', format='table')])], mqttbroker=MQTTConfig(name='hass.h725k.de', port=1883, username=None, password=None), api=None, daemon=None, debuglevel='10', loop=None)
2023-09-08 10:52:10,585:INFO:__init__:main@145: config: {'device': {'id': 'VMIII-5k', 'model': 'VMIII-5k', 'manufacturer': 'Masterpower', 'port': {'type': 'serial', 'path': '/dev/ttyUSB0', 'protocol': 'PI30'}}, 'commands': [{'command': 'QPIGS', 'outputs': [{'type': 'screen', 'format': 'table'}]}], 'mqttbroker': {'name': 'hass.h725k.de', 'port': 1883, 'username': None, 'password': None}, 'debuglevel': 10}
2023-09-08 10:52:10,585:DEBUG:__init__:getPortFromConfig@12: port_config: {'type': 'serial', 'path': '/dev/ttyUSB0', 'protocol': 'PI30'}
2023-09-08 10:52:10,585:DEBUG:__init__:getPortFromConfig@22: portType: serial
2023-09-08 10:52:10,585:DEBUG:serialport:fromConfig@20: building serial port. config:{'type': 'serial', 'path': '/dev/ttyUSB0', 'protocol': 'PI30'}
2023-09-08 10:52:10,585:DEBUG:__init__:get_protocol@21: Protocol PI30
2023-09-08 10:52:10,587:DEBUG:__init__:main@149: Device: unnamed_device, self.identifier='VMIII-5k', self.model='VMIII-5k', self.manufacturer='Masterpower', port: SerialPort: self.path='/dev/ttyUSB0', self.baud=2400, protocol:PI30 protocol handler, self.serialPort=None, self.error=None, commands:[]

2023-09-08 10:52:10,587:DEBUG:__init__:getOutputs@46: processing outputsConfig: [{'type': 'screen', 'format': 'table'}]
2023-09-08 10:52:10,588:DEBUG:__init__:parseOutputConfig@65: parseOutputConfig, config: {'type': 'screen', 'format': 'table'}, topic: None
2023-09-08 10:52:10,588:DEBUG:__init__:parseOutputConfig@87: got dict type outputConfig: {'type': 'screen', 'format': 'table'}
2023-09-08 10:52:10,588:DEBUG:__init__:getFormatfromConfig@25: getFormatfromConfig, formatConfig: table, device: None, topic: None
2023-09-08 10:52:10,588:DEBUG:__init__:getFormatfromConfig@37: getFormatfromConfig, formatType: table
2023-09-08 10:52:10,588:DEBUG:__init__:parseOutputConfig@95: got format: outputs the results to standard out in a table (optionally formatted with line art boxes)
2023-09-08 10:52:10,588:DEBUG:__init__:getOutputClass@23: outputType screen
2023-09-08 10:52:10,589:DEBUG:screen:__init__@10: outputConfig: None, formatter: outputs the results to standard out in a table (optionally formatted with line art boxes)
2023-09-08 10:52:10,589:DEBUG:__init__:parseOutputConfig@97: got output: Output: Screen, outputs the results to standard out in a table (optionally formatted with line art boxes)
2023-09-08 10:52:10,589:DEBUG:command:__init__@58: Command: self.name='QPIGS' self.full_command=None, self.type='basic', [_outs='Output: Screen, outputs the results to standard out in a table (optionally formatted with line art boxes)'], last_run='Not yet run', next_run='08 Sep 2023 10:52:10', trigger: every 60 loops togo: 0, self.command_defn=None
2023-09-08 10:52:10,590:DEBUG:abstractprotocol:get_command_defn@86: Processing command 'QPIGS'
2023-09-08 10:52:10,590:DEBUG:abstractprotocol:get_command_defn@88: Found command QPIGS in protocol b'PI30'
2023-09-08 10:52:10,590:INFO:__init__:main@155: Device: unnamed_device, self.identifier='VMIII-5k', self.model='VMIII-5k', self.manufacturer='Masterpower', port: SerialPort: self.path='/dev/ttyUSB0', self.baud=2400, protocol:PI30 protocol handler, self.serialPort=None, self.error=None, commands:[<powermon.commands.command.Command object at 0x7f34d8ee45b0>]
2023-09-08 10:52:10,590:DEBUG:mqttbroker:fromConfig@19: mqttbroker config: {'name': 'hass.h725k.de', 'port': 1883, 'username': None, 'password': None}
[... 7 lines cut ...]
2023-09-08 10:52:10,591:DEBUG:mqttbroker:subscribe@136: Not connected, connecting
2023-09-08 10:52:10,591:DEBUG:mqttbroker:connect@88: Connecting to hass.h725k.de on port 1883
2023-09-08 10:52:10,591:DEBUG:mqttbroker:connect@95: No mqtt authentication used
2023-09-08 10:52:10,597:DEBUG:mqttbroker:on_connect@64: MqttBroker connection returned result: 0 Connection successful
[... cut ...]
2023-09-08 10:52:11,601:DEBUG:table:format@38: displayData: {'error': ['Command: QPIGS incurred an error or errors during execution or processing', ''], 'error_count': [2, ''], 'error_#0': ['Serial port not available None', ''], 'error_#1': ['failed validity check: response was empty', '']}
2023-09-08 10:52:11,601:DEBUG:table:format@66: width_c=50, line_length=99, width_p=12, width_v=74, width_u=6
2023-09-08 10:52:11,601:DEBUG:table:format@72: width_c=50, line_length=99, width_p=12, width_v=74, width_u=6`

jblance, 2023-09-08:
It's not seeing the serial port. Have you mapped it to the docker container?

hasselk, 2023-09-08:
yes its mapped and working in similiar fashion with mpp-solar

hasselk, 2023-09-08:
serial port in mpp-solar container:
`root@31b1d2ff42e3:/etc/mpp-solar# ls -l /dev/ttyUSB0 
crw-rw-rw- 1 nobody nogroup 188, 0 Sep  8 11:19 /dev/ttyUSB0`

and in the powermon container:
`root@57df08eecfa2:/etc/mpp-solar# ls -l /dev/ttyUSB0 
crw-rw-rw- 1 nobody nogroup 188, 0 Sep  8 11:19 /dev/ttyUSB0`

hasselk, 2023-09-08:
ok i changed the device.py in powermon like this, because the connect war realy NEVER called and thats the reason the port was uninitializes so = none
which for shure is not the right way todo because i think there is no reconnect logic or anything but now the serial connection works ;) or to say it more correctly it connects to the device
`
    def initialize(self):
        log.info("initializing device")
        self.port.connect()
        return
`

hasselk, 2023-09-08:
and now for something complitly different ;-)

ive activated now the mqtt output but getting this Error, any Ideas?

`  outputs:
    - type: screen
      format: table
    - type: mqtt
      format:
        type: hass
        entity_id_prefix: VMIII-5k
        discovery_prefix: home
`

`2023-09-08 15:25:40,243:INFO:__init__:main@114: Power Device Monitoring Utility, version: 0.16.10
2023-09-08 15:25:40,243:INFO:__init__:main@120: Using config file: powermon.yaml
2023-09-08 15:25:40,251:INFO:__init__:main@130: config=BaseConfig(device=DeviceConfig(name=None, id='VMIII-5k', model='VMIII-5k', manufacturer='Masterpower', port=SerialPortConfig(type='serial', path='/dev/ttyUSB0', baud=None, protocol='PI30')), commands=[CommandConfig(command='QPIGS', type=None, trigger=None, outputs=[OutputConfig(type='screen', format='table'), OutputConfig(type='mqtt', format=HassFormatConfig(type='hass', tag=None, draw_lines=None, keep_case=None, remove_spaces=None, extra_info=None, excl_filter=None, filter=None, discovery_prefix='home', entity_id_prefix='VMIII-5k'))])], mqttbroker=MQTTConfig(name='hass.h725k.de', port=1883, username=None, password=None), api=None, daemon=None, debuglevel='10', loop=None)
2023-09-08 15:25:40,251:INFO:__init__:main@145: config: {'device': {'id': 'VMIII-5k', 'model': 'VMIII-5k', 'manufacturer': 'Masterpower', 'port': {'type': 'serial', 'path': '/dev/ttyUSB0', 'protocol': 'PI30'}}, 'commands': [{'command': 'QPIGS', 'outputs': [{'type': 'screen', 'format': 'table'}, {'type': 'mqtt', 'format': {'type': 'hass', 'entity_id_prefix': 'VMIII-5k', 'discovery_prefix': 'home'}}]}], 'mqttbroker': {'name': 'hass.h725k.de', 'port': 1883, 'username': None, 'password': None}, 'debuglevel': 10}
2023-09-08 15:25:40,251:DEBUG:__init__:getPortFromConfig@12: port_config: {'type': 'serial', 'path': '/dev/ttyUSB0', 'protocol': 'PI30'}
2023-09-08 15:25:40,251:DEBUG:__init__:getPortFromConfig@22: portType: serial
2023-09-08 15:25:40,251:DEBUG:serialport:fromConfig@20: building serial port. config:{'type': 'serial', 'path': '/dev/ttyUSB0', 'protocol': 'PI30'}
2023-09-08 15:25:40,251:DEBUG:__init__:get_protocol@21: Protocol PI30
2023-09-08 15:25:40,253:DEBUG:__init__:main@149: Device: unnamed_device, self.identifier='VMIII-5k', self.model='VMIII-5k', self.manufacturer='Masterpower', port: SerialPort: self.path='/dev/ttyUSB0', self.baud=2400, protocol:PI30 protocol handler, self.serialPort=None, self.error=None, commands:[]
2023-09-08 15:25:40,253:DEBUG:__init__:getOutputs@46: processing outputsConfig: [{'type': 'screen', 'format': 'table'}, {'type': 'mqtt', 'format': {'type': 'hass', 'entity_id_prefix': 'VMIII-5k', 'discovery_prefix': 'home'}}]
2023-09-08 15:25:40,253:DEBUG:__init__:parseOutputConfig@65: parseOutputConfig, config: {'type': 'screen', 'format': 'table'}, topic: None
2023-09-08 15:25:40,254:DEBUG:__init__:parseOutputConfig@87: got dict type outputConfig: {'type': 'screen', 'format': 'table'}
2023-09-08 15:25:40,254:DEBUG:__init__:getFormatfromConfig@25: getFormatfromConfig, formatConfig: table, device: None, topic: None
2023-09-08 15:25:40,254:DEBUG:__init__:getFormatfromConfig@37: getFormatfromConfig, formatType: table
2023-09-08 15:25:40,254:DEBUG:__init__:parseOutputConfig@95: got format: outputs the results to standard out in a table (optionally formatted with line art boxes)
2023-09-08 15:25:40,254:DEBUG:__init__:getOutputClass@23: outputType screen
2023-09-08 15:25:40,255:DEBUG:screen:__init__@10: outputConfig: None, formatter: outputs the results to standard out in a table (optionally formatted with line art boxes)
2023-09-08 15:25:40,255:DEBUG:__init__:parseOutputConfig@97: got output: Output: Screen, outputs the results to standard out in a table (optionally formatted with line art boxes)
2023-09-08 15:25:40,255:DEBUG:__init__:parseOutputConfig@65: parseOutputConfig, config: {'type': 'mqtt', 'format': {'type': 'hass', 'entity_id_prefix': 'VMIII-5k', 'discovery_prefix': 'home'}}, topic: None
2023-09-08 15:25:40,255:DEBUG:__init__:parseOutputConfig@87: got dict type outputConfig: {'type': 'mqtt', 'format': {'type': 'hass', 'entity_id_prefix': 'VMIII-5k', 'discovery_prefix': 'home'}}
2023-09-08 15:25:40,255:DEBUG:__init__:getFormatfromConfig@25: getFormatfromConfig, formatConfig: {'type': 'hass', 'entity_id_prefix': 'VMIII-5k', 'discovery_prefix': 'home'}, device: None, topic: None
2023-09-08 15:25:40,256:DEBUG:__init__:getFormatfromConfig@37: getFormatfromConfig, formatType: hass
2023-09-08 15:25:40,256:DEBUG:__init__:parseOutputConfig@95: got format: Format: hass
2023-09-08 15:25:40,256:DEBUG:__init__:getOutputClass@23: outputType mqtt
Traceback (most recent call last):
  File "/usr/local/bin/powermon", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/site-packages/powermon/__init__.py", line 152, in main
    command = Command.fromConfig(commandConfig)
  File "/usr/local/lib/python3.10/site-packages/powermon/commands/command.py", line 43, in fromConfig
    outputs = getOutputs(config.get("outputs", ""))
  File "/usr/local/lib/python3.10/site-packages/powermon/outputs/__init__.py", line 55, in getOutputs
    _outputs.append(parseOutputConfig(outputConfig))
  File "/usr/local/lib/python3.10/site-packages/powermon/outputs/__init__.py", line 96, in parseOutputConfig
    _output = getOutputClass(outputType, _format)
  File "/usr/local/lib/python3.10/site-packages/powermon/outputs/__init__.py", line 27, in getOutputClass
    output_class = MQTT(outputConfig, topic, mqtt_broker, _format)
TypeError: Can't instantiate abstract class MQTT with abstract method process
`

jblance, 2023-09-08:
> 2023-09-08 ***@***.***: Power Device Monitoring
> 2023-09-08 ***@***.***: building serial
> 2023-09-08 ***@***.***: Protocol PI30
> 2023-09-08 ***@***.***: Device: unnamed_device,
> None 2023-09-08 ***@***.***: got

Looks like a bunch of bugs and unfinished refactoring atm

Will start looking at fixes tonight

rossandrews, 2023-09-08:
It's very likely this is my fault. I was poking around in the mqtt format to integrate with the API. I switched to making a new format called api_mqtt but may have left mqtt in a bad state. I will investigate.

hasselk, 2023-09-09:
Hi there, first of all Thanks a lot.

and for the first Problem of the missing _.port.connect()_ for the device, it seems that this was done at one time in the main _init_ and moved to _schedule.py_ in libs with commit [moved port.connect from init to schedule](https://github.com/jblance/mpp-solar/commit/531450626f5f5844d78db37c1a5e67354929ee24#diff-9d24fd77a6e36c8609e900ac06d236443333875f6dc9fbdb048dbbe78c4196ae).

Since schedule.py does not exist any longer, there seems to be no more .port.connect() running. Have not found the commit where the schedule.py got removed yet.

oleks85, 2023-09-12:
I had the same issues with serial port and mqtt output. I do not have much experience with python, but got fixed all those issues in my fork. There is pull request with fixes #401 .

jblance, 2023-10-19:
Looks resolved - but note that powermon is undergoing extensive changes so will break and change continuously, use and upgrade with caution

closed 2023-10-19 by jblance