jblance/mpp-solar #404: ISolar SM IV 5.6KW problem with command PIRI

Waldi81, 2023-09-22:
Hi,
I have problem with command PIRI. Another commands are working. 

DEBUG:

pi@TERMINAL:~$ mpp-solar -p /dev/ttyUSB0 -P PI18sv -c PIRI -D
2023-09-22 22:14:13,644:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.62
2023-09-22 22:14:13,644:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-09-22 22:14:13,644:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2023-09-22 22:14:13,644:DEBUG:__init__:main@241: udp port 5555
2023-09-22 22:14:13,644:DEBUG:__init__:main@243: Using Postgres None
2023-09-22 22:14:13,645:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2023-09-22 22:14:13,645:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI18sv"
2023-09-22 22:14:13,647:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-09-22 22:14:13,647:DEBUG:device:__init__@33: __init__ args ()
2023-09-22 22:14:13,647:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI18sv', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7f0d7de7eb80>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-09-22 22:14:13,647:DEBUG:__init__:get_port_type@69: port matches ttyusb
2023-09-22 22:14:13,647:INFO:__init__:get_port@111: Using serialio for communications
2023-09-22 22:14:13,654:DEBUG:__init__:get_protocol@13: Protocol PI18sv
2023-09-22 22:14:13,752:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7f0d7de7ed90>, protocol PI18 protocol handler for Easun SV inverters
2023-09-22 22:14:13,752:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f0d7de7ebb0>, 'PIRI', 'PIRI', 'screen', None, None)]
2023-09-22 22:14:13,753:INFO:__init__:main@407: Looping 1 commands
2023-09-22 22:14:13,753:INFO:__init__:main@417: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7f0d7de7ed90>, protocol: PI18 protocol handler for Easun SV inverters for command: PIRI, tag: PIRI, outputs: screen
2023-09-22 22:14:13,753:INFO:device:run_command@50: Running command PIRI
2023-09-22 22:14:13,753:INFO:pi18:get_full_command@583: Using protocol b'PI18SV' with 46 commands
2023-09-22 22:14:13,753:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'PIRI'
2023-09-22 22:14:13,753:DEBUG:abstractprotocol:get_command_defn@72: Found command PIRI in protocol b'PI18SV'
2023-09-22 22:14:13,753:DEBUG:pi18:get_full_command@593: _cmd is: b'PIRI'
2023-09-22 22:14:13,753:DEBUG:pi18:get_full_command@596: _type is: QUERY
2023-09-22 22:14:13,753:DEBUG:pi18:get_full_command@600: _prefix: ^P007
2023-09-22 22:14:13,753:DEBUG:pi18:get_full_command@610: auto_prefix: ^P007
2023-09-22 22:14:13,753:DEBUG:pi18:get_full_command@614: _pre_cmd: b'^P007PIRI'
2023-09-22 22:14:13,753:DEBUG:pi18:get_full_command@627: CRC: True
2023-09-22 22:14:13,754:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'^P007PIRI'
2023-09-22 22:14:13,754:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0xee 0x38 0xee38
2023-09-22 22:14:13,754:DEBUG:pi18:get_full_command@637: full command: b'^P007PIRI\xee8\r'
2023-09-22 22:14:13,754:INFO:device:run_command@78: full command b'^P007PIRI\xee8\r' for command PIRI
2023-09-22 22:14:13,754:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'PIRI'
2023-09-22 22:14:13,754:DEBUG:abstractprotocol:get_command_defn@72: Found command PIRI in protocol b'PI18SV'
2023-09-22 22:14:13,754:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2023-09-22 22:14:13,756:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-09-22 22:14:14,202:DEBUG:serialio:send_and_receive@30: serial response was: b'^D0892300,243,2300,500,243,5600,5600,480,460,540,424,552,520,8,010,120,1,0,0,9,0,0,0,1,1,00\xee\xe0\r'
2023-09-22 22:14:14,209:DEBUG:device:run_command@95: Send and Receive Response b'^D0892300,243,2300,500,243,5600,5600,480,460,540,424,552,520,8,010,120,1,0,0,9,0,0,0,1,1,00\xee\xe0\r'
2023-09-22 22:14:14,210:INFO:abstractprotocol:decode@245: response passed to decode: b'^D0892300,243,2300,500,243,5600,5600,480,460,540,424,552,520,8,010,120,1,0,0,9,0,0,0,1,1,00\xee\xe0\r'
2023-09-22 22:14:14,210:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'PIRI'
2023-09-22 22:14:14,210:DEBUG:abstractprotocol:get_command_defn@72: Found command PIRI in protocol b'PI18SV'
2023-09-22 22:14:14,210:INFO:abstractprotocol:decode@289: Processing response of type DEFAULT
2023-09-22 22:14:14,210:DEBUG:abstractprotocol:decode@293: trimmed and split responses: [b'2300', b'243', b'2300', b'500', b'243', b'5600', b'5600', b'480', b'460', b'540', b'424', b'552', b'520', b'8', b'010', b'120', b'1', b'0', b'0', b'9', b'0', b'0', b'0', b'1', b'1', b'00']
2023-09-22 22:14:14,211:INFO:abstractprotocol:decode@300: Processing DEFAULT type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 420, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 112, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 342, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
IndexError: list index out of range

Waldi81, 2023-09-23:
Problem solved. 
In the pi18.py file, line 162, add the missing types of batteries and everything works.

jblance, 2023-09-23:
Can you supply your change and I'll update the source code.

Thanks

Waldi81, 2023-09-23:
I added:
`            [
                "option",
                "Battery Type",
                [
                    "AGM",
                    "Flooded",
                    "User",
                    "Pylontech",
                    "TBD",
                    "WECO",
                    "Soltaro",
                    "3rd party Lithium",
                    "LIB",
                    "LIC",
                ]],`

closed 2023-09-23 by Waldi81

sante85, 2024-01-26:
@jblance @Waldi81   what command and protocol suggest for inverter isolar smg II 5.6kw to get status and all setting information?

thanks and best regards

Waldi81, 2024-01-26:
@sante85  I'm using a different inverter but try to google this topic "SMX II 5.6 is compatible with SRNE inverter protocol"