jblance/mpp-solar #406: EASUN SPL5K - Inverter switches to bypass at 2 kW

mmsfla, 2023-09-30:
Dear friends,
Maybe someone is reading along who can help me.
My inverter is the type from EASUN mentioned above and should only use the bypass and the connected network of my house when it exceeds 4000 watts or 5000 VA. However, it does it at 2000 watts. What am I doing wrong?

My configuration:
4 * 12 Volt / 100 Ah battery
approx. 1700 watts of solar

Settings SPL5k:

pi@grafana:~ $ mpp-solar --getsettings
Command: QFLAG - Flag Status inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
ac_input_voltage              	230.0          	V   
ac_input_current              	21.7           	A   
ac_output_voltage             	230.0          	V   
ac_output_frequency           	50.0           	Hz  
ac_output_current             	21.7           	A   
ac_output_apparent_power      	5000           	VA  
ac_output_active_power        	4000           	W   
battery_voltage               	48.0           	V   
battery_recharge_voltage      	44.0           	V   
battery_under_voltage         	42.0           	V   
battery_bulk_charge_voltage   	56.4           	V   
battery_float_charge_voltage  	54.0           	V   
battery_type                  	AGM            	    
max_ac_charging_current       	10             	A   
max_charging_current          	20             	A   
input_voltage_range           	Appliance      	    
output_source_priority        	Solar first    	    
charger_source_priority       	Solar first    	    
max_parallel_units            	6              	units
machine_type                  	Off Grid       	    
topology                      	transformerless	    
output_mode                   	single machine output	    
battery_redischarge_voltage   	52.0           	V   
pv_ok_condition               	As long as one unit of inverters has connect PV, parallel system will consider PV OK	    
pv_power_balance              	PV input max power will be the sum of the max charged power and loads power	    
overload_restart              	enabled        	    
over_temperature_restart      	enabled        	    
lcd_backlight                 	enabled        	    
primary_source_interrupt_alarm	enabled        	    
buzzer                        	disabled       	    
overload_bypass               	disabled       	    
power_saving                  	disabled       	    
lcd_reset_to_default          	disabled       	    
record_fault_code             	disabled

jblance, 2023-10-19:
you would get more responses if you posted on a solar forum regarding this, this is just a codebase that attempts to talk to the inverter and get information (and for some inverters allows you to change settings)

i dont fully grasp your problem, but at a guess your batteries might be sagging at 2KW to the point that the inverter switches to AC, from my understanding the inverter has a max output of 4KW after which it may fail completely, even in bypass

closed 2023-10-19 by jblance

mmsfla, 2023-10-21:
I have now been able to identify the problem.
When I bought the battery I ignored the name C100.
So my batteries are "AGM 100Ah C100", which means:
100 Ah at 100 hour discharge.
This explains the switching to bypass when the load is high.