jblance/mpp-solar #408: Voltronic Axpert 8kw

sante85, 2023-10-02:
hi,

this inverter have 2 MPPT DEVICES.

with command
 mpp-solar -p /dev/ttyUSB0 --getstatus

expose datas of one of two.

how to do this?

sante85, 2023-10-02:
i have sended this command QPIGS2

 mpp-solar -p /dev/ttyUSB0 -c QPIGS2

but not response get.

SolPiLog get a response

jblance, 2023-10-04:
the default protocol doesnt have that command, try `mpp-solar -p /dev/ttyUSB0 -c QPIGS2 -P pi30max` or better yet post the result of `mpp-solar -p /dev/ttyUSB0 --getDeviceId`

sante85, 2023-10-09:
with python  3.9.2
it is very strange because at launch: mpp-solar -p /dev/ttyUSB0 -c QPIGS2 -P pi30max get this error
 package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/pi30max.py", line 3, in <module>
    from .pi30 import pi30
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/pi30.py", line 3, in <module>
    from mppsolar.protocols.abstractprotocol import AbstractProtocol
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 20, in <module>
    class ProtocolDTO(BaseModel):
  File "pydantic/main.py", line 186, in pydantic.main.ModelMetaclass.__new__
**TypeError: Argument 'bases' has incorrect type (expected list, got tuple)**

why?

thanks

jblance, 2023-10-09:
what version are you using?
can you post a debug run

jblance, 2023-10-09:
can you update to the latest? (will need python 3.10 which may be a problem)

sante85, 2023-10-09:
> can you post a debug run

dbg mode:

mpp-solar -p /dev/ttyUSB0 --getstatus -D
2023-10-09 22:35:46,900:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.62
2023-10-09 22:35:46,901:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-10-09 22:35:46,902:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2023-10-09 22:35:46,902:DEBUG:__init__:main@241: udp port 5555
2023-10-09 22:35:46,902:DEBUG:__init__:main@243: Using Postgres None
2023-10-09 22:35:46,903:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2023-10-09 22:35:46,903:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2023-10-09 22:35:46,909:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-10-09 22:35:46,910:DEBUG:device:__init__@33: __init__ args ()
2023-10-09 22:35:46,910:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x760c8b50>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-10-09 22:35:46,911:DEBUG:__init__:get_port_type@69: port matches ttyusb
2023-10-09 22:35:46,911:INFO:__init__:get_port@111: Using serialio for communications
2023-10-09 22:35:46,933:DEBUG:__init__:get_protocol@13: Protocol PI30
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 353, in main
    device = device_class(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/mppsolar.py", line 7, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 37, in __init__
    self._protocol = get_protocol(get_kwargs(kwargs, "protocol"))
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/__init__.py", line 19, in get_protocol
    proto_module = importlib.import_module("mppsolar.protocols." + protocol_id, ".")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/pi30.py", line 3, in <module>
    from mppsolar.protocols.abstractprotocol import AbstractProtocol
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 20, in <module>
    class ProtocolDTO(BaseModel):
  File "pydantic/main.py", line 186, in pydantic.main.ModelMetaclass.__new__
TypeError: Argument 'bases' has incorrect type (expected list, got tuple)

rossandrews, 2023-10-09:
Is the toDTO method called at all in the mppsolar folder? I thought it was just a powermon thing?

sante85, 2023-10-10:
with python 3.11 and 0.16 run right.

IF I RUN 
mppsolar -p /dev/ttyUSB0 -P PI30 -c QPIGS -o mqtt --mqtttopic aaa

not consider "aaa" with prefix

MOSQUITO OUTPUT SUBSCRIBE

QPIGS/status/ac_input_voltage/value 223.2
QPIGS/status/ac_input_voltage/unit V
QPIGS/status/ac_input_frequency/value 49.9
QPIGS/status/ac_input_frequency/unit Hz
QPIGS/status/ac_output_voltage/value 223.2
QPIGS/status/ac_output_voltage/unit V
QPIGS/status/ac_output_frequency/value 49.9
QPIGS/status/ac_output_frequency/unit Hz
QPIGS/status/ac_output_apparent_power/value 0
QPIGS/status/ac_output_apparent_power/unit VA
.....

jblance, 2023-10-10:
If you are trying to influence the topic I think it's the 'tag' option for
this output module

sante85, 2023-10-10:
i have tried but in this case mqtt not run properly

mppsolar -p /dev/ttyUSB0 -P PI30MAX -c QPIGS2
Command: QPIGS2 - General Status Parameters inquiry 2
--------------------------------------------------------------------------------
Parameter          Value                Unit
pv2_input_current  0.0                  A       {'icon': 'mdi:solar-power', 'device-class': 'current'}
pv2_input_voltage  0.0                  V       {'icon': 'mdi:solar-power', 'device-class': 'voltage'}
pv2_charging_power 0                    W       {'icon': 'mdi:solar-power', 'device-class': 'power', 'state_class': 'measurement'}
--------------------------------------------------------------------------------

mppsolar -p /dev/ttyUSB0 -P PI30MAX -c QPIGS2 -o mqtt -T 92932208100911

mosquito output

92932208100911/status/validity_check/value Error: Invalid response CRCs

why?

sante85, 2023-10-10:
only output mqtt not trigger data.

json format is ok
 mppsolar -p /dev/ttyUSB0 -P PI30MAX -c QPIGS2 -o json
{"_command": "QPIGS2", "_command_description": "General Status Parameters inquiry 2", "pv2_input_current": 0.0, "pv2_input_voltage": 0.0, "pv2_charging_power": 0}

jblance, 2023-10-10:
> i have tried but in this case mqtt not run properly
>
> ## mppsolar -p /dev/ttyUSB0 -P PI30MAX -c QPIGS2
> Command: QPIGS2 - General Status Parameters inquiry 2
> ## Parameter Value Unit
> pv2_input_current [...]

the device response was likely incomplete - are any other connections to the device?
try running the command a few times